evcc-io/evcc #14343: evcc vehicle liefert unexpected status: 400 (Bad Request) nach update auf 0.127.0

mborgelt, 2024-06-13:
### Describe the bug

Nach dem Update auf 0.127.0 liefert ein 'evcc vehicle':  
```
[main  ] ERROR 2024/06/13 18:55:31 creating vehicle citigo-iv failed: cannot create vehicle type 'template': cannot create vehicle type 'skoda': unexpected status: 400 (Bad Request)

### Steps to reproduce

1. Update auf  0.127.0
2. evcc vehicle

### Configuration details

```yaml
vehicles:
- type: template
  template: skoda
  title: BI-MB430
  vin: TMBZXXXXXXXXXXXXXXX
  user: skoda@xxxxx.xxx
  password: ***redacted****
  capacity: 36
  phases: 2
  priority: 2  
  name: citigo-iv
  identifiers:
    - Citigo-iv
  mode: pv
```

### Log details

```text
[main  ] INFO 2024/06/13 18:55:30 evcc 0.127.0 (3735c8c0)
[main  ] INFO 2024/06/13 18:55:30 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/13 18:55:30 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2024/06/13 18:55:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT * FROM `settings` 42 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT  fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT  fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "% device_id       %") -1 <nil>
[db    ] TRACE 2024/06/13 18:55:30 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[skoda ] TRACE 2024/06/13 18:55:30 GET https://identity.vwgroup.io/oidc/v1/authorize?client_id=7f045eee-7003-4379-9968-9355ed2adb06%40apps_vw-dilab_com&code_challenge=DiGy0x5RBtaiM_eJI-FpRYxn2_aFYFWDaLbhupCoHTM&code_challenge_method=S256&redirect_uri=myskoda%3A%2F%2Fredirect%2Flogin%2F&response_type=code+id_token&scope=address+badge+birthdate+cars+driversLicense+dealers+email+mileage+mbb+nationalIdentifier+openid+phone+profession+profile+vin&nonce=qwiYMwwomnnwbnrCyIZpxdYMkjzPyGdhLuwaJlyDnwV&state=78a3054a-04b1-4b81-b7d1-a565834a591a
[skoda ] TRACE 2024/06/13 18:55:30 
GET /oidc/v1/authorize?client_id=7f045eee-7003-4379-9968-9355ed2adb06%40apps_vw-dilab_com&code_challenge=DiGy0x5RBtaiM_eJI-FpRYxn2_aFYFWDaLbhupCoHTM&code_challenge_method=S256&redirect_uri=myskoda%3A%2F%2Fredirect%2Flogin%2F&response_type=code+id_token&scope=address+badge+birthdate+cars+driversLicense+dealers+email+mileage+mbb+nationalIdentifier+openid+phone+profession+profile+vin&nonce=qwiYMwwomnnwbnrCyIZpxdYMkjzPyGdhLuwaJlyDnwV&state=78a3054a-04b1-4b81-b7d1-a565834a591a HTTP/1.1
Host: identity.vwgroup.io
User-Agent: Go-http-client/1.1
Accept-Encoding: gzip

HTTP/2.0 302 Found
Apigw-Requestid: ZUKkdjJPliAEJ9Q=
Cache-Control: no-cache, no-store, max-age=0, must-revalidate
Date: Thu, 13 Jun 2024 16:55:30 GMT
Expires: 0
Location: https://identity.vwgroup.io/signin-service/v1/signin/7f045eee-7003-4379-9968-9355ed2adb06@apps_vw-dilab_com?relayState=ccfc2326f3b93f3e9e8235867173417ed082f6bf
Pragma: no-cache
Referrer-Policy: origin
Set-Cookie: vcap_journey=ff6d8ae3-3870-4c0f-a23e-0f8f39a64d1a; Path=/; Secure
Strict-Transport-Security: max-age=31536000; includeSubDomains; preload
[... 223 lines cut ...]
Content-Type: application/x-www-form-urlencoded
Accept-Encoding: gzip

auth_code=eyJraWQiOiI0ODEyODgzZi05Y2FiLTQwMWMtYTI5OC0wZmEyMTA5Y2ViY2EiLCJhbGciOiJSUzI1NiJ9.eyJzdWIiOiI1MWQ1OTE3My1iMmFhLTRmOGUtOTA3NC1jODNhZmZjODI4MzkiLCJhdWQiOiI3ZjA0NWVlZS03MDAzLTQzNzktOTk2OC05MzU1ZWQyYWRiMDZAYXBwc192dy1kaWxhYl9jb20iLCJhY3IiOiJodHRwczovL2lkZW50aXR5LnZ3Z3JvdXAuaW8vYXNzdXJhbmNlL2xvYS0yIiwic2NwIjoiYWRkcmVzcyBiYWRnZSBiaXJ0aGRhdGUgY2FycyBkcml2ZXJzTGljZW5zZSBkZWFsZXJzIGVtYWlsIG1pbGVhZ2UgbWJiIG5hdGlvbmFsSWRlbnRpZmllciBvcGVuaWQgcGhvbmUgcHJvZmVzc2lvbiBwcm9maWxlIHZpbiIsImFhdCI6ImlkZW50aXR5a2l0IiwiaXNzIjoiaHR0cHM6Ly9pZGVudGl0eS52d2dyb3VwLmlvIiwianR0IjoiYXV0aG9yaXphdGlvbl9jb2RlIiwiZXhwIjoxNzE4Mjk4MDMxLCJpYXQiOjE3MTgyOTc3MzEsIm5vbmNlIjoicXdpWU13d29tbm53Ym5yQ3lJWnB4ZFlNa2p6UHlHZGhMdXdhSmx5RG53ViIsImp0aSI6Ijc4NjQ3ZDAxLTc1YTQtNDBlOC1hYTBmLTNkYjRhYjZjNjE3NSJ9.lUlJ2orrq4LRZ93X8LeMF-8O8dhy94HP3QWCZaxpnMpaUy1okarvrujSVabH75xMDhv7rGtarcCIiX9htG4jcgr8u0YpAArQyRf0_3Phj4AhG-GBK4ipVOTnNd-eo9dJLqPJqMIK-wXKdlaJrvbfwTbaHtV8ZRROehW5jfNfiKAkMoJryjVyEjLPeG6Oe-lKf4vl2cTtTmJESLar3XW62qF39iTMJa-pador6CfhZ8LCNJWi3U01MXUD2dTJnwgVOgxR3cwuZxK0zKNsqxv0UdHc1ZUipULLI-4grOtbQHiVuU0Miw4XxjbO6TvmijSdYB1WMh1n74-Mu0EB_Lyj_2ZJsU12boTaX0XTKCdP70mBXVjr2sa78HAotXXBb8PYBIbf-uX-yo-oyr2hlVQ04EeUo93Va0M23L4Fkp92Aoq6EU1SfcvZA2OAHgU5o7E7NOQTHHFQozdvQDNuu2UShZE2ryq3sUWXJD3aW16EZ3RTH3E11HTLljJIBfrvbxtbFBsJiPBoji8mvcdhzT4aRPiMPFrWkfcbXvFiD2nGg5T5bVGnn95LaHTGf5euWQAM5xLiPVxZ1cK-01nm7FkolGwlayyv5JEbY5gkFKT9BYdnCkoP32Pr8gclrvKAx_4LH4MOOoNV-GIY203FpSPUN5iOkfqYNWHK6knnsrIVA2E&brand=skoda&code=eyJraWQiOiI0ODEyODgzZi05Y2FiLTQwMWMtYTI5OC0wZmEyMTA5Y2ViY2EiLCJhbGciOiJSUzI1NiJ9.eyJzdWIiOiI1MWQ1OTE3My1iMmFhLTRmOGUtOTA3NC1jODNhZmZjODI4MzkiLCJhdWQiOiI3ZjA0NWVlZS03MDAzLTQzNzktOTk2OC05MzU1ZWQyYWRiMDZAYXBwc192dy1kaWxhYl9jb20iLCJhY3IiOiJodHRwczovL2lkZW50aXR5LnZ3Z3JvdXAuaW8vYXNzdXJhbmNlL2xvYS0yIiwic2NwIjoiYWRkcmVzcyBiYWRnZSBiaXJ0aGRhdGUgY2FycyBkcml2ZXJzTGljZW5zZSBkZWFsZXJzIGVtYWlsIG1pbGVhZ2UgbWJiIG5hdGlvbmFsSWRlbnRpZmllciBvcGVuaWQgcGhvbmUgcHJvZmVzc2lvbiBwcm9maWxlIHZpbiIsImFhdCI6ImlkZW50aXR5a2l0IiwiaXNzIjoiaHR0cHM6Ly9pZGVudGl0eS52d2dyb3VwLmlvIiwianR0IjoiYXV0aG9yaXphdGlvbl9jb2RlIiwiZXhwIjoxNzE4Mjk4MDMxLCJpYXQiOjE3MTgyOTc3MzEsIm5vbmNlIjoicXdpWU13d29tbm53Ym5yQ3lJWnB4ZFlNa2p6UHlHZGhMdXdhSmx5RG53ViIsImp0aSI6Ijc4NjQ3ZDAxLTc1YTQtNDBlOC1hYTBmLTNkYjRhYjZjNjE3NSJ9.lUlJ2orrq4LRZ93X8LeMF-8O8dhy94HP3QWCZaxpnMpaUy1okarvrujSVabH75xMDhv7rGtarcCIiX9htG4jcgr8u0YpAArQyRf0_3Phj4AhG-GBK4ipVOTnNd-eo9dJLqPJqMIK-wXKdlaJrvbfwTbaHtV8ZRROehW5jfNfiKAkMoJryjVyEjLPeG6Oe-lKf4vl2cTtTmJESLar3XW62qF39iTMJa-pador6CfhZ8LCNJWi3U01MXUD2dTJnwgVOgxR3cwuZxK0zKNsqxv0UdHc1ZUipULLI-4grOtbQHiVuU0Miw4XxjbO6TvmijSdYB1WMh1n74-Mu0EB_Lyj_2ZJsU12boTaX0XTKCdP70mBXVjr2sa78HAotXXBb8PYBIbf-uX-yo-oyr2hlVQ04EeUo93Va0M23L4Fkp92Aoq6EU1SfcvZA2OAHgU5o7E7NOQTHHFQozdvQDNuu2UShZE2ryq3sUWXJD3aW16EZ3RTH3E11HTLljJIBfrvbxtbFBsJiPBoji8mvcdhzT4aRPiMPFrWkfcbXvFiD2nGg5T5bVGnn95LaHTGf5euWQAM5xLiPVxZ1cK-01nm7FkolGwlayyv5JEbY5gkFKT9BYdnCkoP32Pr8gclrvKAx_4LH4MOOoNV-GIY203FpSPUN5iOkfqYNWHK6knnsrIVA2E&code_verifier=fikNBDbGjr2IrVOctyCHGig-m2inUyOoFU5isfepkhM&id_token=eyJraWQiOiI0ODEyODgzZi05Y2FiLTQwMWMtYTI5OC0wZmEyMTA5Y2ViY2EiLCJhbGciOiJSUzI1NiJ9.eyJzdWIiOiI1MWQ1OTE3My1iMmFhLTRmOGUtOTA3NC1jODNhZmZjODI4MzkiLCJlbWFp [... cut ...]
[main  ] ERROR 2024/06/13 18:55:31 creating vehicle citigo-iv failed: cannot create vehicle type 'template': cannot create vehicle type 'skoda': unexpected status: 400 (Bad Request)
.Soc:        vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'skoda': unexpected status: 400 (Bad Request)
Capacity:    36.0kWh
Identifiers: [Citigo-iv]
Features:    [Offline Retryable]
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.127.0 (3735c8c0)

StefanSchoof, 2024-06-13:
Siehe auch https://github.com/evcc-io/evcc/discussions/14327

GrimmiMeloni, 2024-06-13:
Danke Stefan - das hilft.

Hier gibt es in der Tat eine Wechselwirkung mit dem Change für den Enyaq. Wir fragen jetzt mit dem MySkoda Client das Auth Token an, wollen es dann aber nicht beim MySkoda Server, sondern beim alten SkodaConnect/VW abholen. Das geht dann schief. 

Ich denke das beste ist, wir stellen alle anderen Skodas jetzt auch direkt auf MySkoda um.
@StefanSchoof nutzt Du evcc dockerized, und könntest einen build testen wenn ich Dir einen Container baue?

GrimmiMeloni, 2024-06-13:
Auf den zweiten Blick - kannst Du einfach mal das Template von `skoda` auf `enyaq` wechseln?

StefanSchoof, 2024-06-13:
Ich habe das Problem nicht. Habe nur versucht die Diskussion und Issue zu verknüpfen.

GrimmiMeloni, 2024-06-13:
@andig Wir könnten jetzt hier den Auth Flow auf den vom Enyaq (also MySkoda) umstellen. Scheinbar nutzt aber die bisherige Skoda Implementierung nochmal eine ganz andere API (also auch nicht SkodaConnect). Diese nutzt dann aber auch wiederum eine andere MBB Token Source. Da weiß ich wiederum nicht, ob diese auch mit den Tokens die der MySkoda AuthServer auswirft umgehen kann.

Bevor wir das anfassen würde ich erstmal einen Test von einem betroffenen User mit dem Enyaq (MySkoda) abwarten. Vielleicht müssen wir hier gar nichts machen und können das ganze auf die eine MySkoda basierte Implementierung vereinheitlichen.

mborgelt, 2024-06-13:
Ich hab mal das Template enyaq für meinen citigo-iv ausprobiert. Damit klappt die Authentifizierung aber das Fahrzeug wird natürlich nicht gefunden.
```
{"vehicles":[],"errors":[{"type":"NOT_ONLY_MOD4_VEHICLES","description":"User’s garage contains vehicles other than the currently supported MOD3 and MOD4 vehicles. The unsupported vehicles were not returned."}]}
[main  ] ERROR 2024/06/13 21:05:18 creating vehicle citigo-iv failed: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': cannot find vehicle, got: []
.Soc:        vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': cannot find vehicle, got: []
Capacity:    36.0kWh
Identifiers: [Citigo-iv]
Features:    [Offline Retryable]

degit007, 2024-06-13:
@GrimmiMeloni Hi Michael, danke.

Mit type=enyaq funktioniert die Anmeldung für meinen Skoda Superb wieder und ich kann in evcc den Ladestand des Auto auslesen.

GrimmiMeloni, 2024-06-13:
Danke Euch. 

OK, das bedeutet also, daß ältere(?) Fahrzeuge wie der Citigo nicht über die neue MySkoda Schnittstelle laufen. Somit können wir also aktuelle den Auth Server auch nicht teilen.

PR #14349 (2024-06-13): Fix: move MySkoda implementation into own package
- Revert #14186
- move reverted changes into own package `skoda/myskoda`
- adjust Enyaq template to use above `myskoda` package
- fix #14343
closed 2024-06-13 by andig