evcc-io/evcc #14344: Renault alternativ wakeup bringt Fehlermeldung.

DaEffendi, 2024-06-13:
### Describe the bug

Bereits seit einiger Zeit funktioniert der alternative Wakeup nicht mehr. Evtl ist auch der "normale" Wakeup betroffen.
Die Versionen 0.126.6 und die aktuelle 0.127.0 funktionieren nicht. Die Versionen 0.126.4 und 0.126.5 vor dem Renault API change leider auch nicht.

Ich bekomme folgende Fehlermeldung sobald der Timer den wakeup call startet.

unexpected status: 415 (Unsupported Media Type)

### Steps to reproduce

1. evcc vehicle -w in der Konsole ausgeführt bringt unten aufgeführte Fehlermeldung.

2.evcc vehicle -a (dies sollte der alternativ wakeup "start charge" sein) bringt unten aufgeführte Fehlermeldung

3.
...

### Configuration details

```yaml
Mein Fahrzeug benötigt den alternative Wakeup. Dieser ist in der yaml konfiguriert. Hier wurde nichts verändert. Der alternative wakeup hatte über Monate bestens funktioniert.

alternativewakeup: true
```

### Log details

```text
1: evcc vehicle -w

[renault] TRACE 2024/06/13 19:22:34 POST https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/7389967b-6970-4855-bd01-e49ffb5c0c90/kamereon/kca/car-adapter/v1/cars/***/actions/charging-start?country=DE
[renault] TRACE 2024/06/13 19:22:34 {"data":{"attributes":{"action":"start"},"type":"ChargingStart"}}
--
{"type":"FUNCTIONAL","messages":[{"code":"err.func.wired.unsupportedMediaType","message":"Content-Type 'application/octet-stream' is not supported"}],"errors":[{"errorCode":"err.func.wired.unsupportedMediaType","errorMessage":"Content-Type 'application/octet-stream' is not supported"}],"error_reference":"FUNCTIONAL"}
[main  ] ERROR 2024/06/13 19:22:34 wakeup: unexpected status: 415 (Unsupported Media Type)

2: evcc vehicle -a

[renault] TRACE 2024/06/13 19:24:49 POST https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/7389967b-6970-4855-bd01-e49ffb5c0c90/kamereon/kca/car-adapter/v1/cars/***/actions/charging-start?country=DE
[renault] TRACE 2024/06/13 19:24:50 {"data":{"attributes":{"action":"start"},"type":"ChargingStart"}}
--
{"type":"FUNCTIONAL","messages":[{"code":"err.func.wired.unsupportedMediaType","message":"Content-Type 'application/octet-stream' is not supported"}],"errors":[{"errorCode":"err.func.wired.unsupportedMediaType","errorMessage":"Content-Type 'application/octet-stream' is not supported"}],"error_reference":"FUNCTIONAL"}
[main  ] ERROR 2024/06/13 19:24:50 start charge: unexpected status: 415 (Unsupported Media Type)
```

### What type of operating system are you running?

Linux

### Version

0.126.4 - 0.127.0 getestet

GrimmiMeloni, 2024-06-14:
Da wurde scheinbar die API von Renault verändert. :( 
Gibt es bereits ein anderes Projekt das die Integration angepasst hat, wo wir abschauen können?

andig, 2024-06-14:
Closed in 03edf16240380ea8bd05a56d0f8251e8e346edbd. Schöne Beschäftigungstherapie.

DaEffendi, 2024-06-14:
> Da wurde scheinbar die API von Renault verändert. :( 
> Gibt es bereits ein anderes Projekt das die Integration angepasst hat, wo wir abschauen können? 

https://github.com/chrisstaite/ZEServices

Hier! Damit läuft es. Danke.

StefanSchoof, 2024-06-15:
> Closed in 03edf16240380ea8bd05a56d0f8251e8e346edbd. Schöne Beschäftigungstherapie.

@andig Irgendwie ist das nicht verlinkt und ich sehe im master branch keinen Commit der passt. Ist da was schief gegangen?

andig, 2024-06-15:
Aber jetzt!

DaEffendi, 2024-06-16:
Hallo,
hier als Feedback!
Ich nehme an, das soll mit 0.127.2 gefixt sein?
Leider immer noch fehlerhaft!
Danke

[renault] TRACE 2024/06/16 20:48:39 POST https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/7389967b-6970-4855-bd01-e49ffb5c0c90/kamereon/kca/car-adapter/v1/cars/***/actions/charging-start?country=DE
[renault] TRACE 2024/06/16 20:48:39 {"data":{"attributes":{"action":"start"},"type":"ChargingStart"}}
--
{"type":"FUNCTIONAL","messages":[{"code":"err.func.wired.unsupportedMediaType","message":"Content-Type 'application/octet-stream' is not supported"}],"errors":[{"errorCode":"err.func.wired.unsupportedMediaType","errorMessage":"Content-Type 'application/octet-stream' is not supported"}],"error_reference":"FUNCTIONAL"}
[renault] TRACE 2024/06/16 20:48:39 GET https://accounts.eu1.gigya.com/accounts.login?apiKey=3_7PLksOyBRkHv126x5WhHb-5pqC1qFR8pQjxSeLB6nhAnPERTUlwnYoznHSxwX668&loginID=***&password=***

[renault] TRACE 2024/06/16 20:48:39 POST https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/7389967b-6970-4855-bd01-e49ffb5c0c90/kamereon/kca/car-adapter/v1/cars/***/actions/charging-start?country=DE
[renault] TRACE 2024/06/16 20:48:39 {"data":{"attributes":{"action":"start"},"type":"ChargingStart"}}
--
{"type":"FUNCTIONAL","messages":[{"code":"err.func.wired.unsupportedMediaType","message":"Content-Type 'application/octet-stream' is not supported"}],"errors":[{"errorCode":"err.func.wired.unsupportedMediaType","errorMessage":"Content-Type 'application/octet-stream' is not supported"}],"error_reference":"FUNCTIONAL"}
[main  ] ERROR 2024/06/16 20:48:39 start charge: unexpected status: 415 (Unsupported Media Type)

andig, 2024-06-16:
Ging es in diesem Issue nicht um den hvacStatus? Hab ich mich davertan?

andig, 2024-06-16:
Ansonsten bleibt die Fragen iben bestehen. Es braucht Informationen…

DaEffendi, 2024-06-16:
Nein, es geht um das alternative Wakeup. Das Auto wird von evcc leider nicht mehr geweckt. 
Wenn Evcc das wakeup anstößt bekomme ich diese Fehlermeldung "unexpected status: 415 (Unsupported Media Type)"

DaEffendi, 2024-06-16:
> > Da wurde scheinbar die API von Renault verändert. :( 
> > Gibt es bereits ein anderes Projekt das die Integration angepasst hat, wo wir abschauen können? 
> 
> https://github.com/chrisstaite/ZEServices
> 
> Hier! Damit läuft es. Danke.

Ich kann nur sagen, mit diesem Projekt lässt sich die Ladung starten. Ich denke die nutzen die gleichen Endpoints wie ihr.  Es hat sich scheinbar 1 Jahr nichts getan und funktioniert immer noch. Wurde bei Evcc was verändert?

schnakomo, 2024-06-19:
Ich kann den bug #14344 von DaEffendi bestätigen. Ich verwende den "normalen" Wakeup (ohne "alternativewakeup: true" in der yaml-Datei) und bekomme auch die Fehlermeldung: "unexpected status: 415 (Unsupported Media Type)".

gaborgast, 2024-06-19:
Ich hab mit dem Dacia Spring gleiches Problem

[lp-1 ] ERROR 2024/06/19 11:13:58 wake-up vehicle: unexpected status: 415 (Unsupported Media Type)
mit Version 127.2

andig, 2024-06-19:
"Ich auch" liefert leider keine neuen Erkenntnisse.

Wir brauchen die "korrekten" Aufrufe. Z.B. indem jemand eine funktionierende Lösung (siehe oben) mal mittels MITM Proxy traced.

DaEffendi, 2024-06-19:
Kann man das aus dem Code des anderen Projektes (link oben) nicht herauslesen. Wenn ich nach Start charge suche bekomme ich etliche Ergebnisse.  
Leider bin ich Laie auf dem Gebiet und kann somit nicht zur Lösung beitragen. :-(

andig, 2024-06-30:
Keine neuen Erkenntnisse- wir schließen. Vielleicht heilt Renault das Problem ja selbst...

closed 2024-06-30 by andig

mrhund, 2024-06-30:
Also oben in der Fehlermeldung steht: "Content-Type 'application/octet-stream' is not supported". 
In dem anderen Projekt wird der Content-Type einmal auf "application/x-www-form-urlencoded" und einmal auf "application/vnd.api+json" gesetzt, wobei ich auf letztes tippen würde, weil dieser in der Datei "VehicleApi.kt" verwendet wird.

Ich hoffe das reicht, dass jemand von den Contributors weiterschauen kann.

andig, 2024-06-30:
Das hilft- gerne mal probieren!

CommoCruise, 2024-07-19:
Ich habe weiterhin das Problem, das unser Dacia Spring aus dem Tiefschlaf nicht wieder aufwacht.
Hat dieses Problem noch jemand anderes?

premultiply, 2024-07-19:
Was sagt der Hersteller dazu?

gaborgast, 2024-07-19:
Ich habe ebenfalls dieses Problem. Und der Ladestand wird auch nicht mehr
angezeigt.
Schade

CommoCruise, 2024-07-19:
> Was sagt der Hersteller dazu?

Habe ich noch nicht angefragt, dazu bräuchte ich den request für den wakeup. Kann den jemand nennen?
Ich hoffe den exakten dann im log zu finden.

premultiply, 2024-07-19:
Die Frage ist: Warum wacht das Fahrzeug nicht auf wenn eine angeschlossene Wallbox die Ladefreigabe erteilt?

CommoCruise, 2024-07-19:
> Die Frage ist: Warum wacht das Fahrzeug nicht auf wenn eine angeschlossene Wallbox die Ladefreigabe erteilt?

Das Fahrzeug macht auf und lädt auch wenn man die Wallbox anklemmt.
Geht aber nach ein paar Stunden nicht laden in den Tiefschlaf und muss dann zum Laden mit einem wakeup call geweckt werden.
Und dieser wakeup call scheint hier nicht zu funktionieren. 

Aktuell kann ich den wakeup nur manuell über die dacia app triggern.

gaborgast, 2024-07-19:
Und es hat schon ein paar Monate lange funktioniert und mit einem der
letzten Updates nicht mehr

CommoCruise, 2024-07-19:
Korrekt  @gaborgast es hat in der Vergangenheit bereits funktioniert. 
Wenn wir es wieder richten können, wäre es schön einen Test dafür hinzuzufügen, damit das nicht wieder hinten runter fällt.

andig, 2024-07-19:
> Und es hat schon ein paar Monate lange funktioniert und mit einem der
letzten Updates nicht mehr

Dann sollte Renault das Update zurück drehen und/oder eine Schnittstellendoku veröffentlichen. Ihr seid die Kunden- setzt die Hersteller unter Druck!

premultiply, 2024-07-19:
>> Die Frage ist: Warum wacht das Fahrzeug nicht auf wenn eine angeschlossene Wallbox die Ladefreigabe erteilt?

> Das Fahrzeug macht auf und lädt auch wenn man die Wallbox anklemmt.
> Geht aber nach ein paar Stunden nicht laden in den Tiefschlaf und muss dann zum Laden mit einem wakeup call geweckt werden.

Das hantieren mit irgendeiner App oder deren undokumentierten API ist für absolute Basisfunktionen weder praxistauglich noch marktüblich, war bestimmt nicht Teil des Verkaufsgespräch und der Vertragsbedingungen und ist daher überraschend und stellt einen Mangel des Produkts dar. Workarounds mit allerhöchstens sehr begrenzter Haltbarkeit helfen wenig.

Die Frage an den Hersteller bleibt: **Warum wacht das Fahrzeug nicht auf wenn eine angeschlossene Wallbox die Ladefreigabe erteilt**?
Dazu gibt es sogar einschlägige Testreihen und und Defacto-Standards. Die Frage wäre auch, wie das Fahrzeug diese in diesem Softwarezustand bestehen konnte.