evcc-io/evcc #14347: Fatal Error in 0.127.0 regarding "template". wallbox does not work // cannot create eebus

deThommy, 2024-06-13:
It seems that there is a problem with the "template". With the version 0.127 also my wallbox does not work and the error looks very similar:

cannot create charger 'wallboxElli': cannot create charger type 'template': cannot create charger type 'eebus': eebus not configured

I hope it helps.

_Originally posted by @pierangelof in https://github.com/evcc-io/evcc/discussions/14327#discussioncomment-9766408_

deThommy, 2024-06-13:
I have the exact same problem after updating to 0.127.0.

Elli Charger Pro, no changes to my hardware whatsoever...

Fatal Error and EVCC won't start now. Before the update on 0.126.4 everything was fine and has been since February. Can I help with more data?

deThommy, 2024-06-13:
Bin gerade hierüber gestolpert #14321

Schließe das hier daher wieder. @pierangelof seems like it has been fixed and will be in live in the new release

closed 2024-06-13 by deThommy

deThommy, 2024-06-14:
Updated to 0.127.1 just now - everything is working as intended for me

pierangelof, 2024-06-14:
Same for me. Many thanks to the developers!