evcc-io/evcc #14348: Wird Laden einmal gestoppt (bspw. weil Wolke vor der Sonne), kann es nicht wieder aufgenommen werden

cdyna43, 2024-06-13:
### Describe the bug

Hallo allerseits!

Ich habe eine PV-Anlage (Fronium WR und SmartMeter) und zwei Wallbox Pulsar. EVCC läuft auf RP4. 

Fahrzeuge: Ein VW ID.3 und ein BMW 330e

Grundsätzlich funktioniert alles.

ABER:

Wenn der Ladevorgang einmal unterbrochen ist (manuell oder weil nicht genug Überschuss da ist), ist es nicht möglich, ihn "over-the-air" erneut zu starten. Die Wallbox (oder das Fahrzeug?) geht offenbar davon aus, dass der Ladevorgang beendet ist (so die Anzeige in der Wallbox-App). 
Zum "Wiederbeleben" sind mit zwei Möglichkeiten bekannt: 
1. Auto ab- und wieder anstecken.
2. Wallbox OCPP auf ChargeHQ umstellen. Dann "wacht" die Wallbox wieder auf und lässt sich wieder steuern.

Schalte ich die Wallbox über die Wallbox-App (also OCPP deaktiviert), gibt es das Problem nicht. Wenn ich dort unterbreche steht da "Ladegerät bereit" (oder so ähnlich, kann gerade nicht nachschauen) und eben nicht "Laden beendet". 
Auch mit ChargeHQ funktioniert alles, dort wird der Ladevorgang also auch nach "OCPP-Stoppen" wieder aufgenommen. 

Das Auto (hier der ID.3) gibt als Fehlermeldung "Fehler beim Laden Bitte versuchen Sie es noch mal"; 

Das ganz wird alle paar Sekunden wiederholt, wenn EVCC versucht, den Ladevorgang neu zu starten...

Aufgrund der Symptome ist meine Laien-Einschätzung also, dass EVCC das falsche Signal zur Wallbox schickt, also "beenden" anstatt "unterbrechen" und auch das "Neustart"-Signal das falsche ist (weil es bei ChargeHQ ja funktioniert...)
 

### Steps to reproduce

1. EVCC auf min+PV (oder auf schnell oder auf PV bei ausreichend Sonne) 
2. Auto anstecken 
3.Ladevorgang über EVCC stoppen (auf aus oder weil PV und nicht ausreichend Sonne) 
4. Versuch, Laden wieder zu starten (durch Umstellen auf min+PV oder schnell oder weil wieder genug Sonne) 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxxxxx

interval: 30s # control cycle interval

meters:
- type: template
  template: sunspec-inverter 
  id: 240  
  host: 192.168.xx. Xx
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sunspec-inverter 
  id: 1  
  host: 192.168.xx.xx
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
[... 297 lines cut ...]
Jun 13 20:35:42 raspberrypi evcc[33638]: [site  ] DEBUG 2024/06/13 20:35:42 ----
Jun 13 20:35:42 raspberrypi evcc[33638]: [lp-1  ] DEBUG 2024/06/13 20:35:42 charge power: 0W
Jun 13 20:35:42 raspberrypi evcc[33638]: [lp-1  ] DEBUG 2024/06/13 20:35:42 charge currents: [0 0 0]A
Jun 13 20:35:42 raspberrypi evcc[33638]: [lp-2  ] DEBUG 2024/06/13 20:35:42 charge power: 0W
Jun 13 20:35:42 raspberrypi evcc[33638]: [lp-2  ] DEBUG 2024/06/13 20:35:42 charge currents: [0 0 0]A
Jun 13 20:35:43 raspberrypi evcc[33638]: [site  ] DEBUG 2024/06/13 20:35:43 pv power: 196W
Jun 13 20:35:43 raspberrypi evcc[33638]: [site  ] DEBUG 2024/06/13 20:35:43 grid meter: 2678W
Jun 13 20:35:44 raspberrypi evcc[33638]: [site  ] DEBUG 2024/06/13 20:35:44 grid powers: [-23 2673 28]W
Jun 13 20:35:45 raspberrypi evcc[33638]: [site  ] DEBUG 2024/06/13 20:35:45 grid currents: [-0.485 10.1 0.331]A
Jun 13 20:35:45 raspberrypi evcc[33638]: [site  ] DEBUG 2024/06/13 20:35:45 site power: 2678W
Jun 13 20:35:45 raspberrypi evcc[33638]: [lp-1  ] DEBUG 2024/06/13 20:35:45 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
Jun 13 20:35:45 raspberrypi evcc[33638]: [lp-1  ] DEBUG 2024/06/13 20:35:45 charge total import: 179.136kWh
Jun 13 20:35:45 raspberrypi evcc[33638]: [lp-1  ] DEBUG 2024/06/13 20:35:45 charger status: B
Jun 13 20:35:45 raspberrypi evcc[33638]: [lp-1  ] DEBUG 2024/06/13 20:35:45 pv charge current: min 6A > 0A (2678W @ 1p, battery: false)
^C
cdyna@raspberrypi:~ $
```

### What type of operating system are you running?

Linux

### Version

0.127.0

GrimmiMeloni, 2024-06-14:
Das dürfte am Fahrzeug liegen. Ich nehme an Du hast mit dem ID3 getestet? Wenn ja, welche MEB Firmware?

cdyna43, 2024-06-14:
ID.3 und BMW 545e.

Mit ChargeHQ funktioniert es bei beiden problemlos, was mich denken lässt, dass weder das Auto noch die Wallbox das Problem ist...

cdyna43, 2024-06-14:
So, ich habe es noch einmal versucht. Das (lange) Log findet sich nachstehend. 

Laden über EVCC nicht möglich. Immer "Ladebereit. Warte auf Fahrzeug". 

OCCP in der Wallbox auf ChargeHQ - ich kann den Ladevorgang starten und stoppen, alles funktioniert. OCCP zurück auf EVCC - wieder "Ladevorgang beendet" in der Wallbox-App und "Ladebereit. Warte auf Fahrzeug" in EVCC. 

Ich habe mehrfach zwischen ChargeHQ und EVCC hin- und hergewechselt. Immer das gleiche Ergebnis. ChargeHQ hat kein Problem, EVCC schafft es aber nicht. 
War alles mit dem ID.3 angeschlossen an LP1. An der Software des Fahrzeugs liegt es also nicht grundsätzlich (aber vielleicht an der Kompatibilität mit EVCC...). 

Jun 14 10:47:24 raspberrypi evcc[33638]: [lp-1  ] DEBUG 2024/06/14 10:47:24 charge total import: 179.165kWh
Jun 14 10:47:24 raspberrypi evcc[33638]: [lp-1  ] DEBUG 2024/06/14 10:47:24 !! session: start=179.099 stop=179.165 chargedEnergy=0.034
Jun 14 10:47:24 raspberrypi systemd[1]: evcc.service: Deactivated successfully.
Jun 14 10:47:24 raspberrypi systemd[1]: Stopped evcc.service - evcc.
Jun 14 10:47:24 raspberrypi systemd[1]: evcc.service: Consumed 1min 43.437s CPU time.
Jun 14 10:48:12 raspberrypi systemd[1]: Started evcc.service - evcc.
Jun 14 10:48:12 raspberrypi evcc[38371]: [main  ] INFO 2024/06/14 10:48:12 evcc 0.127.1
Jun 14 10:48:12 raspberrypi evcc[38371]: [main  ] INFO 2024/06/14 10:48:12 using config file: /etc/evcc.yaml
Jun 14 10:48:12 raspberrypi evcc[38371]: [db    ] INFO 2024/06/14 10:48:12 using sqlite database: /var/lib/evcc/evcc.db
Jun 14 10:48:12 raspberrypi evcc[38371]: [main  ] INFO 2024/06/14 10:48:12 listening at :7070
Jun 14 10:48:40 raspberrypi evcc[38371]: [acd067-1] DEBUG 2024/06/14 10:48:40 waiting for chargepoint: 5m0s
Jun 14 10:48:40 raspberrypi evcc[38371]: [e86243-1] DEBUG 2024/06/14 10:48:40 waiting for chargepoint: 5m0s
Jun 14 10:49:00 raspberrypi evcc[38371]: [ocpp  ] DEBUG 2024/06/14 10:49:00 charge point connected: e86243Jun 14 10:49:30 raspberrypi evcc[38371]: [ocpp  ] DEBUG 2024/06/14 10:49:30 charge point connected: acd067Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] DEBUG 2024/06/14 10:49:37 charge total import: 179.169kWh
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] DEBUG 2024/06/14 10:49:37 charge total import: 2428.563kWh
Jun 14 10:49:37 raspberrypi evcc[38371]: [site  ] INFO 2024/06/14 10:49:37 site config:
Jun 14 10:49:37 raspberrypi evcc[38371]: [site  ] INFO 2024/06/14 10:49:37   meters:      grid ✓ pv ✓ battery ✗
Jun 14 10:49:37 raspberrypi evcc[38371]: [site  ] INFO 2024/06/14 10:49:37     grid:      power ✓ energy ✓ currents ✓
Jun 14 10:49:37 raspberrypi evcc[38371]: [site  ] INFO 2024/06/14 10:49:37     pv 1:      power ✓ energy ✓ currents ✗
Jun 14 10:49:37 raspberrypi evcc[38371]: [site  ] INFO 2024/06/14 10:49:37   vehicles:
Jun 14 10:49:37 raspberrypi evcc[38371]: [site  ] INFO 2024/06/14 10:49:37     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] INFO 2024/06/14 10:49:37 loadpoint 1:
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] INFO 2024/06/14 10:49:37   mode:        pv
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] INFO 2024/06/14 10:49:37   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] INFO 2024/06/14 10:49:37   meters:      charge ✓
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] INFO 2024/06/14 10:49:37     charge:    power ✓ energy ✓ currents ✓
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] INFO 2024/06/14 10:49:37 loadpoint 2:
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] INFO 2024/06/14 10:49:37   mode:        pv
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] INFO 2024/06/14 10:49:37   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] INFO 2024/06/14 10:49:37   meters:      charge ✓
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] INFO 2024/06/14 10:49:37     charge:    power ✓ energy ✓ currents ✓
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] DEBUG 2024/06/14 10:49:37 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] DEBUG 2024/06/14 10:49:37 phase timer inactive
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] DEBUG 2024/06/14 10:49:37 pv timer inactive
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] DEBUG 2024/06/14 10:49:37 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] DEBUG 2024/06/14 10:49:37 phase timer inactive
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] DEBUG 2024/06/14 10:49:37 pv timer inactive
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] INFO 2024/06/14 10:49:37 vehicle updated: unknown -> ID.3
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] DEBUG 2024/06/14 10:49:37 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
Jun 14 10:49:37 raspberrypi evcc[38371]: [site  ] DEBUG 2024/06/14 10:49:37 ----
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] DEBUG 2024/06/14 10:49:37 charge power: 0W
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-1  ] DEBUG 2024/06/14 10:49:37 charge currents: [0 0 0]A
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] DEBUG 2024/06/14 10:49:37 charge power: 0W
Jun 14 10:49:37 raspberrypi evcc[38371]: [lp-2  ] DEBUG 2024/06/14 10:49:37 charge currents: [0 0 0]A
[... cut ...]
Jun 14 11:38:11 raspberrypi evcc[38371]: [site  ] DEBUG 2024/06/14 11:38:11 grid powers: [-165 -61 -164]W
Jun 14 11:38:12 raspberrypi evcc[38371]: [site  ] DEBUG 2024/06/14 11:38:12 grid currents: [-0.914 -1.32 -0.773]A
Jun 14 11:38:12 raspberrypi evcc[38371]: [site  ] DEBUG 2024/06/14 11:38:12 site power: -393W
Jun 14 11:38:12 raspberrypi evcc[38371]: [lp-2  ] DEBUG 2024/06/14 11:38:12 charge total import: 2428.563kWh
Jun 14 11:38:12 raspberrypi evcc[38371]: [lp-2  ] DEBUG 2024/06/14 11:38:12 charger status: A

GrimmiMeloni, 2024-06-14:
```
Jun 14 10:58:59 raspberrypi evcc[38371]: [ocpp ] DEBUG 2024/06/14 10:58:59 charge point disconnected: acd067
Jun 14 10:59:07 raspberrypi evcc[38371]: [site ] DEBUG 2024/06/14 10:59:07 ----
Jun 14 10:59:07 raspberrypi evcc[38371]: [lp-1 ] ERROR 2024/06/14 10:59:07 charge power: timeout
Jun 14 10:59:08 raspberrypi evcc[38371]: [lp-1 ] ERROR 2024/06/14 10:59:08 charge currents: timeout
```

Der Charger disconnected.

cdyna43, 2024-06-14:
Ja, weil ich ihn auf das OCPP von ChargeHQ umstellen. Weil er über EVCC nicht mehr zum Laden zu bewegen ist.
Das Umstellen ist zum Testen, ob es an Wallbox oder Auto liegt. Stelle ich nicht auf ChargeHQ um, bleibt Wallbox verbunden, lädt aber nicht.

premultiply, 2024-06-14:
#12861 könnte vielleicht die Lösung für das dabei zugrunde liegende Problem sein.

/cc @andig

andig, 2024-06-14:
Ja, klingt identisch. Jetzt müssen wir den nur noch fertig machen.

cdyna43, 2024-06-14:
Sorry, wenn ich etwas dumm frage: Das Problem ist erkannt/bekannt und die Experten arbeiten dran? Falls nicht und ich mir die Lösung aus dem verlinkten Thread erschließen soll, bitte ich um einen Hinweis für Dummys....

Ich bin gerade nach Hause gekommen und habe beide Autos aus- und wieder eingesteckt. Es gab einige Fehlermeldungen und das Problem ist immer noch da (was zu erwarten war). Soll ich noch einmal mein Log posten?

andig, 2024-07-04:
Die angedachte Lösung ist ja https://github.com/evcc-io/evcc/pull/12861. Es scheint aber Boxen zu geben, die einen Vorgabestrom von 0 (=disable) ignorieren. Das würde in Konsequenz bedeuten, dass wir zwei unterschiedliche Implementierungen benötigen würden?

andig, 2024-07-06:
Could you try https://github.com/evcc-io/evcc/pull/14733? For Docker users it's available using the `ocpp` tag.

closed 2024-07-07 by andig

cdyna43, 2024-07-10:
Heißt "completed", dass das Problem gelöst sein soll? Ist es nämlich leider nicht. Habe gerade EVCC auf 0.128.2 aktualisiert. Keine Veränderung. Sobald das Laden gestoppt wird, kann man es über EVCC nicht mehr starten, weder automatisch, noch indem man manuell auf Schnelladen schaltet.

andig, 2024-07-11:
Completed heisst dass es einen Lösungsvorschlag gibt. Hast Du den denn ausprobiert? Falls es nicht geht- gibts ein Logfile? Config richtig?

cdyna43, 2024-07-11:
Ich habe EVCC aktualisiert. Wenn das nicht der Lösungsvorschlag ist, dann habe ich ihn leider nicht verstanden, sorry.
Config ist richtig in dem Sinne, dass alles funktioniert - außer eben dem Neustart, wenn der Ladevorgang unterbrochen wurde. Gibt es hierzu irgendetwas in der Config?

Log ist hier:
```

[main  ] INFO 2024/07/10 20:14:14 evcc 0.128.2

[main  ] INFO 2024/07/10 20:14:14 using config file: /config/evcc.yaml
[db    ] INFO 2024/07/10 20:14:14 using sqlite database: /data/evcc.db
[main  ] INFO 2024/07/10 20:14:14 listening at :7070
[e86243-1] DEBUG 2024/07/10 20:14:42 waiting for chargepoint: 5m0s
[acd067-1] DEBUG 2024/07/10 20:14:42 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/07/10 20:15:26 charge point connected: acd067
[ocpp  ] DEBUG 2024/07/10 20:16:04 charge point connected: e86243
[e86243-1] DEBUG 2024/07/10 20:16:13 hijacking transaction: 1720635176
[lp-1  ] DEBUG 2024/07/10 20:16:21 charge total import: 270.425kWh
[lp-2  ] DEBUG 2024/07/10 20:16:21 charge total import: 2484.777kWh
[site  ] INFO 2024/07/10 20:16:21 site config:
[site  ] INFO 2024/07/10 20:16:21   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/07/10 20:16:21     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2024/07/10 20:16:21     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/07/10 20:16:21   vehicles:
[site  ] INFO 2024/07/10 20:16:21     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/07/10 20:16:21 loadpoint 1:
[lp-1  ] INFO 2024/07/10 20:16:21   mode:        pv
[lp-1  ] INFO 2024/07/10 20:16:21   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/07/10 20:16:21   meters:      charge ✓
[lp-1  ] INFO 2024/07/10 20:16:21     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/07/10 20:16:21 loadpoint 2:
[lp-2  ] INFO 2024/07/10 20:16:21   mode:        pv
[lp-2  ] INFO 2024/07/10 20:16:21   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/07/10 20:16:21   meters:      charge ✓
[lp-2  ] INFO 2024/07/10 20:16:21     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/07/10 20:16:21 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/10 20:16:21 phase timer inactive
[lp-1  ] DEBUG 2024/07/10 20:16:21 pv timer inactive
[lp-2  ] DEBUG 2024/07/10 20:16:21 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-2  ] DEBUG 2024/07/10 20:16:21 phase timer inactive
[lp-2  ] DEBUG 2024/07/10 20:16:21 pv timer inactive
[lp-2  ] INFO 2024/07/10 20:16:21 vehicle updated: unknown -> ID.3
[lp-2  ] DEBUG 2024/07/10 20:16:21 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
[site  ] WARN 2024/07/10 20:16:21 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/07/10 20:16:21 ----
[lp-1  ] DEBUG 2024/07/10 20:16:21 charge power: 0W
[lp-1  ] DEBUG 2024/07/10 20:16:21 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/07/10 20:16:21 charge power: 0W
[lp-2  ] DEBUG 2024/07/10 20:16:21 charge currents: [0 0 0]A
[site  ] DEBUG 2024/07/10 20:16:21 pv power: 129W
[site  ] DEBUG 2024/07/10 20:16:22 grid meter: 436W
[site  ] DEBUG 2024/07/10 20:16:23 grid powers: [56 164 14]W
[site  ] DEBUG 2024/07/10 20:16:24 grid currents: [0.615 1.49 0.307]A
[site  ] DEBUG 2024/07/10 20:16:24 site power: 436W
[lp-1  ] DEBUG 2024/07/10 20:16:24 charge total import: 270.425kWh
[lp-1  ] DEBUG 2024/07/10 20:16:24 charger status: A
[lp-1  ] INFO 2024/07/10 20:16:24 car disconnected
[lp-1  ] DEBUG 2024/07/10 20:16:24 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/10 20:16:24 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/07/10 20:16:24 set session soc limit: 0
[lp-1  ] DEBUG 2024/07/10 20:16:24 set session energy limit: 0
[site  ] DEBUG 2024/07/10 20:16:36 ----
[lp-1  ] DEBUG 2024/07/10 20:16:36 charge power: 0W
[lp-1  ] DEBUG 2024/07/10 20:16:36 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/07/10 20:16:36 charge power: 0W
[lp-2  ] DEBUG 2024/07/10 20:16:36 charge currents: [0 0 0]A
[site  ] DEBUG 2024/07/10 20:16:36 pv power: 131W
[site  ] DEBUG 2024/07/10 20:16:37 grid meter: 234W
[site  ] DEBUG 2024/07/10 20:16:38 grid powers: [56 162 15]W
[site  ] DEBUG 2024/07/10 20:16:39 grid currents: [0.615 1.47 0.3]A
[site  ] DEBUG 2024/07/10 20:16:39 site power: 234W
[... 1045 lines cut ...]
[site  ] DEBUG 2024/07/10 20:33:54 site power: 287W
[lp-1  ] DEBUG 2024/07/10 20:33:54 charge total import: 270.425kWh
[lp-1  ] DEBUG 2024/07/10 20:33:54 charger status: A
[site  ] DEBUG 2024/07/10 20:34:06 ----
[lp-1  ] DEBUG 2024/07/10 20:34:06 charge power: 0W
[lp-1  ] DEBUG 2024/07/10 20:34:06 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/07/10 20:34:06 charge power: 0W
[lp-2  ] DEBUG 2024/07/10 20:34:06 charge currents: [0 0 0]A
[site  ] DEBUG 2024/07/10 20:34:06 pv power: 38W
[site  ] DEBUG 2024/07/10 20:34:07 grid meter: 286W
[site  ] DEBUG 2024/07/10 20:34:08 grid powers: [30 173 80]W
[site  ] DEBUG 2024/07/10 20:34:09 grid currents: [0.488 1.52 0.458]A
[site  ] DEBUG 2024/07/10 20:34:09 site power: 286W
[lp-2  ] DEBUG 2024/07/10 20:34:09 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-2  ] DEBUG 2024/07/10 20:34:09 charge total import: 2484.880kWh
[lp-2  ] DEBUG 2024/07/10 20:34:09 charger status: B
[lp-2  ] ERROR 2024/07/10 20:34:09 max charge current 16A: set charging profile: Rejected
[site  ] DEBUG 2024/07/10 20:34:21 ----
[lp-1  ] DEBUG 2024/07/10 20:34:21 charge power: 0W
[lp-1  ] DEBUG 2024/07/10 20:34:21 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/07/10 20:34:21 charge power: 0W
[lp-2  ] DEBUG 2024/07/10 20:34:21 charge currents: [0 0 0]A
[site  ] DEBUG 2024/07/10 20:34:21 pv power: 41W
[site  ] DEBUG 2024/07/10 20:34:22 grid meter: 279W
[site  ] DEBUG 2024/07/10 20:34:23 grid powers: [29 171 80]W

andig, 2024-07-11:
> Wenn das nicht der Lösungsvorschlag ist

Der Vorschlag ist:

> Could you try https://github.com/evcc-io/evcc/pull/14733

Dort steht:

> This PR adds an OCPP AutoStart mode. If `autostart: true`

Das müsste also bitte konfiguriert werden. Und wenn _das_ nicht geht brauchts bitte ein Ocpp trace log.

cdyna43, 2024-07-11:
Vielen Dank für die Geduld mit mir. Ich hielt das für interne Kommentare unter euch Profis und dachte, das würde im Update angepasst werden für Laien wie mich. 
Ich schaue mal, ob ich das hinbekomme..
Mein Verständnis: in der evcc.yaml unter chargers "autostart: true" ergänzen, richtig?

andig, 2024-07-11:
Genau bei deinem Charger, ja. 

> das würde im Update angepasst werden für Laien wie mich.

Auch das klappt leider nicht überall- Ocpp ist eine Wundertüte. Deshalb muss es einzeln konfiguriert werden.

cdyna43, 2024-07-11:
Sorry, bin wohl doch zu blöd.

Config:
chargers:
- type: template
  template: pulsarplus 
  stationid: acd067  
  connector: 1  
  connecttimeout: 5m  
  timeout: 2m  
  name: wallbox4
  getconfiguration: false
  autostart: true

gibt mir:
``
[main  ] INFO 2024/07/11 20:29:09 listening at :7070
[main  ] FATAL 2024/07/11 20:29:37 cannot create charger 'wallbox4': cannot create charger type 'template': invalid key: autostart
``

andig, 2024-07-11:
template: ocpp

cdyna43, 2024-07-11:
Ok. Keine Fehler und manuelles Starten scheint zu funktionieren. Jetzt muss ich nur noch bis morgen auf Sonne warten...

Vielen Dank!

andig, 2024-07-11:
@cdyna43 could your give https://github.com/evcc-io/evcc/pull/14824 a try? Fixed a bug that I could reproduce. Der Stand hier wird so noch nicht funktionieren.

cdyna43, 2024-07-11:
Happy to do so. You'll need to walk me through whatever you want me to do, though, I am afraid.