evcc-io/evcc #14353: SMA BYD Speicher wird trotz Entladesperre im Modus "Schnell" entladen

julred, 2024-06-14:
### Describe the bug

Hallo,

aktuell habe ich das Problem, dass trotz entladesperre, evcc die Batterie entlädt.
[image]
[image]

Ich habe einen SMA Hybrid SE 8 Wechselrichter mit einem 7.7kWh BYD Speicher

### Steps to reproduce

1. Auto anstecken
2. Modus auf Schnell laden aktivieren

### Configuration details

```yaml
root@evcc:/etc# cat evcc.yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  lp-1: trace
  
# unique installation id
plant: ***

interval: 30s # control cycle interval

sponsortoken: ***

telemetry: false

influx:
  url: http://192.168.178.16:8086
  database: evcc 
  token: ***
  org: home

mqtt:
  broker: 192.168.178.15:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: mqtt
  password: ***

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.178.129
  name: grid1
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.77  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.178.128  
[... 392 lines cut ...]
[lp-2  ] DEBUG 2024/06/14 09:21:47 pv charge current: 0A = 0A + -6.97A (4809W @ 3p)
[site  ] DEBUG 2024/06/14 09:22:16 ----
[lp-1  ] DEBUG 2024/06/14 09:22:16 charge power: 7032W
[lp-2  ] DEBUG 2024/06/14 09:22:16 charge power: 1W
[site  ] DEBUG 2024/06/14 09:22:17 pv power: 2385W
[site  ] DEBUG 2024/06/14 09:22:17 battery soc: 75%
[site  ] DEBUG 2024/06/14 09:22:17 battery power: 4926W
[site  ] DEBUG 2024/06/14 09:22:17 grid meter: -10W
[site  ] DEBUG 2024/06/14 09:22:17 grid powers: [1167 1166 -2342]W
[site  ] DEBUG 2024/06/14 09:22:17 grid currents: [5.38 5.15 -9.87]A
[site  ] DEBUG 2024/06/14 09:22:17 site power: 4916W
[lp-1  ] DEBUG 2024/06/14 09:22:17 !! session: chargeRater.chargedEnergy=2.6 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/14 09:22:17 charge total import: 932.440kWh
[lp-1  ] DEBUG 2024/06/14 09:22:17 charger status: C
[site  ] DEBUG 2024/06/14 09:22:25 set battery discharge control: false
[site  ] DEBUG 2024/06/14 09:22:25 set battery discharge control: true
```

### What type of operating system are you running?

Linux

### Version

V0.127.1

premultiply, 2024-06-14:
SMA hat es durch ein missglücktes automatisches Firmware-Update des SHM kaputt gemacht. Wir können da nichts tun bis SMA das repariert hat.

Bitte an den SMA Support wenden.

Tombra1889, 2024-06-14:
Leider, stürzt zudem evcc jetzt ab, wenn mit der neuen SHM Firmware die Batteriesperre deaktiviert werden soll:
[image]

premultiply, 2024-06-14:
Das ist ggf. ein anderes Problem. Bitte neues Issue dazu. Danke.

premultiply, 2024-06-14:
Ich mache mal hier zu, da wir hier nichts tun können und auf einen Bugfix von SMA warten müssen.

closed 2024-06-14 by premultiply

Hooch76, 2024-07-03:
An alle, die von dem Problem wegen der neuen SHM-Firmware betroffen sind: weitere Infos hier: https://www.facebook.com/share/p/fQA6EQReZwoQBDyb/