evcc-io/evcc #14355: Neue URL für die Integration vom Cupra Born

JonisK, 2024-06-14:
### Describe the bug

Die Integration des Cupra Born funktioniert seit etwa einem Tag nicht mehr.
Log: `[main ] ERROR 2024/06/14 09:52:47 creating vehicle grauer_born failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': cannot get vehicles: unexpected status: 404 (Not Found) `
Zur gleichen Zeit wurde auch in der HACS Integration für das gleiche Auto ein Fehler gefunden: https://community.home-assistant.io/t/cupra-born-ev-integration/580016/79

Ein fix ist auch bereits verfügbar. Und zwar musste dort eine URL angepasst werden: https://github.com/daernsinstantfortress/WeConnect-Cupra-python/commit/d61af9873cfc8a4dfcea16d56601e5d2e6dded19
Vielleicht ist es für evcc der gleiche Fehler und der Fix von daern wird auch hier funktionieren?

### Steps to reproduce

Cupra Born in die YAML eintragen und EVCC neu starten,
die Fehlermeldung taucht beim Neustart auf.

### Configuration details

```yaml
vehicles:
  - name: grauer_born
    type: template
    template: cupra
    title: Cupra Born
    user: ***
    password: ***
```

### Log details

```text
[main ] ERROR 2024/06/14 09:52:47 creating vehicle grauer_born failed: cannot create vehicle type 'template': cannot create vehicle type 'cupra': cannot get vehicles: unexpected status: 404 (Not Found)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.127.1

JonisK, 2024-06-14:
Ich denke mal das `v1` in [dieser Zeile ](https://github.com/evcc-io/evcc/blob/9ff704e763dc8cbe99b43d8691ca30c64fc94494/vehicle/seat/cupra/api.go#L46)sollte `v2` sein?

vistalba, 2024-06-14:
Gleiches problem. Habe es gestern nach dem update auf 0.127.0 festgestellt. Rollback auf 0.126.6 hat jedoch auch nichts mehr gebracht. Somit wohl das gleiche Problem.

gz2k2, 2024-06-14:
Habe das selbe Problem. Denke mal, das hat mit den Umstellungen der App usw zu tun

djanknecht, 2024-06-14:
Habe das Problem auch - die Änderung auf /v2 funktioniert schon mal im lokalen Test. PR folgt irgendwann...

vistalba, 2024-06-14:
@djanknecht Wie konntest du das ändern? Hast du die Datei im Docker Container gefunden und angepasst?

djanknecht, 2024-06-14:
@vistalba nein im Code, und dann kompiliert. Ich habe dazu PR https://github.com/evcc-io/evcc/pull/14359 angelegt, den nun jemand vom Projekt approved muss.

closed 2024-06-14 by andig

vistalba, 2024-06-23:
@djanknecht Ich habe seither keinen Kilometerstand mehr im Export der Ladevorgänge. Kannst du das bei dir auch Mal prüfen? Ist das auch ein Side-Effect von dem URL/API Change?

djanknecht, 2024-06-23:
> [Verfolgungsbild][https://github.com/notifications/beacon/ACPGVOKEE6MKFM5TYTNL2ATZI2Z2VA5CNFSM6AAAAABJJ325FCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUCHPLIU.gif]

Hi, ich hatte vorher keine Kilometerstände und jetzt auch nicht. Wie hast Du das hinbekommen?

vistalba, 2024-06-23:
@djanknecht 

Nichts spezielles. 

```
vehicles:
- type: template
  template: cupra
  title: Born
  user: [REDACTED]
  password: [REDACTED]
  capacity: 58
  mode: pv
  phases: 3
  name: Born
  icon: car
```

Wenn ich dann in der Ansicht "Ladevorgänge" das CSV exportiere, hatte ich bis zum 27.04.2024 immer schön den Kilometerstand vom Fahrzeug mit drin. Seither ist der aber weg.

MatzeBu, 2024-07-25:
Hallo, 
bei mir geht es erneut seid vorgestern nicht mehr. 
Sowohl in evcc, als auch in Home Assistent. Keine Verbindung.

therealmadmaxx, 2024-07-28:
Hallo,
bei mir ist auch Seite Donnerstag/Freitag der Wurm drin. Cupra App lieferte immer Daten, Home Assistent ging aber nach kurzer Zeit wieder, nur evcc bekommt keine Daten...

evcc  vehicle
.Soc:          unexpected status: 404 (Not Found)
Capacity:      58.0kWh
Charge status: unexpected status: 404 (Not Found)
Range:         unexpected status: 404 (Not Found)
Odometer:      unexpected status: 404 (Not Found)
Finish time:   unexpected status: 404 (Not Found)
Climater:      unexpected status: 404 (Not Found)
Limit Soc:     unexpected status: 404 (Not Found)

Grüße
Marcus

djanknecht, 2024-07-28:
@therealmadmaxx  bei mir funktioniert es mit der Version 0.129 wieder einwandfrei. @vistalba im Log wird auch der Kilometerstand wieder angezeigt, probierts mal aus.

MatzeBu, 2024-07-28:
Bei mir läuft's nach dem Update wieder 👍