evcc-io/evcc #14356: EEBUS Wallbox - Charge power: data not available

augard, 2024-06-14:
### Describe the bug

I am unable to charge car with new version 127.x, previously the same settings worked fine. Now i get error in logs "charge power: data not available" and "charge currents: not available".

### Steps to reproduce

1. start charing with car

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: warning
levels:
  cache: error

# unique installation id
plant: e54ca51a41c43d04f487b552158402752fc587cdf15340a89167bb662087424f

meters:
- name: grid_1
  type: template
  template: solax
  usage: grid
  modbus: tcpip
  id: 1
  host: 192.168.0.200
  port: 502

- name: pv_1
  type: template
  template: solax
  usage: pv
  modbus: tcpip
  id: 1
  host: 192.168.0.200
  port: 502

- name: battery_1
  type: template
  template: solax
  usage: battery
  modbus: tcpip
  id: 1
  host: 192.168.0.200
  port: 502

vehicles:
- type: template
  template: seat
  title: Cupra Formentor
  user: user
  password: password
  vin: vin
  capacity: 12.8
  name: vehicle_1

eebus:
  interfaces:
   - eth0
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          ...
          -----END EC PRIVATE KEY-----
[... 747 lines cut ...]
[lp-1  ] ERROR 2024/06/14 11:20:30 charge power: data not available
[eebus ] DEBUG 2024/06/14 11:20:30 Send: d:_i:47859_Elli-Wallbox-2022A0RQG6 notify 178 DeviceDiagnosisHeartbeatData
[lp-1  ] ERROR 2024/06/14 11:20:30 charge currents: not available
[site  ] DEBUG 2024/06/14 11:20:31 pv power: 2154W
[site  ] DEBUG 2024/06/14 11:20:31 battery soc: 78%
[site  ] DEBUG 2024/06/14 11:20:31 battery power: 1966W
[site  ] DEBUG 2024/06/14 11:20:31 grid meter: -0W
[site  ] DEBUG 2024/06/14 11:20:31 site power: 2316W
[lp-1  ] DEBUG 2024/06/14 11:20:31 charger status: B
[eebus ] DEBUG 2024/06/14 11:20:31 Send: d:_i:47859_Elli-Wallbox-2022A0RQG6 notify 179 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/06/14 11:20:32 Send: d:_i:47859_Elli-Wallbox-2022A0RQG6 notify 180 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/06/14 11:20:33 Send: d:_i:47859_Elli-Wallbox-2022A0RQG6 notify 181 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/06/14 11:20:33 Send: d:_i:47859_Elli-Wallbox-2022A0RQG6 notify 182 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/06/14 11:20:34 Send: d:_i:47859_Elli-Wallbox-2022A0RQG6 notify 183 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/06/14 11:20:35 Send: d:_i:47859_Elli-Wallbox-2022A0RQG6 notify 184 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/06/14 11:20:36 Send: d:_i:47859_Elli-Wallbox-2022A0RQG6 notify 185 DeviceDiagnosisHeartbeatData
```

### What type of operating system are you running?

Linux

### Version

0.127.1

An-Wi, 2024-06-14:
Wallbox restarted already?
👉 https://github.com/evcc-io/evcc/issues/14321#issuecomment-2167618024

augard, 2024-06-14:
> Wallbox restarted already? 👉 [#14321 (comment)](https://github.com/evcc-io/evcc/issues/14321#issuecomment-2167618024)

No, that din't help

augard, 2024-06-14:
Ah, i got it running, i needed to shutdown the container. Restart the wallbox. And then start it again. When i restarted it while evcc was running it din't work.

StefanSchoof, 2024-06-14:
With the elli Box always restart and update evcc only when no car is connected. Otherwise there is a high risk that the eebus Stack of the elli wallbox crashes.

closed 2024-06-14 by augard