evcc-io/evcc #14368: Keine Aktualisierung der Werte in evcc, wenn mit der neuen SMA Firmware die Entladesperre deaktiviert wird

Tombra1889, 2024-06-14:
Leider, stürzt zudem evcc jetzt ab, wenn mit der neuen SHM Firmware die Batteriesperre deaktiviert werden soll:
[image]
Also im Detail: wenn die Sperre aufgehoben wird (was nicht möglich ist), findet keine Aktualisierung mehr statt.
Ich nutze die Version V0.127.1

_Originally posted by @Tombra1889 in https://github.com/evcc-io/evcc/issues/14353#issuecomment-2168506398_

premultiply, 2024-06-14:
Bitte Details und Logfile usw. dazu was es ermöglicht das nachzuvollziehen und bestensfalls zu reproduzieren.
Für lustige Suchspiele haben wir keine Kapazitäten.

Tombra1889, 2024-06-14:
```
[site ] DEBUG 2024/06/14 20:56:47 set battery discharge control: false
[site ] DEBUG 2024/06/14 20:56:49 ----
[lp-1 ] DEBUG 2024/06/14 20:56:49 charge power: 0W
[lp-1 ] DEBUG 2024/06/14 20:56:49 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/06/14 20:56:49 charge power: 25W
[lp-2 ] DEBUG 2024/06/14 20:56:49 charge currents: [0.209 0.096 0.034]A
[lp-2 ] DEBUG 2024/06/14 20:56:50 set session soc limit: 21
[lp-2 ] DEBUG 2024/06/14 20:57:00 set session soc limit: 21
[lp-2 ] DEBUG 2024/06/14 20:57:10 set session soc limit: 21
[lp-2 ] DEBUG 2024/06/14 20:57:20 set session soc limit: 21
```

danach passiert nichts mehr.

andig, 2024-06-14:
Hier fehlen leider alle angefragten Informationen. Close?

Tombra1889, 2024-06-14:
Ich kann leider nicht mehr beschreiben. Es handelt sich um einen SHM mit der aktuellen Firmware. Die Batteriesperre, also der Schalter in evcc ist eingeschaltet. Wenn ich diesen ausschalte, macht evcc keinen mucks mehr und ich muss den Prozess neustarten.

andig, 2024-06-14:
Und dann? Neuer Prozess ohne Batteriesperre geht? Config? Trace log?

Tombra1889, 2024-06-14:
Ne der Zustand bleibt nach dem Neustart auf: 
[image]
Wenn man nicht an den Schalter geht funktioniert evcc, sonst stürzt es direkt ab.

```
[site ] DEBUG 2024/06/14 22:26:34 set battery discharge control: false
[modbus] TRACE 2024/06/14 22:26:34 modbus: send 11 5b 00 00 00 0b 03 10 9c d7 00 02 04 00 00 03 23
[modbus] TRACE 2024/06/14 22:26:34 modbus: recv 11 5b 00 00 00 06 03 10 9c d7 00 02
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.87: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.87: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.87: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.87: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 login for 192.168.66.88:9522
[sma ] TRACE 2024/06/14 22:26:35 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.87: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 requestValues for 192.168.66.88:9522: 0x5100 0x263F00 0x263FFF
[sma ] TRACE 2024/06/14 22:26:35 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 requestValues for 192.168.66.88:9522: 0x5100 0x295A00 0x295AFF
[sma ] TRACE 2024/06/14 22:26:35 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 requestValues for 192.168.66.88:9522: 0x5100 0x411E00 0x4120FF
[sma ] TRACE 2024/06/14 22:26:35 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 requestValues for 192.168.66.88:9522: 0x5100 0x464000 0x4642FF
[sma ] TRACE 2024/06/14 22:26:35 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 requestValues for 192.168.66.88:9522: 0x5100 0x464800 0x4655FF
[sma ] TRACE 2024/06/14 22:26:35 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 requestValues for 192.168.66.88:9522: 0x5100 0x465700 0x4657FF
[sma ] TRACE 2024/06/14 22:26:35 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:35 requestValues for 192.168.66.88:9522: 0x5100 0x491E00 0x495DFF
[sma ] TRACE 2024/06/14 22:26:35 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 requestValues for 192.168.66.88:9522: 0x5180 0x214800 0x2148FF
[sma ] TRACE 2024/06/14 22:26:36 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 requestValues for 192.168.66.88:9522: 0x5180 0x416400 0x4164FF
[sma ] TRACE 2024/06/14 22:26:36 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 recv 192.168.66.87: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 requestValues for 192.168.66.88:9522: 0x5200 0x237700 0x2377FF
[sma ] TRACE 2024/06/14 22:26:36 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 requestValues for 192.168.66.88:9522: 0x5380 0x251E00 0x251EFF
[sma ] TRACE 2024/06/14 22:26:36 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 requestValues for 192.168.66.88:9522: 0x5380 0x451F00 0x4521FF
[sma ] TRACE 2024/06/14 22:26:36 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 requestValues for 192.168.66.88:9522: 0x5400 0x260100 0x2622FF
[sma ] TRACE 2024/06/14 22:26:36 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 recv 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma ] TRACE 2024/06/14 22:26:36 requestValues for 192.168.66.88:9522: 0x5400 0x462E00 0x462FFF
[sma ] TRACE 2024/06/14 22:26:36 send 192.168.66.88: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[... cut ...]
[semp ] TRACE 2024/06/14 22:26:39 send: {XMLName:{Space: Local:} Xmlns:urn:schemas-upnp-org:device-1-0 SpecVersion:{Major:1 Minor:0} Device:{DeviceType:urn:schemas-simple-energy-management-protocol:device:Gateway:1 FriendlyName:evcc Manufacturer:github.com/evcc-io/evcc ModelName:EVCC SEMP Server 0.127.1 UDN:uuid:3ff31bcb-2a80-11ef-9579-e45f011a0327 PresentationURL:http://192.168.66.193:7070 ServiceDefinition:{Xmlns:urn:schemas-simple-energy-management-protocol:service-1-0 Server:http://192.168.66.193:7070 BasePath:/semp Transport:HTTP/Pull ExchangeFormat:XML WsVersion:1.1.0} ServiceList:[]}}
```

andig, 2024-06-14:
Was stürzt da ab??? Meinst du es hängt?

Tombra1889, 2024-06-14:
Man kann andere Button zwar noch betätigen, aber es passiert quasi von der Steuerung und Anzeige nichts mehr.

andig, 2024-06-15:
Also deadlock. Dann bitte analog https://github.com/evcc-io/evcc/issues/12093#issuecomment-1929337104.

Tombra1889, 2024-06-15:
Ich hoffe ich habe es korrekt gemacht: 

> <span style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-size: medium; float: none; display: inline !important;">/debug/pprof/</span><br style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0);"><p style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0);">Set debug=1 as a query parameter to export in legacy text format</p><br style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0);"><span style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-size: medium; float: none; display: inline !important;">Types of profiles available:</span>
> Count | Profile
> -- | --
> 152 | allocs
> 0 | block
> 0 | cmdline
> 55 | goroutine
> 152 | heap
> 0 | mutex
> 0 | profile
> 11 | threadcreate
> 0 | trace
> 
> <a href="http://192.168.66.193:7070/debug/pprof/goroutine?debug=2" style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px;">full goroutine stack dump</a><span style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-size: medium; float: none; display: inline !important;"></span><br style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0);"><p style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0);">Profile Descriptions:</p><ul style="font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none; caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0);"><li><div class="profile-name" style="display: inline-block; width: 6rem;">allocs:<span class="Apple-converted-space"> </span></div><span class="Apple-converted-space"> </span>A sampling of all past memory allocations</li><li><div class="profile-name" style="display: inline-block; width: 6rem;">block:<span class="Apple-converted-space"> </span></div><span class="Apple-converted-space"> </span>Stack traces that led to blocking on synchronization primitives</li><li><div class="profile-name" style="display: inline-block; width: 6rem;">cmdline:<span class="Apple-converted-space"> </span></div><span class="Apple-converted-space"> </span>The command line invocation of the current program</li><li><div class="profile-name" style="display: inline-block; width: 6rem;">goroutine:<span class="Apple-converted-space"> </span></div><span class="Apple-converted-space"> </span>Stack traces of all current goroutines. Use debug=2 as a query parameter to export in the same format as an unrecovered panic.</li><li><div class="profile-name" style="display: inline-block; width: 6rem;">heap:<span class="Apple-converted-space"> </span></div><span class="Apple-converted-space"> </span>A sampling of memory allocations of live objects. You can specify the gc GET parameter to run GC before taking the heap sample.</li><li><div class="profile-name" style="display: inline-block; width: 6rem;">mutex:<span class="Apple-converted-space"> </span></div><span class="Apple-converted-space"> </span>Stack traces of holders of contended mutexes</li><li><div class="profile-name" style="display: inline-block; width: 6rem;">profile:< [... cut ...]
> mutex:  Stack traces of holders of contended mutexes
> profile:  CPU profile. You can specify the duration in the seconds GET parameter. After you get the profile file, use the go tool pprof command to investigate the profile.
> threadcreate:  Stack traces that led to the creation of new OS threads
> trace:  A trace of execution of the current program. You can specify the duration in the seconds GET parameter. After you get the trace file, use the go tool trace command to investigate the trace.

Tombra1889, 2024-06-15:
> goroutine 830 [running]:
> runtime/pprof.writeGoroutineStacks({0x3624c00, 0x4001470fc0})
> 	runtime/pprof/pprof.go:743 +0x6c
> runtime/pprof.writeGoroutine({0x3624c00?, 0x4001470fc0?}, 0x7f6ef30ee0?)
> 	runtime/pprof/pprof.go:732 +0x2c
> runtime/pprof.(*Profile).WriteTo(0x5f80c50?, {0x3624c00?, 0x4001470fc0?}, 0xc?)
> 	runtime/pprof/pprof.go:369 +0x148
> net/http/pprof.handler.ServeHTTP({0x4002548ca1, 0x9}, {0x3724340, 0x4001470fc0}, 0x4000d2a7e0)
> 	net/http/pprof/pprof.go:267 +0x444
> net/http/pprof.Index({0x3724340, 0x4001470fc0}, 0x4000d2a7e0?)
> 	net/http/pprof/pprof.go:384 +0xc4
> net/http.HandlerFunc.ServeHTTP(0x5fe59c0?, {0x3724340?, 0x4001470fc0?}, 0x4000cb1360?)
> 	net/http/server.go:2166 +0x38
> net/http.(*ServeMux).ServeHTTP(0x4000d2a6c0?, {0x3724340, 0x4001470fc0}, 0x4000d2a7e0)
> 	net/http/server.go:2683 +0x1a4
> github.com/gorilla/mux.(*Router).ServeHTTP(0x400081c000, {0x3724340, 0x4001470fc0}, 0x4000d2a000)
> 	github.com/gorilla/mux@v1.8.1/mux.go:212 +0x194
> net/http.serverHandler.ServeHTTP({0x400246e660?}, {0x3724340?, 0x4001470fc0?}, 0x6?)
> 	net/http/server.go:3137 +0xbc
> net/http.(*conn).serve(0x4001000fc0, {0x372b830, 0x4000ab5cb0})
> 	net/http/server.go:2039 +0x508
> created by net/http.(*Server).Serve in goroutine 1
> 	net/http/server.go:3285 +0x3f0
> 
> goroutine 1 [IO wait]:
> internal/poll.runtime_pollWait(0x7f6f2be2d0, 0x72)
> 	runtime/netpoll.go:345 +0xa0
> internal/poll.(*pollDesc).wait(0x13?, 0x2bc2c?, 0x0)
> 	internal/poll/fd_poll_runtime.go:84 +0x28
> internal/poll.(*pollDesc).waitRead(...)
> 	internal/poll/fd_poll_runtime.go:89
> internal/poll.(*FD).Accept(0x400050d280)
> 	internal/poll/fd_unix.go:611 +0x250
> net.(*netFD).accept(0x400050d280)
> 	net/fd_unix.go:172 +0x28
> net.(*TCPListener).accept(0x4000e472c0)
> 	net/tcpsock_posix.go:159 +0x28
> net.(*TCPListener).Accept(0x4000e472c0)
> 	net/tcpsock.go:327 +0x2c
> net/http.(*Server).Serve(0x400129e000, {0x3723f20, 0x4000e472c0})
> 	net/http/server.go:3255 +0x2a8
> net/http.(*Server).ListenAndServe(0x400129e000)
> 	net/http/server.go:3184 +0x84
> github.com/evcc-io/evcc/cmd.runRoot(0x5fb0a40, {0x2a97c1f?, 0x7?, 0x2a8abda?})
> 	github.com/evcc-io/evcc/cmd/root.go:310 +0x1b3c
> github.com/spf13/cobra.(*Command).execute(0x5fb0a40, {0x40000a8090, 0x1, 0x1})
> 	github.com/spf13/cobra@v1.8.0/command.go:987 +0x828
> github.com/spf13/cobra.(*Command).ExecuteC(0x5fb0a40)
> 	github.com/spf13/cobra@v1.8.0/command.go:1115 +0x344
> github.com/spf13/cobra.(*Command).Execute(...)
> 	github.com/spf13/cobra@v1.8.0/command.go:1039
> github.com/evcc-io/evcc/cmd.Execute()
> 	github.com/evcc-io/evcc/cmd/root.go:107 +0x24
> main.main()
> 	github.com/evcc-io/evcc/main.go:44 +0x38
> 
> goroutine 6 [select]:
> google.golang.org/grpc/internal/grpcsync.(*CallbackSerializer).run(0x4000b1b5d0, {0x372b868, 0x40005f8960})
> 	google.golang.org/grpc@v1.63.2/internal/grpcsync/callback_serializer.go:76 +0xc8
> created by google.golang.org/grpc/internal/grpcsync.NewCallbackSerializer in goroutine 1
> 	google.golang.org/grpc@v1.63.2/internal/grpcsync/callback_serializer.go:52 +0x118
> 
> goroutine 7 [select]:
> google.golang.org/grpc/internal/grpcsync.(*CallbackSerializer).run(0x4000b1b600, {0x372b868, 0x40005f89b0})
> 	google.golang.org/grpc@v1.63.2/internal/grpcsync/callback_serializer.go:76 +0xc8
> created by google.golang.org/grpc/internal/grpcsync.NewCallbackSerializer in goroutine 1
> 	google.golang.org/grpc@v1.63.2/internal/grpcsync/callback_serializer.go:52 +0x118
> 
> goroutine 8 [select]:
> google.golang.org/grpc/internal/grpcsync.(*CallbackSerializer).run(0x4000b1b630, {0x372b868, 0x40005f8a00})
[... 628 lines cut ...]
> 	github.com/evcc-io/evcc/core/site_battery.go:46 +0x98
> github.com/evcc-io/evcc/core.(*Site).SetBatteryDischargeControl(0x400120fa40, 0x0)
> 	github.com/evcc-io/evcc/core/site_api.go:347 +0xfc
> github.com/evcc-io/evcc/server.(*HTTPd).RegisterSiteHandlers.boolHandler.handler[...].func30(0x4002187d01)
> 	github.com/evcc-io/evcc/server/http_site_handler.go:117 +0x94
> net/http.HandlerFunc.ServeHTTP(0x276bae0?, {0x3728720?, 0x40023d7620?}, 0x1b?)
> 	net/http/server.go:2166 +0x38
> github.com/gorilla/handlers.(*cors).ServeHTTP(0x40006c1170, {0x3728720, 0x40023d7620}, 0x4000ce26c0)
> 	github.com/gorilla/handlers@v1.5.2/cors.go:136 +0x5bc
> github.com/gorilla/handlers.CompressHandler.CompressHandlerLevel.func1({0x3724340, 0x40000e3c00}, 0x4000ce26c0)
> 	github.com/gorilla/handlers@v1.5.2/compress.go:141 +0x4e8
> net/http.HandlerFunc.ServeHTTP(0x276bae0?, {0x3724340?, 0x40000e3c00?}, 0xc?)
> 	net/http/server.go:2166 +0x38
> github.com/evcc-io/evcc/server.jsonHandler.func1({0x3724340, 0x40000e3c00}, 0x4000ce26c0)
> 	github.com/evcc-io/evcc/server/http_site_handler.go:73 +0xf0
> net/http.HandlerFunc.ServeHTTP(0x4000ce2480?, {0x3724340?, 0x40000e3c00?}, 0x22488?)
> 	net/http/server.go:2166 +0x38
> github.com/gorilla/mux.(*Router).ServeHTTP(0x400081c000, {0x3724340, 0x40000e3c00}, 0x4000ce2000)
> 	github.com/gorilla/mux@v1.8.1/mux.go:212 +0x194
> net/http.serverHandler.ServeHTTP({0x4001d41440?}, {0x3724340?, 0x40000e3c00?}, 0x6?)
> 	net/http/server.go:3137 +0xbc
> net/http.(*conn).serve(0x4001e69440, {0x372b830, 0x4000ab5cb0})
> 	net/http/server.go:2039 +0x508
> created by net/http.(*Server).Serve in goroutine 1
> 	net/http/server.go:3285 +0x3f0

andig, 2024-06-15:
Kannst du die goroutines als Datei anhängen?

Tombra1889, 2024-06-15:
[goroutine.txt](https://github.com/user-attachments/files/15846197/goroutine.txt)
Musste nochmal eine neue erstellen, hatte den Service schon neugestartet.

andig, 2024-06-15:
Sieht aus wie Deadlock in SetBatteryDischargeControl. Ich schaus an, danke! Dachte egtl, der wäre gelöst.

andig, 2024-06-15:
SetBatteryDischargeControl aquires lock, calls applyBatteryMode calls SetBatteryMode which wants to aquire lock. Bamm.

closed 2024-06-15 by andig