evcc-io/evcc #14371: Fahrzeug Porsche

Mestro82, 2024-06-14:
### Describe the bug

Seit kurzem kann ich bei einem Bekannten nicht mehr die Infos aus dem Porsche Portal anzeigen.
Hat es vll mit dem Screenshot zu tun?

### Steps to reproduce

1. Status Fahrzeug zeigt keine Information mehr
[image]
 an
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: porsche
    title: # Wird in der Benutzeroberfläche angezeigt (optional)
    user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: W... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 50 # Akkukapazität in kWh (optional)
```

### Log details

```text
pi@raspi:~ $ evcc vehicle
[main  ] INFO 2024/06/14 21:58:29 evcc 0.126.4
[main  ] INFO 2024/06/14 21:58:29 using config file: /etc/evcc.yaml
[main  ] ERROR 2024/06/14 21:58:31 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'porsche': unexpected status: 403 (Forbidden)
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'porsche': unexpected status: 403 (Forbidden)
Capacity: 93.0kWh
Features: [Offline Retryable]
```

### What type of operating system are you running?

Linux

### Version

0.126.4

NoMoreInfo, 2024-06-15:
Bei mir klappt es seit ein paar Tagen auch nicht mehr. Heute ein paar Mal das Captcha im Browser machen müssen bevor ich mich wieder ohne einloggen konnte. EVCC neu gestartet, aber (noch) klappt es nicht.

psattl, 2024-06-15:
Bei mir ist auch keine Verbindung mehr möglich:
[main ] ERROR 2024/06/15 14:27:42 creating vehicle taycan2 failed: cannot create vehicle type 'template': cannot create vehicle type 'porsche': unexpected status: 403 (Forbidden). 

Version 0.127.1

AndaleR, 2024-06-15:
Porsche gehört ja auch irgendwie zu VW - und die haben ihr System umgestellt. War beim Skoda Enyaq auch so - und hier wurde es beim Update Donnerstag/Freitag behoben. Denke, dass es bei Porsche dann auch ein solches Problem wird und mit einem der nächsten Updates behoben werden kann.

SwissFanboy, 2024-06-17:
Habe dasselbe Problem: [](https://github.com/evcc-io/evcc/discussions/14404)

normanmaurer, 2024-06-17:
Es scheint die API hat sich geändert... https://github.com/CJNE/ha-porscheconnect/issues/242

Der Entwickler hat angefangen den code umzuschreiben das es wieder geht:

https://github.com/CJNE/pyporscheconnectapi/commits/new_api/

Vielleicht sind da ja schon die Änderungen dabei die hier benötigt werden

Gandalf-1, 2024-06-19:
Auch bei mir klappt es plötzlich seit ein paar Tagen nicht mehr, keine Verbindung mehr möglich. EVCC Updates und Neustarts haben das Problem nicht gelöst.

**Configuration details**
```
- type: template
  template: porsche 
  title: Porsche Taycan  
  icon: car  
  user: **********  
  password: *********  
  vin: **********  
  capacity: 71  
  phases: 3  
  cache: 15m  
  mode: now  
  name: ev3
```

```
**Log details**
[main  ] ERROR 2024/06/19 11:12:12 creating vehicle ev3 failed: cannot create vehicle type 'template': cannot create vehicle type 'porsche': unexpected status: 403 (Forbidden)

ev3
---
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'porsche': unexpected status: 403 (Forbidden)
Capacity: 71.0kWh
Features: [Offline Retryable]
```

**What type of operating system are you running?**
Linux

**Version**
0.127.2

damienmc3, 2024-06-21:
Hatte es auch schon hier gemeldet:

https://github.com/evcc-io/evcc/discussions/12445

andig, 2024-06-22:
Depends on https://github.com/CJNE/ha-porscheconnect/issues/242

Jimnaldo, 2024-06-27:
Is there a solution yet?

normanmaurer, 2024-07-02:
FYI with the changes that were done in https://github.com/CJNE/pyporscheconnectapi/tree/new_api it's at least possible to "read" states again. So maybe this is a good starting point at least and will allow evcc to do everything needed except changing charging levels etc

normanmaurer, 2024-07-02:
This is with the code installed and used in homeassistant: 

[image]

andig, 2024-07-02:
Looks like a full rewrite. We'll need a MITM trace of the vehicle communication or someone must re-engineer what the HA plugin does.

normanmaurer, 2024-07-02:
@andig if you can give me some details how I could produce so a trace I would be happy to help

andig, 2024-07-02:
Sorry, not using HA…

normanmaurer, 2024-07-04:
Maybe @fredriklj can share such a trace as I suspect he had to obtain one as well for his work... OSS for the win :)

Mestro82, 2024-07-19:
Any news ok that topic?

andig, 2024-07-19:
Closed in https://github.com/evcc-io/evcc/commit/cd27ee2ce1a37fa86c6603f5ff59c2820a0a7a84. Nobody has found a way to update- hence we're deprecating the Porsche API. May come back at a later time.

closed 2024-07-19 by andig

NoMoreInfo, 2024-07-19:
Sounds like it is solved for OpenWB!? 
https://forum.openwb.de/viewtopic.php?t=1943&start=30

snapscanii, 2024-07-28:
https://developer.porsche.com/request-access

devdems, 2024-07-28:
@snapscanii they will not give you login access to this page. I think most of us already tried this.

diehappy69, 2024-10-13:
Hi,
myPorsche funktioniert über IoBroker. Kann jemand sagen, ob dann ein porsche template erzeugt werden kann oder wie ich mit dem generische Vehicle und der Iobroker Verbindung die Daten rüberziehen kann?

Wichtig ist, dass das Fahrzeug nur während des Ladevorgangs gepollt wird, ansonsten geht es nicht in den Tiefschlaf und die 12V Batterie und dann die HV Batter entleert sich (das hatte ich leider schon einmal).

Grüße und danke für die HIlfe

andig, 2024-10-13:
Wir sind mit Porsche im Kontakt. Schauen wir mal, ob da was bei raus kommt...

MarkusGH, 2024-11-03:
@andig: Looks like pyporscheconnectapi has a working implementation (beware the rate limit):
https://github.com/CJNE/pyporscheconnectapi/issues/46

normanmaurer, 2024-11-06:
I am currently using a combination of https://github.com/CJNE/pyporscheconnectapi/issues/46, shell scripting and mqtt to integrate my taycan. Works "well enough" until we have a native support again.

MarkusGH, 2024-11-26:
> I am currently using a combination of [CJNE/pyporscheconnectapi#46](https://github.com/CJNE/pyporscheconnectapi/issues/46), shell scripting and mqtt to integrate my taycan. Works "well enough" until we have a native support again.

@normanmaurer: Could you share that?

WolfImBusch, 2024-11-26:
Hi @MarkusGH 

maybe I can help out describing my solution divided in the following three steps

1) Install the files from the repository https://github.com/CJNE/ha-porscheconnect according to the instructions "manual installation". Important: use the branch **"new_api"** since the changes are not currently not committed to the "main" branch. Please check if information of you car appear in Home Assistant
2) Push the information of your car from Home Assistant via MQTT (e.g. via Statestream (https://www.home-assistant.io/integrations/mqtt_statestream)
3) Adapt the vehicle configuration in evcc.yaml as follows

```
vehicles:
- name: PAG
  type: custom
  title: Cayenne
  capacity: 21.8
  soc:
    source: mqtt
    topic: homeassistant/sensor/state_of_charge/state
  range:
    source: mqtt
    topic: homeassistant/sensor/remaining_range_electric/state
```
The Home Assistant plugin updates the information approximately every 30 minutes so target SOC charging is "challenging". But hey - much better than before ... :-)

Fingers crossed! :-)

normanmaurer, 2024-11-27:
@MarkusGH I basically do the same as @WolfImBusch with the difference that I use a shell script for 2.

```
#!/usr/bin/bash

DIR="${BASH_SOURCE%/*}"
if [[ ! -d "$DIR" ]]; then DIR="$PWD"; fi
. "$DIR/config.sh"

CAPABILITIES=`/home/norman/pyporsche/bin/porschecli capabilities -a`

for VIN in ${VINS[@]}; do
    MEASUREMENTS=`echo $CAPABILITIES | jq ". | select(.vin==\"$VIN\")" | jq '.measurements'`
    CURRENT_CHARGE=`echo $MEASUREMENTS | jq '.[] | select(.key=="BATTERY_LEVEL")' | jq ."value.percent"`
    CURRENT_RANGE=`echo $MEASUREMENTS | jq '.[] | select(.key=="E_RANGE")' | jq ."value.kilometers"`
    CURRENT_MILEAGE=`echo $MEASUREMENTS | jq '.[] | select(.key=="MILEAGE")' | jq ."value.kilometers"`
    CURRENT_MIN_SOC=`echo $MEASUREMENTS | jq '.[] | select(.key=="CHARGING_PROFILES")' | jq ".value.list[] | select(.name==\"Home\")" | jq '.minSoc'`
    CURRENT_CHARGING_STATE=`echo $MEASUREMENTS | jq '.[] | select(.key=="BATTERY_CHARGING_STATE")' | jq ."value.status"`

    /usr/bin/mosquitto_pub -r -h $MQTT_HOST -u $MQTT_USER -P $MQTT_PASSWORD -t $MQTT_TOPIC/$VIN/battery_level -m $CURRENT_CHARGE
    /usr/bin/mosquitto_pub -r -h $MQTT_HOST -u $MQTT_USER -P $MQTT_PASSWORD -t $MQTT_TOPIC/$VIN/range -m $CURRENT_RANGE
    /usr/bin/mosquitto_pub -r -h $MQTT_HOST -u $MQTT_USER -P $MQTT_PASSWORD -t $MQTT_TOPIC/$VIN/mileage -m $CURRENT_MILEAGE
    /usr/bin/mosquitto_pub -r -h $MQTT_HOST -u $MQTT_USER -P $MQTT_PASSWORD -t $MQTT_TOPIC/$VIN/min_soc -m $CURRENT_MIN_SOC
    /usr/bin/mosquitto_pub -r -h $MQTT_HOST -u $MQTT_USER -P $MQTT_PASSWORD -t $MQTT_TOPIC/$VIN/charging_state -m $CURRENT_CHARGING_STATE
done
```

MarkusGH, 2024-11-27:
@normanmaurer:
Amazing! That saves so much time.
Can you please post your vehicle configuration in evcc.yaml also ?

WolfImBusch, 2024-11-28:
Hi @normanmaurer 

I'm sure you already know that meanwhile a new version for the HA Plugin has been officially released - yeah! 😄

Is there a special reason why you collect/send the current odometer information to evcc?

Thank's for a short hint! 🙏

ivoks, 2024-11-29:
Thanks for this! I copied @WolfImBusch's approach:

  soc:
    source: mqtt
    topic: homeassistant/sensor/taycan_state_of_charge/state
  limitsoc:
    source: mqtt
    topic: homeassistant/number/taycan_target_state_of_charge/state
  climater:
    source: mqtt
    topic: homeassistant/switch/taycan_remote_climatisation/state
  status:
    source: mqtt
    topic: homeassistant/sensor/taycan_charging_status/state
  range:
    source: mqtt
    topic: homeassistant/sensor/taycan_remaining_range_electric/state
  odometer:
    source: mqtt
    topic: homeassistant/sensor/taycan_mileage/state

normanmaurer, 2024-11-29:
> Hi @normanmaurer
> 
> I'm sure you already know that meanwhile a new version for the HA Plugin has been officially released - yeah! 😄
> 
> Is there a special reason why you collect/send the current odometer information to evcc?

Because why not ;) 
> 
> Thank's for a short hint! 🙏

normanmaurer, 2024-11-29:
@MarkusGH yes same as all the MQTT_ ones

MarkusGH, 2024-12-14:
@normanmaurer: Does it still work for you? Here it stopped working. See: https://github.com/CJNE/pyporscheconnectapi/issues/66

normanmaurer, 2024-12-15:
@MarkusGH yeah still works for me

MarkusGH, 2024-12-15:
@normanmaurer: Could you please post the output from `porschecli -d capabilities -a`?

ivoks, 2024-12-29:
@andig any news on communication with Porsche? Should we wait? Should we work on the implementation ourselves?

Thanks!

andig, 2024-12-31:
Ongoing process- no ETA. If you want to have a quicker solution go for it...