evcc-io/evcc #14373: Set password dialog appears on invalid evcc.yaml

naltatis, 2024-06-14:
### Describe the bug

follow up to https://github.com/evcc-io/evcc/issues/14241

When you try to start evcc with a broken evcc.yaml file it presents the "Set Admin Password" dialoge. The reason is, that it hasn't established a database connection but parts of the code (auth endpoints) rely on reading from the database.

Wanted behavior:
- Establish database connection even if evcc.yaml fails. In the future database location should only be defined by env or cli parameter.
- settings api should return error when read access fails (no connection, db locked, ...)

\cc @andig 

### Steps to reproduce

1. use a working configuration with configured password
2. introduce a syntax error in evcc.yaml
3. restart evcc
4. open the UI
5. you'll see a set admin password prompt

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.127.1

andig, 2024-06-15:
Wenn die DB nicht initialisiert ist können wir ja überhaupt gar nichts tun. Sollten wir das vllt. als einen speziellen `fatal` Error anzeigen? Das würde weitere Änderungen ersparen. Hier macht es ja auch keinen Sinn, die Configoberläche auz zu machen?

naltatis, 2024-06-19:
> Hier macht es ja auch keinen Sinn, die Configoberläche auz zu machen?

Ja, können wir auch machen. Dann brächten wir hier wirklich zwei Klassen von Fehlern. Die durch die UI behebbaren (üblicherweise Geräte, Integrationen, ...) und die katastrophalen (evcc.yaml unlesbar, keine db verbindung, ...).

@andig Wie würdest du die API mäßig unterscheidbar machen? Aktuell läuft das ja alles unter `fatal`.