evcc-io/evcc #14376: Deye High Voltage 3p Hybrid Inverters need different template

Robwagi, 2024-06-15:
**Is your feature request related to a problem? Please describe.**
When using Modbus RTU via RS485 to USB on Deye 3p hybrid HIGH VOLTAGE inverters some of the received values are to small by factor 10.

**Describe the solution you'd like**
Additional template for deye high voltage 3p hybrid inverter variants.

**Describe alternatives you've considered**
Using custom template could be an option, but its to complicated for me.

[MODBUSRTU三相高压储能通信规约V104（高压）(3)(1)1111_SG01-HP3-AM2.pdf](https://github.com/user-attachments/files/15844917/MODBUSRTU.V104.3.1.1111_SG01-HP3-AM2.pdf)

andig, 2024-06-15:
@premultiply die HV brauchen also einfach andere Templates als die NV bzw. Mini-WRs?

Robwagi, 2024-06-15:
> @premultiply die HV brauchen also einfach andere Templates als die NV bzw. Mini-WRs?

Ansonsten hat bei mir alles genau so funktioniert

Robwagi, 2024-06-22:
[deye-hybrid-hp3.yaml.zip](https://github.com/user-attachments/files/15940919/deye-hybrid-hp3.yaml.zip)

Ich hab mal ein template erstellt. Eigentlich müsste es so passen. Leider habe ich es nicht geschafft lokal zu bauen und das ganze zu testen.

andig, 2024-06-23:
Klasse- magst Du einen Pull Request draus machen? Lokal bauen ist in der Readme beschrieben- alles kein Hexenwerk!

Robwagi, 2024-06-23:
Hab es leider trotz Anleitung auf meinem Mac nicht geschafft. VSCode installiert, Go und Node installiert inkl. der VSCode Erweiterungen. Kriege bei den make Befehlen aber leider nur Fehlermeldungen mit Access Denied und es klappt einfach nicht.

andig, 2024-06-23:
> Kriege bei den make Befehlen aber leider nur Fehlermeldungen mit Access Denied und es klappt einfach nicht.

Mach gerne Discussion dazu mit den konkreten Fehlermeldungen auf. Solange Du nicht zwischendurch mit sudo rum `fuchtelst` sollte das eigentlich alles problemlos gehen, gerade auf dem Mac!

PR #14522 (2024-06-30): Add Deye HP3 hybrid inverter
The template should fit all Deye high voltage inverters from the SG01HP3 series.

Fixes #14376
closed 2024-06-30 by andig