evcc-io/evcc #14377: Huawei Luna2000 - "Verhindere Entladung im Schnell-Modus" - Fehler: battery mode: modbus: exception '1' (illegal function), function '6'

dieterlind, 2024-06-15:
### Describe the bug

Wenn die Funktionalität "Verhindere Entladung im Schnell-Modus und bei geplantem Laden" aktiviert wird wird der Fehler "battery mode: modbus: exception '1' (illegal function), function '6'" im Log-File angezeigt.
Die Batterie wird trotz Aktivierung der Funktion im Schnel-Modus entladen.
Batterie: Huawei Luna2000

### Steps to reproduce

1. Aktivierung "Verhindere Entladung im Schnell-Modus und bei geplantem Laden"

### Configuration details

```yaml
- name: battery1
  type: template
  template: huawei-dongle-powersensor
  usage: battery
  modbus: tcpip
  id: 1
  host: 192.168.0.11 # Hostname
  port: 502 # Port
  timeout: 15s # optional
```

### Log details

```text
[lp-1 ] DEBUG 2024/06/15 09:14:26 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/06/15 09:14:26 charge power: 0W
[lp-2 ] DEBUG 2024/06/15 09:14:26 charge currents: [0 0 0]A
[site ] DEBUG 2024/06/15 09:14:26 pv power: 4922W
[site ] DEBUG 2024/06/15 09:14:28 battery soc: 36%
[site ] DEBUG 2024/06/15 09:14:28 battery power: -200W
[site ] DEBUG 2024/06/15 09:14:28 grid meter: 792W
[site ] DEBUG 2024/06/15 09:14:28 grid currents: [3.67 5.17 6.24]A
[site ] DEBUG 2024/06/15 09:14:28 battery has priority at soc 36% (< 70%)
[site ] DEBUG 2024/06/15 09:14:28 site power: 792W
[lp-2 ] DEBUG 2024/06/15 09:14:28 !! session: chargeRater.chargedEnergy=2.0 - chargedAtStartup=0.0
[lp-2 ] DEBUG 2024/06/15 09:14:28 charge total import: 4280.213kWh
[lp-2 ] DEBUG 2024/06/15 09:14:28 charger status: A
[site ] ERROR 2024/06/15 09:14:28 battery mode: modbus: exception '1' (illegal function), function '6'
[site ] DEBUG 2024/06/15 09:14:41 ----
[openwb-2.0] TRACE 2024/06/15 09:14:41 modbus: send 3c 33 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/06/15 09:14:41 modbus: recv 3c 33 00 00 00 07 01 04 04 00 00 00 00
[lp-1 ] DEBUG 2024/06/15 09:14:41 charge power: 0W
[openwb-2.0] TRACE 2024/06/15 09:14:41 modbus: send 3c 34 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/06/15 09:14:41 modbus: recv 3c 34 00 00 00 09 01 04 06 00 00 00 00 00 00
[lp-1 ] DEBUG 2024/06/15 09:14:41 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/06/15 09:14:41 charge power: 0W
[lp-2 ] DEBUG 2024/06/15 09:14:41 charge currents: [0 0 0]A
[site ] DEBUG 2024/06/15 09:14:41 pv power: 4923W
[site ] DEBUG 2024/06/15 09:14:43 battery soc: 36%
[site ] DEBUG 2024/06/15 09:14:43 battery power: -200W
[site ] DEBUG 2024/06/15 09:14:43 grid meter: 803W
[site ] DEBUG 2024/06/15 09:14:43 grid currents: [3.72 5.15 6.23]A
[site ] DEBUG 2024/06/15 09:14:43 battery has priority at soc 36% (< 70%)
[site ] DEBUG 2024/06/15 09:14:43 site power: 803W
[lp-1 ] DEBUG 2024/06/15 09:14:43 !! session: chargeRater.chargedEnergy=14.9 - chargedAtStartup=0.0
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: send 3c 35 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: recv 3c 35 00 00 00 07 01 04 04 00 47 40 52
[lp-1 ] DEBUG 2024/06/15 09:14:43 charge total import: 4669.522kWh
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: send 3c 36 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: recv 3c 36 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: send 3c 37 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: recv 3c 37 00 00 00 05 01 04 02 00 01
[lp-1 ] DEBUG 2024/06/15 09:14:43 charger status: B
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: send 3c 38 00 00 00 06 01 04 27 b0 00 0a
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: recv 3c 38 00 00 00 17 01 04 14 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: send 3c 39 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: recv 3c 39 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: send 3c 3a 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/06/15 09:14:43 modbus: recv 3c 3a 00 00 00 05 01 04 02 00 01
[lp-1 ] DEBUG 2024/06/15 09:14:43 limitSoc reached: 91.0% > 90%
[site ] ERROR 2024/06/15 09:14:43 battery mode: modbus: exception '1' (illegal function), function '6'
[site ] DEBUG 2024/06/15 09:14:56 ----
[openwb-2.0] TRACE 2024/06/15 09:14:56 modbus: send 3c 3b 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/06/15 09:14:56 modbus: recv 3c 3b 00 00 00 07 01 04 04 00 00 00 00
[lp-1 ] DEBUG 2024/06/15 09:14:56 charge power: 0W
[openwb-2.0] TRACE 2024/06/15 09:14:56 modbus: send 3c 3c 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/06/15 09:14:56 modbus: recv 3c 3c 00 00 00 09 01 04 06 00 00 00 00 00 00
[lp-1 ] DEBUG 2024/06/15 09:14:56 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/06/15 09:14:56 charge power: 0W
[lp-2 ] DEBUG 2024/06/15 09:14:56 charge currents: [0 0 0]A
[site ] DEBUG 2024/06/15 09:14:56 pv power: 4937W
[site ] DEBUG 2024/06/15 09:14:58 battery soc: 36%
[site ] DEBUG 2024/06/15 09:14:58 battery power: -200W
[site ] DEBUG 2024/06/15 09:14:58 grid meter: 808W
[site ] DEBUG 2024/06/15 09:14:58 grid currents: [3.66 5.13 6.26]A
[site ] DEBUG 2024/06/15 09:14:58 battery has priority at soc 36% (< 70%)
[site ] DEBUG 2024/06/15 09:14:58 site power: 808W
[lp-2 ] DEBUG 2024/06/15 09:14:58 !! session: chargeRater.chargedEnergy=2.0 - chargedAtStartup=0.0
[lp-2 ] DEBUG 2024/06/15 09:14:58 charge total import: 4280.213kWh
[lp-2 ] DEBUG 2024/06/15 09:14:58 charger status: A
[site ] ERROR 2024/06/15 09:14:58 battery mode: modbus: exception '1' (illegal function), function '6'
```

### What type of operating system are you running?

Docker container

### Version

0.127.1

uwejanssen19, 2024-06-16:
Batterie sperren bei Schnelladen funktioniert bei 0.127.1 auch nicht mehr. In die Logs habe ich allerdings nicht hineingeschaut.
Habe auch LUNA2000.
Revert auf 0.126.6 funktioniert wieder.

andig, 2024-06-29:
@uwejanssen19 Sehr hilfreich- irgendwie scheint das aber nicht zu passen da es keine Änderung gab:

    git log 0.126.6...0.127.1 -- templates/definition/meter/huawei-sun2000-dongle-powersensor.yaml

bleibt leer. 

Hier findet sich auch nichts: https://github.com/evcc-io/evcc/commits/master/templates/definition/meter/huawei-sun2000-dongle-powersensor.yaml

dieterlind, 2024-07-08:
Still not working with 128.1

Standard-Log (nach ERROR gefiltert):
```
[site ] ERROR 2024/07/08 10:04:00 battery mode: modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 10:04:14 modbus: send 6a 1e 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 10:04:14 modbus: recv 6a 1e 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:14 modbus: send 6a 1f 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 10:04:14 modbus: recv 6a 1f 00 00 00 09 01 04 06 00 00 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: send 6a 20 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: recv 6a 20 00 00 00 07 01 04 04 00 47 c2 f9
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: send 6a 21 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: recv 6a 21 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: send 6a 22 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: recv 6a 22 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: send 6a 23 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: recv 6a 23 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: send 6a 24 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/08 10:04:15 modbus: recv 6a 24 00 00 00 05 01 04 02 00 00
[site ] ERROR 2024/07/08 10:04:15 battery mode: modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 10:04:29 modbus: send 6a 25 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 10:04:29 modbus: recv 6a 25 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:29 modbus: send 6a 26 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 10:04:29 modbus: recv 6a 26 00 00 00 09 01 04 06 00 00 00 00 00 00
[site ] ERROR 2024/07/08 10:04:30 battery mode: modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 10:04:44 modbus: send 6a 27 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 10:04:44 modbus: recv 6a 27 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:44 modbus: send 6a 28 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 10:04:44 modbus: recv 6a 28 00 00 00 09 01 04 06 00 00 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: send 6a 29 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: recv 6a 29 00 00 00 07 01 04 04 00 47 c2 f9
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: send 6a 2a 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: recv 6a 2a 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: send 6a 2b 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: recv 6a 2b 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: send 6a 2c 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: recv 6a 2c 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: send 6a 2d 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/08 10:04:45 modbus: recv 6a 2d 00 00 00 05 01 04 02 00 00
[site ] ERROR 2024/07/08 10:04:45 battery mode: modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 10:04:59 modbus: send 6a 2e 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 10:04:59 modbus: recv 6a 2e 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 10:04:59 modbus: send 6a 2f 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 10:04:59 modbus: recv 6a 2f 00 00 00 09 01 04 06 00 00 00 00 00 00
[site ] ERROR 2024/07/08 10:05:00 battery mode: modbus: exception '1' (illegal function), function '6' 
```

Modbus-Log:
```
[modbus] TRACE 2024/07/08 10:04:15 modbus: send 51 cd 00 00 00 06 01 03 90 89 00 02
[modbus] TRACE 2024/07/08 10:04:15 modbus: recv 51 cd 00 00 00 07 01 03 04 ff ff ff de
[modbus] TRACE 2024/07/08 10:04:15 modbus: send 51 ce 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/07/08 10:04:15 modbus: recv 51 ce 00 00 00 07 01 03 04 00 00 00 00
[modbus] TRACE 2024/07/08 10:04:15 modbus: send 51 cf 00 00 00 06 01 03 90 8c 00 01
[modbus] TRACE 2024/07/08 10:04:15 modbus: recv 51 cf 00 00 00 05 01 03 02 00 5a
[modbus] TRACE 2024/07/08 10:04:15 modbus: send 51 d0 00 00 00 06 01 06 b7 fc 00 00
[modbus] TRACE 2024/07/08 10:04:15 modbus: recv 51 d0 00 00 00 03 01 86 01
[modbus] TRACE 2024/07/08 10:04:30 modbus: send 51 d1 00 00 00 06 01 03 90 89 00 02
[modbus] TRACE 2024/07/08 10:04:30 modbus: recv 51 d1 00 00 00 07 01 03 04 ff ff ff de
[modbus] TRACE 2024/07/08 10:04:30 modbus: send 51 d2 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/07/08 10:04:30 modbus: recv 51 d2 00 00 00 07 01 03 04 00 00 00 00
[modbus] TRACE 2024/07/08 10:04:30 modbus: send 51 d3 00 00 00 06 01 03 90 8c 00 01
[modbus] TRACE 2024/07/08 10:04:30 modbus: recv 51 d3 00 00 00 05 01 03 02 00 5a
[modbus] TRACE 2024/07/08 10:04:30 modbus: send 51 d4 00 00 00 06 01 06 b7 fc 00 00
[modbus] TRACE 2024/07/08 10:04:30 modbus: recv 51 d4 00 00 00 03 01 86 01
[modbus] TRACE 2024/07/08 10:04:45 modbus: send 51 d5 00 00 00 06 01 03 90 89 00 02
[modbus] TRACE 2024/07/08 10:04:45 modbus: recv 51 d5 00 00 00 07 01 03 04 ff ff ff de
[modbus] TRACE 2024/07/08 10:04:45 modbus: send 51 d6 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/07/08 10:04:45 modbus: recv 51 d6 00 00 00 07 01 03 04 00 00 00 00
[... cut ...]
[modbus] TRACE 2024/07/08 10:05:00 modbus: send 51 da 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/07/08 10:05:00 modbus: recv 51 da 00 00 00 07 01 03 04 00 00 00 00
[modbus] TRACE 2024/07/08 10:05:00 modbus: send 51 db 00 00 00 06 01 03 90 8c 00 01
[modbus] TRACE 2024/07/08 10:05:00 modbus: recv 51 db 00 00 00 05 01 03 02 00 5a
[modbus] TRACE 2024/07/08 10:05:00 modbus: send 51 dc 00 00 00 06 01 06 b7 fc 00 00
[modbus] TRACE 2024/07/08 10:05:00 modbus: recv 51 dc 00 00 00 03 01 86 01 
```

andig, 2024-07-08:
Jetzt müssten wir raus finden, ob das schonmal ging, siehe https://github.com/evcc-io/evcc/issues/14377#issuecomment-2198126176. Bisher haben wir keine gute Indikation. Kann jemand die letzte funktionierende Version rausfinden?

dieterlind, 2024-07-08:
Die letzte funktionierende Version kann ich dir leider nicht nennen ... 
aber spannenderweise bekomme ich die Fehler jetzt auch, wenn die Funktion "Verhindere Entladung im Schnell-Modus" deaktiviert ist:

```
[watchdog] ERROR 2024/07/08 18:41:30 modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 18:41:44 modbus: send 8e 7f 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 18:41:44 modbus: recv 8e 7f 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:41:44 modbus: send 8e 80 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 18:41:44 modbus: recv 8e 80 00 00 00 09 01 04 06 00 00 00 00 00 00
[watchdog] ERROR 2024/07/08 18:41:45 modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 18:41:59 modbus: send 8e 81 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 18:41:59 modbus: recv 8e 81 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:41:59 modbus: send 8e 82 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 18:41:59 modbus: recv 8e 82 00 00 00 09 01 04 06 00 00 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: send 8e 83 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: recv 8e 83 00 00 00 07 01 04 04 00 47 c2 f9
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: send 8e 84 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: recv 8e 84 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: send 8e 85 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: recv 8e 85 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: send 8e 86 00 00 00 06 01 04 27 83 00 01
[watchdog] ERROR 2024/07/08 18:42:00 modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: recv 8e 86 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: send 8e 87 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/08 18:42:00 modbus: recv 8e 87 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:14 modbus: send 8e 88 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 18:42:14 modbus: recv 8e 88 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:14 modbus: send 8e 89 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 18:42:14 modbus: recv 8e 89 00 00 00 09 01 04 06 00 00 00 00 00 00
[watchdog] ERROR 2024/07/08 18:42:15 modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 18:42:29 modbus: send 8e 8a 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 18:42:29 modbus: recv 8e 8a 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:29 modbus: send 8e 8b 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 18:42:29 modbus: recv 8e 8b 00 00 00 09 01 04 06 00 00 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: send 8e 8c 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: recv 8e 8c 00 00 00 07 01 04 04 00 47 c2 f9
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: send 8e 8d 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: recv 8e 8d 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: send 8e 8e 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: recv 8e 8e 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: send 8e 8f 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: recv 8e 8f 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: send 8e 90 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/08 18:42:30 modbus: recv 8e 90 00 00 00 05 01 04 02 00 00
[watchdog] ERROR 2024/07/08 18:42:30 modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 18:42:44 modbus: send 8e 91 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 18:42:44 modbus: recv 8e 91 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:44 modbus: send 8e 92 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 18:42:44 modbus: recv 8e 92 00 00 00 09 01 04 06 00 00 00 00 00 00
[watchdog] ERROR 2024/07/08 18:42:45 modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 18:42:59 modbus: send 8e 93 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 18:42:59 modbus: recv 8e 93 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:42:59 modbus: send 8e 94 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 18:42:59 modbus: recv 8e 94 00 00 00 09 01 04 06 00 00 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:43:00 modbus: send 8e 95 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/07/08 18:43:00 modbus: recv 8e 95 00 00 00 07 01 04 04 00 47 c2 f9
[openwb-2.0] TRACE 2024/07/08 18:43:00 modbus: send 8e 96 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/08 18:43:00 modbus: recv 8e 96 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:43:00 modbus: send 8e 97 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/08 18:43:00 modbus: recv 8e 97 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:43:00 modbus: send 8e 98 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/08 18:43:00 modbus: recv 8e 98 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/08 18:43:00 modbus: send 8e 99 00 00 00 06 01 04 27 82 00 01
[... cut ...]
[watchdog] ERROR 2024/07/08 18:43:00 modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/08 18:43:14 modbus: send 8e 9a 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/08 18:43:14 modbus: recv 8e 9a 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/08 18:43:14 modbus: send 8e 9b 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/08 18:43:14 modbus: recv 8e 9b 00 00 00 09 01 04 06 00 00 00 00 00 00
[watchdog] ERROR 2024/07/08 18:43:15 modbus: exception '1' (illegal function), function '6'
```

andig, 2024-07-08:
Du könntest aber downgraden- oben gabs ja schon eine Idee, welche zuletzt funktioniert hatte. Konnte allerdings niemand bestätigen.

dieterlind, 2024-07-09:
Ich hab jetzt mal ein downgrade 0.126.6 gemacht.
Fehler ist trotzdem da:

```
[site ] ERROR 2024/07/09 13:20:15 battery mode: modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/09 13:20:29 modbus: send 00 17 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/09 13:20:29 modbus: recv 00 17 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/09 13:20:29 modbus: send 00 18 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/09 13:20:29 modbus: recv 00 18 00 00 00 09 01 04 06 00 00 00 00 00 00
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: send 00 19 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: recv 00 19 00 00 00 07 01 04 04 00 47 c2 f9
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: send 00 1a 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: recv 00 1a 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: send 00 1b 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: recv 00 1b 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: send 00 1c 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: recv 00 1c 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: send 00 1d 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/07/09 13:20:31 modbus: recv 00 1d 00 00 00 05 01 04 02 00 00
[site ] ERROR 2024/07/09 13:20:31 battery mode: modbus: exception '1' (illegal function), function '6'
[openwb-2.0] TRACE 2024/07/09 13:20:44 modbus: send 00 1e 00 00 00 06 01 04 27 74 00 02
[openwb-2.0] TRACE 2024/07/09 13:20:44 modbus: recv 00 1e 00 00 00 07 01 04 04 00 00 00 00
[openwb-2.0] TRACE 2024/07/09 13:20:44 modbus: send 00 1f 00 00 00 06 01 04 27 7b 00 03
[openwb-2.0] TRACE 2024/07/09 13:20:44 modbus: recv 00 1f 00 00 00 09 01 04 06 00 00 00 00 00 00
[site ] ERROR 2024/07/09 13:20:46 battery mode: modbus: exception '1' (illegal function), function '6' 
```

andig, 2024-07-09:
> Standard-Log (nach ERROR gefiltert):

Bitte den relevanten Logteil in richtiger Reihenfolge. Wie soll man sonst herausfinden, zu welchem Modbus Befehl der Fehler gehört? OpenWB ist uninteressant.

andig, 2024-07-09:
> Revert auf 0.126.6 funktioniert wieder.

@uwejanssen19 konnten wir nicht bestätigen, es gibt auch keine relevanten Änderungen. Passt dein Test?

andig, 2024-07-09:
Für ich ist leider unklar, was hier auf Modbus versucht wird. Die Logs sind nicht aussagekräftig.

uwejanssen19, 2024-07-10:
**Leider** nicht reproduzierbar mit 0.127.1 und auch nicht mit derzeit aktueller 0.128.2
(Ging schon mal nicht mit 0.127.1)
Meine Umgebung: 

RPI 4 mit
Linux openhabian 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

Wallbox: Elli Wallbox typ connect max. 11 kW
Vehicle: BYD Atto3 
Inverter1: HUAWEI SUN 2000 8000 Wp
Storage: LUNA 2000 10kWh
Inverter2: OPEN DTU 2 x 800Wp
Inverter3: AHOY DTU 1x 300Wp

Habe mich aufgerafft und versucht, den Fehler (Batterie entladen nicht gesperrt bei Schnelladen) zu reproduzieren.
Leider kein Erfolg.
Ich werde das weiter beobachten.

Ich habe ja keine Ahnung von der Software, aber:

Ist die beobachtete Exception wirklich Teil der battery mode Programmierung?
Es kommt ja vor dass Exceptions auftreten und nicht gefangen werden also ein falscher Text ausgegen wird.

Viel Erfolg weiterhin und Gratulation zu diesem phantastischen Projekt!
Uwe Janssen

andig, 2024-07-10:
Aktuell also kein reproduzierbares Fehlerbild. Vielleicht tritt der Fehler auch nur in bestimmten Umschaltszenarien auf?

Man dann das auch bewusst auf der Kommandozeile probieren:

    evcc meter <battery-name> --log trace,db:error --battery-mode <mode> --battery-mode-wait 2m

Mit mode: (normal, hold, charge). Dann mal alle durchprobieren. Das wait ist notwenig weil der Luna einen Watchdog hat, ggf. tritt es also nicht sofort auf.

uwejanssen19, 2024-07-10:
> Aktuell also kein reproduzierbares Fehlerbild. Vielleicht tritt der Fehler auch nur in bestimmten Umschaltszenarien auf?
>
> Man dann das auch bewusst auf der Kommandozeile probieren:
>
> ```
> evcc meter [...]
Verstehe nicht ganz, was da probiert wird.
1. Soll der evcc service im Hintergrund weiterlaufen?
2. simuliert das beschriebene Kommando das Laden?

uwejanssen19, 2024-07-10:
> 1. lieber nicht- dann ist es sortenrein. 2) nein, aber Modus `hold` wird beim Laden gesetzt, sollte also gleichwertig sein.

Habs probiert, log file unter:

https://drive.google.com/drive/folders/1tqbfR6VykTiEBWJoHvN0roMQsdpzMgWW?usp=sharing

andig, 2024-07-10:
Bevor ich lang suche: gibts Fehler? Bei welchem Modus?

uwejanssen19, 2024-07-11:
Kein  FehlerVon meinem/meiner Galaxy gesendet

andig, 2024-07-11:
Dann haben:

- kein Logfile das den Fehler zeigen würde
- keine Fehler im Test

Bei Euch beiden im Log taucht aber die OpenWB auf. Ist es vllt. ein Fehler mit der OpenWB?

Ohne weitere Erkenntnisse würde ich hier zu machen da kein Fehler erkennbar.

dieterlind, 2024-07-13:
Hallo!

Ich tu mir schwer, zu beurteilen welche Teil des logs zu welchem Fehler gehört.
Fakt ist, sobald ich die Funktion aktiviere erhalte ich Fehler derart im Logfile:

`[site ] ERROR 2024/07/13 10:13:18 battery mode: modbus: exception '1' (illegal function), function '6'`

Vor jedem Fehler stehen 2 Modbus-Einträge:

```
[modbus] TRACE 2024/07/13 10:14:03 modbus: send 05 14 00 00 00 06 01 06 b7 fc 00 00
[modbus] TRACE 2024/07/13 10:14:03 modbus: recv 05 14 00 00 00 03 01 86 01
[site ] ERROR 2024/07/13 10:14:03 battery mode: modbus: exception '1' (illegal function), function '6'
[modbus] TRACE 2024/07/13 10:14:18 modbus: send 05 18 00 00 00 06 01 06 b7 fc 00 00
[modbus] TRACE 2024/07/13 10:14:18 modbus: recv 05 18 00 00 00 03 01 86 01
[site ] ERROR 2024/07/13 10:14:18 battery mode: modbus: exception '1' (illegal function), function '6'
[modbus] TRACE 2024/07/13 10:14:33 modbus: send 05 1c 00 00 00 06 01 06 b7 fc 00 00
[modbus] TRACE 2024/07/13 10:14:33 modbus: recv 05 1c 00 00 00 03 01 86 01
[site ] ERROR 2024/07/13 10:14:33 battery mode: modbus: exception '1' (illegal function), function '6'
```

andig, 2024-07-13:
@dieterlind prima- Du hast es geschafft, das erste verwertbare Logfile zu produzieren :). Das Problem ist Register `B7FC`, also 47100:

```
      - case: 1 # normal
        set:
          source: const
          value: 0 # stop
          set:
            source: modbus
            {{- include "modbus" . | indent 10 }}
            register:
              address: 47100 # Forcible charge/discharge
              type: writesingle
              encoding: uint16
```

Wir schaffen es nicht, da Modus "normal" rein zu schreiben. Jetzt müsste jemand mit Luna mal ausprobieren ob das an der Reihenfolge liegt bzw. _warum_ das nicht klappt.

andig, 2024-07-17:
Mangels Test geschlossen.

closed 2024-07-17 by andig

dieterlind, 2024-07-18:
Hi! Sorry für die späte Antwort ... wie kann ich das am besten testen?? Eine Lösung hierfür wäre wohl für alle Besitzer einer Luna2000 hilfreich ...

andig, 2024-07-18:
Es muss getestet werden, wie bei Huawei die drei Batteriemodi normal/ hold/ charge umgestellt werden können. Die Implementierung aus https://github.com/evcc-io/evcc/pull/12449 scheint nicht oder nicht immer zu funktionieren. Es muss rausgefunden werden, warum es anscheinend bei einigen Anwendern funktioniert, bei anderen aber nicht.

/cc @mucki

mucki12, 2024-07-18:
Sorry, da ich im Sommer nie schnelles Laden benötige habe ich das noch nicht bemerkt. Kann ich aber natürlich gerne mal testen. Mir sind allerdings keine Änderungen an dem Huawei Modbus Register bekannt und bei der damaligen Umsetzung funktionierte das definitiv…

dieterlind, 2024-07-28:
Hab mir das heute noch einmal angeschaut ... 
Ausgangssituation

- Register 47100 (forciblechareordischarge) war auf 0
- Register 47087 (chargefromgrid) war auf 1
- Register 47246 (chargeordischargesettingmode) war auf 1 (von einem anderen Test zuvor)
- Register 47247 (forciblechargepower) war auf 5000

Geändert:

- Register 47100 (forciblechareordischarge) auf 1 gesetzt
- Register 47246 (chargeordischargesettingmode) auf 0 gesetzt (47083 (chargingandischargingperiod war auf 200 und blieb unverändert)

Da hat sich nicht wirklich was getan.

Dann zusätzlich geändert
- Register 47247 (forciblechargepower) war auf 4500

Damit hat eine Ladung mit ca. 1 kW begonnen ... allerdings nicht mehr

Dann noch 

- Register 47087 (chargefromgridfunction) auf false und 1-2 Minuten später wieder auf true

Siehe da, Batterie lädt mit 4500 W (wie zuvor eingestellt)

Danach 

- Register 47247 (forciblechargepower) war auf 5000

Und lädt jetzt auf 5000 W

Ich meine, dass nach Einstellung der Parameter für 47100 bzw. 47246 der Parameter für 47087 noch einmal auf false und anschließend auf true gesetzt werden muss, damit das wirklich funktioniert.

Oder konnten andere etwas Anderes beobachten??

andig, 2024-07-28:
Die Frage hier ist ja erstmal, wo der Fehler her kommt und warum? Funktionalität würde dann m.E. der zweite Schritt.

mucki12, 2024-08-15:
Bin aus dem Urlaub zurück und habe die Batteriesperre kurz getestet. Der Luna Hausspeicher wird wie gewünscht im Modus "Schnell" und entsprechender Nutzung der Funktion"Verhindere Entladung im Schnell-Modus und bei geplantem Laden." gesperrt und evcc verhindert wie gewünscht die Entladung.
Wenn die Funktion "Verhindere Entladung im Schnell-Modus und bei geplantem Laden." wieder deaktiviert wird, wird der Luna wieder wie gewünscht für die Ladung genutzt.

Kann den Grundfehler somit nicht nachstellen und habe daher auf Logs verzichtet....

Texter2024, 2024-09-23:
Ich habe das Problem auch. Luna entlädt sich obwohl gesperrt. Hier ist der log file dazu:

Ich hoffe dieser Threat ist noch offen und ihr könnt mir helfen. Jetzt wo die TAge wieder kürzer werden, ist das Thema mehr von Interesse. Es hat monatelang einwandfrei funktioniert. Bis auf Systemupdates habe ich nichts verändert. 

[image]

Huawei Luna 2000 10kWh Speicher
Fahrzeug MG4

evcc: v0.130.12

[lp-1 ] INFO 2024/09/23 19:37:10 start charging ->
[lp-1 ] DEBUG 2024/09/23 19:37:10 wake-up timer: stop
[lp-1 ] DEBUG 2024/09/23 19:37:10 soc estimated: 82.33% (vehicle: 82.20%)
[lp-1 ] DEBUG 2024/09/23 19:37:10 vehicle soc: 82%
[lp-1 ] DEBUG 2024/09/23 19:37:10 vehicle soc limit: 60%
[lp-1 ] DEBUG 2024/09/23 19:37:10 vehicle range: 281km
[site ] DEBUG 2024/09/23 19:37:10 ----
[lp-1 ] DEBUG 2024/09/23 19:37:10 charge power: 3542W
[lp-1 ] DEBUG 2024/09/23 19:37:10 charge currents: [14.7 0.009 0.012]A
[site ] DEBUG 2024/09/23 19:37:18 set battery mode: hold
[site ] DEBUG 2024/09/23 19:37:20 pv power: 0W
[site ] DEBUG 2024/09/23 19:37:21 battery soc: 87%
[site ] DEBUG 2024/09/23 19:37:21 battery power: 4087W
[site ] DEBUG 2024/09/23 19:37:22 grid meter: 13W
[site ] DEBUG 2024/09/23 19:37:24 grid currents: [10.7 -5.5 -5.01]A
[site ] DEBUG 2024/09/23 19:37:25 site power: 4200W
[lp-1 ] DEBUG 2024/09/23 19:37:25 detected active phases: 1p
[lp-1 ] DEBUG 2024/09/23 19:37:25 charge total import: 1825.846kWh
[lp-1 ] DEBUG 2024/09/23 19:37:25 charger status: C
[lp-1 ] DEBUG 2024/09/23 19:37:25 soc estimated: 82.33% (vehicle: 82.20%)
[lp-1 ] DEBUG 2024/09/23 19:37:25 vehicle soc: 82%
[lp-1 ] DEBUG 2024/09/23 19:37:25 vehicle soc limit: 60%
[lp-1 ] DEBUG 2024/09/23 19:37:25 vehicle range: 281km
[site ] DEBUG 2024/09/23 19:37:40 ----
[lp-1 ] DEBUG 2024/09/23 19:37:40 charge power: 3542W
[lp-1 ] DEBUG 2024/09/23 19:37:40 charge currents: [14.7 0.009 0.012]A
[site ] DEBUG 2024/09/23 19:37:45 pv power: 0W
[site ] DEBUG 2024/09/23 19:37:49 battery soc: 86%
[site ] DEBUG 2024/09/23 19:37:49 battery power: 4127W
[site ] DEBUG 2024/09/23 19:37:50 grid meter: -0W
[site ] DEBUG 2024/09/23 19:37:52 grid currents: [10.7 -5.51 -5.02]A
[site ] DEBUG 2024/09/23 19:37:53 site power: 4227W
[lp-1 ] DEBUG 2024/09/23 19:37:53 detected active phases: 1p
[lp-1 ] DEBUG 2024/09/23 19:37:53 charge total import: 1825.846kWh
[lp-1 ] DEBUG 2024/09/23 19:37:53 charger status: C
[lp-1 ] DEBUG 2024/09/23 19:37:53 soc estimated: 82.33% (vehicle: 82.20%)
[lp-1 ] DEBUG 2024/09/23 19:37:53 vehicle soc: 82%
[lp-1 ] DEBUG 2024/09/23 19:37:53 vehicle soc limit: 60%
[lp-1 ] DEBUG 2024/09/23 19:37:53 vehicle range: 281km
[lp-1 ] DEBUG 2024/09/23 19:37:57 set charge mode: off
[site ] DEBUG 2024/09/23 19:37:57 ----
[lp-1 ] DEBUG 2024/09/23 19:37:57 charge power: 3542W
[lp-1 ] DEBUG 2024/09/23 19:37:57 charge currents: [14.7 0.009 0.012]A
[site ] DEBUG 2024/09/23 19:38:03 set battery mode: normal
[site ] DEBUG 2024/09/23 19:38:10 pv power: 0W
[site ] DEBUG 2024/09/23 19:38:17 battery soc: 86%
[site ] DEBUG 2024/09/23 19:38:17 battery power: 4130W
[site ] DEBUG 2024/09/23 19:38:19 grid meter: -0W
[site ] DEBUG 2024/09/23 19:38:23 grid currents: [10.6 -5.52 -5.02]A
[site ] DEBUG 2024/09/23 19:38:25 site power: 4230W
[lp-1 ] DEBUG 2024/09/23 19:38:25 detected active phases: 1p
[lp-1 ] DEBUG 2024/09/23 19:38:25 charge total import: 1825.846kWh
[lp-1 ] DEBUG 2024/09/23 19:38:25 charger status: C
[lp-1 ] DEBUG 2024/09/23 19:38:25 soc estimated: 82.33% (vehicle: 82.20%)
[lp-1 ] DEBUG 2024/09/23 19:38:25 vehicle soc: 82%
[lp-1 ] DEBUG 2024/09/23 19:38:25 vehicle soc limit: 60%
[lp-1 ] DEBUG 2024/09/23 19:38:25 vehicle range: 281km
[lp-1 ] DEBUG 2024/09/23 19:38:30 charger disable
[lp-1 ] DEBUG 2024/09/23 19:38:30 wake-up timer: stop
[site ] DEBUG 2024/09/23 19:38:30 ----
[lp-1 ] DEBUG 2024/09/23 19:38:30 charge power: 333W
[lp-1 ] DEBUG 2024/09/23 19:38:30 charge currents: [12.9 0.009 0.012]A
[site ] DEBUG 2024/09/23 19:38:31 pv power: 0W
[site ] DEBUG 2024/09/23 19:38:33 battery soc: 86%
[site ] DEBUG 2024/09/23 19:38:33 battery power: 4131W
[site ] DEBUG 2024/09/23 19:38:37 grid meter: 43W
[site ] DEBUG 2024/09/23 19:38:47 grid currents: [1.08 -0.7 -0.41]A
[site ] DEBUG 2024/09/23 19:38:49 site power: 4274W
[lp-1 ] DEBUG 2024/09/23 19:38:49 detected active phases: 1p
[lp-1 ] DEBUG 2024/09/23 19:38:49 charge total import: 1825.847kWh
[lp-1 ] DEBUG 2024/09/23 19:38:49 charger status: B
[lp-1 ] INFO 2024/09/23 19:38:49 stop charging <-

andig, 2024-09-24:
> Kann den Grundfehler somit nicht nachstellen und habe daher auf Logs verzichtet....

Es gibt keine neuen Erkenntnisse.

> Bis auf Systemupdates habe ich nichts verändert.

Das kannst Du dann schnell verifizieren indem Du eine ältere Version installierst.

CiNcH83, 2025-10-25:
Gibt es etwas, das ich hier ausprobieren könnte? Ich habe jetzt mal durch die Modis geschalten und keine Fehler feststellen können.

```
Pi@raspberrypi5:~ $ sudo docker exec -it evcc sh
/app # evcc meter LUNA2000 -b hold
[main  ] INFO 2025/10/25 18:13:13 using config file: /etc/evcc.yaml
[db    ] INFO 2025/10/25 18:13:13 using sqlite database: /root/.evcc/evcc.db
/app # evcc meter LUNA2000 -b normal
[main  ] INFO 2025/10/25 18:13:22 using config file: /etc/evcc.yaml
[db    ] INFO 2025/10/25 18:13:22 using sqlite database: /root/.evcc/evcc.db
/app # evcc meter LUNA2000 -b charge
[main  ] INFO 2025/10/25 18:14:51 using config file: /etc/evcc.yaml
[db    ] INFO 2025/10/25 18:14:51 using sqlite database: /root/.evcc/evcc.db
/app # evcc meter LUNA2000 -b hold
[main  ] INFO 2025/10/25 18:15:17 using config file: /etc/evcc.yaml
[db    ] INFO 2025/10/25 18:15:17 using sqlite database: /root/.evcc/evcc.db
/app # evcc meter LUNA2000 -b normal
[main  ] INFO 2025/10/25 18:18:22 using config file: /etc/evcc.yaml
[db    ] INFO 2025/10/25 18:18:22 using sqlite database: /root/.evcc/evcc.db

```

Bei `charge` hat er auch Vollgas aus dem Netz geladen...

Ich habe jeweils die aktuellen FW-Versionen installiert. Als Dongle kommt bei mir der wesentlich performantere [02312QMV-004](https://support.huawei.com/enterprise/en/doc/EDOC1100173984/a09cc8a3/communication-networking-of-the-sdonglea-05-wlan-fe-02312qmv-004) zum Einsatz, der im Betrieb deutlich weniger problematisch ist als mein voriger 005.