evcc-io/evcc #14380: failed configuring site but debug meter shows output

infinus, 2024-06-15:
### Describe the bug

Moving to Rpi native install from docker trying to install using same info in exiting working evcc.yaml and getting error:

failed configuring site
missing either grid or pv meter

### Steps to reproduce

1.using same evcc.yaml in docker works fine
2.evcc meter shows output for both grid1 and pv1
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: trace  
  
# unique installation id  
plant: *****  
  
interval: 10s # control cycle interval  
  
sponsortoken: *****  
  
meters:  
- type: template  
  template: enphase  
  usage: grid  
  host: 192.168.1.225  
  token: *****  
  name: grid1  
- type: template  
  template: enphase  
  usage: pv  
  host: 192.168.1.225  
  token: *****  
  name: pv1  
  
chargers:  
- type: template  
  template: twc3  
  host: 192.168.1.108  
  name: TWC  
  
vehicles:  
- name: Infinus  
  type: custom  
  capacity: 74  
  chargeenable:  
    source: http  
    uri: "http://raspberrypi.local:8080/api/1/vehicles/5YJ3E1EC6MF849726/command/{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}"  
    method: POST  
    body: ""  
  maxcurrent: # set charger max current (A)  
    source: http  
    uri: http://raspberrypi.local:8080/api/1/vehicles/5YJ3E1EC6MF849726/command/set_charging_amps  
    method: POST  
    body: '{"charging_amps": "{{.maxcurrent}}"}'  
  wakeup: # vehicle wake up command  
    source: http  
    uri: http://raspberrypi.local:8080/api/1/vehicles/5YJ3E1EC6MF849726/command/wake_up  
    method: POST  
[... 111 lines cut ...]
</body>
</html>
[http  ] TRACE 2024/06/15 01:22:25 GET https://192.168.1.225/production.json
[http  ] TRACE 2024/06/15 01:22:26 {"production":[{"type":"inverters","activeCount":27,"readingTime":0,"wNow":0,"whLifetime":367426},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1718439745,"wNow":0.0,"whLifetime":19164374.133,"varhLeadLifetime":36369.988,"varhLagLifetime":2864315.352,"vahLifetime":22202140.438,"rmsCurrent":4.542,"rmsVoltage":245.634,"reactPwr":549.934,"apprntPwr":557.751,"pwrFactor":0.0,"whToday":1.133,"whLastSevenDays":479849.133,"vahToday":768.438,"varhLeadToday":0.988,"varhLagToday":756.352}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1718439745,"wNow":652.496,"whLifetime":20094808.824,"varhLeadLifetime":4248809.748,"varhLagLifetime":-2683577.192,"vahLifetime":22556258.047,"rmsCurrent":13.757,"rmsVoltage":245.593,"reactPwr":-240.8,"apprntPwr":1689.273,"pwrFactor":0.39,"whToday":2575.824,"whLastSevenDays":26.824,"vahToday":3014.047,"varhLeadToday":1022.748,"varhLagToday":0.0},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1718439745,"wNow":652.496,"whLifetime":932684.566,"varhLeadLifetime":4285179.736,"varhLagLifetime":180738.16,"vahLifetime":22556258.047,"rmsCurrent":9.216,"rmsVoltage":245.593,"reactPwr":-790.734,"apprntPwr":1131.579,"pwrFactor":0.58,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
[http  ] TRACE 2024/06/15 01:22:26 GET http://192.168.1.225/production.json
[http  ] TRACE 2024/06/15 01:22:26 <html>
<head><title>301 Moved Permanently</title></head>
<body>
<center><h1>301 Moved Permanently</h1></center>
<hr><center>openresty/1.17.8.1</center>
</body>
</html>
[http  ] TRACE 2024/06/15 01:22:26 GET https://192.168.1.225/production.json
[http  ] TRACE 2024/06/15 01:22:27 {"production":[{"type":"inverters","activeCount":27,"readingTime":0,"wNow":0,"whLifetime":367426},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1718439746,"wNow":0.0,"whLifetime":19164374.133,"varhLeadLifetime":36369.988,"varhLagLifetime":2864315.688,"vahLifetime":22202140.778,"rmsCurrent":4.542,"rmsVoltage":245.607,"reactPwr":550.18,"apprntPwr":557.614,"pwrFactor":0.0,"whToday":1.133,"whLastSevenDays":479849.133,"vahToday":768.778,"varhLeadToday":0.988,"varhLagToday":756.688}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1718439746,"wNow":655.13,"whLifetime":20094809.224,"varhLeadLifetime":4248810.231,"varhLagLifetime":-2683577.528,"vahLifetime":22556258.738,"rmsCurrent":13.763,"rmsVoltage":245.555,"reactPwr":-239.133,"apprntPwr":1689.768,"pwrFactor":0.39,"whToday":2576.224,"whLastSevenDays":27.224,"vahToday":3014.738,"varhLeadToday":1023.231,"varhLagToday":0.0},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1718439746,"wNow":655.13,"whLifetime":932684.965,"varhLeadLifetime":4285180.219,"varhLagLifetime":180738.16,"vahLifetime":22556258.738,"rmsCurrent":9.222,"rmsVoltage":245.555,"reactPwr":-789.314,"apprntPwr":1132.155,"pwrFactor":0.58,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
Power:  0W
Energy: 19164.4kWh
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.126.6

andig, 2024-06-15:
Please add error log

infinus, 2024-06-15:
```
Jun 15 07:16:53 raspberrypi systemd[1]: Started evcc.service - evcc.
Jun 15 07:16:57 raspberrypi evcc[6608]: [main  ] INFO 2024/06/15 07:16:57 evcc 0.126.6
Jun 15 07:16:57 raspberrypi evcc[6608]: [main  ] INFO 2024/06/15 07:16:57 using config file: /etc/evcc.yaml
Jun 15 07:17:10 raspberrypi evcc[6608]: [main  ] ERROR 2024/06/15 07:17:10 creating vehicle Infinus failed: cannot create vehicle type 'custom': soc: missing mqtt broker configuration
Jun 15 07:17:10 raspberrypi evcc[6608]: [main  ] FATAL 2024/06/15 07:17:10 failed configuring site: missing either grid or pv meter
Jun 15 07:17:10 raspberrypi evcc[6608]: [main  ] FATAL 2024/06/15 07:17:10 will attempt restart in: 5m0s
```

I'm still working on the MQTT part for the custom vehicle. but the grid should be pretty straightforward.

closed 2024-06-15 by infinus

infinus, 2024-06-15:
Forgot site config which is why it would not start.....