evcc-io/evcc #14382: Vorschlag: mehrere Tarife und Verknüpfung mit Vehicles

TobiasHuber1980, 2024-06-15:
Wäre es bitte möglich mehrere Tarife einzuführen und diese mit Autos zu verknüpfen?
also so in der Art:

```
tariffs:
  - name: tariff_1
    currency: EUR
    grid:
      type: fixed
      price: 0.2343
    feedin:
      type: fixed
      price: 0.064

  - name: tariff_2
    currency: EUR
    grid:
      type: fixed
      price: 0.39
    feedin:
      type: fixed
      price: 0.15
	  

vehicles:
  - name: intern_1
    title: interner Mitarbeiter
    type: template
    template: offline
    identifiers:
      - 040c121a
    tariffs:
      - tariff_2	
	  
  - name: extern_2
    title: externer Mitarbeiter
    type: template
    template: offline
    identifiers:
      - 04ac125f
    tariffs:
      - tariff_2
```

andig, 2024-06-15:
Gibts einen Usecase?

TobiasHuber1980, 2024-06-15:
Usecase: Abrechnung mit internen/externen Mitarbeitern.

interner Mitarbeiter soll den Strom zum Selbstkostenpreis bekommen, externer Mitarbeiter zu einem anderen/höheren Preis.

StefanSchoof, 2024-06-15:
Sollen die externen einen Fixen Preis pro kwh oder hängt der von dem Solaranteil ab?

StefanSchoof, 2024-06-15:
Kann man das nicht Recht mit ein bisschen Excel oder ähnliches über den CSV Export lösen?

TobiasHuber1980, 2024-06-15:
> Sollen die externen einen Fixen Preis pro kwh oder hängt der von dem Solaranteil ab?

schon abhängig vom Solaranteil.
ja klar kann man das auch mit excel und csv machen, aber vielleicht findet diese funktion auch den weg in evcc

TobiasHuber1980, 2024-06-15:
und die Mitarbeiter haben zum Teil zugriff auf evcc, da ist es halt einfach schöner, wenn jeder in den Ladevorgängen seinen Preis gleich sieht und ich nicht den Umweg über csv/excel gehen muss.

andig, 2024-06-15:
> schon abhängig vom Solaranteil.

Das erscheint mir abhängig vom Abrechnungsmodell zu sein, insofern nicht universell.

VolkerK62, 2024-06-15:
@TobiasHuber1980   wenn es nur ein Ladepunkt ist, hätte ich ne Idee. Aber du hast mehrere, oder?

TobiasHuber1980, 2024-06-15:
> @TobiasHuber1980 wenn es nur ein Ladepunkt ist, hätte ich ne Idee. Aber du hast mehrere, oder?

ja aktuell vier stück

TobiasHuber1980, 2024-06-15:
Mir ist noch ein Usecase eingefallen, vielleicht wird es auch etwas für die Allgemeinheit.
Abrechnung Dienstwagen und privates Auto:

```
tariffs:
  - name: tariff_1
    currency: EUR
    grid:
      type: fixed
      price: 0.309
    feedin:
      type: fixed
      price: 0.08

  - name: tariff_2
    currency: EUR
    grid:
      type: fixed
      price: 0.42 # echter Strompreis, oder was auch immer mit dem Arbeitgeber vereinbart ist.
	  
vehicles:
  - name: Privatauto
    title: Privatauto
    tariffs:
      - tariff_1	
	  
  - name: Dienstwagen
    title: Dienstwagen
    tariffs:
      - tariff_2
```

andig, 2024-06-15:
Und was macht der Planner damit?

TobiasHuber1980, 2024-06-15:
> Und was macht der Planner damit?

der planner plant anhand der hinterlegten tarife? keine ahnung :-)

TobiasHuber1980, 2024-06-17:
Viellicht kann man ja beim Planner als default tariff_1 hinterlegen....?

andig, 2024-06-30:
Ich würde hier schließen. Abrechnung ist aktuell kein Fokus.

closed 2024-06-30 by andig