evcc-io/evcc #14384: Templates in the uri of http-plugin not parsed since version 0.127.0

wimaha, 2024-06-15:
### Describe the bug

I use templates in the http-plugin like:
```
enable:
     source: http
     uri: "http://192.168.178.27:8080/api/1/vehicles/LRW3E7FS7PC681392/command/{{if .enable}}charge_start{{else}}charge_stop{{end}}"
     method: POST
```

or like 
```
chargeenable:
      source: http
      uri: http://192.168.178.27:8080/api/1/vehicles/LRW3E7FS7PC681392/command/{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}
      method: POST
```

Since version 0.127.0 the if command is not parsed. I would expect that I receive the url like `http://192.168.178.27:8080/api/1/vehicles/LRW3E7FS7PC681392/command/charge_start', but I receive the url like 'http://192.168.178.27:8080/api/1/vehicles/LRW3E7FS7PC681392/command/{{if .chargeenable}}charge_start{{else}}charge_stop{{end}}'

With version 0.126.6 everything is working fine.

### Steps to reproduce

1. Create custom charger oder custom vehicle
2. Use the http-plugin and use if-condition in the field uri as described in the [docs](https://docs.evcc.io/en/docs/reference/plugins#http-readwrite)

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Docker container

### Version

0.127.1

VolkerK62, 2024-06-15:
I have similar with a shelly and it is still working with 0.127.1
```
enable:
    source: http
    uri: http://192.168.178.88/relay/0?turn={{if .enable}}on{{else}}off{{end}}
```

wimaha, 2024-06-15:
Maybe it‘s a problem with post requests?

andig, 2024-06-15:
@wimaha did it work with 125?

wimaha, 2024-06-15:
@andig Version 0.125.0 also doesn't work.
I found: the if condition is working when it is used auf a query string (after "?"). So this is why the example of @VolkerK62  is working and my example is not.

So:
`uri: 'http://localhost:8080/relay/0?turn={{if.enable}}on{{else}}off{{end}}'`
-->produces:
```http://localhost:8080/relay/0?turn=on```

and
```uri: http://localhost:8080/api/1/vehicles/VIN/command/{{if.enable}}charge_start{{else}}charge_stop{{end}}```
--> produces
```http://localhost:8080/api/1/vehicles/VIN/command/%7B%7Bif.enable%7D%7Dcharge_start%7B%7Belse%7D%7Dcharge_stop%7B%7Bend%7D%7D```

For me this is a bug. At least this behavior is not documented and it breaks the compatibility to my Tesla BLE module.

closed 2024-06-16 by andig

wimaha, 2024-06-16:
Danke @andig!