evcc-io/evcc #14387: Easee-Wallbox: Keeps charing on one Phase

MathysDev, 2024-06-15:
### Describe the bug

Easee Wallbox does not change to three phases anymore since few days (Probably with an release)

### Steps to reproduce

1. Easee Wallbox
2. Enough Energy to change to 3 phases
3. change is initiated
4. EVCC gets charger error

### Configuration details

```yaml
interval: 50s # control cycle interval

vehicles:
- type: template
  template: hyundai
  title: IQ5
  user: ***
  password: ***
  capacity: 72
  cache: 60m
  phases: 3
  language: de
  minCurrent: 8
  maxCurrent: 20
  name: IQ5
  mode: pv
- name: Velos
  type: template
  template: offline
  title: Velos
  capacity: 1
  minCurrent: 2
  maxCurrent: 10
- name: Zero
  type: template
  template: offline
  title: Zero
  capacity: 12
  phases: 2
  minCurrent: 6
  maxCurrent: 16

chargers:
- name: Easee
  type: template
  template: easee
  user: michael@mmhu.ch
  password: Siw2n7JwiwNj
  charger: EC45YS6D
- name: Velo
  type: template
  template: mystrom
  host: myStrom1
  integrateddevice: true
  icon: bike

loadpoints:
- title: Einstellhalle
  charger: Easee
  vehicle: IQ5
  mode: pv
  phases: 0
  enable:
   threshold: 0
   delay: 2m
  guardduration: 7m
```

### Log details

```text
Jun 15 13:30:25 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:30:25 ----
Jun 15 13:30:27 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:30:27 pv power: 22520W
Jun 15 13:30:29 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:30:29 grid meter: -15006W
Jun 15 13:30:29 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:30:29 site power: -15006W
Jun 15 13:31:15 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:31:15 ----
Jun 15 13:31:17 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:31:17 pv power: 22520W
Jun 15 13:31:19 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:31:19 grid meter: -15006W
Jun 15 13:31:19 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:31:19 site power: -15006W
Jun 15 13:31:19 raspi evcc[2120323]: [lp-1  ] ERROR 2024/06/15 13:31:19 vehicle soc: timeout
Jun 15 13:32:05 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:32:05 ----
Jun 15 13:32:07 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:32:07 pv power: 22520W
Jun 15 13:32:09 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:32:09 grid meter: -15006W
Jun 15 13:32:09 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:32:09 site power: -15006W
Jun 15 13:32:55 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:32:55 ----
Jun 15 13:32:57 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:32:57 pv power: 22520W
Jun 15 13:32:59 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:32:59 grid meter: -15006W
Jun 15 13:32:59 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:32:59 site power: -15006W
Jun 15 13:32:59 raspi evcc[2120323]: [lp-1  ] ERROR 2024/06/15 13:32:59 vehicle soc: timeout
Jun 15 13:32:59 raspi evcc[2120323]: [lp-1  ] WARN 2024/06/15 13:32:59 charger logic error: phases mismatch (got 1, expected 3)
Jun 15 13:33:45 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:33:45 ----
Jun 15 13:33:47 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:33:47 pv power: 22515W
Jun 15 13:33:49 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:33:49 grid meter: -17331W
Jun 15 13:33:49 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:33:49 site power: -17331W
Jun 15 13:34:35 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:34:35 ----
Jun 15 13:34:37 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:34:37 pv power: 22515W
Jun 15 13:34:39 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:34:39 grid meter: -17331W
Jun 15 13:34:39 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:34:39 site power: -17331W
Jun 15 13:34:39 raspi evcc[2120323]: [lp-1  ] ERROR 2024/06/15 13:34:39 vehicle soc: timeout
Jun 15 13:35:25 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:35:25 ----
Jun 15 13:35:27 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:35:27 pv power: 22250W
Jun 15 13:35:29 raspi evcc[2120323]: [site  ] DEBUG 2024/06/15 13:35:29 grid meter: -15566W
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.127.1

MathysDev, 2024-06-15:
New logs with phase change etc.
```
Jun 15 13:42:12 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:42:12 charge power: 3611W
Jun 15 13:42:12 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:42:12 charge currents: [14.9 0.015 0.008]A
Jun 15 13:43:02 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:02 charge power: 3611W
Jun 15 13:43:02 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:02 charge currents: [14.9 0.015 0.008]A
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 detected active phases: 1p
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 charge total import: 4891.091kWh
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 charger status: C
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 !! active phases: 1p = min(1p measured 3p vehicle 0p physical 0p charger)
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 available power 20992W > 5520W min 3p threshold
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 phase scale3p in 0s
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 switched phases: 3p
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 phase timer inactive
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
Jun 15 13:43:07 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:07 pv charge current: 30.5A = 5.33A + 25.2A (-17381W @ 3p)
Jun 15 13:43:52 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:52 charge power: 3611W
Jun 15 13:43:52 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:43:52 charge currents: [14.9 0.015 0.008]A
Jun 15 13:44:42 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:42 charge power: 3611W
Jun 15 13:44:42 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:42 charge currents: [14.9 0.015 0.008]A
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 detected active phases: 1p
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 !! session: chargeRater.chargedEnergy=0.6 - chargedAtStartup=0.0
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 charge total import: 4891.694kWh
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 charger status: C
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 vehicle status: C
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 vehicle soc: 71%
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 vehicle soc limit: 100%
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 vehicle range: 329km
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] WARN 2024/06/15 13:44:46 charger logic error: phases mismatch (got 1, expected 3)
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 available power 8904W > 5520W min 3p threshold
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 start phase scale3p timer
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 phase scale3p in 2m0s
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
Jun 15 13:44:46 raspi evcc[2222584]: [lp-1  ] DEBUG 2024/06/15 13:44:46 pv charge current: 39A = 16A + 23A (-5293W @ 1p)

```

andig, 2024-06-15:
/cc @GrimmiMeloni

GrimmiMeloni, 2024-06-16:
_@MathysDev FYI - your Easee password was leaked through your shared config._

Without trace level logging from the `easee` category, we cannot analyse this.
Please adjust the logging section in `evcc.yaml` accordingly, e.g.:

```
log: debug
levels:
  cache: error
  easee: trace
```

MathysDev, 2024-06-16:
Thank you for the answer. Changed the log-level and retry...

MathysDev, 2024-06-17:
Was not able to reproduce with 127.2 today...

MathysDev, 2024-06-22:
I close this issue because was not able to reproduce yet...

MathysDev, 2024-07-05:
Reproduced the problem today
```

Jul 05 12:25:27 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:27 charge power: 2874W
Jul 05 12:25:27 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:27 charge currents: [11.9 0.017 0.009]A
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 detected active phases: 1p
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 !! session: chargeRater.chargedEnergy=5.1 - chargedAtStartup=0.0
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 charge total import: 4953.562kWh
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 charger status: C
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] WARN 2024/07/05 12:25:31 vehicle soc: timeout (ignored by estimator)
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 soc estimated: 76.44% (vehicle: 70.00%)
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 vehicle soc: 76%
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] ERROR 2024/07/05 12:25:31 vehicle soc limit: timeout
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] ERROR 2024/07/05 12:25:31 vehicle range: timeout
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 !! active phases: 1p = min(1p measured 3p vehicle 1p physical 0p charger)
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 available power 13288W > 5520W min 3p threshold
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 start phase scale3p timer
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 phase scale3p in 2m0s
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 !! active phases: 1p = min(1p measured 3p vehicle 1p physical 0p charger)
Jul 05 12:25:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:31 pv charge current: 58.3A = 13A + 45.3A (-10414W @ 1p)
Jul 05 12:25:31 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:31 POST https://api.easee.com/api/chargers/EC45YS6D/settings
Jul 05 12:25:32 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:32 {"dynamicChargerCurrent":16}
Jul 05 12:25:32 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:32 ProductUpdate EC45YS6D: (2024-07-05 10:25:30 +0000 UTC) DYNAMIC_CHARGER_CURRENT 16
Jul 05 12:25:33 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:33 CommandResponse EC45YS6D: {SerialNumber:EC45YS6D ID:48 Timestamp:2024-07-05 10:25:31.93797 +0000 UTC DeliveredAt:2024-07-05 10:25:30.728 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638557719319379700}
Jul 05 12:25:33 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:25:33 max charge current: 16A
Jul 05 12:25:33 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:33 ProductUpdate EC45YS6D: (2024-07-05 10:25:31 +0000 UTC) REASON_FOR_NO_CURRENT 25
Jul 05 12:25:51 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:51 ProductUpdate EC45YS6D: (2024-07-05 10:25:49 +0000 UTC) REASON_FOR_NO_CURRENT 78
Jul 05 12:25:51 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:51 ProductUpdate EC45YS6D: (2024-07-05 10:25:49 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L1 16
Jul 05 12:25:51 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:51 ProductUpdate EC45YS6D: (2024-07-05 10:25:49 +0000 UTC) IN_CURRENT_T3 14.586999893188477
Jul 05 12:25:51 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:51 ProductUpdate EC45YS6D: (2024-07-05 10:25:49 +0000 UTC) OUTPUT_CURRENT 16
Jul 05 12:25:52 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:52 ProductUpdate EC45YS6D: (2024-07-05 10:25:50 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L1 14.586999893188477
Jul 05 12:25:52 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:52 ProductUpdate EC45YS6D: (2024-07-05 10:25:50 +0000 UTC) IN_CURRENT_T2 15.057999610900879
Jul 05 12:25:52 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:25:52 ProductUpdate EC45YS6D: (2024-07-05 10:25:50 +0000 UTC) TOTAL_POWER 3.5999999046325684
Jul 05 12:26:17 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:26:17 charge power: 3600W
Jul 05 12:26:17 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:26:17 charge currents: [14.6 0.017 0.009]A
Jul 05 12:27:00 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:27:00 POST https://api.easee.com/api/chargers/EC45YS6D/commands/poll_lifetimeenergy
Jul 05 12:27:00 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:27:00 {"device":"EC45YS6D","commandId":2,"ticks":638557720208737920}
Jul 05 12:27:01 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:27:01 ProductUpdate EC45YS6D: (2024-07-05 10:26:59 +0000 UTC) LIFETIME_ENERGY 4954.13929083334
Jul 05 12:27:01 raspi evcc[2863027]: [easee ] TRACE 2024/07/05 12:27:01 CommandResponse EC45YS6D: {SerialNumber:EC45YS6D ID:2 Timestamp:2024-07-05 10:27:00.873792 +0000 UTC DeliveredAt:2024-07-05 10:26:59.76 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638557720208737920}
Jul 05 12:27:07 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:27:07 charge power: 3600W
Jul 05 12:27:07 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:27:07 charge currents: [14.6 0.017 0.009]A
Jul 05 12:27:11 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:27:11 detected active phases: 1p
Jul 05 12:27:11 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:27:11 !! session: chargeRater.chargedEnergy=5.7 - chargedAtStartup=0.0
Jul 05 12:27:11 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:27:11 charge total import: 4954.139kWh
[... cut ...]
Jul 05 12:35:27 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:35:27 charge currents: [14.6 0.017 0.009]A
Jul 05 12:35:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:35:31 detected active phases: 1p
Jul 05 12:35:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:35:31 !! session: chargeRater.chargedEnergy=6.0 - chargedAtStartup=0.0
Jul 05 12:35:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:35:31 charge total import: 4954.429kWh
Jul 05 12:35:31 raspi evcc[2863027]: [lp-1  ] DEBUG 2024/07/05 12:35:31 charger status: C

```

GrimmiMeloni, 2024-07-17:
Reopening. I will check the provided log after my vacation.

GrimmiMeloni, 2024-07-20:
> Reproduced the problem today

Based on the log, it seems that the loadpoint never asks the Charger to switch phases - but I can tell from the code it does.
There's only one way to get through the switch phase logic in the Easee implementation silently, i.e. without any requests. It would mean the system is controlling the phases on Charger, not Circuit level, and that it deems the change to be unnecessary (i.e. system already running on 3p).
I would like to see the rest of the log, in particular the startup sequence with first contact messages to Easee. It should reveal wether the circuit can be determined or not.

MathysDev, 2024-07-20:
Logs after retart today
`Jul 20 11:11:39 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:39 POST https://api.easee.com/api/accounts/login
Jul 20 11:11:39 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:39 {"userName":"***","password":"[Passwort]"}
Jul 20 11:11:39 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:39 GET https://api.easee.com/api/chargers/[ChargerID]/site
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 {"uuid":"*","id":*,"siteKey":"*","name":"*","levelOfAccess":1,"address":{"street":"*","buildingNumber":"52","zip":"*","area":"*","country":{"id":"CH","name":"Switzerland","phonePrefix":0},"latitude":null,"longitude":null,"altitude":null},"contactInfo":{"installerName":null,"installerPhoneNumber":null,"ownerName":"*","ownerPhoneNumber":"*","company":null},"costPerKWh":0.2905,"costPerKwhExcludeVat":0.2687,"currencyId":"CHF","siteType":100,"ratedCurrent":100.0,"vat":8.1,"partnerId":20,"installerId":null,"installerUserId":null,"supportId":null,"installerAlias":null,"useDynamicMaster":false,"circuits":[{"id":344327,"uuid":"ded685f3-9d8a-4034-bf3b-b62735b04758","siteId":360590,"circuitPanelId":1,"panelName":"1","ratedCurrent":63.0,"fuse":null,"chargers":[{"id":"ECLFK77A","name":"*","color":1,"createdOn":"2022-11-07T19:40:35.884631","updatedOn":"2024-03-05T13:00:23.052276","backPlate":{"id":"8072B5DAEE4F04","masterBackPlateId":"8072B5DA5FB604","name":"*","features":[]},"levelOfAccess":3,"productCode":100,"userRole":1,"isTemporary":false},{"id":"EC7PTDV9","name":"Ladestation 2","color":null,"createdOn":"2021-11-17T05:40:24.557909","updatedOn":"2023-02-22T08:05:12.19096","backPlate":{"id":"8072B5DAB2C704","masterBackPlateId":"8072B5DA5FB604","name":"*","features":[]},"levelOfAccess":null,"productCode":100,"userRole":1,"isTemporary":false},{"id":"[ChargerID]","name":"Ladestation 1","color":1,"createdOn":"2021-11-15T18:00:29.625163","updatedOn":"2023-02-22T08:05:12.197045","backPlate":{"id":"8072B5DA5FB604","masterBackPlateId":"8072B5DA5FB604","name":"","features":[]},"levelOfAccess":null,"productCode":100,"userRole":1,"isTemporary":false}],"masterBackplate":{"id":"8072B5DA5FB604","masterBackPlateId":"8072B5DA5FB604","name":"","features":[]},"useDynamicMaster":false,"parentCircuitId":null}],"equalizers":[],"createdOn":"2022-06-09T11:32:29.004427","updatedOn":"2024-04-23T05:45:56.303388","userRole":1,"allowedSiteActions":["AllowToConfigureLevelOfAccess"],"support":null,"regulations":[]}
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 POST https://streams.easee.com/hubs/chargers/negotiate
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 {"negotiateVersion":0,"connectionId":"*","availableTransports":[{"transport":"WebSockets","transferFormats":["Text","Binary"]},{"transport":"ServerSentEvents","transferFormats":["Text"]},{"transport":"LongPolling","transferFormats":["Text","Binary"]}]}
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-07-20 04:02:40 +0000 UTC) CHARGER_OFFLINE_REASON 0
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:21 +0000 UTC) LOCAL_PRE_AUTHORIZE_ENABLED true
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:21 +0000 UTC) LOCAL_AUTHORIZE_OFFLINE_ENABLED true
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:21 +0000 UTC) ALLOW_OFFLINE_TX_FOR_UNKNOWN_ID true
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:35 +0000 UTC) DETECTED_POWER_GRID_TYPE 1
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:52 +0000 UTC) CIRCUIT_MAX_CURRENT_P1 63
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:52 +0000 UTC) CIRCUIT_MAX_CURRENT_P2 63
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:52 +0000 UTC) CIRCUIT_MAX_CURRENT_P3 63
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:52 +0000 UTC) LOCK_CABLE_PERMANENTLY false
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:52 +0000 UTC) IS_ENABLED true
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-06-27 01:06:34 +0000 UTC) WI_FI_SSID SEL
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-07-20 01:54:14 +0000 UTC) PHASE_MODE 1
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:20 +0000 UTC) LED_STRIP_BRIGHTNESS 25
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:52 +0000 UTC) ENABLE_IDLE_CURRENT false
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:21 +0000 UTC) LOCAL_AUTHORIZATION_REQUIRED false
Jul 20 11:11:40 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:40 ProductUpdate [ChargerID]: (2024-04-23 05:45:21 +0000 UTC) AUTHORIZATION_REQUIRED false
[... cut ...]
Jul 20 11:11:41 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:41 CommandResponse [ChargerID]: {SerialNumber:[ChargerID] ID:2 Timestamp:2024-07-20 09:11:41.04001 +0000 UTC DeliveredAt:2024-07-20 09:11:40.823 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638570635010400100}
Jul 20 11:11:41 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:11:41 ProductUpdate [ChargerID]: (2024-07-20 09:11:40 +0000 UTC) LIFETIME_ENERGY 5026.000266666678
Jul 20 11:13:06 raspi evcc[2449036]: [easee ] TRACE 2024/07/20 11:13:06 ProductUpdate [ChargerID]: (2024-07-20 09:13:05 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L3 0.007000000216066837`

MaTr74, 2024-07-22:
I can reproduce the problem. Unfortunately, I don't own an EV by myself so I can get the logs only if a guest of our hotel is plugging in his car. So far I can report, that I tested without phase switching because the charge mode was set to fast and that the problem only exist if EVCC is in between. If I'm stopping EVCC, the charger is charging on 3 phases. Here is my config and log (only debug level, yet).

[image]
[image]

Version:
`evcc 0.128.3`

Configuration:
```
loadpoints:                                                                                                                                                                                
   - title: Wallbox-links                                                                                                                                                                   
     charger: easee-master                                                                                                                                                                  
     mode: minpv                                                                                                                                                                            
     enable:                                                                                                                                                                                
       threshold: 0                                                                                                                                                                         
       delay: 1m                                                                                                                                                                            
     disable:                                                                                                                                                                               
       threshold: 6000                                                                                                                                                                      
       delay: 3m                                                                                                                                                                            
     phases: 3                                                                                                                                                                              
     priority: 1                                                                                                                                                                            
     mincurrent: 6                                                                                                                                                                          
     maxcurrent: 32     
```

Charging is starting with full speed on 3 phases and is then switching to 1 phase.

```
[lp-1 ] DEBUG 2024/07/21 14:50:26 charge power: 0W
[lp-1 ] DEBUG 2024/07/21 14:50:26 charge currents: [0.009 0.009 0.009]A
[lp-1 ] DEBUG 2024/07/21 14:50:55 charge power: 17193W
[lp-1 ] DEBUG 2024/07/21 14:50:55 charge currents: [24.2 24.2 24.3]A
[lp-1 ] DEBUG 2024/07/21 14:50:55 charge total import: 1825.960kWh
[lp-1 ] DEBUG 2024/07/21 14:50:55 charger status: C
[lp-1 ] INFO 2024/07/21 14:50:55 car connected
[lp-1 ] DEBUG 2024/07/21 14:50:55 pv timer elapse
[lp-1 ] DEBUG 2024/07/21 14:50:55 pv timer inactive
[lp-1 ] DEBUG 2024/07/21 14:50:55 charge total import: 1825.960kWh
[lp-1 ] INFO 2024/07/21 14:50:55 start charging ->
[lp-1 ] DEBUG 2024/07/21 14:50:55 wake-up timer: stop
[lp-1 ] DEBUG 2024/07/21 14:50:55 charger vehicle id: 360F24B1
[lp-1 ] WARN 2024/07/21 14:50:55 charger logic error: disabled but charging
[lp-1 ] DEBUG 2024/07/21 14:50:58 charger enable
[lp-1 ] DEBUG 2024/07/21 14:50:59 max charge current: 32A
[lp-1 ] DEBUG 2024/07/21 14:51:25 charge power: 1450W
[lp-1 ] DEBUG 2024/07/21 14:51:25 charge currents: [6.03 0.008 0.009]A
[lp-1 ] DEBUG 2024/07/21 14:51:55 charge power: 1450W
[lp-1 ] DEBUG 2024/07/21 14:51:55 charge currents: [6.03 0.008 0.009]A
[lp-1 ] DEBUG 2024/07/21 14:52:25 charge power: 1450W
[lp-1 ] DEBUG 2024/07/21 14:52:25 charge currents: [6.03 0.008 0.009]A
[lp-1 ] DEBUG 2024/07/21 14:52:25 detected active phases: 1p
[lp-1 ] DEBUG 2024/07/21 14:52:25 charge total import: 1825.960kWh
[lp-1 ] DEBUG 2024/07/21 14:52:25 charger status: C
[lp-1 ] WARN 2024/07/21 14:52:25 charger logic error: current mismatch (got 6A, expected 32A)
[lp-1 ] WARN 2024/07/21 14:52:25 charger logic error: phases mismatch (got 1, expected 3)
[lp-1 ] DEBUG 2024/07/21 14:52:25 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/07/21 14:52:25 switched phases: 3p
[lp-1 ] DEBUG 2024/07/21 14:52:25 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/07/21 14:52:55 charge power: 1450W
[lp-1 ] DEBUG 2024/07/21 14:52:55 charge currents: [6.03 0.008 0.009]A
```

GrimmiMeloni, 2024-07-23:
Thanks for pointing out that peak. I missed it as it is before evcc even realizes something is charging.

To me this looks like another case of the Easee just doing its thing without any control from the outside.

As you can see from the log, the charge starts uncontrolled.

```
[lp-1 ] WARN 2024/07/21 14:50:55 charger logic error: disabled but charging
```

This is a known issue/behavior with Easee, so we can ignore it for now.

What is more interesting, is this message, also confirmed by the graph you shared:
```
[lp-1 ] WARN 2024/07/21 14:52:25 charger logic error: current mismatch (got 6A, expected 32A)
```

Also, your comment here is interesting regarding the configuration:
> I configured that to be sure there is nothing that wants switch to 1 phase, to exclude any mistake in configuration that leads to the observed behaviour.

Indeed, this configuration cannot guarantee that _something_ asks for a phase switch. All it guarantees is that evcc does never ask the charger to switch phases, which it also based on the log correctly does NOT. 

My read is, that the charger changed its current on its own. We have seen cases like this in the past. 
Do you by any chance have multiple Chargers?
If so, do you use the Easee Equalizer, or any other means for load balancing between the chargers?
Are there other components in your setup with API level access to your Easee chargers? We know that Tibber (even when "deactivated") interacts with the API and causes trouble.

If none of the above is the case, we would need a full debug log including trace level data from the Easee.

GrimmiMeloni, 2024-07-23:
@MathysDev 

> Logs after restart today

Thanks. It shows there are 3 chargers on the circuit:

* `ECLFK77A`
* `EC7PTDV9`
* `[ChargerID]` (aka. `Ladestation 1`)

_Since you obfuscated a lot of data 🙄, I have to _guess_ that `[ChargerID]` is actually the charger you showed the config for ealier, i.e. `EC45YS6D`. (If it is not, the startup shows a completely different site - which I assume is not the case here.)_

So this matches my assumption. As we see multiple chargers, we are not controlling on circuit level. We are trying to switch phase modes on the chargers instead. On bootup the charger is in PHASE_MODE 1, i.e. it is configured to fixed 1p.
Evcc might/will change that during runtime. I would need to see a full log (from start) with the same trace level logging for a failed phase switch to further confirm this.

MaTr74, 2024-07-23:
Hi @GrimmiMeloni,

thanks for your detailed reply! We have a second charger in place which is performing load balancing together with the charger used. The charger used in this case is the master. But, the second charger wasn't in use at this time. The circuit is configured to use up to 25 A, which should not lead to a 1-phase-request of one of the chargers. Furthermore, I have a openHAB-thing configured, which isn't doing any write task on the API as far as I know. 

The most interesting thing is my observation that when I turn off EVCC, everything works fine. So I would assume, that EVCC is requesting the change to 1 phase. I will start EVCC again with the trace log, but I don't know when the next guest car will be connected to the charger.

GrimmiMeloni, 2024-07-23:
@MaTr74 you should move your issue with a reference to our exchange here to a new discussion or issue. By now it has become clear, that we are mixing problems in here:

@MathysDev does not see a phase scale _**up**_ 1p->3p
@MaTr74 is seeing unrequested scale **_down_** 3p->1p and DCC

These are different problems. @MathysDev's could in theory be caused by #14996, but that would not explain what @MaTr74 is seeing.

eviw, 2024-07-23:
Hi, I also have the issue, that scale **up** 1p->3p does not work. Scale down 3p->1p works.
I attach my logs as well. 
[evcc_log_since_start.txt](https://github.com/user-attachments/files/16350363/evcc_log_since_start.txt)
[evcc_yaml.txt](https://github.com/user-attachments/files/16350374/evcc_yaml.txt)

GrimmiMeloni, 2024-07-23:
@eviw do you also have more than one Easee on the same curcuit?

eviw, 2024-07-24:
No, I have just this one Easee Home.

GrimmiMeloni, 2024-07-24:
> No, I have just this one Easee Home.

Then your case is different. Please open a new issue and include a trace level log of the missing switch for analysis.

MathysDev, 2024-07-24:
> So this matches my assumption. As we see multiple chargers, we are not controlling on circuit level. We are trying to switch phase modes on the chargers instead. On bootup the charger is in [...]

I try to get this log. Today I do not have errors with easee.
I assume that the error is only reproducable in specific conctellation of the other chargers (connected/charging?)

MathysDev, 2024-07-24:
managed to reproduce the error again. The other two chargers are not connected to a car.

```
Jul 24 12:06:38 raspi evcc[3450991]: [main  ] INFO 2024/07/24 12:06:38 evcc 0.128.4
Jul 24 12:06:38 raspi evcc[3450991]: [main  ] INFO 2024/07/24 12:06:38 using config file: /etc/evcc.yaml
Jul 24 12:06:39 raspi evcc[3450991]: [main  ] INFO 2024/07/24 12:06:39 listening at :7070
Jul 24 12:06:39 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:39 POST https://api.easee.com/api/accounts/login
Jul 24 12:06:39 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:39 {"userName":"***","password":"*"}
Jul 24 12:06:39 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:39 GET https://api.easee.com/api/chargers/EC45YS6D/site
Jul 24 12:06:39 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:39 {"uuid":"56ff8068-7a68-4b6b-9e64-e3bb15f7ead4","id":360590,"siteKey":"ZP53-S422","name":"* MFH *","levelOfAccess":1,"address":{"street":"*","buildingNumber":"52","zip":"*","area":"*","country":{"id":"CH","name":"Switzerland","phonePrefix":0},"latitude":null,"longitude":null,"altitude":null},"contactInfo":{"installerName":null,"installerPhoneNumber":null,"ownerName":"*","ownerPhoneNumber":"*","company":null},"costPerKWh":0.2905,"costPerKwhExcludeVat":0.2687,"currencyId":"CHF","siteType":100,"ratedCurrent":100.0,"vat":8.1,"partnerId":20,"installerId":null,"installerUserId":null,"supportId":null,"installerAlias":null,"useDynamicMaster":false,"circuits":[{"id":344327,"uuid":"ded685f3-9d8a-4034-bf3b-b62735b04758","siteId":360590,"circuitPanelId":1,"panelName":"1","ratedCurrent":63.0,"fuse":null,"chargers":[{"id":"ECLFK77A","name":"* * 44","color":1,"createdOn":"2022-11-07T19:40:35.884631","updatedOn":"2024-03-05T13:00:23.052276","backPlate":{"id":"8072B5DAEE4F04","masterBackPlateId":"8072B5DA5FB604","name":"* * 44","features":[]},"levelOfAccess":3,"productCode":100,"userRole":1,"isTemporary":false},{"id":"EC7PTDV9","name":"Ladestation 2 (*)","color":null,"createdOn":"2021-11-17T05:40:24.557909","updatedOn":"2023-02-22T08:05:12.19096","backPlate":{"id":"8072B5DAB2C704","masterBackPlateId":"8072B5DA5FB604","name":"*","features":[]},"levelOfAccess":null,"productCode":100,"userRole":1,"isTemporary":false},{"id":"EC45YS6D","name":"Ladestation 1 (Mathys)","color":1,"createdOn":"2021-11-15T18:00:29.625163","updatedOn":"2023-02-22T08:05:12.197045","backPlate":{"id":"8072B5DA5FB604","masterBackPlateId":"8072B5DA5FB604","name":"","features":[]},"levelOfAccess":null,"productCode":100,"userRole":1,"isTemporary":false}],"masterBackplate":{"id":"8072B5DA5FB604","masterBackPlateId":"8072B5DA5FB604","name":"","features":[]},"useDynamicMaster":false,"parentCircuitId":null}],"equalizers":[],"createdOn":"2022-06-09T11:32:29.004427","updatedOn":"2024-04-23T05:45:56.303388","userRole":1,"allowedSiteActions":["AllowToConfigureLevelOfAccess"],"support":null,"regulations":[]}
Jul 24 12:06:39 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:39 POST https://streams.easee.com/hubs/chargers/negotiate
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 {"negotiateVersion":0,"connectionId":"M7AAIWOmNs4fdaV7vpfhEg","availableTransports":[{"transport":"WebSockets","transferFormats":["Text","Binary"]},{"transport":"ServerSentEvents","transferFormats":["Text"]},{"transport":"LongPolling","transferFormats":["Text","Binary"]}]}
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-07-20 04:02:40 +0000 UTC) CHARGER_OFFLINE_REASON 0
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:21 +0000 UTC) LOCAL_PRE_AUTHORIZE_ENABLED true
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:21 +0000 UTC) LOCAL_AUTHORIZE_OFFLINE_ENABLED true
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:52 +0000 UTC) CIRCUIT_MAX_CURRENT_P1 63
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:35 +0000 UTC) DETECTED_POWER_GRID_TYPE 1
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:21 +0000 UTC) ALLOW_OFFLINE_TX_FOR_UNKNOWN_ID true
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:52 +0000 UTC) CIRCUIT_MAX_CURRENT_P2 63
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:52 +0000 UTC) CIRCUIT_MAX_CURRENT_P3 63
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:52 +0000 UTC) LOCK_CABLE_PERMANENTLY false
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:52 +0000 UTC) IS_ENABLED true
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-04-23 05:45:52 +0000 UTC) ENABLE_IDLE_CURRENT false
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-06-27 01:06:34 +0000 UTC) WI_FI_SSID SEL
Jul 24 12:06:40 raspi evcc[3450991]: [easee ] TRACE 2024/07/24 12:06:40 ProductUpdate EC45YS6D: (2024-07-24 09:56:49 +0000 UTC) PHASE_MODE 2
[... cut ...]
Jul 24 12:41:48 raspi evcc[3450991]: [lp-1  ] DEBUG 2024/07/24 12:41:48 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
Jul 24 12:41:48 raspi evcc[3450991]: [lp-1  ] DEBUG 2024/07/24 12:41:48 !! active phases: 3p = min(0p measured 3p vehicle 3p physical 0p charger)
Jul 24 12:41:48 raspi evcc[3450991]: [lp-1  ] DEBUG 2024/07/24 12:41:48 pv charge current: 27A = 5.33A + 21.6A (-14924W @ 3p)
Jul 24 12:42:34 raspi evcc[3450991]: [lp-1  ] DEBUG 2024/07/24 12:42:34 charge power: 3599W
Jul 24 12:42:34 raspi evcc[3450991]: [lp-1  ] DEBUG 2024/07/24 12:42:34 charge currents: [15 0.016 0.01]A

```

GrimmiMeloni, 2024-07-26:
@MathysDev thanks for the additional logs. During the restart the charge was already going. Would have been nice to see how the session is setup. Background: The charger reports to be in PhaseMode 2 (Auto) - which matches everything so far. What is unclear though is how we end up with a 1p session in the first place. As the charger is in Auto Mode, I would expect it to go for 3p right from the start.

That all said, I will try around with those switch settings with my local unit later today, to see if I can replicate the problem. Stay tuned.

GrimmiMeloni, 2024-08-08:
I looked into #14996 and confirmed that the phase mode setting we do is correct.
However, it is essential pause/resume the charge for the settings to take effect - and as we have seen here, there's an edge case that will not trigger the pause/resume if the charger already is in auto mode.
As stated above, we have a somewhat strange situation here, that a 1p charge is going, although the charger is in phaseMode 2. Maybe some external influence is at play here, changing the phaseMode underneath evcc.

That all said - I think the fix is straight forward. I will decouple the reconfiguration from the pause/resume, i.e. pause/resume will always happen when a phase switch is requested on charger level - independent of wether we reconfigure the charger or not. PR will be there in a moment.

Peacecamper, 2025-01-06:
Ich habe ein ähnliches Problem. Bei mir wird regelmäßig die L1-Phase ausgelassen, sprich nicht belastet. Ich bin mir allerdings nicht sicher, ob das an EVCC oder an der Easee-Wallbox oder ggf. sogar an meiner Elektroinstallation liegt. Da ich außer EVCC-Updates aber eigentlich nichts verändert habe und es immer wieder auch mit allen drei Phasen funktioniert, würde ich erstmal EVCC verdächtigen.

`# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070
  
  #mqtt message broker
mqtt:
  broker: 192.168.1.10:1883 # euer MQTT-Broker Adresse mit Port
  topic: evcc # root topic for publishing, set empty to disable
  user: mqtt_user
  password: nr158dmrwsPJF7qhxzOr

log: debug
levels:
  cache: error

interval: 30s # control cycle interval

meters:
- type: template
  template: sunspec-inverter 
  id: 71  
  host: 192.168.1.109  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.1.156  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid2

chargers:
- type: template
  template: easee 
  user:   
  password: 
  charger:  
  name: wallbox4
- type: template
  template: easee 
  user: 
  password: 
  charger: 
  name: wallbox5

vehicles:
- type: template
  template: enyaq
  title: Winston
  user:
  password: 
  vin: 
  capacity: 77
  name: ev3

loadpoints:
- title: Wallbox 1
  charger: wallbox4
  mode: minpv
- title: Wallbox 2
  charger: wallbox5
  mode: now
[... 369 lines cut ...]
[lp-2  ] DEBUG 2025/01/06 11:15:37 charge power: 0W
[lp-2  ] DEBUG 2025/01/06 11:15:37 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/01/06 11:15:37 charge power: 6895W
[lp-1  ] DEBUG 2025/01/06 11:15:37 charge currents: [0 15.7 15.6]A
[site  ] DEBUG 2025/01/06 11:15:37 grid power: 6370W
[site  ] DEBUG 2025/01/06 11:15:37 grid powers: [-230 3230 3370]W
[site  ] DEBUG 2025/01/06 11:15:37 grid currents: [-1.42 14.6 15.1]A
[site  ] DEBUG 2025/01/06 11:15:37 pv power: 1180W
[site  ] DEBUG 2025/01/06 11:15:37 site power: 6370W
[lp-1  ] DEBUG 2025/01/06 11:15:37 detected active phases: 2p
[lp-1  ] DEBUG 2025/01/06 11:15:37 charge total import: 2857.446kWh
[lp-1  ] DEBUG 2025/01/06 11:15:37 charger status: C
[lp-1  ] DEBUG 2025/01/06 11:15:37 soc estimated: 69.00% (vehicle: 69.00%)
[lp-1  ] DEBUG 2025/01/06 11:15:37 vehicle soc: 69%
[lp-1  ] DEBUG 2025/01/06 11:15:37 vehicle soc limit: 80%
[lp-1  ] DEBUG 2025/01/06 11:15:37 vehicle range: 204km
[site  ] DEBUG 2025/01/06 11:16:07 ----
[lp-2  ] DEBUG 2025/01/06 11:16:07 charge power: 0W
[lp-1  ] DEBUG 2025/01/06 11:16:07 charge power: 6895W
[lp-1  ] DEBUG 2025/01/06 11:16:07 charge currents: [0 15.7 15.6]A
[lp-2  ] DEBUG 2025/01/06 11:16:07 charge currents: [0 0 0]A
[site  ] DEBUG 2025/01/06 11:16:07 grid power: 6400W
[site  ] DEBUG 2025/01/06 11:16:07 grid powers: [-230 3240 3390]W
[site  ] DEBUG 2025/01/06 11:16:07 grid currents: [-1.4 14.6 15.2]A
`

mfuchs1984, 2025-01-06:
@Peacecamper das liegt am Auto, machen alle VW Derivate aktuell, siehe bspw. 
#17318
#17238 
https://www.meinid.com/thread/5372-ac-laden-id4-schaltet-beim-laden-von-3-auf-2-phasen-alfen-eve-single-wallbox/?pageNo=8
https://www.meinid.com/thread/8031-volkswagen-damaged-the-on-board-charger-in-the-id-3-1st-plus-with-the-ota-3-7-up/

Peacecamper, 2025-01-06:
> @Peacecamper das liegt am Auto, machen alle VW Derivate aktuell, siehe bspw. #17318 #17238 [...]

Danke für die schnelle Antwort!

PR #15306 (2024-08-08): Easee: always pause/resume when phase switching on charger level
fix #14387