evcc-io/evcc #14395: Renault "vehicle is not connected to driver"

gandalfb, 2024-06-15:
### Describe the bug

I am trying to include a Zoe which is shown within myrenault.

2 vehicles are linked to the account.

Home Assistant is showing the data with the Renault device.

`evcc-1    | [main  ] ERROR 2024/06/15 20:06:09 creating vehicle zoe failed: cannot create vehicle type 'template': cannot create vehicle type 'renault': vehicle is not connected to driver`

```
vehicles:
- type: template
  template: renault
  title: red Zoe
  vin: xxx
  capacity: 52
  name: zoe
  icon: car
  user: xxx
  password: xxx
```

Thank you, I found a discussion with this, but no solution.

Regards,
Gandalf

### Steps to reproduce

1. Add a Renault with 2 vehicles to myRenault(1 Zoe, 1 Kadjar)
2. Add evcc config as shown
3. Start evcc and receive the error

### Configuration details

```yaml
vehicles:
- type: template
  template: renault
  title: red Zoe
  vin: xxx
  capacity: 52
  name: zoe
  icon: car
  user: xxx
  password: xxx
```

### Log details

```text
evcc-1    | [main  ] INFO 2024/06/15 20:26:31 evcc 0.127.1
...
evcc-1    | [main  ] ERROR 2024/06/15 20:06:09 creating vehicle zoe failed: cannot create vehicle type 'template': cannot create vehicle type 'renault': vehicle is not connected to driver
```

### What type of operating system are you running?

Linux

### Version

0.127.1

andig, 2024-06-15:
Please send vehicle credentials to info@evcc.io

gandalfb, 2024-06-15:
> Please send vehicle credentials to [info@evcc.io](mailto:info@evcc.io)

Thank you. I sent it.

aikitori, 2024-06-17:
I have the same problem...
Getting data via Node-Red works.

https://github.com/evcc-io/evcc/discussions/5215

gandalfb, 2024-06-19:
So, together with Home Assistant and a mqtt I am able to somewhat - hopefully - get a workaround and even saving resources on Renault API (seems to be rather strict)

Walkaround: use mqtt to publish data from HA to evcc

evcc config
```
  - name: zoe
    type: custom
    title: red Zoe # display name for UI
    capacity: 52 # byttery capacity (kWh)
    soc: # battery soc (%)
      source: mqtt
      topic: ev/red_zoe/batterie
      timeout: 10m
    range: # optional electric range (km)
      source: mqtt
      topic: ev/red_zoe/batterieautonomie
      timeout: 10m
    status: # FahrzeugStatus (A..F), A= Not connected, B= connected, C= charging, F= error
      source: mqtt
      topic: ev/red_zoe/status
      timeout: 10m
    odometer: # Odometer (km)
      source: mqtt
      topic: ev/red_zoe/kilometerstand
      timeout: 10m
    onIdentify: # set defaults when vehicle is identified
      mode: pv
      #minSoC: 15 # set through plan
      #targetSoC: 80 # set through plan
```

home assistant automation
```
alias: EV zoe to mqtt
description: sends EV zoe data to mqtt
trigger:
  - platform: time_pattern
    minutes: /5
condition: []
action:
  - service: mqtt.publish
    data:
      topic: ev/red_zoe/batterie
      payload_template: "{{states.sensor.batterie.state}}"
  - service: mqtt.publish
    data:
      topic: ev/red_zoe/batterieautonomie
      payload_template: "{{states.sensor.batterieautonomie.state}}"
  - service: mqtt.publish
    data:
      topic: ev/red_zoe/kilometerstand
      payload_template: "{{states.sensor.kilometerstand.state}}"
  - service: mqtt.publish
    data:
      topic: ev/red_zoe/status
      payload_template: >-
        {% set l = states('sensor.ladezustand') %}
        {% set s = states('sensor.steckerzustand') %}
        {% if l == "charge_in_progress" %} C
        {% elif l == "charge_error" %} F
        {% elif s == "unplugged" %} A
        {% elif s == "plugged" %} B
        {% elif s == "plug_error" %} F
        {% else %} A
        {% endif %}
mode: single
```

Of course I am not quite sure yet, if my "mapping" to Status works. At first glance recognizing the car works.

`FahrzeugStatus (A..F), A= Not connected, B= connected, C= charging, F= error`

closed 2024-06-22 by andig