evcc-io/evcc #14403: evcc tries to switch to three phases but vehicle only supports one

fussel132, 2024-06-16:
### Describe the bug

Hi there,
so I've set up evcc yesterday and it appears to work fine. However, evcc tries to change the wallbox to three-phase operation each minute although I've configured it to only use one phase for the selected vehicle. evcc is set to automatically switch between one and three phases. The wallbox refuses to change phases resulting in evcc logging a charger logic error (phases mismatch, got 1, expected 3). See logs below.
Is this due to misconfiguration or is it a bug within evcc?
Thanks in advance!

### Steps to reproduce

1. Create a vehicle only using single phase charging
2. Set loadpoint to automatically switch between phase operation
3. Review wallbox logs to identify errors with phase switching

### Configuration details

```yaml
network:
  schema: http
  host: xxx
  port: xxx

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

meters:
- name: pv-g
  type: template
  template: hoymiles-opendtu 
  usage: pv  
  host: xxx
- name: smartmeter
  type: template
  template: tasmota-sml
  usage: grid
  host: xxx

chargers:
- type: template
  template: tinkerforge-warp3 
  host: xxx
  port: xxx
  user: xxx
  password: xxx
  topic: warp3
  timeout: 30s  
  name: warp3

vehicles:
- type: template
  template: bmw
  title: BMW i3
  icon: car
  user: xxx
  password: xxx
  vin: xxx
  capacity: 22
  phases: 1
  cache: 5m
  mode: pv
  region: EU
  name: i3

loadpoints:
- title: Wallbox
  charger: warp3
  mode: pv
  # Values are deprecated and set via UI, see https://docs.evcc.io/docs/reference/configuration/loadpoints#phases
  #phases: 0 # 0=Auto
  #minCurrent: 6

site:
  title: My Home
  meters:
    pv:
      - pv-g
    grid: smartmeter
```

### Log details

```text
[site ] DEBUG 2024/06/16 03:12:02 ----
[lp-1 ] DEBUG 2024/06/16 03:12:02 charge power: 709W
[lp-1 ] DEBUG 2024/06/16 03:12:02 charge currents: [3.16 0 0]A
[site ] DEBUG 2024/06/16 03:12:02 pv power: 0W
[site ] DEBUG 2024/06/16 03:12:02 grid meter: 1041W
[site ] DEBUG 2024/06/16 03:12:02 site power: 1041W
[lp-1 ] DEBUG 2024/06/16 03:12:02 detected active phases: 1p
[lp-1 ] DEBUG 2024/06/16 03:12:02 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/16 03:12:02 charge total import: 18.020kWh
[lp-1 ] DEBUG 2024/06/16 03:12:02 charger status: C
[lp-1 ] WARN 2024/06/16 03:12:02 charger logic error: phases mismatch (got 1, expected 3)
[lp-1 ] DEBUG 2024/06/16 03:12:02 !! active phases: 1p = min(0p measured 1p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/06/16 03:12:02 switched phases: 3p
[lp-1 ] DEBUG 2024/06/16 03:12:02 !! active phases: 1p = min(0p measured 1p vehicle 3p physical 0p charger)

And the wallbox lox has a lot of those entries:
2024-06-16 03:12:01,817 | evse_v2 | Requested phase switch but can't switch at the moment.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.127.1

andig, 2024-06-16:
I don‘t think we‘re preventing switch to 3p just because the vehicle wouldn‘t use it. Maybe we should? @GrimmiMeloni ?

Why the warp doesn‘t like switching and errors is a separate question though, not sure.

GrimmiMeloni, 2024-06-16:
It depends on the mode you are in.

- For PV Mode we definetely do consider the vehicle phases.
- Fast Charging does NOT consider the vehicle phases.

The log snippet / data shared is unfortunately insufficient to know which mode this was tried in.

Additionally: The shown logic error will NOT trigger a phase switch, as the loadpoint follows the charger in such situations and updates its internal representation to match the charger. I therefore assume fastMode is to blame for this. And yes, I agree we should not try to switch if the connected vehicle is configured with fixed phases.

fussel132, 2024-06-16:
Oh sorry, yes I was using fast mode, I think the issue is not present when using pv or pvmin mode.

@andig I think the warp3 doesn't like it as the car only charges with single phase and after switching to three it automatically switches back as only one phase is being used. I could switch that off as a workaround for the "problem" right? evcc would request three phases (because apparently fast mode doesn't consider the vehicle's maximum phase count) and the warp3 would just be set to three but only charges with one due to the connected vehicles limitation.

Otherwise when using fast mode I would manually need to set single phase in the UI settings. That's a little inconvenient in my eyes, I think it would be great if fast mode also considers the (for the current vehicle) configured phase count!

andig, 2024-06-16:
> I think the warp3 doesn't like it as the car only charges with single phase and after switching to three it automatically switches back as only one phase is being used

I dont think so

andig, 2024-06-16:
Closing as works as expected.

GrimmiMeloni, 2024-06-16:
Bad timing. I just put a change for fastvmode together. 🤷‍♂️

andig, 2024-06-16:
From the PR:

> The vehicle phases settings is only a hint. Imho it does not make sense to treat it as a restriction. And even if it were I'd still consider fast mode exactly that. If vehicle is capable of 1p only it will only charge 1p, regardless if we set 3p.

fussel132, 2024-06-16:
> > I think the warp3 doesn't like it as the car only charges with single phase and after switching to three it automatically switches back as only one phase is being used
> 
> I dont think so

@andig Well that's how I understand this option:

[image]

It says that if the internal electricity meter is being used to detect how many phases will be used for charging and if only one is in use it switches to single-phase causing the logic issue within evcc (evcc expects three but gets one).

@GrimmiMeloni That looks promising, would be very cool if this gets integrated soon. Because right now, if there is no issue with my configuration I would not say that things work as expected.

andig, 2024-06-16:
@fussel132 das ist eine ziemlich sinnlose Funktion da das Auto eh 1p lädt. Wozu soll das gut sein?

/cc @rtrbt

fussel132, 2024-06-16:
@andig Ja ich glaub ich versteh jetzt was du meinst, der Fast Mode setzt die Wallbox einfach auf ihren maximal möglichen Ausgang. Dann wäre es wieder unnötig, die Beschränkungen für das Fahrzeug zu beachten. Ich hatte das nur so verstanden: wenn ich evcc sage, das Auto kann nur einphasig, dann wird auch im Fast Mode nur versucht einphasig zu laden. Wenn sich die Einstellung aber auf die Wallbox und nicht das Auto bezieht, dann ist das wirklich unnötig.

rtrbt, 2024-06-17:
> das ist eine ziemlich sinnlose Funktion da das Auto eh 1p lädt. Wozu soll das gut sein?

Die Funktion hat im Moment nur einen Use-Case: Wir arbeiten gerade an einer neuen Lastmanagement-Version, die pro Phase steuert. Wenn man ein einphasig ladendes Auto an eine dreiphasige Wallbox anschließt, die umschaltbar ist, dann kann die Box auf einphasiges Laden wechseln (ohne CP-Trennung und den ganzen Tanz, das ist nur, dass wir das L2-L3-Schütz wegschalten) und ab dann kann sich das Lastmanagement darauf verlassen, dass diese Wallbox nicht auf L2 oder L3 Strom ziehen wird. Der kann dann also auf andere Wallboxen umverteilt werden.

Netter Nebeneffekt: Im Webinterface steht dann auch einphasig.

Damit ich das Problem richtig verstanden habe: EVCC stolpert in dem Fall darüber, dass Umschalten auf dreiphasig nicht mehr klappt?

andig, 2024-06-17:
Genau. Wir schalten auf 3p und lesen dann nach ob das geklappt hat. Hier sehen wir jetzt aber nur 1p, entsprechend Fehlermeldung und der erneute Versuch auf 3p zu schalten, zugegebenerweise ohne dass das für ein 1p Fahrzeug notwendig wäre.

Wir haben also den Klassiker konkurrierender Steuerungen. 

Ich sehe folgende Lösungsmöglichkeiten:

1. Funktion in Warp wird durch Anwender deaktiviert. Es kann nur einer steuern. Wie?
2. Funktion in Warp ist inaktiv wenn es nur eine Box gibt- Schadenspotential schonmal deutlich reduziert.
3. Wir erkennen die Situation (wie?) und gaukeln 3p vor weil es ja "im Prinzip" 3p gewesen wäre. Nicht schön.
4. Wir verzichten darauf, bei 1p Fahrzeugen im FAST wie PV Modus auf 3p zu schalten. Hat dann den Nachteil, dass ein falsch erkanntes Fahrzeug auch bei 1p Ladung verharren wird auch wenn es schneller könnte. Würde das Problem auch nur für den 1p->3p Übergang lösen. Sollte die Box schon in 3p sein würde sich dadurch nichts ändern und das Problem besteht weiter.

@GrimmiMeloni 4. ist nicht ganz dein Vorschlag- ich würde einfach _gar nicht_ schalten.

/cc @premultiply

rtrbt, 2024-06-17:
> Funktion in Warp wird durch Anwender deaktiviert. Es kann nur einer steuern. Wie?

Unter Wallbox -> Einstellungen den Punkt "Automatischer Phasenwechsel" deaktivieren.
Alternatividee: EVCC kann das selbst deaktivieren: https://docs.warp-charger.com/docs/mqtt_http/api_reference/evse#evse_phase_auto_switch_warp3

> Funktion in Warp ist inaktiv wenn es nur eine Box gibt- Schadenspotential schonmal deutlich reduziert.

Wir benutzen das laut meinem Kollegen auch für die PV-Überschussreglung -> wird nicht passieren

> Wir erkennen die Situation (wie?) und gaukeln 3p vor weil es ja "im Prinzip" 3p gewesen wäre. Nicht schön.

In evse/low_level_state gibt es bei WARP2 und WARP3 das Feld phases_info. Das ist 1, wenn das EVSE von drei- auf einphasig umgeschaltet hat, weil das Auto 1p lädt. https://docs.warp-charger.com/docs/mqtt_http/api_reference/evse#evse_low_level_state_warp3

andig, 2024-06-17:
> Wir benutzen das laut meinem Kollegen auch für die PV-Überschussreglung -> wird nicht passieren

@rtrbt siehe mein Argument oben. Wozu? Das ist bei 1p Fahrzeugen ein nop!

andig, 2024-06-17:
> Alternatividee: EVCC kann das selbst deaktivieren: https://docs.warp-charger.com/docs/mqtt_http/api_reference/evse#evse_phase_auto_switch_warp3

@rtrbt Link funktioniert nicht, auch nicht der Inline-Link wenn ich Warp3 auf der Seite auswähle.

rtrbt, 2024-06-18:
> siehe mein Argument oben. Wozu? Das ist bei 1p Fahrzeugen ein nop! Ansonsten- wenn ihr Überschussregelung macht tut evcc das nicht. Dann gibts auch keine Problem :)

@andig Das ist bei uns im Moment halt nicht gekoppelt. Die Idee war, dass selbst wenn das Gerät (andere Wallbox oder Energy Manager) neustartet, die Information "da lädt ein einphasiges Auto" nicht verloren geht bzw. neu erkannt werden muss, wenn die Box selbst einfach L2 und L3 wegnimmt. Das ist aus Auto-Sicht ein NOP aber wir speichern praktisch die Information "in Hardware".

Bevor ihr losrennt um irgendetwas zu tun: Als Übergangslösung kann @fussel132 die "Auto­ma­tisch­er Phasen­wechsel"-Option deaktivieren. Wir sind wie gesagt dabei, das ganze Lastmanagement (daran hängt auch die Überschussregelung) komplett umzuschreiben. Es zeichnet sich gerade ab, dass wir den Automatischen Phasenwechsel dann möglicherweise nicht mehr brauchen. Wenn das der Fall ist, würden wir das Feature rauswerfen oder zumindest das Default auf Aus ändern.

> Link funktioniert nicht, auch nicht der Inline-Link wenn ich Warp3 auf der Seite auswähle.

Hm in der Tat ist das in Chrome kaputt. Die Hardware-Auswahl ist sehr gehackt, die wollte ich für Docusaurus 3.3 nochmal robuster machen, hatte aber noch keine Zeit dafür.

andig, 2024-06-19:
> Das ist aus Auto-Sicht ein NOP aber wir speichern praktisch die Information "in Hardware".

Yep, verwirrt aber halt das andere steuernde Gerät. Also am Besten warnen: in `evse/phase_auto_switch` finde ich ein JSON mit `enabled: true` oder wie wäre das über MQTT? Das gibts nur bei Warp3- woran würde ich die erkennen?

rtrbt, 2024-06-20:
> Also am Besten warnen: in evse/phase_auto_switch finde ich ein JSON mit enabled: true

Genau

> Das gibts nur bei Warp3- 

Gibt's bei WARP2 auch, ist aber immer false.

> woran würde ich die erkennen?

Eigentlich braucht das keine Erkennung. Entweder die Box publisht auf .../evse/phase_auto_switch oder eben nicht. Alternativ: in .../info/name gibt es das Feld "type". Das ist bei einer WARP3 "warp3". Damit kannst du WARP2 + Energy Manager von WARP3 unterscheiden.

andig, 2024-06-20:
Danke- erstmal Lösung über Doku: https://github.com/evcc-io/evcc/pull/14494

closed 2024-06-20 by andig