evcc-io/evcc #14408: Smartpricelimit: Show in UI if/when cheap charging is possible

Horgul123, 2024-06-16:
If I set a smartpricelimit and EVCC is in PV mode, I don't see in the UI when cheap charging will be possible (if it starts e.g. in 2 hours) or if it's possible at the moment. 

Would like to see in the UI if/when "cheap" charging is possible.

EVCC v0.127.1

naltatis, 2024-06-19:
@andig Currently we don't have this info in the UI (without polling tariffs REST API). I'd suggest that we introduce a new loadpoint datafield (similar to the existing `planProjectedStart`) that indicates if and when smartCost charging will be active in the upcoming hours that we can see right now. This date field code be called `smartCostNextStart` or `smartCostProjectedStart` if we want to be in sync with plan API.

andig, 2024-06-19:
I dont think that smart cost is necessarily plannable, e.g. not for dynamic tariffs. Isn‘t this issue about having an active indicator?

naltatis, 2024-06-19:
> if it starts e.g. in 2 hours

For tariffs with forecast (like Tibber) we can look into the future and know if the price will drop under the configured limit any time soon.

naltatis, 2024-06-19:
The **active indicator** based on `smartCostActive` is already present.

andig, 2024-06-19:
> or if it's possible at the moment.

Seems we‘re not using that?

naltatis, 2024-06-20:
@Horgul123 just to avoid confusion. You mentioned `1. if` & `2. when` smart charging is possible.

**1.1. if cheap charging is possible**

Charging depending on price or co2 is always possible when a supporting tariff is configured. I don't think we need an extra indicator in the main UI for this, since this does not change without reconfiguration. If such a tariff exists, can be seen in the loadpoint modal.

[image]

**1.2. if cheap charging, currently active**

We also have a status message that's shows when fast charging due to cheap price is activated.

[image]

**2. when will it be possible?**

This one we currently dont show in the main ui. It's only visible in the loadpoint settings modal. I'd suggest we'll add this information also as a status message when a cheap time window appears in the foreseeable future.

Does this match with your intention of this issue?

Horgul123, 2024-06-21:
@naltatis 
1.1: cheap charging possible: I don't mean if I configured a tariff which allows it. I mean: Is it possible NOW based on the current price and my setting (e.g.: current price 25ct, limit 26ct -> Cheap charing is available now. As  far as I know this is not displayed in the main UI when no car is connected. My idea: e.g. on Monday I set a limit of 15ct. On Monday and Tuesday the price is always >15ct. Then Wednesday noon the price drops below 15ct. When my car is not connected (e.g. rainy day and no PV energy available) I would maybe miss this chance. When it is displayed in the main UI (e.g. "Cheap charging possible now -> Please connect your car") I would have a change to see this hint and connect my car to the wallbox.
And a hint like "cheap charging possible in x hours at 14:00 " would be great, too. This would increase the chance to connect the car.

1.2:  Yes, I know it's displayed when the car is connected. But see 1.1.: It's only visible when the car is connected

2. A status message in the main UI is want I thought about (see 1.1)

So yes, such a message in the main UI is want I really would like to see. I think this could help other people, too.
At the moment I check the Tibber priceinformation to know when I need to plug in my car

naltatis, 2024-06-21:
Ok, thanks for the clarification. The status message when not connected was not on my radar. I guess it's because i always plugin when I come home. But I think its a good idea to add cheap charging availability (now or in x hours) event if no vehicle is connected.

Horgul123, 2024-06-21:
Thanks a lot for accepting this idea :)

I normally plug in only if I want to charge because I tend to forget that the car is plugged in and then I'm wondering we the car didn't start :P

closed 2024-07-07 by andig