evcc-io/evcc #14415: Lastmanagement; Werte werden teilwise nicht aufgerundet

TobiasHuber1980, 2024-06-16:
hello, mir ist gerade was bzgl. lastmanagement aufgefallen, irgendwie rundet evcc die werte manchmal nicht korrekt.

`circuit-loadpoints] DEBUG 2024/06/14 17:41:35 current: 31.993A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:42:05 power: 21389.472999999998W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:42:05 current: 31.998A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:42:35 power: 21362.698W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:42:35 current: 32.001000000000005A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:43:05 power: 21366.184999999998W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:43:05 current: 32.009A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:43:35 power: 21308.007W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:43:35 current: 31.994A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:44:05 power: 21273.064W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:44:05 current: 31.829A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:44:35 power: 21288.359W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:44:35 current: 31.852A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:45:05 power: 21282.735W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:45:05 current: 31.851A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:45:35 power: 21286.658W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:45:35 current: 31.907A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:46:05 power: 21302.401W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:46:05 current: 31.884999999999998A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:46:35 power: 21308.064W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:46:35 current: 31.922A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:47:05 power: 21316.438000000002W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:47:05 current: 31.93A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:47:35 power: 21308.809999999998W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:47:35 current: 31.921A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:48:05 power: 21322.447W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:48:05 current: 31.962A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:48:35 power: 21312.995W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:48:35 current: 32.022999999999996A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:49:05 power: 21271.081W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:49:05 current: 31.994999999999997A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:49:35 power: 21306.465W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:49:35 current: 32.017A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:50:05 power: 21291.987W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:50:05 current: 32A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:50:35 power: 21253.045W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:50:35 current: 32.002A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:51:05 power: 10606.942W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:51:05 current: 16.002A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:51:35 power: 10609.761W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:51:35 current: 15.997A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:52:05 power: 10604.547W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:52:05 current: 15.991A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:52:35 power: 10602.762W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:52:35 current: 15.992A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:53:05 power: 10606.409W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:53:05 current: 16.013A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:53:35 power: 10603.79W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:53:35 current: 16.013A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:54:05 power: 10614.506W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:54:05 current: 16.028A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:54:35 power: 10626.821W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:54:35 current: 16A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:55:05 power: 10634.282W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:55:05 current: 15.991A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:55:35 power: 10615.944W`
`[circuit-loadpoints] DEBUG 2024/06/14 17:55:35 current: 16.002A`
`[circuit-loadpoints] DEBUG 2024/06/14 17:56:05 power: 10630.397W`

[Slack Message](https://evccgroup.slack.com/archives/D05KCKH0QJX/p1718382591435199)

andig, 2024-06-16:
Was soll da wo gerundet werden? Was ist das Problem?

andig, 2024-06-16:
Logformat in https://github.com/evcc-io/evcc/commit/477141e7f1fe7320cd20c10a05f0097fba2c8866 angepasst.

closed 2024-06-16 by andig

TobiasHuber1980, 2024-06-16:
huhu, ich glaube das passt so auch nicht:
```
[circuit-main] DEBUG 2024/06/16 18:46:45 power: -1.69e+04W
[circuit-main] DEBUG 2024/06/16 18:46:45 current: 0A
[circuit-main] DEBUG 2024/06/16 18:47:15 power: -1.66e+04W
[circuit-main] DEBUG 2024/06/16 18:47:15 current: 0A
[circuit-main] DEBUG 2024/06/16 18:47:44 power: -1.63e+04W
[circuit-main] DEBUG 2024/06/16 18:47:44 current: 0A
[circuit-main] DEBUG 2024/06/16 18:48:15 power: -1.59e+04W
[circuit-main] DEBUG 2024/06/16 18:48:15 current: 0A
[circuit-main] DEBUG 2024/06/16 18:48:45 power: -1.58e+04W
[circuit-main] DEBUG 2024/06/16 18:48:45 current: 0A
```
sollten doch ca. -16 kW oder -16.000 W sein.

andig, 2024-06-16:
Nochmal geändert