evcc-io/evcc #14417: LM und Wallbox ohne mA-Regelung. System schwingt um maxcurrent

VolkerK62, 2024-06-16:
Es ist für mich persönlich nicht so tragisch, ich will aber trotzdem darauf hinweisen:
Der Wallbox Circuit ist auf 20A begrenzt.
Meine Wallbox kann nur in vollen A-Schritten geregelt werden. Das wird vom LM nicht beachtet. 
Dadurch, dass mein Auto immer 0,5-0,8 A weniger nimmt, als freigegeben, regelt LM nach und das System schwingt zwischen 20A und 21 A hin und her.

Vor `mode: now` waren 6A freigegeben, Auto nimmt nur 5,5A, also ist 20,5A das neue max charge current.
Die Wallbox macht daraus 20A und das Auto nimmt 19,3A.
Also sind noch 0,7A bis zur 20A Grenze übrig, was zur Folge hat, das dann 21,2A (20,5 + 0,7) freigegeben werden.
Die Wallbox macht daraus 21A und das Auto nimmt 20,3A. Was overcurrent zur Folge hat.
Es wird um 0,3A heruntergeregelt auf 20,9A (21,2 - 0,3).
Die Wallbox macht daraus 20A, das Auto nimmt 19,3, und es fängt wieder von vorne an.

```
Jun 15 09:31:52 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:52 ----
Jun 15 09:31:53 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:53 charge power: 1265W
Jun 15 09:31:53 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:53 charge currents: [5.5 0 0]A
Jun 15 09:31:53 raspberrypi evcc[8014]: [circuit-carport] DEBUG 2024/06/15 09:31:53 power: 1265W
Jun 15 09:31:53 raspberrypi evcc[8014]: [circuit-carport] DEBUG 2024/06/15 09:31:53 current: 5.5A
Jun 15 09:31:53 raspberrypi evcc[8014]: [circuit-main] DEBUG 2024/06/15 09:31:53 power: 990.31W
Jun 15 09:31:53 raspberrypi evcc[8014]: [circuit-main] DEBUG 2024/06/15 09:31:53 current: 4.4A
Jun 15 09:31:53 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:53 pv power: 5121W
Jun 15 09:31:53 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:53 battery soc: 29%
Jun 15 09:31:53 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:53 battery power: -643W
Jun 15 09:31:53 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:53 set charge mode: now
Jun 15 09:31:53 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:53 pv timer reset
Jun 15 09:31:53 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:53 pv timer inactive
Jun 15 09:31:53 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:53 grid meter: 696W
Jun 15 09:31:53 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:53 grid powers: [244 -435 887]W
Jun 15 09:31:53 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:53 grid currents: [2.02 -2.04 4.4]A
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 battery has priority at soc 29% (< 90%)
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 aux power 1: 0W
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 aux power: 0W
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 site power: 846W
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 set battery mode: hold
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 ----
Jun 15 09:31:54 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:54 charge power: 1265W
Jun 15 09:31:54 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:54 charge currents: [5.5 0 0]A
Jun 15 09:31:54 raspberrypi evcc[8014]: [circuit-carport] DEBUG 2024/06/15 09:31:54 power: 1266W
Jun 15 09:31:54 raspberrypi evcc[8014]: [circuit-carport] DEBUG 2024/06/15 09:31:54 current: 5.5A
Jun 15 09:31:54 raspberrypi evcc[8014]: [circuit-main] DEBUG 2024/06/15 09:31:54 power: -603.11W
Jun 15 09:31:54 raspberrypi evcc[8014]: [circuit-main] DEBUG 2024/06/15 09:31:54 current: 3.39A
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 pv power: 5121W
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 battery soc: 29%
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 battery power: -392W
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 grid meter: -603W
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 grid powers: [-359 -783 539]W
Jun 15 09:31:54 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:54 grid currents: [-2.1 -3.39 3.1]A
Jun 15 09:31:55 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:55 battery has priority at soc 29% (< 90%)
Jun 15 09:31:55 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:55 aux power 1: 0W
Jun 15 09:31:55 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:55 aux power: 0W
Jun 15 09:31:55 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:55 site power: -453W
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 charge voltages: [230 0 0]V
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 detected connected phases: 1p
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 detected active phases: 1p
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 charge total import: 1865.400kWh
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 charger status: C
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 soc estimated: 75.12% (vehicle: 75.00%)
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 vehicle soc: 75%
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 vehicle soc limit: 100%
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 vehicle range: 345km
Jun 15 09:31:55 raspberrypi evcc[8014]: [circuit-carport] DEBUG 2024/06/15 09:31:55 validate current: 6A -> 20.5A <= 20A at 5.5A: capped at 20.5A
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
Jun 15 09:31:55 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:55 max charge current: 20.5A
Jun 15 09:31:57 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:31:57 ----
Jun 15 09:31:58 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:58 charge power: 4384W
Jun 15 09:31:58 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:31:58 charge currents: [19.3 0 0]A
Jun 15 09:31:58 raspberrypi evcc[8014]: [circuit-carport] DEBUG 2024/06/15 09:31:58 power: 4384W
Jun 15 09:31:58 raspberrypi evcc[8014]: [circuit-carport] DEBUG 2024/06/15 09:31:58 current: 19.3A
Jun 15 09:31:58 raspberrypi evcc[8014]: [circuit-main] DEBUG 2024/06/15 09:31:58 power: 2239.42W
[... 274 lines cut ...]
Jun 15 09:33:13 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:13 pv power: 5550W
Jun 15 09:33:13 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:13 battery soc: 29%
Jun 15 09:33:13 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:13 battery power: -1001W
Jun 15 09:33:13 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:13 grid meter: 200W
Jun 15 09:33:13 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:13 grid powers: [-809 -1764 2773]W
Jun 15 09:33:13 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:13 grid currents: [-3.57 -7.58 12.6]A
Jun 15 09:33:14 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:14 battery has priority at soc 29% (< 90%)
Jun 15 09:33:14 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:14 aux power 1: 0W
Jun 15 09:33:14 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:14 aux power: 0W
Jun 15 09:33:14 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:14 site power: 350W
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 charge voltages: [228 0 0]V
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 detected connected phases: 1p
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 detected active phases: 1p
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 charge total import: 1865.500kWh
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 charger status: C
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 soc estimated: 75.25% (vehicle: 75.00%)
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 vehicle soc: 75%
[... cut ...]
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 vehicle range: 345km
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 max charge current: 20A
Jun 15 09:33:14 raspberrypi evcc[8014]: [lp-1  ] DEBUG 2024/06/15 09:33:14 set charge mode: pv
Jun 15 09:33:14 raspberrypi evcc[8014]: [site  ] DEBUG 2024/06/15 09:33:14 ----
```

andig, 2024-06-16:
Wo sitzen denn deine Zähler?

andig, 2024-06-16:
Die Frage scheint zu sein, wie/wo das max charge current in Richtung Wallbox aktiv wird und ob hier die 1A Schritte berücksichtigt werden? Der Minderverbauch des Fahrzeugs spielt doch keine Rolle für das Problem, oder?

VolkerK62, 2024-06-16:
Der Zähler sitzt direkt in der Wallbox.

Der Minderverbrauch des Autos sorgt dafür, dass nachgeregelt wird.
So wie ich das beobachte wird immer das Delta aus dem actual current und der Grenze (20A) nachgeregelt.
Bei einer mA Regelung würde hier nach ein paar Intervallen genau 20A actual current rauskommen.
Bei 1A-Wallboxen dürfte hier nur hochgeregelt werden, wenn dieses Delta größer 1A ist.

andig, 2024-06-16:
@VolkerK62 wäre doch die Frage, ob wir die beiden Aufrufe nicht umdrehen sollten, oder den cap 2x durchführen:

```go
// setLimit applies charger current limits and enables/disables accordingly
func (lp *Loadpoint) setLimit(chargeCurrent float64) error {
	// full amps only?
	if _, ok := lp.charger.(api.ChargerEx); !ok || lp.vehicleHasFeature(api.CoarseCurrent) {
		chargeCurrent = math.Trunc(chargeCurrent)
	}

	// apply circuit limits
	if lp.circuit != nil {
		currentLimit := lp.circuit.ValidateCurrent(lp.chargeCurrent, chargeCurrent)

		activePhases := lp.ActivePhases()
		powerLimit := lp.circuit.ValidatePower(lp.chargePower, currentToPower(chargeCurrent, activePhases))
		currentLimitViaPower := powerToCurrent(powerLimit, activePhases)

		chargeCurrent = min(currentLimit, currentLimitViaPower)
	}
}
```

Willst Du das mal lokal ausprobieren? Nach der letzten Zeile kommt sonst nämlich kein ganzzahliger Wert mehr raus :O

VolkerK62, 2024-06-16:
>Willst Du das mal lokal ausprobieren?

da muss ich erstmal schauen, dass ich das "selber bauen" hinbekomme.
Hab es vor einiger Zeit mal versucht und bin kläglich gescheitert.

VolkerK62, 2024-06-16:
Ich bin mir nicht sicher, ob das `chargeCurrent = math.Trunc(chargeCurrent)` Wirkung hat, denn das macht die Wallbox ja auch so. Egal ob 20,1A oder 20,9A kommen, sie macht 20A draus.

Müsste man nicht beim Delta eingreifen?
Ich vermute aber, hier liegt die Info über "fully amps" nicht vor?
https://github.com/evcc-io/evcc/blob/8633283ab720cd0a35fbac0424ba7d173d24c7b4/core/circuit.go#L288

premultiply, 2024-06-16:
> Also sind noch 0,7A bis zur 20A Grenze übrig, was zur Folge hat, das dann 21,2A (20,5 + 0,7) freigegeben werden.
> Die Wallbox macht daraus 21A und das Auto nimmt 20,3A. Was overcurrent zur Folge hat.

Ich denke dass ist das eigentliche Problem hier: Es müsste jeweils das Limit des Ciruits UND des bzw. der Loadpoints (min-Funktion) berücksichtigt werden. Die sonst übliche Delta-Regelung muss hier auch limitiert werden.
Der Loadpoint darf nie einen höheren Strom einstellen (= dem Auto signalisieren) wie das numerische Limit des Loadpoints und des Circuits.

andig, 2024-06-17:
@VolkerK62 Dein Log in lesbar(er):

```
[site  ] DEBUG 2024/06/15 09:31:52 ----
[lp-1  ] DEBUG 2024/06/15 09:31:53 charge power: 1265W
[lp-1  ] DEBUG 2024/06/15 09:31:53 charge currents: [5.5 0 0]A
[circuit-carport] DEBUG 2024/06/15 09:31:53 power: 1265W
[circuit-carport] DEBUG 2024/06/15 09:31:53 current: 5.5A
[circuit-main] DEBUG 2024/06/15 09:31:53 power: 990.31W
[circuit-main] DEBUG 2024/06/15 09:31:53 current: 4.4A
[site  ] DEBUG 2024/06/15 09:31:53 pv power: 5121W
[site  ] DEBUG 2024/06/15 09:31:53 battery soc: 29%
[site  ] DEBUG 2024/06/15 09:31:53 battery power: -643W
[lp-1  ] DEBUG 2024/06/15 09:31:53 set charge mode: now
[lp-1  ] DEBUG 2024/06/15 09:31:53 pv timer reset
[lp-1  ] DEBUG 2024/06/15 09:31:53 pv timer inactive
[site  ] DEBUG 2024/06/15 09:31:53 grid meter: 696W
[site  ] DEBUG 2024/06/15 09:31:53 grid powers: [244 -435 887]W
[site  ] DEBUG 2024/06/15 09:31:53 grid currents: [2.02 -2.04 4.4]A
[site  ] DEBUG 2024/06/15 09:31:54 battery has priority at soc 29% (< 90%)
[site  ] DEBUG 2024/06/15 09:31:54 aux power 1: 0W
[site  ] DEBUG 2024/06/15 09:31:54 aux power: 0W
[site  ] DEBUG 2024/06/15 09:31:54 site power: 846W
[site  ] DEBUG 2024/06/15 09:31:54 set battery mode: hold
[site  ] DEBUG 2024/06/15 09:31:54 ----
[lp-1  ] DEBUG 2024/06/15 09:31:54 charge power: 1265W
[lp-1  ] DEBUG 2024/06/15 09:31:54 charge currents: [5.5 0 0]A
[circuit-carport] DEBUG 2024/06/15 09:31:54 power: 1266W
[circuit-carport] DEBUG 2024/06/15 09:31:54 current: 5.5A
[circuit-main] DEBUG 2024/06/15 09:31:54 power: -603.11W
[circuit-main] DEBUG 2024/06/15 09:31:54 current: 3.39A
[site  ] DEBUG 2024/06/15 09:31:54 pv power: 5121W
[site  ] DEBUG 2024/06/15 09:31:54 battery soc: 29%
[site  ] DEBUG 2024/06/15 09:31:54 battery power: -392W
[site  ] DEBUG 2024/06/15 09:31:54 grid meter: -603W
[site  ] DEBUG 2024/06/15 09:31:54 grid powers: [-359 -783 539]W
[site  ] DEBUG 2024/06/15 09:31:54 grid currents: [-2.1 -3.39 3.1]A
[site  ] DEBUG 2024/06/15 09:31:55 battery has priority at soc 29% (< 90%)
[site  ] DEBUG 2024/06/15 09:31:55 aux power 1: 0W
[site  ] DEBUG 2024/06/15 09:31:55 aux power: 0W
[site  ] DEBUG 2024/06/15 09:31:55 site power: -453W
[lp-1  ] DEBUG 2024/06/15 09:31:55 charge voltages: [230 0 0]V
[lp-1  ] DEBUG 2024/06/15 09:31:55 detected connected phases: 1p
[lp-1  ] DEBUG 2024/06/15 09:31:55 detected active phases: 1p
[lp-1  ] DEBUG 2024/06/15 09:31:55 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/15 09:31:55 charge total import: 1865.400kWh
[lp-1  ] DEBUG 2024/06/15 09:31:55 charger status: C
[lp-1  ] DEBUG 2024/06/15 09:31:55 soc estimated: 75.12% (vehicle: 75.00%)
[lp-1  ] DEBUG 2024/06/15 09:31:55 vehicle soc: 75%
[lp-1  ] DEBUG 2024/06/15 09:31:55 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/06/15 09:31:55 vehicle range: 345km
[circuit-carport] DEBUG 2024/06/15 09:31:55 validate current: 6A -> 20.5A <= 20A at 5.5A: capped at 20.5A
[lp-1  ] DEBUG 2024/06/15 09:31:55 !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/06/15 09:31:55 max charge current: 20.5A
[site  ] DEBUG 2024/06/15 09:31:57 ----
[lp-1  ] DEBUG 2024/06/15 09:31:58 charge power: 4384W
[lp-1  ] DEBUG 2024/06/15 09:31:58 charge currents: [19.3 0 0]A
[circuit-carport] DEBUG 2024/06/15 09:31:58 power: 4384W
[circuit-carport] DEBUG 2024/06/15 09:31:58 current: 19.3A
[circuit-main] DEBUG 2024/06/15 09:31:58 power: 2239.42W
[circuit-main] DEBUG 2024/06/15 09:31:58 current: 16.62A
[site  ] DEBUG 2024/06/15 09:31:58 pv power: 5121W
[site  ] DEBUG 2024/06/15 09:31:58 battery soc: 29%
[site  ] DEBUG 2024/06/15 09:31:58 battery power: -127W
[site  ] DEBUG 2024/06/15 09:31:58 grid meter: 2298W
[site  ] DEBUG 2024/06/15 09:31:58 grid powers: [-609 -796 3703]W
[site  ] DEBUG 2024/06/15 09:31:59 grid currents: [-2.68 -3.46 16.6]A
[site  ] DEBUG 2024/06/15 09:31:59 battery has priority at soc 29% (< 90%)
[site  ] DEBUG 2024/06/15 09:31:59 aux power 1: 0W
[site  ] DEBUG 2024/06/15 09:31:59 aux power: 0W
[... 264 lines cut ...]
[site  ] DEBUG 2024/06/15 09:33:13 pv power: 5550W
[site  ] DEBUG 2024/06/15 09:33:13 battery soc: 29%
[site  ] DEBUG 2024/06/15 09:33:13 battery power: -1001W
[site  ] DEBUG 2024/06/15 09:33:13 grid meter: 200W
[site  ] DEBUG 2024/06/15 09:33:13 grid powers: [-809 -1764 2773]W
[site  ] DEBUG 2024/06/15 09:33:13 grid currents: [-3.57 -7.58 12.6]A
[site  ] DEBUG 2024/06/15 09:33:14 battery has priority at soc 29% (< 90%)
[site  ] DEBUG 2024/06/15 09:33:14 aux power 1: 0W
[site  ] DEBUG 2024/06/15 09:33:14 aux power: 0W
[site  ] DEBUG 2024/06/15 09:33:14 site power: 350W
[lp-1  ] DEBUG 2024/06/15 09:33:14 charge voltages: [228 0 0]V
[lp-1  ] DEBUG 2024/06/15 09:33:14 detected connected phases: 1p
[lp-1  ] DEBUG 2024/06/15 09:33:14 detected active phases: 1p
[lp-1  ] DEBUG 2024/06/15 09:33:14 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/06/15 09:33:14 charge total import: 1865.500kWh
[lp-1  ] DEBUG 2024/06/15 09:33:14 charger status: C
[lp-1  ] DEBUG 2024/06/15 09:33:14 soc estimated: 75.25% (vehicle: 75.00%)
[lp-1  ] DEBUG 2024/06/15 09:33:14 vehicle soc: 75%
[lp-1  ] DEBUG 2024/06/15 09:33:14 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/06/15 09:33:14 vehicle range: 345km
[lp-1  ] DEBUG 2024/06/15 09:33:14 !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
[lp-1  ] DEBUG 2024/06/15 09:33:14 max charge current: 20A
[lp-1  ] DEBUG 2024/06/15 09:33:14 set charge mode: pv
[site  ] DEBUG 2024/06/15 09:33:14 ----
```

VolkerK62, 2024-06-17:
Oh, irgendwas ist immer 😄 
war schon froh, dass ich die LP´s 2 bis 6  rausgegrept bekommen habe

andig, 2024-06-17:
@VolkerK62 leider ist das Logging leicht kaputt, mit 85c436393bea7dd5707756581a50a2250c004625 behoben. Ich sehe:

```
[lp-1  ] DEBUG 2024/06/15 09:31:53 charge currents: [5.5 0 0]A
[lp-1  ] DEBUG 2024/06/15 09:31:53 set charge mode: now
[circuit-carport] DEBUG 2024/06/15 09:31:55 validate current: 6A -> 20.5A <= 20A at 5.5A: capped at 20.5A
[lp-1  ] DEBUG 2024/06/15 09:31:55 max charge current: 20.5A
```

--> hier müsste nochmal gerundet werden, denn sonst wird 20.5A immer von 20A im Loadpoint abweichen. Geändert in 06fe501edc747f3bf91a0469db9df54044ed9ad4.

```
[lp-1  ] DEBUG 2024/06/15 09:31:58 charge currents: [19.3 0 0]A
[lp-1  ] DEBUG 2024/06/15 09:32:03 charge currents: [19.4 0 0]A
```

Hier passiert dann nach meinem Veständnis der Fehler:

```
[circuit-carport] DEBUG 2024/06/15 09:32:14 validate current: 20.5A -> 21.2A <= 20A at 19.3A: capped at 21.2A
[lp-1  ] DEBUG 2024/06/15 09:32:14 max charge current: 21.2A
```

Loadpoint will von 20.5 auf 21.2 laut Circuit. Tatsächlich will we aber von 20.0 auf 21.2. Fehler ist genau die fehlender Rundung auf ganze Ampere.

Vorschlag: in `master` nochmal nachtesten, dann gibts auch ein besseres Log.

VolkerK62, 2024-06-17:
ok. Kann ich morgen machen.

by the way, das Nightly ist letzte Nacht nicht durchgelaufen.

VolkerK62, 2024-06-18:
@andig  So funktioniert es besser.
Vermutlich aber nur solange, wie die Differenz zwischen Freigabe und tatsächlichem Strom kleiner als 1A ist.
Anbei das Log (in schön) 😄 

```
 [site  ] DEBUG 2024/06/18 09:22:02 ----
 [lp-1  ] DEBUG 2024/06/18 09:22:02 charge power: 2861W
 [lp-1  ] DEBUG 2024/06/18 09:22:02 charge currents: [12.4 0 0]A
 [lp-1  ] DEBUG 2024/06/18 09:22:02 set charge mode: now
 [circuit-carport] DEBUG 2024/06/18 09:22:03 power: 2863W
 [circuit-carport] DEBUG 2024/06/18 09:22:03 current: 12.4A
 [circuit-main] DEBUG 2024/06/18 09:22:03 power: -1427.4W
 [circuit-main] DEBUG 2024/06/18 09:22:03 current: 6.8A
 [site  ] DEBUG 2024/06/18 09:22:03 pv power: 4711W
 [site  ] DEBUG 2024/06/18 09:22:03 battery soc: 66%
 [site  ] DEBUG 2024/06/18 09:22:03 battery power: 118W
 [site  ] DEBUG 2024/06/18 09:22:03 grid meter: -1427W
 [site  ] DEBUG 2024/06/18 09:22:03 grid powers: [-1343 -1521 1440]W
 [site  ] DEBUG 2024/06/18 09:22:03 grid currents: [-5.75 -6.59 6.79]A
 [site  ] DEBUG 2024/06/18 09:22:03 aux power 1: 0W
 [site  ] DEBUG 2024/06/18 09:22:03 aux power: 0W
 [site  ] DEBUG 2024/06/18 09:22:03 site power: -309W
 [site  ] DEBUG 2024/06/18 09:22:04 ----
 [lp-1  ] DEBUG 2024/06/18 09:22:04 charge power: 2861W
 [lp-1  ] DEBUG 2024/06/18 09:22:04 charge currents: [12.4 0 0]A
 [circuit-carport] DEBUG 2024/06/18 09:22:04 power: 2861W
 [circuit-carport] DEBUG 2024/06/18 09:22:04 current: 12.4A
 [circuit-main] DEBUG 2024/06/18 09:22:04 power: -1429.7W
 [circuit-main] DEBUG 2024/06/18 09:22:04 current: 6.78A
 [site  ] DEBUG 2024/06/18 09:22:04 pv power: 4711W
 [site  ] DEBUG 2024/06/18 09:22:04 battery soc: 66%
 [site  ] DEBUG 2024/06/18 09:22:04 battery power: 134W
 [site  ] DEBUG 2024/06/18 09:22:05 grid meter: -1430W
 [site  ] DEBUG 2024/06/18 09:22:05 grid powers: [-1344 -1523 1437]W
 [site  ] DEBUG 2024/06/18 09:22:05 grid currents: [-5.75 -6.59 6.78]A
 [site  ] DEBUG 2024/06/18 09:22:05 aux power 1: 0W
 [site  ] DEBUG 2024/06/18 09:22:05 aux power: 0W
 [site  ] DEBUG 2024/06/18 09:22:05 site power: -296W
 [lp-1  ] DEBUG 2024/06/18 09:22:05 charge voltages: [231 0 0]V
 [lp-1  ] DEBUG 2024/06/18 09:22:05 detected connected phases: 1p
 [lp-1  ] DEBUG 2024/06/18 09:22:05 detected active phases: 1p
 [lp-1  ] DEBUG 2024/06/18 09:22:05 !! session: chargeRater.chargedEnergy=0.1 - chargedAtStartup=0.0
 [lp-1  ] DEBUG 2024/06/18 09:22:05 charge total import: 1865.600kWh
 [lp-1  ] DEBUG 2024/06/18 09:22:05 charger status: C
 [lp-1  ] DEBUG 2024/06/18 09:22:05 soc estimated: 72.12% (vehicle: 72.00%)
 [lp-1  ] DEBUG 2024/06/18 09:22:05 vehicle soc: 72%
 [lp-1  ] DEBUG 2024/06/18 09:22:05 vehicle soc limit: 100%
 [lp-1  ] DEBUG 2024/06/18 09:22:05 vehicle range: 332km
 [circuit-carport] DEBUG 2024/06/18 09:22:05 validate current: 13A -> 32A <= 20A at 12.4A: capped at 20.6A
 [lp-1  ] DEBUG 2024/06/18 09:22:05 !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
 [lp-1  ] DEBUG 2024/06/18 09:22:05 max charge current: 20A
 [site  ] DEBUG 2024/06/18 09:22:07 ----
 [lp-1  ] DEBUG 2024/06/18 09:22:07 charge power: 4417W
 [lp-1  ] DEBUG 2024/06/18 09:22:07 charge currents: [19.3 0 0]A
 [circuit-carport] DEBUG 2024/06/18 09:22:08 power: 4440W
 [circuit-carport] DEBUG 2024/06/18 09:22:08 current: 19.4A
 [circuit-main] DEBUG 2024/06/18 09:22:08 power: 146.08W
 [circuit-main] DEBUG 2024/06/18 09:22:08 current: 13.7A
 [site  ] DEBUG 2024/06/18 09:22:08 pv power: 4711W
 [site  ] DEBUG 2024/06/18 09:22:08 battery soc: 66%
 [site  ] DEBUG 2024/06/18 09:22:08 battery power: 202W
 [site  ] DEBUG 2024/06/18 09:22:08 grid meter: 146W
 [site  ] DEBUG 2024/06/18 09:22:08 grid powers: [-1339 -1521 3006]W
 [site  ] DEBUG 2024/06/18 09:22:08 grid currents: [-5.71 -6.59 13.7]A
 [site  ] DEBUG 2024/06/18 09:22:08 aux power 1: 0W
 [site  ] DEBUG 2024/06/18 09:22:08 aux power: 0W
 [site  ] DEBUG 2024/06/18 09:22:08 site power: 1348W
 [lp-1  ] DEBUG 2024/06/18 09:22:08 charge voltages: [229 0 0]V
 [lp-1  ] DEBUG 2024/06/18 09:22:08 detected connected phases: 1p
 [lp-1  ] DEBUG 2024/06/18 09:22:08 detected active phases: 1p
[... 202 lines cut ...]
 [circuit-main] DEBUG 2024/06/18 09:23:08 power: 180.09W
 [circuit-main] DEBUG 2024/06/18 09:23:08 current: 13.9A
 [site  ] DEBUG 2024/06/18 09:23:08 pv power: 4565W
 [site  ] DEBUG 2024/06/18 09:23:08 battery soc: 66%
 [site  ] DEBUG 2024/06/18 09:23:08 battery power: 126W
 [site  ] DEBUG 2024/06/18 09:23:08 grid meter: 180W
 [site  ] DEBUG 2024/06/18 09:23:08 grid powers: [-1430 -1454 3064]W
 [site  ] DEBUG 2024/06/18 09:23:08 grid currents: [-6.11 -6.31 14]A
 [site  ] DEBUG 2024/06/18 09:23:08 aux power 1: 0W
 [site  ] DEBUG 2024/06/18 09:23:08 aux power: 0W
 [site  ] DEBUG 2024/06/18 09:23:08 site power: 1306W
 [lp-1  ] DEBUG 2024/06/18 09:23:09 charge voltages: [228 0 0]V
 [lp-1  ] DEBUG 2024/06/18 09:23:09 detected connected phases: 1p
 [lp-1  ] DEBUG 2024/06/18 09:23:09 detected active phases: 1p
 [lp-1  ] DEBUG 2024/06/18 09:23:09 !! session: chargeRater.chargedEnergy=0.2 - chargedAtStartup=0.0
 [lp-1  ] DEBUG 2024/06/18 09:23:09 charge total import: 1865.700kWh
 [lp-1  ] DEBUG 2024/06/18 09:23:09 charger status: C
 [lp-1  ] DEBUG 2024/06/18 09:23:09 soc estimated: 72.25% (vehicle: 72.00%)
 [lp-1  ] DEBUG 2024/06/18 09:23:09 vehicle soc: 72%
 [lp-1  ] DEBUG 2024/06/18 09:23:09 vehicle soc limit: 100%
 [lp-1  ] DEBUG 2024/06/18 09:23:09 vehicle range: 332km
 [circuit-carport] DEBUG 2024/06/18 09:23:09 validate current: 20A -> 32A <= 20A at 19.3A: capped at 20.7A
 [lp-1  ] DEBUG 2024/06/18 09:23:09 !! active phases: 1p = min(1p measured 0p vehicle 1p physical 0p charger)
 [site  ] DEBUG 2024/06/18 09:23:12 ----
```

andig, 2024-06-18:
> Vermutlich aber nur solange, wie die Differenz zwischen Freigabe und tatsächlichem Strom kleiner als 1A ist.

Warum? Es wird immer auf Delta geregelt. Nur war hier das Regeldelta ein anderes als das physische. Das ist jetzt behoben.

VolkerK62, 2024-06-18:
Annahme: 
Es würde tatsächlich nur mit 18,3A geladen werden, dann würde diese Zeile doch so aussehen, oder?
` [circuit-carport] DEBUG 2024/06/18 09:23:09 validate current: 20A -> 32A <= 20A at 18.3A: capped at 21.7A`
Was passiert dann bei `capped at 21.7A`?
Wird dann auf 21A hochgeregelt?

andig, 2024-06-18:
Ja, und das wäre ja auch korrekt. Wenn noch 1.7A Platz sind dann darf das Auto ruhig +1A nehmen.

andig, 2024-06-18:
Logging in https://github.com/evcc-io/evcc/commit/97bb90c68b3310055082b860c31b9f488ffa0160 nochmals optimiert um die Rechnung nachvollziehbarer zu machen.

VolkerK62, 2024-06-18:
.

closed 2024-06-18 by VolkerK62