evcc-io/evcc #14418: MinCurrent setting not working for Elli wallbox

ilion1, 2024-06-16:
### Describe the bug

While Elli is working as expected in all modes and also MaxCurrent setting results in adjusted current, the MinCurrent setting is ignored and will always charge with 6A regardless of changed setting. This has been the case for many versions and I do not recall whether it has ever been working correctly. Today I tried once more with version 0.127.2 with the same result that any change to MinCurrent is ignored.

### Steps to reproduce

Fresh start of EVCC with following steps and timestamp (trace logs are attached):

1. connect car with modus PV: 18:10:45
2. change mode to NOW: 18:11:30
3. change max current to 12A: 18:12:30 --> change successful
4. change max current to 16A: 18:13:45 --> change successful
5. change mode to MinPV: 18:14:30
6. change min current to 10A: 18:15:30 --> change ignored, stays on 6A
7. change min current to 8A: 18:16:30 --> change ignored, stays on 6A
8. change min current to 6A: 18:17:30
9. disconnect car: 18:18:00

[evcc-logs.zip](https://github.com/user-attachments/files/15859093/evcc-logs.zip)

### Configuration details

```yaml
see attached files
```

### Log details

```text
see attached files
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.127.2

andig, 2024-06-16:
Missing config

ilion1, 2024-06-16:
Config is attached like written above.
Here once more:
```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
#  eebus: trace

# unique installation id
plant: xxx

interval: 10s # control cycle interval

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.189.38  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.189.38  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.189.38  
  port: 8080  
  name: battery3

chargers:
- type: template
  template: ellipro 
  ski: D9A2 03D9 E918 F950 EC97 E593 7251 E179 79A3 905E  
  ip: 192.168.189.34  
  name: wallbox5

loadpoints:
- title: Elli Wallbox
  charger: wallbox5
  mode: pv

site:
  title: Zu Hause
  residualPower: 100
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          xxx
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          xxx
          -----END CERTIFICATE-----

```

andig, 2024-06-16:
Please share relevant trace log here inline.

ilion1, 2024-06-16:
OK, here you go, but truncated because too long. Full log downloadable (see above)

```
[main  ] INFO 2024/06/16 18:10:13 evcc 0.127.2
[main  ] INFO 2024/06/16 18:10:13 using config file: /home/pi/evcc/evcc.yaml
[db    ] INFO 2024/06/16 18:10:13 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/06/16 18:10:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT * FROM `settings` 4 <nil>
[eebus ] INFO 2024/06/16 18:10:13 Local SKI:  1d5be7dbd5197baaed9616d379fc6d3d2dd5ebcc
[eebus ] DEBUG 2024/06/16 18:10:13 mdns: announce
[eebus ] DEBUG 2024/06/16 18:10:13 mdns: using avahi
[eebus ] DEBUG 2024/06/16 18:10:13 starting websocket server on :4712
[db    ] TRACE 2024/06/16 18:10:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[main  ] INFO 2024/06/16 18:10:13 listening at :7070
[db    ] TRACE 2024/06/16 18:10:13 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2024/06/16 18:10:13 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[eebus ] TRACE 2024/06/16 18:10:13 registering ski: d9a203d9e918f950ec97e5937251e17979a3905e
[eebus ] DEBUG 2024/06/16 18:10:13 mdns: start search
[eebus ] DEBUG 2024/06/16 18:10:13 ski: d9a203d9e918f950ec97e5937251e17979a3905e name: Elli-Wallbox-2125A1G53O brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2125A1G53O register: false host: wallbox-2125A1G53O.local port: 4712 addresses: [192.168.189.34]
[eebus ] DEBUG 2024/06/16 18:10:13 delaying connection to d9a203d9e918f950ec97e5937251e17979a3905e by 1.969s to minimize double connection probability
[eebus ] DEBUG 2024/06/16 18:10:15 incoming connection request from d9a203d9e918f950ec97e5937251e17979a3905e
[eebus ] TRACE 2024/06/16 18:10:15 Recv: d9a203d9e918f950ec97e5937251e17979a3905e ship init
[eebus ] TRACE 2024/06/16 18:10:15 Send: d9a203d9e918f950ec97e5937251e17979a3905e ship init
[eebus ] TRACE 2024/06/16 18:10:15 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"connectionHello":[{"phase":"ready"},{"waiting":60000}]}
[eebus ] TRACE 2024/06/16 18:10:15 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"connectionHello":[{"phase":"ready"},{"waiting":60000}]}
[eebus ] TRACE 2024/06/16 18:10:15 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"messageProtocolHandshake":[{"handshakeType":"announceMax"},{"version":[{"major":1},{"minor":0}]},{"formats":[{"format":["JSON-UTF8"]}]}]}
[eebus ] TRACE 2024/06/16 18:10:15 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"messageProtocolHandshake":[{"handshakeType":"select"},{"version":[{"major":1},{"minor":0}]},{"formats":[{"format":["JSON-UTF8"]}]}]}
[eebus ] TRACE 2024/06/16 18:10:15 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"messageProtocolHandshake":[{"handshakeType":"select"},{"version":[{"major":1},{"minor":0}]},{"formats":[{"format":["JSON-UTF8"]}]}]}
[eebus ] TRACE 2024/06/16 18:10:15 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"connectionPinState":[{"pinState":"none"}]}
[... cut ...]
[eebus ] TRACE 2024/06/16 18:16:55 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"data":[{"header":[{"protocolId":"ee1.0"}]},{"payload":{"datagram":[{"header":[{"specificationVersion":"1.1.1"},{"addressSource":[{"device":"d:_i:EVCC_HEMS-EVCC-3639656234373032"},{"entity":[1]},{"feature":1}]},{"addressDestination":[{"device":"d:_i:47859_Elli-Wallbox-2125A1G53O"},{"entity":[1,1]},{"feature":1001}]},{"msgCounter":508},{"cmdClassifier":"notify"}]},{"payload":[{"cmd":[[{"deviceDiagnosisHeartbeatData":[{"timestamp":"2024-06-16T16:16:55Z"},{"heartbeatCounter":460},{"heartbeatTimeout":"PT4S"}]}]]}]}]}}]}
```

andig, 2024-06-16:
@DerAndereAndi hast du eine Idee? Die 8A kommen laut log nicht an, es bleibt bei 6A?

DerAndereAndi, 2024-06-16:
Siehe diese Diskussion: https://github.com/evcc-io/evcc/issues/11280#issuecomment-1870514593

andig, 2024-06-17:
Das hatten wir doch nach langer Diskussion dann mittels https://github.com/evcc-io/evcc/pull/11311 gelöst was alle Parteien zufrieden zu stellen schien? Es hatte sogar Deinen Daumen in https://github.com/evcc-io/evcc/issues/11280#issuecomment-1871230515.

andig, 2024-06-18:
@ilion1 Um was für ein Fahrzeug geht es denn überhaupt?

ilion1, 2024-06-18:
An der Wallbox laden ein KIA e-Niro (BJ 22, 11 kW-3 phasig) sowie ein Mercedes Hybrid (7,4 kW-2 phasig). Ich habe häufiger den Bedarf schneller als Min+PV (nur 1,4 kW pro Phase) mit ca. 6-7 kW unter Nutzung des vollen Speichers zu laden, wobei 16 A dann zu viel ist und Netzstrom benötigt würde. Das Feature ist für mich nicht super wichtig, da ich alternativ auf Modus "NOW" gehe und den maximalen Strom begrenze. Wenn dann allerdings mehr PV-Leistung verfügbar ist, wird wegen MaxCurrent dennoch nicht schneller geladen.

andig, 2024-06-29:
> Ich habe häufiger den Bedarf schneller als Min+PV (nur 1,4 kW pro Phase) mit ca. 6-7 kW unter Nutzung des vollen Speichers zu laden, wobei 16 A dann zu viel ist und Netzstrom benötigt würde

Du änderst das dann übers API bzw. über die UI, richtig?

Und laut deinem Issue scheint das Problem _nur_ zu bestehen, wenn Du im `MinPV` Modus bist, nicht aber sonst???

andig, 2024-06-29:
> Ah, und zeitlich danach kam https://github.com/evcc-io/evcc/issues/11280#issuecomment-1889506809. Scheint als gäbe es dieses Problem jetzt immer noch.

@DerAndereAndi letztlich scheinen wir das in https://github.com/evcc-io/evcc/pull/12866 behoben zu haben, siehe Testmatrix im PR.

DerAndereAndi, 2024-06-29:
Ich kenne die Details nicht mehr und auch nicht den aktuellen Code. Ich erinnere mich nur grob an die damalige Diskussion.

Bei mir funktoniert es, wenn im Vehicle eine höherer Min-Current konfiguriert wird. Hier in dem Beispiel geht es aber doch um die Änderung per API Call. Evtl. wird das nicht berücksichtigt?

andig, 2024-06-29:
Wartet auf Feedback von @ilion1 https://github.com/evcc-io/evcc/issues/14418#issuecomment-2198144708

andig, 2024-07-01:
@ilion1 ohne Input von Dir nichts weiter zu tun. Laut Tests müsste das gehen. Wenn Du den Fall nochmal provozieren kannst bräuchten wir bitte gleichzeitig den Output von /api/state.

ilion1, 2024-07-01:
Sorry, ich war zwei Tage offline.
Richtig: Es geht um den Modus Min+PV, für den ich manuell höhere Leistung einstellen möchte. Das mache ich über die experimentelle Funktion im GUI für Ladestrom am Ladepunkt. Wie man in den Logs sieht, wird der Strom erhöht aber die Box bleibt bei 6A. Hier ist die Ausgabe von /api/state, falls das hilft. Ich nutze aber keine API sondern nur die Weboberfläche. Übrigens funktioniert es mit einer ABLemh1 korrekt.
```
{
  "result": {
    "auth": {

    },
    "battery": [
      {
        "power": 170,
        "soc": 87,
        "controllable": false
      }
    ],
    "batteryCapacity": 0,
    "batteryDischargeControl": false,
    "batteryEnergy": 0,
    "batteryMode": "unknown",
    "batteryPower": 170,
    "batterySoc": 87,
    "bufferSoc": 100,
    "bufferStartSoc": 0,
    "currency": "EUR",
    "fatal": null,
    "greenShareHome": 1,
    "greenShareLoadpoints": 0.046,
    "gridConfigured": true,
    "gridPower": 49,
    "homePower": 0,
    "influx": {
      "url": "",
      "database": "",
      "org": "",
      "user": ""
    },
    "interval": 10,
    "loadpoints": [
      {
        "chargeCurrent": 6,
        "chargeCurrents": [5.35, 5.39, 5.25],
        "chargeDuration": 560,
        "chargePower": 3677.7,
        "chargeRemainingDuration": 17688,
        "chargeRemainingEnergy": 18070.052,
        "chargedEnergy": 575.316,
        "chargerFeatureHeating": false,
        "chargerFeatureIntegratedDevice": false,
        "chargerIcon": null,
        "chargerPhases1p3p": false,
        "chargerPhysicalPhases": null,
        "charging": true,
        "connected": true,
        "connectedDuration": 0,
        "disableThreshold": 0,
        "effectiveLimitSoc": 100,
        "effectiveMaxCurrent": 16,
        "effectiveMinCurrent": 6,
        "effectivePlanSoc": 0,
        "effectivePlanTime": null,
        "effectivePriority": 0,
        "enableThreshold": 0,
        "enabled": true,
        "limitEnergy": 0,
        "limitSoc": 0,
        "maxCurrent": 16,
        "minCurrent": 10,
        "mode": "minpv",
        "phaseAction": "inactive",
        "phaseRemaining": 0,
[... 186 lines cut ...]
    "vehicles": {
      "W213": {
        "title": "Mercedes E-Klasse",
        "capacity": 10,
        "limitSoc": 70
      },
      "ev4": {
        "title": "KIA e-Niro",
        "icon": "car",
        "capacity": 64,
        "limitSoc": 100
      },
      "ev5": {
        "title": "E-Bike",
        "icon": "car",
        "capacity": 0.5,
        "features": [
          "Offline"
        ]
      }
    },
    "version": "0.127.3"
  }
}
```

andig, 2024-07-02:
LP1 zeigt das Problem:

```
"minCurrent": 10,
"effectiveMinCurrent": 6,
"vehicleName": "ev4",
```

Am Auto gibt es soweit ich sehe keine weiteren Einstellungen.

> Ich nutze aber keine API sondern nur die Weboberfläche. 

Hättest Du noch einen Screenshot, wo Du das machst?

Passiert das _nur_ im minPV Modus?

ilion1, 2024-07-02:
Es passiert mit beiden Autos, die regelmäßig geladen werden gleich. Am Mercedes Hybrid (2-Phasen max 16A) kann man nichts einstellen. Am KIA e-Niro kann man auch am Auto runterstellen. Dann wird der Ladestrom durch das Auto fest begrenzt unabhängig davon, ob die Wallbox mehr liefern kann.
Eigentlich kommt das nur im Modus minPV zum Tragen. Bei PV würde das Laden wahrscheinlich erst starten, wenn ausreichend Leistung vorhanden ist, so dass statt 6A z.B. mit 10A gestartet würde. Das habe ich aber nicht ausprobiert, da es sehr schwer zu testen ist mit schwankender PV-Leistung. Grundsätzlich ist das Ziel, das Auto schneller als mit 6A zu laden. Wenn ich im Modus Now den maximalen Strom begrenze, funktioniert das zwar aber bei steigender PV-Leistung würde diese nicht benutzt.
Hier der Screenshot der Einstellung: minPV + z.B. 10A 
[image]

andig, 2024-07-02:
@DerAndereAndi welche Limits würde die Elli hier schicken wenn ein KIA angeschlossen ist:

```go
minLimits, maxLimits, _, err := c.uc.OpEV.CurrentLimits(evEntity)
if err != nil {
	if err == eebusapi.ErrDataNotAvailable {
		err = api.ErrNotAvailable
	}
	return minMax{}, err
}
```

gar keine oder Defaultwerte?

DerAndereAndi, 2024-07-02:
min 6A und max 16A

andig, 2024-07-02:
@DerAndereAndi könnten wir das so ändern, dass wir die Limits (in dem Fall ja IEC) nur dann schicken wenn sie tatsächlich vom Fahrzeug kommen, wir also im ISO Modus sind?

DerAndereAndi, 2024-07-02:
Das finde ich nicht gut, denn es verschiebt das Problem nur auf ISO Fahrzeuge/Wallboxen anstatt es für alle zu lösen. Aber auch bei ISO ist nicht immer der Wertebereich vom Auto bekannt.

andig, 2024-07-05:
@ilion1 current workaround would be to configure `minCurrent` on vehicle. Unfortunately we don't have an api for doing that yet.

andig, 2024-07-05:
> Das finde ich nicht gut, denn es verschiebt das Problem nur auf ISO Fahrzeuge/Wallboxen anstatt es für alle zu lösen. 

@DerAndereAndi ich sehe grad keine Rundumglücklichlösung. Für IEC würde es das Problem schonmal lösen- denn da erwarten/brauchen wir auch keine Limits vom Charger.

> Aber auch bei ISO ist nicht immer der Wertebereich vom Auto bekannt.

Was passiert in dem Fall? Lässt der sich im Protokoll identifizieren und dann ebenfalls keine Limits schicken?

ilion1, 2024-07-05:
> @ilion1 current workaround would be to configure `minCurrent` on vehicle. Unfortunately we don't have an api for doing that yet.

Not a real problem for me. I just wanted to report the bug as it may affect others or might be related to other issues.

chrikoch, 2024-08-18:
Ich häng mich hier mal mit dran, da es mir scheint dass ich das selbe Problem habe. Ich habe eine Elli Wallbox und u.a. einen Renault Twingo. Da der Twingo minimal 8A braucht zum Laden habe ich nicht nur am vehicle sondern auch am loadpoint den mincurrent auf 8A gesetzt. Da die Renault-API erfahrungsgemäß auch mal rumzickt möchte ich damit erreichen, dass mit 8A geladen wird, auch falls der Twingo nicht erkannt wurde. Sprich ein Gastfahrzeug soll auch mit 8A laden.

Hab es gerade mit der ganz aktuellen 0.130.0 versucht, trace log anbei. 

Hier sieht man, dass der Ladestrom trotz mincurrent 8 am lp geringer ist:

[lp-1  ] DEBUG 2024/08/18 13:45:23 max charge current: 6.7A

Erst nachdem der Twingo erkannt wurde, wird es auf 8A gesetzt, und erst dann fängt der Twingo auch wirklich an zu laden:

[lp-1  ] DEBUG 2024/08/18 13:45:54 max charge current: 8A

[evcc.log](https://github.com/user-attachments/files/16648242/evcc.log)

anonymisierte Config:

```
network:
  schema: https
  host: xxxxxxxxxxxxx
  port: 7070

site:
  - title: Home
    meters:
      grid: Grid
      pv: PV
      battery: Battery
    residualPower: 200

interval: 30s

database:
  type: sqlite
  dsn: /var/evcc/evcc.db

loadpoints:
  - title: Elli Carport
    charger: Elli
    mode: minpv #now
    mincurrent: 8 # set to 8, because Twingo needs it, so it is 8 in case the vehicle is not identified correctly

chargers:
  - name: Elli
    type: template
    template: elliconnect
    ski: xxxxxxxxxxxxxxxxxxxxxxxxxx
    ip: xxxxxxxxxxxxxxx

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.30
  co2:
    type: grünstromindex
    zip: xxxxxxxxxx

meters:
  - name: Grid
    type: template
    template: fronius-solarapi-v1
    usage: grid
    host: xxxxxxxxxxxxx
  - name: PV
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: xxxxxxxxxxxxx
  - name: Battery
    type: template
    template: fronius-solarapi-v1
    usage: battery
    host: xxxxxxxxx

vehicles:
  - name: Tequila
    title: Tequila
    type: template
    template: renault
    capacity: 22
    user: xxxxxxxxxxxx
    password: xxxxxxxxxxxxxxxxx
    mincurrent: 8 #is also set as default in the loadpoint, but set it here to be sure
    #targetSoc: 80
  - name: Tesla
    title: 42
    type: template
    template: tesla
    accessToken:xxxxxxxxxxxx
    refreshToken: xxxxxxxxxxxxxxxxxxxxxxxxxx
    capacity: 60 # Akkukapazität in kWh (optional)
    mincurrent: 6 #tesla supports min 6A

eebus:
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
      xxxxxxxxxxxxxxxxxxxxxxxxxx
      -----END CERTIFICATE-----

    private: |
      -----BEGIN EC PRIVATE KEY-----
      xxxxxxxxxxxxxxxxxxxxxxxxxx
      -----END EC PRIVATE KEY-----

```

P.S.: Ein riesen Dankeschöne für eure Arbeit und dieses tolle Projekt!!!