evcc-io/evcc #14419: No Connection to Skoda Enyaq within EVCC

GitDaniel24, 2024-06-16:
### Describe the bug

Within the EVCC App i get a crossed cloud with the error message "Vehicle not reachable". Reboot of EVCC NOK. Timeout is set to 10 sec.
If i check the logs i found the following:

[main ] ERROR 2024/06/16 18:11:05 creating vehicle Enyaq failed: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': Post "https://mysmob.api.connect.skoda-auto.cz/api/v1/authentication/exchange-authorization-code?tokenType=CONNECT": dial tcp: lookup mysmob.api.connect.skoda-auto.cz: i/o timeout

### Steps to reproduce

1.Add a Templeate skoda Enyaq 
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: Enyaq
    type: template
    template: enyaq
    title: Skoda Enyaq
    user: username
    password: password
    vin: vin
    capacity: 82
    timeout: 10s
```

### Log details

```text
[main ] ERROR 2024/06/16 18:11:05 creating vehicle Enyaq failed: cannot create vehicle type 'template': cannot create vehicle type 'enyaq': Post "https://mysmob.api.connect.skoda-auto.cz/api/v1/authentication/exchange-authorization-code?tokenType=CONNECT": dial tcp: lookup mysmob.api.connect.skoda-auto.cz: i/o timeout
```

### What type of operating system are you running?

Docker container

andig, 2024-06-16:
Io timeout sounds like network issue

/cc @GrimmiMeloni

GitDaniel24, 2024-06-16:
On skoda side?! Network/internet on my side works without any problems…

andig, 2024-06-16:
Whatever. An io timeout is just that.

GrimmiMeloni, 2024-06-16:
Yeah. DNS lookup times out. Some issue outside evcc. Likely will fix itself after some time.

closed 2024-06-16 by andig