evcc-io/evcc #14424: MG ZS EV - Fehler beim erkennen der Fahrzeuges

skipperkm, 2024-06-17:
### Describe the bug

Hallo,

bei der Erkennung des Fahrzeuges oder auch im laufenden Betrieb (habe die Erkennung jetzt umgangen) erhalte ich häufig folgende Fehlermeldung:

Wallbox: vehicle odometer: Get \"https://gateway-mg-eu.soimt.com/api.app/v1/vehicle/charging/mgmtData?vin=4b0227f31b2b5b07c93968216b3783528d2baa61efb3cfeca76f670f3772b1d8\": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

Diese Meldung kommt auch auch bezogen auf andere Werte (SoC, Range)

Woran kann das liegen?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- name: wallbox1
    type: template
    template: cfos
    usage: charge
    # Modbus TCP
    modbus: tcpip
    id: 2
    host: 172.16.98.74
    port: 4702

chargers:
  - name: wb1
    type: template
    template: cfos
    host: 172.16.98.74

loadpoints:
  - title: Wallbox
    charger: wb1
    mode: off
    phases: 0
    vehicle: mgzsev

vehicles:
  - name: mgzsev
    type: template
    template: mg
    title: Auto # Wird in der Benutzeroberfläche angezeigt (optional)
    user: ******@********.de # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: ********** # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: **************** # Erforderlich
    capacity: 70 # Akkukapazität in kWh
```

### Log details

```text
[lp-1 ] INFO 2024/06/17 09:28:09 vehicle updated: unknown -> B-KK 863
[lp-1 ] DEBUG 2024/06/17 09:28:09 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[site ] DEBUG 2024/06/17 09:28:09 ----
[lp-1 ] DEBUG 2024/06/17 09:28:09 charge power: 0W
[site ] DEBUG 2024/06/17 09:28:09 pv power: 2354W
[site ] DEBUG 2024/06/17 09:28:09 battery soc: 100%
[site ] DEBUG 2024/06/17 09:28:09 battery power: 0W
[site ] DEBUG 2024/06/17 09:28:09 grid meter: -1974W
[site ] DEBUG 2024/06/17 09:28:09 site power: -1874W
[mg ] DEBUG 2024/06/17 09:28:09 StatInvaild. Starting query
[mg ] DEBUG 2024/06/17 09:28:19 Getting event id failed
[lp-1 ] ERROR 2024/06/17 09:28:19 vehicle odometer: Get "https://gateway-mg-eu.soimt.com/api.app/v1/vehicle/charging/mgmtData?vin=4b0227f31b2b5b07c93968216b3783528d2baa61efb3cfeca76f670f3772b1d8": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1 ] DEBUG 2024/06/17 09:28:19 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/17 09:28:20 charger status: A
[lp-1 ] INFO 2024/06/17 09:28:20 car disconnected
[lp-1 ] DEBUG 2024/06/17 09:28:20 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/17 09:28:20 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/17 09:28:20 set session soc limit: 0
[lp-1 ] DEBUG 2024/06/17 09:28:20 set session energy limit: 0
[lp-1 ] DEBUG 2024/06/17 09:28:20 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/17 09:28:23 set phases: 1
[site ] DEBUG 2024/06/17 09:28:23 ----
[lp-1 ] DEBUG 2024/06/17 09:28:23 charge power: 0W
[site ] DEBUG 2024/06/17 09:28:23 pv power: 2328W
[site ] DEBUG 2024/06/17 09:28:23 battery soc: 100%
[site ] DEBUG 2024/06/17 09:28:23 battery power: 0W
[site ] DEBUG 2024/06/17 09:28:23 grid meter: -1944W
[site ] DEBUG 2024/06/17 09:28:23 site power: -1844W
[lp-1 ] ERROR 2024/06/17 09:28:23 vehicle odometer: Get "https://gateway-mg-eu.soimt.com/api.app/v1/vehicle/charging/mgmtData?vin=4b0227f31b2b5b07c93968216b3783528d2baa61efb3cfeca76f670f3772b1d8": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[lp-1 ] DEBUG 2024/06/17 09:28:23 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/17 09:28:23 charger status: A
[site ] DEBUG 2024/06/17 09:28:52 ----
[lp-1 ] DEBUG 2024/06/17 09:28:52 charge power: 0W
[site ] DEBUG 2024/06/17 09:28:52 pv power: 2316W
[site ] DEBUG 2024/06/17 09:28:52 battery soc: 100%
[site ] DEBUG 2024/06/17 09:28:52 battery power: 0W
[site ] DEBUG 2024/06/17 09:28:53 grid meter: -1949W
[site ] DEBUG 2024/06/17 09:28:53 site power: -1849W
[lp-1 ] DEBUG 2024/06/17 09:28:53 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/17 09:28:53 charger status: A
[site ] DEBUG 2024/06/17 09:29:22 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.127.2

andig, 2024-06-17:
Looks like api is horribly slow. How should we handle this?

skipperkm, 2024-06-17:
Keine Ahnung, ich dache ihr habt eine Idee was zu tun ist. Vielleicht hilft ein einstellbarer Timeout für das Template?
Wie lange dauert es bis der Timeout zuschlägt?

andig, 2024-06-18:
Maybe @kscholty knows who has built the MG integration.

kscholty, 2024-06-18:
That really looks like a slow  server on MG's side. I have that sometimes as well.
I'll create a pull request to slightly extend it.

closed 2024-06-30 by andig

kscholty, 2024-07-27:
@andig Ist das Problem eigentlich behoben? Ich konnte keinen zugehörigen Fix finden. 
Soll ich danach nochmal schauen?

Gorw3n, 2025-07-13:
hi all,
I get ""The service is not available,please try again later." when trying to communicate towards
https://gateway-mg-eu.soimt.com/api.app/v1/
does the integration now use another endpoint? 
We got an MG-4 and it would be wonderful to be able to include the car in evcc.