evcc-io/evcc #14429: Mercedes: Documentation misleading after change #13812

ReneNulschDE, 2024-06-17:
### Describe the bug

The change #13812 introduced a new [documentation page](https://docs.evcc.io/docs/devices/vehicles#mercedes-benz). The new output is misleading. User are creating the yaml entry based on the output, but the token structure is wrong. I'm not sure why it was changed.

Actual help text:
```
    region: EMEA
    accessToken: # Siehe https://docs.evcc.io/docs/devices/vehicles#mercedes
    refreshToken: # Siehe https://docs.evcc.io/docs/devices/vehicles#mercedes
    vin: V...
```
Required help text:
Actual output:
```
    region: ...
    tokens:
      access: ...
      refresh: ...
    vin: ...
```

### Steps to reproduce

See [here](https://github.com/evcc-io/evcc/discussions/12581#discussioncomment-9791069) for example

### Configuration details

```yaml
N/A
```

### Log details

```text
N/A
```

### What type of operating system are you running?

Linux

deadrabbit87, 2024-06-17:
What also should me mentioned: E-Mail-Adress only in lower case. 

If there is one big case, you'll get an error.

VolkerK62, 2024-06-17:
the syntax differs whether you use `type: mercedes` or `type: template` (`template: mercedes`).

ReneNulschDE, 2024-06-17:
What is the prefered way from core team perspective `type: mercedes` or `type: template` (`template: mercedes`)?

andig, 2024-06-17:
Templates have meta data. The UI-based config will always use templates without forbidding to create a custom vehicle via YAML. The token generator could check if the type is template or different, it actually already does. It just doesn't use that information for printing the resulting tokens.

stefaanbolle, 2024-06-18:
If you define the mercedes via the UI, how do you generate a token? evcc token xxx gives an error since xxx is still unknown until validated and validating fails due to no tokens

andig, 2024-06-18:
/cc @naltatis we‘ll need to find a solution for this. Save broken vehicles?

closed 2024-06-18 by andig

ReneNulschDE, 2024-06-18:
The problem is not solved. Looks like that the template mode does not work even with the correct config. I'll try to debug it and write on slack in case of further dev-questions...