evcc-io/evcc #14430: Hoymiles DTU-PRO

andig, 2024-06-17:
Hallo, ich habe die selbe Anforderung. Hier schon mal die Modbus Spezifikation für alle Hoymiles DTU-PRO.
[Technical-Note-Modbus-implementation-using-3Gen-DTU-Pro-V1.2.pdf](https://github.com/evcc-io/evcc/files/14826692/Technical-Note-Modbus-implementation-using-3Gen-DTU-Pro-V1.2.pdf)

_Originally posted by @Sushifix in https://github.com/evcc-io/evcc/discussions/11302#discussioncomment-8974951_

premultiply, 2024-06-20:
Das Gerät unterstützt laut Doku SunSpec.
Man muss halt jeden Micro-WR einzeln als Meter anlegen...

closed 2024-06-30 by andig

ZaphodB72, 2024-07-18:
Welche Voraussetzungen gibt es noch um den Hoymiles in EVCC einzufügen?
Wenn ich unter Meters die Konfig vom "Sunspec-inverter" eingebe und die IP auf die des Hoymiles ändere, steht im Log nur "Connection Refused" und EVCC startet nicht mehr.

andig, 2024-07-19:
Dann antwortet das Gerät nicht. Config richtig?

ZaphodB72, 2024-07-19:
Wenn ich das mal wüsste… Ich kann ja nur die IP eintragen, die stimmt auf jeden Fall. Eine ID ist optional steht da, hab da nix drin stehen aber auch 0 oder 1 oder 2 und die Seriennr vom WR versucht, immer das gleiche…

andig, 2024-07-19:
Ähhh- wie ist das Ding denn angeschlossen? Nicht über seriellen Adapter? Und lass uns doch bitte nicht raten- wie sieht Deine Config aus?

ZaphodB72, 2024-07-19:
Der 800W hat ja ein eigenes WLAN-Modul, worüber er verbundenen ist. Gewählt habe ich deshalb TCP/IP.
Bin mir halt weder mit ID noch mit Port sicher, hab aber nix am Standard geändert.

meters:
  - name: my_pv
    type: template
    template: sunspec-inverter
    usage: pv4
  
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.48 # Hostname
    port: 502 # Port

Sushifix, 2024-07-19:
Hi, Modbus unterstützt nur die DTU-PRO und auch nur über LAN. Es kann keine Kommunikation zu dem Hoymiles Wechselrichter direkt per MODUS aufgebaut werden. Hast du eine DTU-PRO?

ZaphodB72, 2024-07-19:
Ääähm wie gesagt… ich habe einen HMS-800W-2T und keine zusätzlichen Logger oder Geräte daran angebaut. Im Homeassistant gibt’s ne Integration die ihn mir da anzeigt aber in EVCC bekomme ich die Werte nicht rein…

wyssnet, 2024-07-22:
> Hi, Modbus unterstützt nur die DTU-PRO und auch nur über LAN. Es kann keine Kommunikation zu dem Hoymiles Wechselrichter direkt per MODUS aufgebaut werden. Hast du eine DTU-PRO?

Hallo
Ich habe eine DTU Pro via Lan angeschlossen
Müsste ich den jetzt nicht so auslesen können ?
  - name: my_pv
    type: template
    template: sunspec-inverter
    usage: pv

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.1.217 # Hostname
    port: 502 # Port 

Irgendwie krieg ich das nicht hin
Fehler:
cannot create meter 'my_pv': cannot create meter type 'template': cannot create meter type 'custom': power: not a SunSpec device

Kannst du mir eine tipp geben was ich falsch mache ?
Danke Dir
Dany