evcc-io/evcc #14431: Allow "MaxCurrent" to return "errAsleep", or add status "asleep" to vehicles

Maddimax, 2024-06-17:
**Is your feature request related to a problem? Please describe.**
I have configured my tesla vehicle with part teslalogger (all "read" values via mqtt) and part custom BLE control (via. Http).
When the tesla is asleep, evcc fails to wake it up, as there is no way to tell evcc that the vehicle is asleep (that I could find)

**Describe the solution you'd like**
Either: A way to specify an "asleep" state or a way to create a "failed because asleep" http response.

**Describe alternatives you've considered**
I could always call wakeup in my BLE proxy for any request from evcc, but that seems a bit inefficient.

andig, 2024-06-18:
Question is: how? I don‘t see a good option to add a side channel for errrors? We could support them via text. Receiving ine of the supported values would convert them to errors: `ErrAsleep`, `ErrMustRetry` and `ErrNotAvailable`.

Maddimax, 2024-06-18:
Maybe (pseudocode):

```
if (http.status == 500) {
   if (body == "ErrAsleep") return ErrAsleep;
  ....
}
```

andig, 2024-06-18:
Genau, ich würde es aber nicht an einen Status binden. Oder hilft das? Und es muss natürlich über Websockets und Mqtt identisch laufen.

andig, 2024-06-18:
Bitte mal ausprobieren. Für HTTP ist Status != 200 Voraussetzung dass die Fehler abgefragt würden.

Maddimax, 2024-06-18:
Websockets würde wahrscheinlich gehen, Mqtt hat halt keine response zu nem publish, von daher ginge da wohl nur ein state, ala:

vehicle:
  asleep:
    source: mqtt
    topic: ....

Und dann ist die frage ob man nicht das gleiche auch für andere benutzen sollte anstatt da zu versuchen error handling zu betreiben.

andig, 2024-06-19:
Einfach mal ausprobieren. 3 Fehler x mind. 20 Apis x 4 Geräteklasse auszumultiplizieren wie du das vorschlägst ist keine Option. 
Websockets und mqtt sind async, da muss die Abfrage also vorher passiert sein. Und auch die kann natürlich einen Fehler wie ErrAsleep liefern.

andig, 2024-06-20:
@Maddimax magst Du den PR mal ausprobieren?

andig, 2024-06-22:
ping @Maddimax

Maddimax, 2024-06-22:
Sry, bin unterwegs. Ich Probier es nexte Woche.

Maddimax, 2024-06-23:
Die fehlerbehandlung funktioniert super 👍

FYI: Mein kleiner tesla-ble-proxy: https://github.com/Maddimax/tesla-proxy

andig, 2024-06-23:
Private repo? Anyway seems we can merge?

Maddimax, 2024-06-23:
Ups, public now. +1 from me for merging it.

closed 2024-06-23 by andig

andig, 2024-06-24:
Wenn du Lust hast meld dich gerne mal auf Slack- wir könnten Entwicklerunterstützung gut brauchen 😀