evcc-io/evcc #14438: Mercedes Integration (400) Bad Request

HiroP77, 2024-06-17:
### Describe the bug

Hi,

when I try to use the Mercedes API Integration, after loading up the tokens always get a

**[main ] ERROR 2024/06/17 14:32:18 creating vehicle GLE failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: unexpected status: 400 (Bad Request)**

### Steps to reproduce

Normal installation (works fine with mqtt as a SOC and RANGE-Source). 

1. set up the vehicle to template mercedes
2. generate tokens (region EMEA)
3. insert Tokens in config
4. reload

main ] INFO 2024/06/17 14:32:17 evcc 0.127.2
[main ] INFO 2024/06/17 14:32:17 using config file: /etc/evcc.yaml
[db ] INFO 2024/06/17 14:32:17 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/06/17 14:32:17 listening at :7070
[mercedes] DEBUG 2024/06/17 14:32:18 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/06/17 14:32:18 identity.NewIdentity - token not valid - database token check started
[main ] ERROR 2024/06/17 14:32:18 creating vehicle GLE failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: unexpected status: 400 (Bad Request)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.XXX.de # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: 70aa4cbd33f72a7aXXXXXXX52482c4b96d58a8425b4e011f57db952f8536fd

interval: 30s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NzMyMTg2MDAsImlhdCI6MTY3ODYxMDYwMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6Ikhpcm9QNzcifQ.ZhEsZPJjj784fr-fYHwCirFQ9awIHG0fe3NexGnIMrk

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: 10.0.0.169
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: kostal-plenticore
  id: 71
  host: 10.0.0.169
  port: 1502
  usage: battery
  modbus: tcpip
  name: battery2
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: 10.0.0.169
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid3

chargers:
- type: template
  template: go-e
  host: 10.0.0.49
  name: wallbox5

vehicles:
- name: GLE
  type: template
  template: mercedes
  user: EMAILADRESSS
  region: EMEA
  accessToken: TOKENTOKENTOKEN
  refreshToken: TOKENTOKEN
  capacity: 27

loadpoints:
- title: Stieldorf
  charger: wallbox5
  mode: pv
  vehicle: GLE

site:
- title: Stieldorf
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2
  residualPower: 300

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3777 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0699 # [currency]/kWh
```

### Log details

```text
main ] INFO 2024/06/17 14:32:17 evcc 0.127.2
[main ] INFO 2024/06/17 14:32:17 using config file: /etc/evcc.yaml
[db ] INFO 2024/06/17 14:32:17 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/06/17 14:32:17 listening at :7070
[mercedes] DEBUG 2024/06/17 14:32:18 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/06/17 14:32:18 identity.NewIdentity - token not valid - database token check started
[main ] ERROR 2024/06/17 14:32:18 creating vehicle GLE failed: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: unexpected status: 400 (Bad Request)
```

### What type of operating system are you running?

Linux

### Version

0.127.2

HiroP77, 2024-06-17:
@ReneNulschDE as requested

ReneNulschDE, 2024-06-17:
OK, Thx. So no Docker or HA-Addon...

First please add the database as a fixed config value to the your evcc.yaml.
Add:
```
database:
  type: sqlite
  dsn: /root/.evcc/evcc.db
```

Then
Change the config of the vehicle
```
vehicles:
- name: GLE
  type: mercedes
  user: EMAILADRESSS
  region: EMEA
  tokens:
    accessToken: TOKENTOKENTOKEN
    refreshToken: TOKENTOKEN
  capacity: 27
```

Save the changes and restart the service. Now we have a common starting point for future analysis.

HiroP77, 2024-06-17:
ah, now I see the error. 

the [documentation](https://docs.evcc.io/docs/devices/vehicles#mercedes-benz) speaks of
"
vehicles:
  - name: my_car
    type: template
    template: mercedes
    title: # Wird in der Benutzeroberfläche angezeigt (optional)
    user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    region: EMEA
    **accessToken: # Siehe https://docs.evcc.io/docs/devices/vehicles#mercedes
    refreshToken: # Siehe https://docs.evcc.io/docs/devices/vehicles#mercedes**
    vin: V... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    capacity: 50 # Akkukapazität in kWh (optional) 
"

but evcc expects a different format:

 region: EMEA
  **tokens:
    access: foo
    refresh: bar**
  capacity: 27

BUT: now I get a 
main ] INFO 2024/06/17 17:26:14 evcc 0.127.2
[main ] INFO 2024/06/17 17:26:14 using config file: /etc/evcc.yaml
[db ] INFO 2024/06/17 17:26:14 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/06/17 17:26:14 listening at :7070
[mercedes] DEBUG 2024/06/17 17:26:15 identity.NewIdentity - token not valid - Add expiry
[mercedes] DEBUG 2024/06/17 17:26:15 identity.NewIdentity - token not valid - database token check started
[mercedes] DEBUG 2024/06/17 17:26:16 identity.NewIdentity - token not valid - refreshToken started
**[main ] ERROR 2024/06/17 17:26:16 creating vehicle GLE failed: cannot create vehicle type 'mercedes': token expired**
[lp-1 ] DEBUG 2024/06/17 17:26:16 charge total import: 2793.000kWh

closed 2024-06-17 by HiroP77

ReneNulschDE, 2024-06-17:
Database doesn't need to be vanilla. There are commands to remove invalid tokens from the db. Search for evcc settings get or evcc settings set...