evcc-io/evcc #14440: Custom vehicle support for charge limit and setting max SOC

infinus, 2024-06-17:
Currently using MQTT integration for custom vehicle I can see SOC, Range and status but it would be nice to be able to support seeing charge limit from vehicle. Also if is possible to send charge limit via custom vehicle? I have wake, and set charging amps.

This is in relation to Tesla control via BLE.

StefanSchoof, 2024-06-17:
In https://github.com/evcc-io/evcc/discussions/14252#discussioncomment-9735057 it says that is possible.

infinus, 2024-06-17:
`embed         `mapstructure:",squash"`
		Soc           provider.Config
		LimitSoc      *provider.Config
		Status        *provider.Config
		Range         *provider.Config
		Odometer      *provider.Config
		Climater      *provider.Config
		MaxCurrent    *provider.Config
		GetMaxCurrent *provider.Config
		FinishTime    *provider.Config
		Wakeup        *provider.Config
		ChargeEnable  *provider.Config`

@StefanSchoof  thank you for finding that. I see limitSOC is this to pull limit from MQTT or is this for sending the command ? For example MaxCurrent is used to send the command and there is GetMaxCurrent which can retrieve. Limit SOC should have a get SOC limit so we can both see current limit and send new one.

andig, 2024-06-18:
Sending limits is not supported.

closed 2024-06-18 by andig

infinus, 2024-06-18:
That is why I submitted this as a feature request..... otherwise what is the point of the slider in the app if you cannot set SOC :) Unless you meant that it's not worth pursuing.

andig, 2024-06-18:
We wont add setting limit soc, sorry. We have too many hard-to-maintain vehicle apis already.