evcc-io/evcc #14441: EVCC hängt, mehrmals am Tag restart nötig // Huawei Luna2000

nauti23, 2024-06-17:
### Describe the bug

Mehrmals am Tag kann es passieren, dass gefühlt random EVCC hängt. Das bemerkt man nicht zwingend, im Zweifel saugt dann das Auto den Haus-Akku doch leer oder es beginnt garnicht erst mit dem Laden.

Eine Zeit lang hat es geholfen durch einen cron.job evcc morgen um abends jeweils neu zu starten. aber nun passieren die hänger seit einigen versionen auch einfach so übern tag.

### Steps to reproduce

1. EVCC starten
2. laufen lassen (siehe log z.b. gestern abstecken des fahrzeugs um 8:48)
3. irgendwann feststellen es hängt (nach rückkehr Jun 16 13:12:44 -> hier dann neustart manuell ausgelöst da die werte in der GUI hingen...
...
Im Log sehe ich nichts, alles scheint zu laufen, dann hägnt es und im log sehe ich dann nur die neustarts die ich manuell über die gui auslöse.
das heißt die GUI lädt zeigt aber werte aus der vergangenheit und updated nicht mehr.
wahrscheinlich weiß ich qauch einfach nicht wo ich genau nach dem fehler suchen muss?...

ich hoffe ihr könnt mir hilfreiche tipps oder eine idee geben. gerne hole ich alle logs hervor, sagt mir wo ich suchen muss.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 60s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 2
  host: 192.100.178.70
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 2
  host: 192.100.178.70
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 2  
  host: 192.100.178.70
  port: 502  
  usage: battery
  modbus: tcpip
  name: battery3

chargers:
- type: template
  template: nrgkick-bluetooth
  mac: 00:1E:C0:3D:FC:15
  pin: ************
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: minpv

site:
  title: ***
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.28 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.07 # [currency]/kWh
```

### Log details

```text
Jun 16 08:22:10 evcc evcc[6647]: [lp-1  ] ERROR 2024/06/16 08:22:10 charge currents: Operation failed with ATT error>
Jun 16 08:24:00 evcc evcc[6647]: [lp-1  ] INFO 2024/06/16 08:24:00 stop charging <-
Jun 16 08:24:58 evcc evcc[6647]: [lp-1  ] INFO 2024/06/16 08:24:58 start charging ->
Jun 16 08:39:55 evcc evcc[6647]: [lp-1  ] ERROR 2024/06/16 08:39:55 charge power: Operation failed with ATT error: 0>
Jun 16 08:40:05 evcc evcc[6647]: [lp-1  ] ERROR 2024/06/16 08:40:05 charge currents: find device: discovery timeout >
Jun 16 08:40:20 evcc evcc[6647]: time="2024-06-16T08:40:20+02:00" level=warning msg="Recovering from panic: send on >
Jun 16 08:45:58 evcc evcc[6647]: [lp-1  ] INFO 2024/06/16 08:45:58 stop charging <-
Jun 16 08:48:54 evcc evcc[6647]: [lp-1  ] INFO 2024/06/16 08:48:54 car disconnected
Jun 16 08:48:54 evcc evcc[6647]: [lp-1  ] INFO 2024/06/16 08:48:54 vehicle updated: Xc60 -> unknown
Jun 16 13:12:44 evcc evcc[6647]: [main  ] INFO 2024/06/16 13:12:44 new version available: 0.127.1
Jun 16 16:09:55 evcc systemd[1]: Stopping evcc...
Jun 16 16:09:55 evcc systemd[1]: evcc.service: Succeeded.
Jun 16 16:09:55 evcc systemd[1]: Stopped evcc.
Jun 16 16:09:55 evcc systemd[1]: evcc.service: Consumed 2min 30.345s CPU time.
Jun 16 16:10:06 evcc systemd[1]: Started evcc.
Jun 16 16:10:07 evcc evcc[18419]: [main  ] INFO 2024/06/16 16:10:07 evcc 0.127.1
Jun 16 16:10:07 evcc evcc[18419]: [main  ] INFO 2024/06/16 16:10:07 using config file: /etc/evcc.yaml
Jun 16 16:10:08 evcc evcc[18419]: [db    ] INFO 2024/06/16 16:10:08 using sqlite database: /var/lib/evcc/evcc.db
Jun 16 16:10:08 evcc evcc[18419]: [main  ] INFO 2024/06/16 16:10:08 listening at :7070
Jun 16 16:10:09 evcc evcc[18419]: [lp-1  ] WARN 2024/06/16 16:10:09 deprecated: mincurrent setting is ignored, pleas>
Jun 16 16:10:09 evcc evcc[18419]: [lp-1  ] WARN 2024/06/16 16:10:09 deprecated: maxcurrent setting is ignored, pleas>
Jun 16 16:10:09 evcc evcc[18419]: [lp-1  ] WARN 2024/06/16 16:10:09 deprecated: phases setting is ignored, please re>
Jun 16 16:10:17 evcc evcc[18419]: [site  ] WARN 2024/06/16 16:10:17 battery configured but residualPower is missing >
Jun 16 16:10:17 evcc evcc[18419]: [site  ] INFO 2024/06/16 16:10:17 site config:
Jun 16 16:10:17 evcc evcc[18419]: [site  ] INFO 2024/06/16 16:10:17   meters:      grid ✓ pv ✓ battery ✓
Jun 16 16:10:17 evcc evcc[18419]: [site  ] INFO 2024/06/16 16:10:17     grid:      power ✓ energy ✓ currents ✓
Jun 16 16:10:17 evcc evcc[18419]: [site  ] INFO 2024/06/16 16:10:17     pv 1:      power ✓ energy ✓ currents ✗
Jun 16 16:10:17 evcc evcc[18419]: [site  ] INFO 2024/06/16 16:10:17     battery 1: power ✓ energy ✓ currents ✗ soc ✓>
Jun 16 16:10:17 evcc evcc[18419]: [site  ] INFO 2024/06/16 16:10:17   vehicles:
Jun 16 16:10:17 evcc evcc[18419]: [site  ] INFO 2024/06/16 16:10:17     vehicle 1: range ✓ finish ✓ status ✓ climate>
Jun 16 16:10:17 evcc evcc[18419]: [site  ] INFO 2024/06/16 16:10:17     vehicle 2: range ✓ finish ✗ status ✓ climate>
Jun 16 16:10:17 evcc evcc[18419]: [lp-1  ] INFO 2024/06/16 16:10:17 loadpoint 1:
Jun 16 16:10:17 evcc evcc[18419]: [lp-1  ] INFO 2024/06/16 16:10:17   mode:        minpv
Jun 16 16:10:17 evcc evcc[18419]: [lp-1  ] INFO 2024/06/16 16:10:17   charger:     power ✓ energy ✓ currents ✓ phase>
Jun 16 16:10:17 evcc evcc[18419]: [lp-1  ] INFO 2024/06/16 16:10:17   meters:      charge ✓
Jun 16 16:10:17 evcc evcc[18419]: [lp-1  ] INFO 2024/06/16 16:10:17     charge:    power ✓ energy ✓ currents ✓
Jun 16 16:10:19 evcc systemd[1]: Stopping evcc...
Jun 16 16:10:19 evcc systemd[1]: evcc.service: Succeeded.
Jun 16 16:10:19 evcc systemd[1]: Stopped evcc.
```

### What type of operating system are you running?

Linux

### Version

0.172.1

nauti23, 2024-06-17:
die warnungen mit "deprecated" was die config im yaml angeht sind noch aktuell von gestern, nur falls sich jemand wundert warum das ja garnicht in der yaml steht, habe ich heute raus. denke aber nicht, dass daher der error kommt, denn das hängen hatte ich auch swchon fast regelmäßig bevor das deprecated war.

andig, 2024-06-17:
Bitte upgrade, sollte gelöst sein.

closed 2024-06-17 by andig

nauti23, 2024-06-18:
Es hängt fast immer irgendwann... ich weiß nicht ob es die hängende Bluetoothverbindung oder der Wechselrichter ist der nachts keine Werte liefert...
das ist schon fast immer so. Soltle man mal ne refresh wert des bluetooth chargers hochschrauben? 
im grunde scheint es aber so, wenn der inverter in sleep geht wacht auch EVCC nicht mehr auf wenn er morgen aufwacht.
macht das sinn? :)