evcc-io/evcc #14443: Loadpoint: Loadpoint not scaling down

GrimmiMeloni, 2024-06-17:
### Describe the bug

This is a follow up to my comments:
- https://github.com/evcc-io/evcc/issues/12773#issuecomment-1987263008
- https://github.com/evcc-io/evcc/issues/14307#issuecomment-2172492843

It's a bit lengthy to explain, I'll do my best to be brief.

TL;DR - **_if PhaseGetter is not implemented_**, and charger and loadpoint do not agree on the amount of active phases, the loadpoint always wins (ignoring the physical reality).

This is due to the logic in pvScalePhases(). It calls ActivePhases() which gathers loadpoint, charger, vehicle and returns the minimum Phase count from all these. The problem is, that if any of the above items returns 1, then the loadpoint is not considered scalable.
This becomes problematic if reality (measured phases) and loadpoint are out of sync. If measured phases is correctly reporting 3 phases, but the loadpoint incorrectly claims to have just 1 phase active, the loadpoint will never scale down.

Suggested solutions:
- Do not reset measured phases and instead again similar to syncCharger force the loadpoint to follow
- Consider measuredPhases during Sync Charger as well (if PhaseGetter is not available)
    - Optionally maybe also introduce a sanity check on the charger, ensuring measuredPhases and PhaseGetter agree on the number of phases
- Alternatively PhaseGetter insetead of measuredPhases when determining the need for scaling

### Steps to reproduce

See above

### Configuration details

```yaml
N/A
```

### Log details

```text
N/A
```

### What type of operating system are you running?

Docker container

### Version

0.127.2

GrimmiMeloni, 2024-06-17:
@andig created new issue as suggested to discuss the problem with loadpoint not scaling.

andig, 2024-06-20:
Hi @GrimmiMeloni thanks for the issue. Great find! Here's what I understand:
- problem exists if charger fails to follow 1p switch
- ...and does not implement `api.PhaseGetter` for correction
- ...even though phase currents are available to tell the true story
- it doesn't not affect 1p->3p switches (since it might be a 1p vehicle)

This leads me to the conclusion that:
- we really don't need `api.PhaseGetter` when `api.PhaseCurrents` is already implemented except for the case of validating 1p->3p switches
- ...if we determined active phases the right way (i.e. using measured phase currents)

Agreed sofar? Then we should check:
- if `ActivePhases` is the right place to make the change (and how to do it elegantly)
- even if we did that, we'd still need `api.PhaseGetter` in presence of phase currents for handling stuck 1p->3p switches

Does that sound about right?

GrimmiMeloni, 2024-06-20:
Thanks for putting the time in to follow along my reasoning. 👍 
Yes, you got this right - sounds to me we are in sync.

If we have `api.PhaseGetter` available, using it to sanity check against the chargeCurrents would likely make sense. But I am not sure we can be absolutely sure either, if a diversion between the two is actually a problem. The 1p vehicle (or charger cable) can cause the same situation and it is not an error.

Let me put it this way. If we assume that Chargers implement things correctly than `api.PhaseGetter` is in my opinion the accurate way to determine the physical state of the charger. Using the `chargeCurrents` >= 1A logic could be the fallback for chargers without the `api.PhaseGetter`. 

If I get a vote, I would probably leave the logic in pvScalePhases() as is, and change `syncCharger()` to use measuredPhases as a fallback (if `PhaseGetter` is not available). That way we keep all logic related to "sync loadpoint and charger reality" in a single place. As we do it for `PhaseGetter` implementations already there, it seems reasonable to me to do it there also for the other chargers via the measured currents.

andig, 2024-06-29:
> If we have api.PhaseGetter available, using it to sanity check against the chargeCurrents would likely make sense. But I am not sure we can be absolutely sure either, if a diversion between the two is actually a problem.

Check, lets not do this. If we have the getter we should trust it.

> Using the chargeCurrents >= 1A logic could be the fallback for chargers without the api.PhaseGetter.

Check.

> If I get a vote, I would probably leave the logic in pvScalePhases() as is, and change syncCharger() to use measuredPhases as a fallback (if PhaseGetter is not available)

Check.

> sollten wir das Thema "Ströme" bei https://github.com/evcc-io/evcc/issues/14443 auch noch mit in die Betrachtung nehmen?

I'd also take look at that. Thanks for your input!

closed 2024-06-29 by andig