evcc-io/evcc #14457: Wenn sich der Status eines "custom"-vehicles ändert, wirkt sich das nicht aus

binderth, 2024-06-18:
### Describe the bug

Wenn sich der Status von custom-definierten Fahrzeugen ändert, hat das keine Änderungen für die evcc-Logik, welches Fahrzeug am Loadpoint aktuell lädt. 
Das ist wichtig, bei mehreren custom-definierten Fahrzeugen, dass sich deren aktueller IEC-Status auch auf evcc auswirkt.

### Steps to reproduce

1. Definiere zwei Custom-vehicles
2. Vehicle-Informationen werden z.B. per MQTT geliefert
3. erstes Fahrzeug am loadpoint abstecken
4. zweites Fahrzeug am loadpoint anstecken
5. erstes Fahrzeug bleibt aktiv am Loadpoint

In den Logs kann man erkennen, dass

* "ev6adesso" um 07:31:32 abgesteckt wurde
* der Loadpoint wurde um 07:37:32 auf B gesetzt
* im folgenden API-refresh waren die MQTT-messages noch ev6: A, ev6adesso: B
* um 07:45:00 kamen per MQTT-messages ev6:B, ev6adesso:A
* um 07:48:02 wurde mit dem Ablauf von `soc.poll.mode: always` und `soc.poll.intervall: 5m` erkannt, dass das Fahrzeug auf A gewechselt ist

* aber das Fahrzeug "ev6adesso" bleibt weiterhin aktiv, siehe auch `vehicle.soc: 85`, `vehicle.range: 374`, das sind nämlich die Werte von ev6adesso gewesen. ev6 hatte 34%.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http

  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. Host must be set to the docker host's name
  #host: evcc.local
  host: 0.0.0.0

  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

  # per default, evcc should be available at http://evcc.local:7070

interval: 30s # control cycle interval
log: debug

meters:
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
- type: custom
  name: PVErzeugung
  power:
    source: mqtt
    topic: devices/Kostal/totalPVPower
    timeout: 2m
#  energy:
#    source: mqtt
#    topic: devices/Kostal/totalHomeConsumptionPV
#    timeout: 2m
- type: custom
  name: Netzbezug
  power:
    source: mqtt
    topic: devices/Kostal/totalActivePower
    timeout: 2m
#  energy:
#    source: mqtt
#    topic: devices/Kostal/totalHomeConsumptionGrid
#    timeout: 2m
[... 573 lines cut ...]
[lp-1  ] DEBUG 2024/06/18 07:48:32 charge power: 0W
[lp-1  ] DEBUG 2024/06/18 07:48:32 charge currents: [0 0 0]A
[site  ] DEBUG 2024/06/18 07:48:32 pv power: 2307W
[site  ] DEBUG 2024/06/18 07:48:32 battery soc: 35%
[site  ] DEBUG 2024/06/18 07:48:32 battery power: -1872W
[site  ] DEBUG 2024/06/18 07:48:32 grid meter: 37W
[site  ] DEBUG 2024/06/18 07:48:32 battery has priority at soc 35% (< 85%)
[site  ] DEBUG 2024/06/18 07:48:32 site power: 637W
[lp-1  ] DEBUG 2024/06/18 07:48:32 charge voltages: [234 234 234]V
[lp-1  ] DEBUG 2024/06/18 07:48:32 detected connected phases: 3p
[lp-1  ] DEBUG 2024/06/18 07:48:32 charge total import: 6047.932kWh
[lp-1  ] DEBUG 2024/06/18 07:48:32 charger status: B
[lp-1  ] DEBUG 2024/06/18 07:48:32 next soc poll remaining time: 4m30s
[lp-1  ] DEBUG 2024/06/18 07:48:32 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/06/18 07:48:32 pv charge current: 0A = 0A + -0.923A (637W @ 3p)
[site  ] DEBUG 2024/06/18 07:49:02 ----
```

### What type of operating system are you running?

Docker container

### Version

v0.127.2

andig, 2024-06-19:
Bei diesem Status

```
[main  ] DEBUG 2024/06/18 07:37:32 vehicle status: A (Kia EV6)
[main  ] DEBUG 2024/06/18 07:37:32 vehicle status: B (Kia EV6 adesso)
```

kann nur Adesso erkannt werden.

binderth, 2024-06-19:
Das stimmt. Aber spätestens mit der neuen MQTT-Nachricht und allerspätestens mit dem nächsten SoC-Poll sollte das Fahrzeug aich ändern können

andig, 2024-06-19:
Die Erkennung passiert bei Verbindung. Danach änder sich das Fahrzeug ja nicht mehr.

andig, 2024-06-19:
Das wiederum hat nix mit einem Custom vehicle zu tun. Ich frag mich allerdings warum das sonst so wunderbar funktioniert?

binderth, 2024-06-19:
Bei einem custom vehicle läuft der API-Call asynchron zu evcc.
Bei einem evcc-vehicle habt ihr im Code sicher einen wait for answer eingebaut. Auch hier habt ihr keine Chance den IEC-Change parallel oder gar vor dem Wallbox change zu bekommen.

andig, 2024-06-20:
@binderth du könntest als workaround mal probieren, beim abstecken den alten Status des angeschlossenen Fahrzeugs zu nullen, also einfach auf A. Solange bis es einen neuen gibt. Das sollte als Workaround schon reichen.

binderth, 2024-06-26:
@andig Bin wieder ausm Kurzurlaub zurück. ja, mein Workaround ist, bei jedem Wechsel des Status eines angeschlossenen Fahrzeugs lasse ich per API einen refresh auf den loadpoint machen (PATCH /api/loadpoints/<id>/vehicle). 
Wenn es eine evcc-interne Lösung dafür gäbe, kommt das auch weniger versierten Nutzern zugute. 😊

andig, 2024-07-06:
> Wenn es eine evcc-interne Lösung dafür gäbe, kommt das auch weniger versierten Nutzern zugute.

Wie sollte die denn aussehen?

Hofyyy, 2024-07-06:
@binderth Ich bin auch gespannt wie das aussehen soll. Wir haben auch 2 Autos zwei Wallboxen und manchmal geht die Erkennung nicht optimal. Hatte aber noch keine Zeit zu forschen. Ich hänge mich hier mal mit rein.

binderth, 2024-07-06:
> Wie sollte die denn aussehen?

Wenn per MQTT oder API das aktuelle Fahrzeug nicht mehr den IEC-Lade-/Warte-Status hat, sollte eine erneute Abfrage laufen. 
Wenn "Gastfahrzeug" steht, sollte im eingestellten Rythmus geprüft werden, ob ein anderes Fahrzeug lädt.

andig, 2024-07-06:
> Wenn per MQTT oder API das aktuelle Fahrzeug nicht mehr den IEC-Lade-/Warte-Status hat, sollte eine erneute Abfrage laufen.

Was bedeutet das? Der Ladepunkt kennt kein Mqtt.

binderth, 2024-07-07:
Wenn der Ladepunkt agnostisch ist, wie erfolgt die Zuordnung eines Fahrzeugs zu diesem? Und wenn eben jenes Fahrzeug nun in "A" landet, warum sollte es weiter stur als ladendes Fahrzeug gelten?

(Wenn ihr kein Bock auf custom-vehicles habt, dann sagt das bitte, kommt mir echt mühsam vor)

andig, 2024-07-07:
Ich meine damit: der Ladepunkt weiss nichts von Mqtt oder custom. Für ihn ist ein Auto ein Auto. Du kannst also nicht sagen „bei Mqtt soll das passieren, bei normalen Autos jenes“. 

Die Frage war ernst gemeint: wie soll die Logik aussehen?

binderth, 2024-07-08:
Kenne jetzt weder Code, noch Architektur, aber irgendwo muss es ja eine Zuordnung von Fahrzeugen zu Ladepunkten geben.
Wenn sich nun ein Fahrzeug ändert, das an einem Ladepunkt "gemeldet" ist und nun "A" hat, kann doch ein erneuter Ladepunkt-Check gestartet werden? Ich mach das halt per API über mein smarthome (`PATCH /api/loadpoints//vehicle`), aber das kann ja nicht jeder.

andig, 2024-07-08:
Dann müssten wir anhaltend die Fahrzeug APIs pollen? Leider keine gute Idee :(

binderth, 2024-07-08:
Das ist doch nicht nötig, bzw ist das ja mit dem `poll.mode` bzw `poll.interval` für jeden steuerbar!
Wenns gepusht wird (durch MQTT oder per API), fragt ja auch keiner, woher das kommt!??

Ist nur ziemlich unlogisch, dass evcc den Fahrzeugstatus kennt - aber null drauf reagiert und stur bei einer veralteten Meinung bleibt.

andig, 2024-07-09:
Die Frage war: nach welchem Algorithmus soll das geschehen? Bisher gibts dazu keine Antwort die umsetzbar wäre. Der Algo muss von `custom` oder `mqtt` unabhängig sein da der Loadpoint davon nix weiss.

binderth, 2024-07-10:
Die Antwort ist: sobald ein Fahrzeug den Status ändert!?

VolkerK62, 2024-07-10:
Das kommt ja nicht von alleine. Dazu müsste dann der Status ständig abgefragt werden.

Hofyyy, 2024-07-10:
Ich fasse mal zusammen, ggf. prüfe ich auch mal später den Sourcecode.

Wenn sich der State im Auto ändert (wann auch immer), bekommen wir das mindestens per MQTT gepusht!?, wenn dann das Auto den Loadpoint kennt und gerade verbunden ist könnte man da eine fuktion aufrufen, die das gewünsche verhalten ausführt. Also reset des states etc. Dei Frage ist ob das zu 100% das gewünschte verhalten wäre.

Dann wäre es doch auch nicht schlimm, das erstmal für die APIs anzubieten die Push unterstützen oder nicht?
Beispiel bei Teslamate wäre es ja über StreamingAPI -> MQTT auch nice.

Die APIs die das nicht können, haben halt das "tolle" verhalten noch nicht.

andig, 2024-07-10:
Der Loadpoint ist führend. Devices können/sollen nichts am State des Loadpoints ändern. Schon gar keine generischen Devices (mqtt), denn das würde wieder neue Device-APIs bedeuten.

> Das kommt ja nicht von alleine. Dazu müsste dann der Status ständig abgefragt werden.

Full circle. So ist es und stand schon in https://github.com/evcc-io/evcc/issues/14457#issuecomment-2213362202. Die Soc-Abfragen haben ein eigenes Caching, alle anderen APIs gehen i.A. hart auf das Fahrzeug durch.

Die Frage nach dem Algorithmus kommt nicht von ungefähr ;)

Hofyyy, 2024-07-10:
Ok verstanden.

Aus objektorientierter Sicht ist das Auto ja mal ein Objekt mit einem State (Kabel dran) daher könnte das Auto dem bekannten Loadpoint, da es da ja dran hängt, auch dem Loadpoint sagen dürfen, wenn das Kabel aus ihm entfernt wurde. Da wird ja nicht beliebig im Loadpoint rumgefingert, sondern würde ggf. ein ganz definierte Public methode aufgerufen, oder in Go ein API? Also highlevel ist klar wie man es machen müsste / könnte :-)

andig, 2024-07-20:
Keine neue Erkenntnisse/ kein Algorithmus. Closing.

closed 2024-07-20 by andig