evcc-io/evcc #14458: Wrong current used since latest version

Kenterfie, 2024-06-18:
### Describe the bug

I normally use 6A as min and max value to charge the car (Skoda Citigo), which ends up in ~2.8kW via 2 phases, but for some reason this isn't working anymore since the latest changes properly. The car now use 7.4kW (which i can see in my Wallbox App), but evcc still shows me 2.8kW like it is configured in the configuration file (2 times 6A or 2.8kW). It worked fine in the previous version, but this can not be used because of the Skoda API changes in the last weeks.

### Steps to reproduce

Carport settings are
3 phases
6A max current
6A min current

Car is configured to have 2p conntected

Using PV mode

Using latest myskoda api to get the car information

### Configuration details

```yaml
chargers:
  - name: wallbox
    type: template
    template: pulsarplus
    stationid: xxx
    timeout: 5m # optional

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: car1
    type: template
    title: citigo
    template: skoda
    user: xxx
    password: xxx
    capacity: 36.8
    phases: 2
```

### Log details

```text
[main  ] INFO 2024/06/18 13:38:24 evcc 0.127.2
[main  ] INFO 2024/06/18 13:38:24 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2024/06/18 13:38:24 connecting evcc-2042871804 at tcp://mosquitto:1883
[mqtt  ] DEBUG 2024/06/18 13:38:24 tcp://mosquitto:1883 connected
[main  ] INFO 2024/06/18 13:38:24 listening at :7070
[sn435152-1] DEBUG 2024/06/18 13:38:24 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/06/18 13:39:01 charge point connected: sn435152
[ocpp  ] TRACE 2024/06/18 13:39:01 received JSON message from sn435152: [2,"2146372956","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 0,"errorCode": "NoError","status": "Available","timestamp": "2024-06-18T11:39:01Z"}]
[ocpp  ] TRACE 2024/06/18 13:39:01 received JSON message from sn435152: [2,"516009609","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Finishing","timestamp": "2024-06-18T11:39:01Z"}]
[ocpp  ] TRACE 2024/06/18 13:39:01 sent JSON message to sn435152: [3,"516009609",{}]
[ocpp  ] TRACE 2024/06/18 13:39:02 sent JSON message to sn435152: [2,"4071777330","GetConfiguration",{}]
[ocpp  ] TRACE 2024/06/18 13:39:03 received JSON message from sn435152: [3,"4071777330",{"unknownKey": [],"configurationKey": [{"readonly": false,"key": "AllowOfflineTxForUnknownId","value": "true"},{"readonly": false,"key": "AuthorizationCacheEnabled","value": "true"},{"readonly": false,"key": "AuthorizeRemoteTxRequests","value": "false"},{"readonly": false,"key": "BlinkRepeat","value": "10"},{"readonly": false,"key": "ClockAlignedDataInterval","value": "0"},{"readonly": false,"key": "ConnectionTimeOut","value": "120"},{"readonly": false,"key": "ConnectorPhaseRotation","value": "1.Unknown"},{"readonly": true,"key": "ConnectorPhaseRotationMaxLength","value": "1"},{"readonly": true,"key": "GetConfigurationMaxKeys","value": "50"},{"readonly": false,"key": "HeartbeatInterval","value": "60"},{"readonly": false,"key": "LightIntensity","value": "100"},{"readonly": false,"key": "LocalAuthorizeOffline","value": "true"},{"readonly": false,"key": "LocalPreAuthorize","value": "true"},{"readonly": false,"key": "MaxEnergyOnInvalidId","value": "5000"},{"readonly": false,"key": "MeterValuesAlignedData","value": "Energy.Active.Import.Register"},{"readonly": true,"key": "MeterValuesAlignedDataMaxLength","value": "10"},{"readonly": false,"key": "MeterValuesSampledData","value": "Energy.Active.Import.Register"},{"readonly": true,"key": "MeterValuesSampledDataMaxLength","value": "10"},{"readonly": false,"key": "MeterValueSampleInterval","value": "300"},{"readonly": false,"key": "MinimumStatusDuration","value": "1"},{"readonly": true,"key": "NumberOfConnectors","value": "1"},{"readonly": false,"key": "ResetRetries","value": "0"},{"readonly": false,"key": "StopTransactionOnEVSideDisconnect","value": "true"},{"readonly": false,"key": "StopTransactionOnInvalidId","value": "true"},{"readonly": false,"key": "StopTxnAlignedData","value": ""},{"readonly": true,"key": "StopTxnAlignedDataMaxLength","value": "0"},{"readonly": false,"key": "StopTxnSampledData","value": ""},{"readonly": true,"key": "StopTxnSampledDataMaxLength","value": "0"},{"readonly": true,"key": "SupportedFileTransferProtocols","value": ""},{"readonly": true,"key": "SupportedFeatureProfiles","value": "Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"},{"readonly": true,"key": "SupportedFeatureProfilesMaxLength","value": "6"},{"readonly": false,"key": "TransactionMessageAttempts","value": "3"},{"readonly": false,"key": "TransactionMessageRetryInterval","value": "60"},{"readonly": false,"key": "UnlockConnectorOnEVSideDisconnect","value": "true"},{"readonly": false,"key": "WebSocketPingInterval","value": "50"},{"readonly": false,"key": "LocalAuthListEnabled","value": "true"},{"readonly": true,"key": "LocalAuthListMaxLength","value": "5000000"},{"readonly": true,"key": "SendLocalListMaxLength","value": "450"},{"readonly": true,"key": "ReserveConnectorZeroSupported","value": "false"},{"readonly": true,"key": "ChargeProfileMaxStackLevel","value": "3"},{"readonly": true,"key": "ChargingScheduleAllowedChargingRateUnit","value": "Current"},{"readonly": true,"key": "ChargingScheduleMaxPeriods","value": "100"},{"readonly": true,"key": "ConnectorSwitch3to1PhaseSupported","value": "false"},{"readonly": true,"key": "MaxChargingProfilesInstalled","value": "10"},{"readonly": false,"key": "chargingALimitConn1","value": "16"},{"readonly": false,"key": "AuthEnabled","value": "true"},{"readonly": false,"key": "AuthEnabledOffline","value": "true"},{"readonly": false,"key": "AuthDisabledIdTag","value": "NoAuthorization"},{"readonly": false,"key": "minSoC","value": "9"},{"readonly": false,"key": "maxSoC","value": "97"},{"readonly": true,"key": "SSID","value": ""},{"readonly": true,"key": "WiFiSignalStrength","value": "49"},{"readonly": true,"key": "SerialNumber","value": "435152"},{"readonly": true,"key": "PUK","value": "45230452"},{"readonly": false,"key": "WebSocketUrl","value": ""},{"readonly": false,"key": "WebSocketUser","value": "sn435152"},{"readonly": false,"key": "WebSocketPassword","value": ""},{"readonly": false,"key": "MeterValuesSkipWhenOffline","value": "false"} ]}]
[ocpp  ] TRACE 2024/06/18 13:39:03 sent JSON message to sn435152: [2,"1163418459","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/06/18 13:39:03 received JSON message from sn435152: [3,"1163418459",{"status": "Accepted"}]
[ocpp  ] TRACE 2024/06/18 13:39:03 TriggerMessage MeterValues for sn435152: Accepted
[ocpp  ] TRACE 2024/06/18 13:39:03 received JSON message from sn435152: [2,"1017062965","MeterValues",{"meterValue": [{"timestamp": "2024-06-18T11:39:02Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "426927.1"} ]} ],"connectorId": 1}]
[ocpp  ] TRACE 2024/06/18 13:39:03 sent JSON message to sn435152: [3,"1017062965",{}]
[site  ] INFO 2024/06/18 13:39:06 site config:
[site  ] INFO 2024/06/18 13:39:06   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2024/06/18 13:39:06     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/06/18 13:39:06     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/06/18 13:39:06   vehicles:
[site  ] INFO 2024/06/18 13:39:06     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2024/06/18 13:39:06 loadpoint 1:
[lp-1  ] INFO 2024/06/18 13:39:06   mode:        pv
[lp-1  ] INFO 2024/06/18 13:39:06   charger:     power ✗ energy ✓ currents ✗ phases ✗ wakeup ✗
[... 5 lines cut ...]
[lp-1  ] DEBUG 2024/06/18 13:39:06 !! active phases: 2p = min(0p measured 2p vehicle 3p physical 0p charger)
[site  ] DEBUG 2024/06/18 13:39:06 ----
[lp-1  ] DEBUG 2024/06/18 13:39:06 charge power: 0W
[site  ] DEBUG 2024/06/18 13:39:08 pv power: 8735W
[site  ] DEBUG 2024/06/18 13:39:08 grid meter: -8253W
[site  ] DEBUG 2024/06/18 13:39:08 site power: -8253W
[lp-1  ] DEBUG 2024/06/18 13:39:08 vehicle odometer: 35025km
[lp-1  ] DEBUG 2024/06/18 13:39:08 !! active phases: 2p = min(0p measured 2p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/06/18 13:39:08 charger status: B
[lp-1  ] INFO 2024/06/18 13:39:08 car connected
[lp-1  ] DEBUG 2024/06/18 13:39:08 pv timer elapse
[lp-1  ] DEBUG 2024/06/18 13:39:08 pv timer inactive
[lp-1  ] DEBUG 2024/06/18 13:39:08 !! active phases: 2p = min(0p measured 2p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/06/18 13:39:08 vehicle status: B
[lp-1  ] DEBUG 2024/06/18 13:39:08 vehicle soc: 51%
[lp-1  ] DEBUG 2024/06/18 13:39:08 vehicle range: 99km
```

### What type of operating system are you running?

Docker container

### Version

0.127.2

Kenterfie, 2024-06-18:
To have a intermediate fix for that i switched the current limit also in the Wallbox app to 6A, but this currently limits the use of evcc, because it then not respecting anymore the values i have set in evcc if i want to use more than 6A for some reason.

Micha70, 2024-06-18:
I can confirm that problem. Have the same issue. In evcc see different charge power like in the wallbox app. Even from meters I see is charging almost with maximum power. Also using version  0.127.2

andig, 2024-06-19:
Does downgrading to the old version work?

andig, 2024-06-19:
Please open ticket with Pulsar. Looks a lot like a firmware bug. Closing here until new information available.

closed 2024-06-19 by andig

Kenterfie, 2024-06-19:
Where you can see a firmware bug of the wallbox? A downgrade of evcc was not possible because of the changes to the Skoda API. The wallbox's firmware has not been changed for months and has only been used directly via ocpp (no cloud involved/no auto-update active). So only part which was changed/updated was evcc. @Micha70 is the same for you?

andig, 2024-06-19:
Downgrade for testing is always possible. Pulsar rejects the set current. There are already other discussions on the subject where the problem is visible.

Micha70, 2024-06-19:
I have my setup just working since last week. And in the beginning was working I mean. During sun was never charging with max power. I saw this just yesterday after I updated to last version of EVCC. Will downgrade today to last of 0.126.x for testing.

Kenterfie, 2024-06-20:
Ok, was an issue with the Wallbox itself. Restarted the wallbox multiple times and now it is working as expected again. Lets see how long.

andig, 2024-06-20:
Was there a firmware update?

Kenterfie, 2024-06-20:
No firmware update since ~3 months. I am not sure how much Wallbox is doing in the background without a real firmware change, because the app was changed/updated multiple times in same period. Possibly some kind of deadlock in the ocpp part of the firmware. I have not looked deeper into the problem, but when this occurs more frequently i will possibly spend some time to debug it by my own. I have not looked into the code yet, but has evcc possible a message which would indicate that the ocpp communication is behaving incorrectly? I thought i have everything already enabled to log all kind of ocpp events, but possibly i overlooked something.

andig, 2024-06-20:
You‘ll see Rejected in the ocpp trace.

Micha70, 2024-06-24:
Is working now also on my side. After a reset of the wallbox. Thanks for the hint... 
Is possible there is a conflict if I have the charger also connected to home assistant? I have removed now (made the device inactive)...
And another question, is working also automatic phase switching? On my side is charged always with three phases....

andig, 2024-06-24:
> Is possible there is a conflict if I have the charger also connected to home assistant? I

I do you think the charger would react to different controllers giving different commands and replying to questions with different answers? Apart from those concerns, specifically an OCPP charge point can only ever be connected to a single central system at a time.

Micha70, 2024-06-24:
This I assumed. But even the charger was connected to evcc via OCPP I could manipulate the settings from HA. This concerned me a bit and I disabled the device in HA.