evcc-io/evcc #14460: Vehicle Mercedes does not work with type: template

ReneNulschDE, 2024-06-18:
### Describe the bug

When you configure a vehicle with type: template, template: mercedes, the vehicle does not work. The config value `region` is not transfered from the template to the mercedes component.

### Steps to reproduce

1. Create the following config entry in the evcc.yaml
```
vehicles:
  - name: name-car-1
    type: template
    template: mercedes
    title: title-car-1
    user: test123
    region: EMEA
    accesstoken: at
    refreshtoken: rt
    vin: W1
```
2. Set a break point in the mercedes.go line 39
3. Start the debugger and check that `cc.Region` is empty
4. Change the config to:
```
  - name: name-car-1
    type: mercedes
    title: title-car-1
    user: test@intranet
    region: EMEA
    tokens:
     access: at
     refresh: rt
    vin: W1
```
5. Start the debugger and check that `cc.Region` has the value `EMEA`

Looks like it gets lost in templates.template.RenderTemplate (tmpl.execute, line 347)

### Configuration details

```yaml
See steps to repro
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Version

current master branch

VolkerK62, 2024-06-18:
it is missing within the render part of the template

andig, 2024-06-19:
Region fehlt in https://github.com/evcc-io/evcc/blob/master/templates/definition/vehicle/mercedes.yaml#L39. Hab ich ggf bei der Vereinfachung übersehen.

PR #14465 (2024-06-19): mercedes: add missing region to template
fix #14460
closed 2024-06-19 by andig