evcc-io/evcc #14471: Grünstromindex

NiMa1969, 2024-06-19:
### Describe the bug

Wird die CO2 Vorhersage gem. Doku konfiguriert kommt es zum Fehler beim Start von evcc

https://docs.evcc.io/docs/devices/tariffs#gr%C3%BCnstromindex

co2:
  type: grünstromindex
  zip: meine PLZ # PLZ mit führender Null mit in "" setzen

Sonderzeichen "ü" sind nicht zulässig und "ue" funktioniert nicht mehr seit dem letzten Update

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.28 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

  co2:
    type: gruenstromindex
    zip: 79541 # For post codes with leading zeros, enclose in ""
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.127.2

naltatis, 2024-06-19:
> Sonderzeichen "ü" sind nicht zulässig und "ue" funktioniert nicht mehr seit dem letzten Update
 
Wieso sind Sonderzeichen nicht zulässig?

Folgender Code funktioniert bei mir:

```
tariffs:
  co2:
    type: template
    template: grünstromindex
    zip: 12349
```

naltatis, 2024-06-19:
Um das nachvollziehen zu können, brauchen wir ne genaue Fehlermeldung.

andig, 2024-06-19:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-06-19 by andig

Sammy521, 2024-10-15:
Hallo zusammen,

bei mir hat Grünstromindex gemäß Anleitung bislang funktioniert.

Seit gestern (?) erscheint ein rotes Ausrufezeichen und die Fehlermeldung: 
"planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'grünstromindex': EOF"

Geändert habe ich nichts. Gibt es eine Änderung so dass ich deshalb die YAML anpassen muss?
Oder ist das eine bekannte vorübergehende Störung?

Beste Grüße
Sammy