evcc-io/evcc #14474: Startup with Offline Wallbox(es) in EVCC

pjohde, 2024-06-19:
**Is your feature request related to a problem? Please describe.**
evcc fails to start if any of the connected charge points (Wallboxes) are offline. Currently, evcc waits for all Wallboxes to respond during the startup process. If one Wallbox is offline or unresponsive, evcc does not start and logs a fatal error message, then attempts to restart after a set period, but remains down.

**Describe the solution you'd like**
I would like evccto have the capability to start even if some of the Wallboxes are offline. Specifically, evcc should start without the offline Wallbox(es) and continue operating with the available Wallboxes. This would ensure that the charging system remains functional even if one or more Wallboxes are temporarily offline. A more resilient startup behavior would improve the reliability and user experience of  evcc.

**Describe alternatives you've considered**
An alternative solution might be adding a feature where evccretries to connect to the offline Wallbox(es) in the background while remaining operational with the available Wallboxes.

Additional context
In the current scenario, the evcc logs the following fatal error when a Wallbox is offline:
`[main  ] FATAL 2024/06/19 11:30:35 cannot create charger 'EV_Box_Garage': cannot create charger type 'template': cannot create charger type 'ocpp': timeout [main  ] FATAL 2024/06/19 11:30:35 will attempt restart in: 5m0s`

andig, 2024-06-19:
On the otherhand side doing so would mean that the fauly box stays offline indefinitely. Sofar we have always opted for the seemingly safer path and try to restart instead. We're not planning to change this anytime soon.

pjohde, 2024-06-20:
Wouldn't it be an option for evcc to start as soon as at least one wallbox is on? And for the other wallboxes, it could be monitored in the background to see if they respond.
This way, evcc would be operational, and if an inactive wallbox goes online, it would also change to active status in evcc.

andig, 2024-06-20:
Imagine 2 boxes. The car is connected to the flaky one. Restarting often can fix the problem. Now you don't restart and instead are happy with the box without vehicle. The car will not get charged. Does that sound good?

Imho the real question here is: why would the box not be available or become available when retrying? What is the actual use case? My 2 boxes have been running for 5 years without any single fault...

naltatis, 2024-06-20:
I've created a more general issue describing the desired behavior. I'll close this one as duplicate the the new https://github.com/evcc-io/evcc/issues/14496

closed 2024-06-20 by naltatis

asdfklgash, 2025-04-28:
> Imho the real question here is: why would the box not be available or become available when retrying? What is the actual use case? My 2 boxes have been running for 5 years without any single fault...

My ABL eMH1 responds with "i/o timeout" if no car is connected ... as soon as the car gets connected the wallbox is responding.