evcc-io/evcc #14475: Zoe wird nicht automatisch ausgewählt

kolossboss, 2024-06-19:
### Describe the bug

Wenn ich den Zoe anschließe, dann wird dieser Teilweise nicht automatisch als Zoe erkannt. 
In EVCC wird dann Gastfahrzeug angezeigt und so ein Ladekringel.
Ich kann dann einfach auf die Auto Auswahl klicken und dann ZOE auswählen, und dann startet der Ladevorgang ganz normal.
Das Problem ist nur sporadisch und erst seit ein paar Wochen auffällig. Ich kann daher nicht sagen, ob es durch irgendein Update ausgelöst wurde

### Steps to reproduce

1. Fahrzeug anschließen
2. EVCC zeigt Gast an
3. Anklicken und ZOE auswählen
...

### Configuration details

```yaml
network:

  schema: http

  host: xxxxx

  port: 7070

interval: 15s # control cycle interval

log: warn
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

#plant: xxxxxx

sponsortoken: xxxxxx

meters:
  - name: my_grid
    type: custom
#    usage: grid
    power:
      source: mqtt
      timeout: 4m # don't accept values older than timeout
      scale: -1
      topic: homeassistant/sensor/goodwe_pgrid/state
  #  - name: my_grid
  #    type: template
  #    template: tibber-pulse
  #    usage: grid
  #    token: xxxxxxx
  - name: my_pv1
    type: custom
    power:
      source: mqtt
      timeout: 3m
      topic: homeassistant/sensor/goodwe_ppv/state
  - name: my_pv2
    type: custom
    power:
      source: mqtt
      timeout: 60m
      scale: -1
      topic: homeassistant/sensor/pv_gartenhaus_power/state
  - name: my_battery
    type: custom
#    usage: battery
    power:
[... 251 lines cut ...]
[lp-1 ] DEBUG 2024/06/19 15:56:38 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/06/19 15:56:38 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/06/19 15:56:38 pv charge current: 24.4A = 5.4A + 19A (-4373W @ 1p)
[lp-1 ] DEBUG 2024/06/19 15:56:39 max charge current: 24A
[site ] DEBUG 2024/06/19 15:56:53 ----
[lp-1 ] DEBUG 2024/06/19 15:56:54 charge power: 3750W
[lp-1 ] DEBUG 2024/06/19 15:56:54 charge currents: [15.6 0 0]A
[site ] DEBUG 2024/06/19 15:56:54 pv power: 5490W
[site ] DEBUG 2024/06/19 15:56:54 battery soc: 100%
[site ] DEBUG 2024/06/19 15:56:54 battery power: 11W
[site ] DEBUG 2024/06/19 15:56:54 grid meter: -740W
[site ] DEBUG 2024/06/19 15:56:54 site power: -629W
[lp-1 ] DEBUG 2024/06/19 15:56:54 detected active phases: 1p
[lp-1 ] DEBUG 2024/06/19 15:56:54 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/06/19 15:56:54 charge total import: 6046.261kWh
[lp-1 ] DEBUG 2024/06/19 15:56:54 charger status: C
```

### What type of operating system are you running?

Docker container

### Version

v0.127.2

kolossboss, 2024-06-19:
@andig 

klär mich bitte auf. was bedeutet: `vehicle status: A (Zoe)`

Hier wird das Auto verbunden
`[lp-1 ] INFO 2024/06/19 15:53:53 car connected`

Hier klicke ich manuell auf ZOE in der GUI und das laden startet sofort!
`[lp-1 ] INFO 2024/06/19 15:56:17 vehicle updated: unknown -> Zoe`

StefanSchoof, 2024-06-19:
Der Zoe liefert über die API das er gerade nicht verbunden ist und daher geht evcc davon aus, dass er gerade nicht an dieser Wallbox angeschlossen ist.

Wenn du nur ein Auto hast hilft dir vielleicht https://docs.evcc.io/docs/reference/configuration/loadpoints#vehicle