evcc-io/evcc #14486: EVCC Konfiguration zerschossen

tha0o0, 2024-06-20:
### Describe the bug

Habe heute einmal in den Editor reingeklickt nach dem Update auf die neuste Version und dann auf restart gedrückt, seitdem ist meine ganze EVCC config weg und es wird läuft nichts mehr. meine EVCC.YAML ist noch im etc ordner existent aber das System das läuft wenn ich evcc starte ist komplett leer ??

Kann mir da jemand helfen wie ich das zurück bekomme? alles nue einrichten ist echt doof zumal meine Log Daten von 3 Monaten ebenfalls weg sind ...

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
## open evcc at http://192.168.178.101:7070
network:
    schema: http
    host: 192.168.178.101 # .local suffix announces the hostname on MDNS
    port: 7070

log: info
levels:
    site: debug
    lp-1: debug
    lp-2: debug
    cache: error
    db: error
    ev4: trace

# unique installation id
plant: ccb311d9af63a78360482b73614def1564d94acf7816dc870e9f4585dd7ac0ed

interval: 2s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE4MDE0MTEyMDAsImlhdCI6MTcwNjgwMzIwMCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6InRoYTBvMCIsInNyYyI6ImdoIn0.GIRGPaQnH6Fxv9TVI6JiGNH4Ny-YjrBBzGvFMaVCF4U

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
  - type: template
    name: my_grid
    template: shelly-pro-3em
    usage: grid
    host: 192.168.178.68
  - type: template
    name: my_pv1
    template: solaredge-inverter
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.178.35
    port: 1502
  - type: template
    name: my_pv2
    template: fronius-gen24
    usage: pv
    host: 192.168.178.37
    port: 502
  - type: template
    name: my_pv3
    template: hoymiles-opendtu
    usage: pv
    host: 192.168.178.119

chargers:
  - type: template
    template: go-e-v3
    host: 192.168.178.96
    name: wallbox5
  - type: template
    template: shelly
    host: 192.168.178.102
    name: Jeisha
    standbypower: 30

#vehicles:
#    type: template
#    template: hyundai
#    title: IONIQ 6
#    icon: car
#    user: k-klie@gmx.net
#    password: Fr33ky/2510522
#    vin: KMHM341B0PA046640X
#    capacity: 53

loadpoints:
  - title: Carport
    charger: wallbox5
    mode: pv
  - title: Jeisha
    charger: Jeisha
    mode: pv
    phases: 1
#    mincurrent: 6
#    maxcurrent: 16

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: awattar
    region: de # optional
    charges: 0 # optional, additional charges per kWh
    tax: 0.3 # optional, additional tax (0.1 for 10%)
  co2:
    type: grünstromindex
    zip: 21683

  feedin: 0.0755 # rate for feeding excess (pv) energy to the grid
    type: fixed #
    price: 0.09 # [currency]/kWh

site:
  title: Klie Home
  meters:
    grid: my_grid
    pv:
      - my_pv1
      - my_pv2
```

### Log details

```text
dmin@raspi1:~ $ sudo systemctl start evcc
admin@raspi1:~ $ sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; preset: enabled)
     Active: active (running) since Thu 2024-06-20 10:34:15 CEST; 3min 44s ago
   Main PID: 65136 (evcc)
      Tasks: 9 (limit: 9247)
        CPU: 391ms
     CGroup: /system.slice/evcc.service
             └─65136 /usr/bin/evcc

Jun 20 10:34:15 raspi1 systemd[1]: Started evcc.service - evcc.
Jun 20 10:34:16 raspi1 evcc[65136]: [main  ] INFO 2024/06/20 10:34:16 evcc 0.127.2
Jun 20 10:34:16 raspi1 evcc[65136]: [main  ] INFO 2024/06/20 10:34:16 using config file: /etc/evcc.yaml
Jun 20 10:34:16 raspi1 evcc[65136]: [main  ] INFO 2024/06/20 10:34:16 listening at :7070
Jun 20 10:34:16 raspi1 evcc[65136]: [main  ] FATAL 2024/06/20 10:34:16 While parsing config: yaml: line 86:>
Jun 20 10:34:16 raspi1 evcc[65136]: [main  ] FATAL 2024/06/20 10:34:16 will attempt restart in: 15m0s
lines 1-15/15 (END)
```

### What type of operating system are you running?

Linux

### Version

0.127.2

tha0o0, 2024-06-20:
Ich bin echt verzweifelt neu installation hilft nicht meine YAML Datei ist ja noch vorhanden nur das EVCC die Daten garnicht mehr nutzt :(( bitte ich bauche echt hilfe :(

VolkerK62, 2024-06-20:
Was meinst du mit `Habe heute einmal in den Editor reingeklickt`. Welchen Editor?

Dann wird ein Fehler angezeigt (leider unvollständig)
`Jun 20 10:34:16 raspi1 evcc[65136]: [main  ] FATAL 2024/06/20 10:34:16 While parsing config: yaml: line 86:>`

Prüf mal die yaml auf Syntaxfehler 
https://docs.evcc.io/docs/faq#syntax-fehler-in-der-evccyaml-finden

tha0o0, 2024-06-20:
Moin Moin danke für deine Antwort, vielleicht war ich etwas unpräzise.
Mein System lief ja mit besagter YAML einwandfrei seit 3 Monaten. Auch habe ich nichts angepasst. EVCC startet auch nur das nichts mehr von meiner konfig angezeigt wird. Es ist kein Grid-meter konfiguriert kein PV-wechselrichter loadpoint oder irgendwas aber es die siete baut sich ganz normal auf wenn ich meine locale adresse aufrufe

SolarPower2024, 2024-06-20:
Unabhängig von deinem Problem. Lösche lieber mal deinen Sponsortoken aus deinem Beitrag.

VolkerK62, 2024-06-20:
Zeig mal bitte mit einem Screenshot, was du meinst mit `nichts mehr von meiner konfig angezeigt wird`.

tha0o0, 2024-06-20:
[image]
schaut so aus ich bin ratlos

VolkerK62, 2024-06-20:
wenn du jetzt oben rechts auf das Menüzeichen mit dem roten Punkt klickst, was wird da angezeigt?

VolkerK62, 2024-06-20:
Das könnte das Problem sein.
Keine Ahnung, was die Zahlen sollen, aber setz mal ein # davor
`  feedin: 0.0755 # rate for feeding excess (pv) energy to the grid`

Soll dann so aussehen
` feedin: # 0.0755 ....`

tha0o0, 2024-06-20:
[image]
[image]
gerne kommt sowas dann

VolkerK62, 2024-06-20:
Siehe oben, Fehler in `tariffs` bei `feedin`

tha0o0, 2024-06-20:
den habe ich soeben entfernt wie von dir gefordert und gerade noch eine verrückte Zeile bei tarifs: korrigiert welch wunder ich habe meine Daten zurück
[image]

Aber ich kann mir trotzdem nicht erklären warum ich ständig aufgefordert werde ein passwort zu setzen bzw zurück zusetzen ?

tha0o0, 2024-06-20:
ich danke Vielmals für die bisherige Hilfe

VolkerK62, 2024-06-20:
Demnächst findet die Konfiguration nicht mehr per yaml Datei, sondern im WebUI statt.
Deshalb ist dieser Bereich mit Passwort geschützt.

By the way:
>Mein System lief ja mit besagter YAML einwandfrei seit 3 Monaten. Auch habe ich nichts angepasst. 

Mit dem Fehler kann es auch vorher nicht funktioniert haben.

VolkerK62, 2024-06-20:
Dann mach das Issue bitte zu.

tha0o0, 2024-06-20:
ok merkwürdig vielleicht Geistige verwirrung vielen Dank

closed 2024-06-20 by tha0o0