evcc-io/evcc #14496: Startup even if not all wallboxes, meters, integrations are reachable

naltatis, 2024-06-20:
Currently, evcc requires all configured devices to be reachable on startup. When this is not the case (e.g. a meter is not responding) evcc will run in a failsafe mode (control loop and many UI features disabled) and automatically restarts every 15min for retry until everything is reachable.

In some scenarios (unreliable hardware, broken devices), this behavior is problematic and requires temporary reconfiguration of evcc (e.g. disable a device) to get back into a functioning state.

To change this, we have to rework how initialization is done and may need to introduce concepts like device health status, per device retry mechanisms and visualize the affected parts correctly in the UI (e.g. disable loadpoint controls because a charger is not responding for a longer time).
 
\\cc @andig

StefanSchoof, 2024-06-21:
For inverter only (without an battery) there is the extra problem, that many of them go into a power saving mode after sunset and preventing evcc from restarting in the night, see #10972.

Sumpfdotter, 2024-07-08:
Thanks for that description. I add this comment so that this issue can be found by users having that issue as well:
Related log messages, e.g. if eebus wallbox is down during bootup:
`cannot create charger type 'template': cannot create charger type 'eebus': i/o timeout` 
`FATAL will attempt restart in: 15m0s`

armin-ei, 2024-07-15:
I have similar issue with two wallboxes configured in evcc: as soon as one Wallbox is off or offline, then evcc cannot be used anymore for the other wallbox, even though it is online. evcc should be able to start in such case for the wallbox that is in operation. The Workaround to use two different yaml files is not a good WA.

SolarPower2024, 2024-07-18:
Ich musste leider gerade feststellen, dass dieses Verhalten sehr dringend notwendig wäre.

Bei mir hat sich ein Shelly verabschiedet, der als PV Meter diente. (1 von 2 PV meters)
Der shelly wurde über die UI konfiguriert und obwohl das PV Meter rein für die Darstellung dient, Steuerung erfolgt ja via grid meter, startet evcc nicht mehr hoch.
Einzige Lösung: löschen der DB Einträge die mit dem PV Meter zusammen hingen.

Auch ein Failsafe Modus startete nicht. Eigentlich sollte der ja bereits vorhanden sein oder? Oder habe ich das falsch verstanden?

andig, 2024-07-18:
Meine PV läuft seit 14 Jahren ohne einen einzigen Ausfall von irgendeiner Komponente. Eine Shelly ist auch eher Bastelware.

SolarPower2024, 2024-07-18:
> Meine PV läuft seit 14 Jahren ohne einen einzigen Ausfall von irgendeiner Komponente. Eine Shelly ist auch eher Bastelware.

Da stimme ich dir zu.
Ist auch nur das BKW ;-)

umrath, 2024-09-03:
> Auch ein Failsafe Modus startete nicht. Eigentlich sollte der ja bereits vorhanden sein oder? Oder habe ich das falsch verstanden?

Failsafe heißt nur, dass evcc alle paar Minuten aufs neue versucht, die Geräte zu erreichen.
Funktioniert dabei eines nicht, wird der Start abgebrochen und evcc tut gar nichts - außer den Loop nach ein paar Minuten wieder neu zu beginnen.

SolarPower2024, 2024-09-03:
> Failsafe heißt nur, dass evcc alle paar Minuten aufs neue versucht, die Geräte zu erreichen. Funktioniert dabei eines nicht, wird der Start abgebrochen und evcc tut gar nichts - außer den Loop nach [...]

Sehe ich nicht so.
Wenn die ganze config im UI gemacht wird, anstelle der yaml MUSS evcc hochstarten. Sonst kannst du eine kaputte config nie mehr ändern / korrigieren, außer du öffnest die DB mit einem externen Programm und löschst die Einträge raus.
Es muss also ein failsafe mode (oder wie auch immer der dann heißt) existieren, damit Änderungen gemacht werden können. Und das ist ja auch irgendwo diskutiert worden und in der Mache, oder so ähnlich.

naltatis, 2024-09-03:
@umrath @SolarPower2024  Lass uns nicht hier das Thema neu diskutieren. Konkrete Lösungen oder PRs sind natürlich immer willkommen.

andi0b, 2024-09-25:
Wechselrichter im Powersafe-Mode macht gerade im Winter 16 Stunden am Tag permanent rote Fehlermeldungen in der UI. In meinem Fall startet evcc zum Glück trotzdem.

Ich habe auch einen mobilen charger in evcc drinnen, wenn ich den an/abstecke muss ich derzeit die Config immer neu schreiben und evcc neu starten.

> Meine PV läuft seit 14 Jahren ohne einen einzigen Ausfall von irgendeiner Komponente. Eine Shelly ist auch eher Bastelware.
Das ist schön!

voruti, 2024-12-13:
Bei mir ist letztens das Internet ausgefallen und ich habe folgende Meldung im Log beim Start erhalten (Ausschnitt):
```log
[gsi   ] ERROR 2024/12/10 20:45:45 Get ""https://api.corrently.io/v2.0/gsi/prediction?zip=***"": dial tcp 172.67.133.68:443: i/o timeout
[main  ] FATAL 2024/12/10 20:45:45 cannot create tariff type 'grünstromindex': Get ""https://api.corrently.io/v2.0/gsi/prediction?zip=***"": dial tcp 172.67.133.68:443: i/o timeout
[main  ] FATAL 2024/12/10 20:45:45 will attempt restart in: 15m0s
```
Entsprechend lief evcc nicht, obwohl Wallbox, etc. im lokalen Netz noch vollständig verfügbar waren.
Ich nehme an, dies zählt auch zu diesem Issue?

Generell versuche ich, möglichst unabhängig von Cloud-Diensten zu sein und verwende daher gerne Open-Source-Software und offene Komponenten. Mit dem Fehler ist es halt wieder sehr schade, an der Stelle vom Internet abhängig zu sein und das Auto nicht laden zu können.

andi0b, 2024-12-13:
>Ich nehme an, dies zählt auch zu diesem Issue?

Leider ist dieser Issue überwiegend eine bewusste und so gewünschte Design-Entscheidung. EVCC funktioniert ja überwiegend problemlos auch ohne diese Daten. Ich bin am Überlegen einen Sidecar-Container mit einem HTTP-Proxy einzurichten, der einfach auf alle HTTP-Fehler mit einer gecachten 200 OK Response antwortet. Das wird bei einigen Modulen zwar zu Logik-Fehlern führen, aber EVCC kann sich in der Regel sehr gut davon erholen.

Nur beim ersten Start werden alle Fehler als krtisch erachtet und der Start wird verweigert. In vielen Fällen kann man EVCC also nur in gewissen Situationen neu starten, bzw landet in einer gewissen Mieserie. Ich kann beispielsweise EVCC nur während die PV-Anlage Strom produziert starten, im Winter also nur zwischen 8 und 16 Uhr. Wenn ich also nach 16 Uhr EVCC stoppe, kann es sein dass es zu hohen Kosten kommt wenn dann Hausakku oder Auto bei extremen Strompreisen einfach weiterlädt.

umrath, 2024-12-13:
Das ist möglicherweise eine Idee, um die (in meinen Augen) falsche Designentscheidung auszuhebeln: Einen Proxy dazwischen schalten, der grundsätzlich alle Verbindungsanfragen positiv beantwortet, egal ob das Device dahinter antwortet oder nicht.

Ich muss mir mal Gedanken machen, ob sich das irgendwie mit vertretbarem Aufwand umsetzen lässt.

umrath, 2024-12-13:
Es muss ja kein offizieller Patch sein.
Solange der sich relativ einfach auf die Installation anwenden lässt, ist der Aufwand das deutlich geringere Übel als die möglichen Nebenwirkungen durch ein gar nicht startendes evcc.

SolarPower2024, 2024-12-13:
> Das ist möglicherweise eine Idee, um die (in meinen Augen) falsche Designentscheidung auszuhebeln: Einen Proxy dazwischen schalten, der grundsätzlich alle Verbindungsanfragen positiv beantwortet, egal ob das Device dahinter antwortet oder nicht.
> 

Hast du Home Assistant im Einsatz? Ich habe dies damit gelöst. Die Geräte sind in Home Assistant eingebunden und werden per mqtt an evcc geschickt. Sollten die Geräte wie zb mein PV oder grid meter nicht verfügbar sein, so sendet HA halt den Wert 0.
Dadurch habe ich die Fehler für mich abgefangen. 
Das einzige das direkt angebunden ist, ist die Wallbox.

StefanSchoof, 2024-12-13:
Damit Tarifs kein Startup Error erzeugen gab es vor einiger Zeit https://github.com/evcc-io/evcc/pull/16258

umrath, 2024-12-13:
> Hast du Home Assistant im Einsatz? Ich habe dies damit gelöst. Die Geräte sind in Home Assistant eingebunden und werden per mqtt an evcc geschickt. Sollten die Geräte wie zb mein PV oder grid meter nicht verfügbar sein, so sendet HA halt den Wert 0.

Das wäre tatsächlich eine Option.
Dann habe ich zwar noch eine Abhängigkeit an HA - aber damit kann ich wohl leben.
Und ich muss es ja nur für die Wallbox machen, die gern mal den Betrieb einstellt und nicht für alle Geräte. 

Danke für die Idee.

andig, 2024-12-13:
> um die (in meinen Augen) falsche Designentscheidung auszuhebeln

Puh, schon wieder die große Keule...

umrath, 2024-12-13:
Welche andere Wahl habe ich denn?
Dass evcc gar nicht funktioniert, ist die mit Abstand schlechteste Option für mich.
Alles andere ist besser.

andi0b, 2024-12-13:
> > um die (in meinen Augen) falsche Designentscheidung auszuhebeln
> 
> Puh, schon wieder die große Keule...

Ja @andig, ich kenne niemanden der das nicht als falsche Designentscheidung betrachtet. Sorry, ist einfach so, so ehrlich wird man doch noch sein dürfen? Du siehst ja hier, wie gar nicht so wenige hier viele Stunden Lebenszeit investieren dieses Problem zu umgehen.

andig, 2024-12-14:
Wenn es eine Design "entscheidung" wäre dann könnten wir hier als `wontfix` zu machen. Es ist schlicht der Stand der Implementierung. Zu gegebener Zeit werden wir das ändern. Wem das zu langsam geht der darf gerne in den Code schauen und einen Vorschlag machen. Danke.

andig, 2024-12-14:
@umrath @andi0b macht gerne eine Discussion zu alternativen Lösungen die ihr hier verlinken könnt. Hier gehts um das evcc Verhalten.

umrath, 2024-12-14:
> @umrath @andi0b macht gerne eine Discussion zu alternativen Lösungen die ihr hier verlinken könnt. Hier gehts um das evcc Verhalten.

Mir (und ich unterstelle: auch den anderen) geht es exakt um das Verhalten von evcc und nichts anderes.
Ich verstehe daher nicht so recht, wieso unsere Kommentare offtopic sein sollten.
Vielleicht sind sie nicht genehm. Aber offtopic?

Zumindest für einige ist das aktuelle Verhalten ein sehr ernstes Problem und kann potentiell massiv Geld kosten oder zu extrem unschönen Situationen führen.
Entsprechend halte ich die Suche nach einer Mitigation für das (aus user-Sicht) unerwünschte Verhalten durchaus für relevant, da ja ein echter Fix offenbar kompliziert ist, keine Priorität hat und/oder gar nicht von Interesse ist.

Außerdem möchte ich darauf hinweisen, dass ein bereits existierendes Ticket, welches das Ziel hatte, einen Workaround zu finden, mit einem Verweis auf dieses Ticket bereits geschlossen wurde.

dieterxy, 2024-12-14:
Ich habe einen go-e Charger Gemini flex mit 5-poligem roten CEE-Stecker. Diese Wallbox ist dafür ausgelegt, dass man sie auch auf Reisen mitnehmen kann.
Soll ich meinen Heimspeicher nicht vom Grid laden können, nur wenn ich mit der Wallbox unterwegs bin? Das macht doch keinen Sinn.
=> evcc sollte immer starten

Sumpfdotter, 2024-12-16:
As a reader of this discussion, I have perceived different wishes from various participants as to how the system should behave. I would like to contribute with a requirements proposal/clarification so that - assuming the proposal is accepted - the implementation concepts can go in this direction.

Here is the proposal for how evcc should behave, initially regardless of the effort required to create this behaviour. I would like to simplify the system behaviour with the proposal by directing the start case (which is the issue here) directly to an operating state that occurs anyway.

Here is the proposed requirement:

1. if a component connected to evcc fails (or the connection to it fails) while evcc is running, the connection is automatically re-established as soon as the component is available or connectable again.
2.  this also applies if several connected components fail, or even all of them.
3.  the operator of evcc is made aware of the connection statuses or the reasons for errors (this could be log files, status displays or notifications).
4. if connected components are currently not available, this can result in evcc having to degrade the corresponding functions. In the simplest case, evcc stops its entire behaviour when connected components fail until all failed components are reconnected.
5. In addition, gradual and dependency-orientated degradation is aimed for. For example, the local charging control should not be stopped completely if an Internet service for CO2 has failed. However, it is inevitable that charging control must be stopped if the connection to the wallbox is interrupted. Now comes the real point:
6. evcc's start state is defined as equivalent to the state evcc enters if all connected components have failed during operation.

**With requirement 6. evcc does not require its own boot procedure**, but can be booted directly into the ‘all connections are down’ state. From then on, the recovery procedures for this state, which are required here anyway, take effect and, if all connected components are available, all connections are established shortly after the start - without the need for a separate boot procedure.

I personally see the reason for these requirements in the assumption that the affected systems are all systems that are ideally in operation 24/7:
- Wallbox
- PV incl. battery
- EV charging controller (evcc)
- Network components such as routers, switches, ...
- Internet, Internet services ...
- ...

One indication that the systems should not wait for each other when starting up is shown in particular by the coupling of evcc with the wallbox: A wallbox is connected to the HEMS (evcc in this case) and vice versa. If both systems required the other when booting, a connection would never be established in first boot or after each complete shutdown, e.g. after power losses.

I am fully aware that the behaviour described here cannot be implemented easily. With this proposal, I want to work towards a consensus on the desired behaviour.

In other words: Is there a counter-reason why the boot behaviour should deviate from the behaviour when everything around evcc was once down in operation and is now available again? _Personally, I can't see any reason right now, so I'd like to challenge that here._

andig, 2024-12-16:
The reason _right now_ is that the desired behavior is not in place. Nobody debates this is not being the desired state.

Sumpfdotter, 2024-12-16:
My proposal introduces something new: it clearly states that no dedicated boot routine is wished. This is because there are different views on it in this thread, indicating that there are reasons for such a routine without clearly stating them.

> The reason _right now_ is that the desired behavior is not in place. Nobody debates this is not being the desired state.
As I said that is not true. There are serveral comments in this thread of people indicating that it is evcc's "design" to implement a fast fail routine at boot up. In fact, nobody stated the source of that requirement here but also nobody clearly contradicted that this requirement is not present. Examples:

User:
> Das aktuelle Verhalten ist eine bewusste Designentscheidung, so weit ich weiß ist eine Änderung dieses Verhaltens nicht gewünscht, und es gab schon Contributions die dieses Problem zu lösen versuchten. Die Argumentation ist folgende: Es wird erwartet, dass alle Komponenten stabil funktionieren und bei einem Ausfall einer Komponente der Neustart fehlschlägt, damit man Fehler sofort erkennen kann und sie nicht durch die vorhandenen Fehlerkorrekturen unentdeckt bleiben (fail early).

User:
> Ich musste leider gerade feststellen, dass dieses Verhalten sehr dringend notwendig wäre. Bei mir hat sich ein Shelly verabschiedet
Reply:
> Meine PV läuft seit 14 Jahren ohne einen einzigen Ausfall von irgendeiner Komponente. Eine Shelly ist auch eher Bastelware.
(in other words: dear user, please fix your setup, rather than prioritizing a change in evcc's bootup behavior)

The only comment indicating that the current boot behaviour is not wished it that one:
> Wenn es eine Design "entscheidung" wäre dann könnten wir hier als `wontfix` zu machen. Es ist schlicht der Stand der Implementierung. Zu gegebener Zeit werden wir das ändern. Wem das zu langsam geht der darf gerne in den Code schauen und einen Vorschlag machen. Danke.

But also that one does not clearly state if and why a dedicated boot routine is required.

If we agree on not having such a routine, the migration plan is (not easy) but clear:

1. Introduce a state model (connected/disconnted/...) for depending devices and services and implement it for the corresponding integrations (if not yet implemented)
2. Make states and failiures available for operators (assuminlgy this is already implemented with logfiles and error notifications within the UI)
3. Implement a retry and backoff strategy for reconnecting to failing devices/services (if not yet implemented)
4. Implement a more fine granular degradation of functions e.g. if single wallboxes or internet services are not present (if not yet implemented)
5. Replace bootup procedure by directly entering state "disconnected" for all lintegrations (will trigger retry&backoff right at the beginning)

This plan doesn't make sense if there are reasons for having a boot sequence. And those are indicated, but not stated so far. I don't know any, personally. That's why I'm challenging.

Krusty1977, 2025-04-04:
My Problem is that I user a NRGkick Charger, which is connected to a non smart / non controllable Wallbox via a Type-2 Adapter, please see this video https://youtu.be/yM0Cx7GqP08?t=181 

If there is no Charging process running on the _non small wallbox_ my NRGkick Charger will not have power and so it is not connected via WLAN to evcc. EVCC is not stable if this happens and so it would be better if evcc boots and then shows the charger as disconnected.

As soon as I place my Charging Card to the non smart wallbox, the charging process starts, NRGkick will get power, connects to WIFI and evcc and then it should be controllable via evcc. This does not work 100%

Sonderstorch, 2025-04-24:
I may add to the discussion that this behaviour is dire needed.

In my case, every case of non-starting was related to evcc falling over it’s own feed after updates.

Once, nightly auto-update happened, inverter in power-save, evcc stuck and EV was not charged in the morning (disabled auto-updates in HA since then).

Most recent case was modbus behaviour in 203.2 confused the inverter in a way that I needed to power-cycle it in order to restart evcc (problem repeated, reverted back to 202.1).

The only unstable thing in my setup is evcc itself.

Zwer2k, 2025-07-21:
Durch dieses Verhalten von EVCC, kommt es immer wieder vor, dass das Wasser im Speicher kalt bleibt, da die Heizstabsteuerung außer kraft gesetzt wird. 
Das Problem tritt in Verbindung mit einer "Porsche Mobile Charger Connect" Wallbox die nach wenigen Minuten bei nicht Nutzung in Standby-Modus wechselt (was ja eigentlich gut ist, da es ca 12W spart).
Da EVCC als HA-Addon läuft passiert es leider öfters.

Zusätzlich bleibt EVCC immer wieder mit 100% CPU Auslastung hängen, was vermutlich auch auf Standby von der Wallbox zurückzuführen ist. Muss ich aber noch beobachten, wannes genau passiert.

premultiply, 2025-07-21:
> Durch dieses Verhalten von EVCC, kommt es immer wieder vor, dass das Wasser im Speicher kalt bleibt, da die Heizstabsteuerung außer kraft gesetzt wird.

evcc ist weder eine Heizungssteuerung noch ein Thermostat!
Alles was mit Wärme zu tun hat sollte immer eine unabhängige Steuerung haben.
evcc sollte hier wenn dann nur ergänzend Eingreifen um der Heizungssteuerung verfügbaren Energieüberschuss oder Energiemangel zu signalisieren. Leider wurde dieser Grundsatz in einigen Geräteintegrationen wohl nicht ausreichend beachtet. Wo immer man derzeit irgendwelche Temperaturlimits in evcc konfigurieren kann wird es m.M.m. nach schon kritisch...

Auch Wallboxen sollten idealerweise ein konfigurierbares Fallback-Verhalten haben so dass bei Ausfall der übergeordneten Steuerung ein definiertes Havarieszenario eintritt.

premultiply, 2025-07-21:
Oha.

Zwer2k, 2025-07-21:
> evcc ist weder eine Heizungssteuerung noch ein Thermostat! Alles was mit Wärme zu tun hat sollte immer eine unabhängige Steuerung haben. evcc sollte hier wenn dann nur ergänzend Eingreifen um der [...]

Die Heizung würde ja übernehmen, wenn ich die nicht vor dem Sommer komplett abgedreht hätte. Abgedreht habe ich die aber nur weil durch das unschöne Verhalten von EVCC ab und zu teureren Fernwärmeverbrauch (im Vergleich zu PV) gekommen ist.

SolarPower2024, 2025-07-21:
@Zwer2k du hast folgende Möglichkeit eines Workarounds:
a) du kannst in Home Assistant tracken ob der evcc Container läuft, falls nein lass dir eine Erinnerung ans Handy schicken
b) integrierte die Geräte nicht direkt in evcc sondern in Home Assistant und erstelle in evcc ein custom device die die Daten von HA abfragt (und im Fehlerfall halt zb 0 ausgibt)

andi0b, 2025-07-21:
Ja, die Variante mit externen Integrationen kann ich auch nur empfehlen. Auch das http Plugin von evcc funktioniert dafür oft gut. Seither habe ich keine Stabilitätsprobleme mehr, und das Sponsor Token kann man sich auch sparen :)

andig, 2025-07-21:
Danke @andi0b das ist wirklich extrem wertschätzend. Deinen entsprechenden Kommentar habe ich aus nachvollziehbaren Gründen gelöscht.

umrath, 2025-08-12:
> Das Hauptproblem hier ist eigentlich, dass viele Gerätefehler durch Neustarts verschwinden. Wenn wir als Regel _keinen_ Neustart mehr machen, um mit den verbleibenden Geräten weiter zu laufen, bleiben die inaktiven Geräte genau das- inaktiv.
> 
> Woran machen wir fest, ob Neustart oder weiter machen? Abgrenzung: Geräte später "aktivieren" ist aktuell nicht möglich.

Warum müsst ihr das festmachen?
Überlasst doch den Usern die Entscheidung.
Von mir aus macht alles per default auf "required" und überlasst den user die Option, in den Konfiguration ein "required = false" zu setzen.
Wenn ein Gerät mit "required = false" nicht verfügbar ist, dann einfach fortfahren und Fehler ignorieren, periodisch versuchen neu zu laden.

andig, 2025-08-14:
> Wenn mitten im Betrieb eine Wallbox plötzlich nicht mehr erreichbar ist, wird ja auch kein Neustart erzwungen, sondern sie einfach als "inaktiv" bzw. "nicht antwortend" links liegengelassen. Genauso sollte es auch beim Hochfahren sein.
Wenn das (warum auch immer) gar nicht möglich ist, müsste man es wenigstens einheitlich gestalten,

Das ist ein guter Punkt und würde dafür sprechen, dass wir "muss verfügbar sein" eher als opt-in machen und damit das heutige Verhaltung beim Neustart "umkehren". Neustart sollte auch erfolgen, wenn keine Wallbox mehr erreichbar ist (z.B. Netzwerkfehler...).

manny4566, 2025-09-29:
> Zu welchem Zweck?

Damit der Restart halt schneller passiert, wenn mal eine Verbindung zu einem Backend System nach einem Neustart der ganzen Umgebung nicht geklappt hat.
z.b. MQTT, InfluxDB, Auto API, Wallbox API. 
Auf letztere beiden hat man natürlich wenig Einfluss. 
Klar, auf erstere beiden schon. Da kann ich auch die Reihenfolge so anpassen, dass die Wahrscheinlichkeit gering ist, dass z.b. der MQTT Service nicht da ist und somit evcc nicht startet.
Es wäre aber schon nice auch auf evcc Seite den Intervall umstellen zu können, damit der reconnect schneller versucht wird.
Ich nutze die Daten aus evcc weiter und muss sonst 15min ohne Daten auskommen, wenn mir das nicht vorher auffällt.

manny4566, 2025-09-29:
> > damit der reconnect schneller versucht wird. Ich nutze die Daten aus evcc weiter und muss sonst 15min ohne Daten auskommen, wenn mir das nicht vorher auffällt.
> 
> Wenn dir diese Daten so wichtig sind, dann monitore deine System und löse Aktionen auf Grund von Monitoring aus. Das Problem in dem Fall ist ja nicht evcc, sondern das andere Systeme nicht stabil laufen.

Das hat nichts mit Stabi8lität der anderen Systeme zu tun, sondern mit dem Hardcore Stopp während des Startens von evcc, da es eine Ressource nicht erreicht. Also ja, evcc ist nicht gerade unwichtig bei der Konstellation.

Man kann ja trotzdem mal mehr Konfigurationsfreiheit einbauen. 
Es geht nur um einen kleinen Wert, der sich leicht anpassbar machen lässt. Nicht um irgendeine riesige Anpassung. 
Ihr mit eurer bockigen Haltung blockiert die Entwicklung.... naja, dann halt nicht...

andi0b, 2025-09-29:
> Wenn dir diese Daten so wichtig sind, dann monitore deine System und löse Aktionen auf Grund von Monitoring aus. Das Problem in dem Fall ist ja nicht evcc, sondern das andere Systeme nicht stabil laufen.

Instabilität und Verbindungsabbrüche sind eine inherente Eigenschaft von IoT-Geräten. Sowohl im Home-Automation-Bereich als auch bei größeren Anwendungen. Wenn beispielsweise der Ausfall der Kommunikation eines Smart Meters eines Stromnetzbetreibers zu einem kompletten Shut-down des Backends führen würde, dann hätten wir eine lustige Welt.

markus7h, 2025-11-26:
Aus meiner Sicht ist ein nicht fehlertolerantes nicht Wünschenswert. In Installationen die mehrere Komponenten enthalten wie z.B. Auto, WP, Solaranlage plus die technische Komponenten wie MQTT und Modus Proxy, führt der Ausfall einer Komponente zum Ausfall des gesamten Systems. Auch wenn diese überhaupt nicht betroffen sind. Beispiel der MQTT Server ist nicht erreichbar und eine gerade laufende Ladung des Auto läuft dann solange weiter bis das Auto 100% geladen ist, nur weil ein Update des docker Dienstes erfolgt ist.

Ziel sollte sein, das System so robust wie möglich gegen Störungen zu designen und nicht umgekehrt.

Idealerweise gibt es einen Benachrichtigungskanal der über push den Admin informiert das Fehler im System vorliegen.

andig, 2025-11-26:
> führt der Ausfall einer Komponente zum Ausfall des gesamten Systems. Auch wenn diese überhaupt nicht betroffen sind. Beispiel der MQTT Server ist nicht erreichbar und eine gerade laufende Ladung des [...]

Mit Verlaub, das ist Quatsch. Bei evcc fällt da nichts aus. Aber OT daher es hier keinerlei neue Information gibt.

manny4566, 2025-11-26:
> Mit Verlaub, das ist Quatsch. Bei evcc fällt da nichts aus. Aber OT daher es hier keinerlei neue Information gibt.

Ich glaube was er mit Ausfall meint ist, dass evcc nicht mehr startet und im restart loop ist, wenn eine externe Komponente nicht eingebunden werden kann / erreichbar ist.
Also in dem Moment wo zufällig auch evcc gerade neu startet.

phrozen77, 2026-01-15:
What i'd like to add / put up for discussion:

How about you just disable the respective functions, mark it as unavailable in the UI - no matter if during startup or during normal operation/idle?

- Grid meter unavailable? No problem, mark it with a 🛑, disable "PV" button, add message to error log in UI (the triangle icon) - but keep "Fast" mode available for example.
- Wallbox unavailable? Ok, disable all charging modes but keep all other info, retry periodically
- Battery unavailable? Ok, disable all handling of that but don't prevent charging from PV or grid

All missing connections should be retried periodically, done?

VolkerK62, 2026-01-16:
see also https://github.com/evcc-io/evcc/issues/21144#issuecomment-2869864700

kaindl, 2026-02-28:
> Imho there are 2 key questions that need clear answers:
> 
>     * What are tolerable outages that should _not_ hinder evcc start in terms of meters and chargers?
> 
>     * Is that a general rule or does it need opt-in? If opt-in what's the baseline- "must be online" or "can be offline" to configure against?

I would say: should be configurable.
Default should be "can be offline" for anything that is using power (chargers, cars, heaters, ...) or providing power (inverters, batteries, ...). It should be "must be online" for anything, that is needed to control the available devices, should mostly be the meters.

MvsBluewin, 2026-09-06:
I too think this is critical. It is quite normal for some devices to not be online at the moment EVCC has a restart (every time I do an update of the software) because of maintenance, power saving, or poor connection (powerline ethernet) and everything mentioned above. EVCC should ignore the devices it cannot reach (and grey out any dependents) until they're back up and in the mean time proceed as if these were never entered in the yaml (or UI) without me having to comment out the devices in the yaml

LotharSchoenfeld, 2026-09-18:
I can confirm this issue with a reproducible real-world case using Shelly devices.

### Setup

I am running evcc on a Raspberry Pi 5 with several Shelly devices configured through the evcc UI/database configuration.

If one or more Shelly 1PM devices become unreachable while evcc is already running, evcc handles the situation reasonably well: errors are logged for the affected devices, but the rest of the system continues to operate.

For example:

```text
[lp-1 ] ERROR 2026/09/18 17:50:06 charge power: Post "http://192.168.10.49/rpc/Switch.GetStatus": dial tcp 192.168.10.49:80: connect: no route to host
[lp-2 ] ERROR 2026/09/18 17:50:07 charge power: Post "http://192.168.10.50/rpc/Switch.GetStatus": dial tcp 192.168.10.50:80: connect: no route to host

[site ] DEBUG 2026/09/18 17:50:20 grid power: -1556W
[site ] DEBUG 2026/09/18 17:50:20 battery 1 power: 0W
[site ] DEBUG 2026/09/18 17:50:20 pv 1 power: 3057W
[site ] DEBUG 2026/09/18 17:50:20 battery 1 soc: 100%
[site ] DEBUG 2026/09/18 17:50:20 site power: -1556W

[lp-3 ] DEBUG 2026/09/18 17:50:21 charger status: A
[lp-4 ] DEBUG 2026/09/18 17:50:25 charge power: 1335W
[lp-4 ] DEBUG 2026/09/18 17:50:25 charge currents: [5.92 0 0]A
```

This shows that evcc is able to continue operating while the affected Shelly devices are unavailable.

### Problem after restarting evcc

If evcc is restarted while the same Shelly device is still unreachable, initialization fails:

```text
Fehler beim Starten. Überprüfe deine Konfiguration und starte neu.

charger: [db:29] cannot create template 'db:29':
cannot create charger type 'template:shelly':
cannot create charger type 'shelly':
Get "http://192.168.10.49/shelly":
dial tcp 192.168.10.49:80: connect: no route to host

loadpoint: [db:30] missing charger instance
```

So the behavior is reproducible:

**Device reachable → evcc starts normally**

**Device becomes unreachable while evcc is running → evcc continues operating and reports errors for the affected device**

**Device unreachable + restart of evcc → initialization fails**

This is problematic because a configured device can temporarily become unavailable for many normal reasons: hardware failure, power loss, network problems, firmware updates, etc.

In such a situation, changing or deleting an otherwise valid configuration should ideally not be necessary just to get evcc running again.

### Expected behavior

It would be very helpful if evcc could handle unavailable devices during startup in a similar way to how it handles them during normal operation:

* Start evcc even if individual configured devices are currently unreachable.
* Mark the affected device/loadpoint as offline or unavailable.
* Keep unrelated loadpoints, meters, PV, battery and site functions operational.
* Periodically retry communication with the unavailable device.
* Automatically restore the device/loadpoint when it becomes reachable again.
* Do not require the user to modify or delete the configuration because of a temporary device failure.

The interesting part in this case is that evcc already demonstrates during normal operation that it can tolerate the communication failure. The main difference appears to be how the same failure is handled during initialization.

Could the startup/initialization process be changed so that an unavailable device does not prevent otherwise independent parts of evcc from operating?

Thank you!

asdfklgash, 2026-10-06:
Ich habe einige TP-Link Tapo (P)-Steckdosen, die manchmal nicht eingesteckt sind, da das Gerät - z. B. Thermomix - nicht in der Küche ist. Diese sind als Verbraucher angelegt und blockieren damit einen (Neu)Start von evcc, da die Steckdosen natürlich nicht erreichbar sind.

Zudem ist mir aufgefallen, dass eine Deaktivierung der Steckdose auch nicht funktioniert, wenn diese nicht erreichbar ist, da diese das Gerät neu anlegen will - was aufgrund fehlender Erreichbarkeit nicht geht.

Hätte hier bereits eine Fix vorbereitet, der das gleiche Vorgehen wie für die Tarife umsetzet - Start möglich, neuer Versuch für den Verbraucher-Zähler alle 5 Minuten.

asdfklgash, 2026-10-07:
To make my proposal from above concrete: scope is consumer meters only. They are only monitored, not used for control or §14a dimming. Grid, PV, battery, charger, aux and ext meters keep failing startup as today.

Same pattern as `tariff.Wrapper` (#33894): on a non-config error at startup the consumer meter is replaced by a wrapper. The wrapper reports "meter not available" and retries creating the meter at most every 5 minutes. Once that succeeds, the real meter is used, without a restart.

Just treating the meter as inactive at startup would also avoid the fatal, but then it stays inactive until the next restart. The retry covers that.

Branch with tests: https://github.com/asdfklgash/evcc/tree/consumer-meter-retry

Would a PR along these lines be welcome? The crash when disabling an unavailable device is handled separately in #34504 / #34505.

tolot27, 2026-11-01:
Concrete case and a possible first step, focused on OCPP.

A charger that is only temporarily unreachable at startup (rebooting, firmware update, outage) makes `NewOCPP` wait for the connection and `cp.Setup` (default `connecttimeout` 5 min). On timeout the whole startup fails (`cannot create charger type 'ocpp': timeout`, `missing charger instance`) and is retried after 15 minutes. No loadpoint is controlled in the meantime, including those of chargers that are online. The same happens when a charger drops the connection during setup (#35357).

**Possible approach (OCPP first)**

Return from the OCPP constructor immediately and run the wait for the connection and `cp.Setup` in the background. While pending, status and measurements return `api.ErrNotAvailable` and no writes are issued. When setup completes, the charger calls the loadpoint's `RequestUpdate()` (see #35474), so the loadpoint does not wait for its next cycle. Loadpoints with reachable chargers start immediately, an unreachable one only affects its own loadpoint.

Open points:

- OCPP registers capabilities conditionally (`HasMeasurement`, `PhaseSwitching`) after `cp.Setup`. They must be known at construction, or resolved differently for a pending charger.
- How the error page and `joinErrors` treat a charger that is pending or fails later (visible, but not blocking).
- `chargedAtStartup` is read at loadpoint construction and would be read too early.
- Synchronising access to `c.cp`/`c.conn` between background setup and loadpoint.

Related, but none of them changes the blocking startup: #35357 (setup never resumes after a drop), #35651 (BootNotification trigger on reconnect), #35474 (update loadpoint on charge point status change).

I'm happy to implement this as a PR for OCPP if the direction is acceptable, or in a more generic form if you prefer.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

LotharSchoenfeld, 2026-11-06:
Thanks for the detailed analysis and the proposed solution!

Since my original issue is not specific to OCPP but also affects Shelly devices, I would definitely prefer a more generic approach if feasible.

The main goal should be that a single temporarily unreachable device does not prevent evcc from starting and operating all other available devices and loadpoints.

As demonstrated in my original report, evcc already handles unavailable Shelly devices during normal operation. It would be great to achieve similar behavior during startup, regardless of the charger implementation.

I understand that this may require some architectural changes, especially regarding capability detection and initialization.

If a generic solution is feasible, I think it would provide a significant improvement in robustness for evcc as a whole.

Thanks again for looking into this and for offering to implement a solution!

j-github, 2026-11-07:
@tolot27 

>or in a more generic form if you prefer

The more generic the better