evcc-io/evcc #14497: Separate current for the chargemodes

linuxstony, 2024-06-21:
I don't own a big PV - I only have two solar panels with 600W max. 

My provider is tibber so I can make use of the cheap prices charging and I can support with 600W when charging 

So when I got time I switch to PV mode and reduce the current to 6A 

When I switch to fast because I have to charge fast I have manually change to 16 A.

It would be helpful if you could set the max ampere per charge mode. 

E.g. 6 A for PV to use cheap prices and the sun as much as possible and 16 A when I can't wait.

naltatis, 2024-06-21:
I don't understand. PV (and Min+PV) mode should already operate on your lowest current limit (minCurrent) unless more surplus is available. When you enter a "cheap zone" (tibber price below configured limit) evcc will switch to the configured maximum value (maxCurrent) to use as much cheap energy as possible.

linuxstony, 2024-06-21:
Hmm - when I switched to PV it always charges with 16A approx 3,7 kW. It never reduced to 6A. 

Min current was set to 1 - i now changed to 6 - maybe that was the error on my side. 

> - title: Wallbox
  charger: Wall-E
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  enable:
    threshold: -300
    delay: 3m
  disable:
    threshold: 2100
    delay: 3m   

Also - your comment with the Cheapzone could be an additional part that I missed. 

I was just confused that the amps where always the same while switching from FAST to PV or MinPV.

I apologize if my "issue" is a standard feature and I was just too dumb to realize.

linuxstony, 2024-06-21:
Alright - understanding error on my part. 

Sorry

closed 2024-06-21 by linuxstony