evcc-io/evcc #14501: SOC via openWB2: vehicle not available

martin-zh, 2024-06-21:
### Describe the bug

Mit der openWB1.9 funktioniert die SOC Abfrage via euch eines Cupra Born ohne Probleme.
Nun bin ich auf die neue Version 2.1.4 (release) gewechselt und erhalte folgenden Fehler:
<class 'grpc.RpcError'> (<_InactiveRpcError of RPC that terminated with: status = StatusCode.UNKNOWN details = "vehicle not available" debug_error_string = "UNKNOWN:Error received from peer {created_time:"2024-06-21T15:01:47.032895513+02:00", grpc_status:2, grpc_message:"vehicle not available"}" >,)

Zugangsdaten (Username, PWD, VIN) aus dem alten openWB Konfig File kopiert, Token neu generiert.

### Steps to reproduce

SOC abfragen

### Configuration details

```yaml
SOC Type= Seat Cupra
```

### Log details

```text
<class 'grpc.RpcError'> (<_InactiveRpcError of RPC that terminated with: status = StatusCode.UNKNOWN details = "vehicle not available" debug_error_string = "UNKNOWN:Error received from peer {created_time:"2024-06-21T15:01:47.032895513+02:00", grpc_status:2, grpc_message:"vehicle not available"}" >,)
```

### What type of operating system are you running?

Linux

andig, 2024-06-21:
Bitte Zugangsdaten an info@evcc.io zum testen. Danke!

closed 2024-06-21 by andig

YareCare, 2024-07-06:
Same here. Wurde das Problem gelöst? Bei mir gehts auch nimmer bei nem Skoda mit selber Fehlermeldung. Seit wann kann ich leider nicht genau sagen.

andig, 2024-07-06:
Wurde gelöst. Falls nach Neustart immer noch ein Problem bitte Zugangsdaten an [info@evcc.io](mailto:info@evcc.io) zum testen. Danke!

YareCare, 2024-07-06:
Welche Zugangsdaten werden hier benötigt? Dieser API-Key?