evcc-io/evcc #14508: Custom charger: add more apis

andig, 2024-06-22:
Aktuell kann der custom charger noch keine Leistung. Stattdessen wäre ein separates `meter` am Loadpoint zu verwenden. Das ist nicht ganz offensichtlich und könnten wir hinzu fügen- dann müssten wir uns allerdings fragen, welche der vielen `meter` APIs wir hier noch anbieten wollen?

_Originally posted by @andig in https://github.com/evcc-io/evcc/issues/14504#issuecomment-2183952954_

premultiply, 2024-06-22:
Guter Punkt. Daran hab ich überhaupt nicht gedacht.

andig, 2024-06-22:
Meter hätte heute:

```go
Power    provider.Config
Energy   *provider.Config  // optional
Currents []provider.Config // optional
Voltages []provider.Config // optional
Powers   []provider.Config // optional

// battery
capacity    `mapstructure:",squash"`
battery     `mapstructure:",squash"`
Soc         *provider.Config // optional
LimitSoc    *provider.Config // optional
BatteryMode *provider.Config // optional
```

Davon wäre `BatteryMode` nicht relevant und `LimitSoc` erwarten wir für bestimmte Arten von Battery Control einen Setter. Bliebe also alles über dem Strich.

VolkerK62, 2024-06-22:
Würde damit evtl auch https://github.com/evcc-io/evcc/issues/12276  lösbar?

andig, 2024-06-22:
Nein. Deine Schaltsteckdose ist kein `custom` device.

VolkerK62, 2024-06-22:
Ah, Mist. Erst denken, dann schreiben 😄

premultiply, 2024-06-22:
`Power` und `Energy` sollten schon sein, `Currents` und `Voltages` könnten auch noch rein.
Den Rest halte ich erstmal für nicht relevant.

andig, 2024-06-22:
Na dann brauchts `Powers` auch fürs Vorzeichen, oder? Also alles „über dem Strich“?

premultiply, 2024-06-22:
Wozu beim Charger?
Da gibt es nur eine Richtung: Rein in's Fahrzeug. ⚡️➡️🚗 
Daher immer alles unsigned.

andig, 2024-06-22:
Solange es kein Bidi gibt. Aber verstanden, danke 🙏

closed 2024-07-06 by andig