evcc-io/evcc #14513: Huawei Modbus TCP - Batterie Fehler (battery mode: modbus: exception '1' (illegal function), function '6')

fritzrichter, 2024-06-22:
### Describe the bug

I constantly receive this error message in EVCC. It seems still to work most of the time, but the error pops up again and again.

battery mode: modbus: exception '1' (illegal function), function '6'

### Steps to reproduce

Connect EVCC to Huawei Modbus TCP (hidden behind a modbox proxy)

### Configuration details

```yaml
meters:
- name: grid
  type: template
  template: huawei-dongle-powersensor
  usage: grid

  # Modbus TCP
  modbus: tcpip
  id: 1
  host: iobroker
  port: 502
  timeout: 15s

- name: pv
  type: template
  template: huawei-dongle-powersensor
  usage: pv

  # Modbus TCP
  modbus: tcpip
  id: 1
  host: iobroker
  port: 502
  timeout: 15s

- name: battery
  type: template
  template: huawei-dongle-powersensor
  usage: battery

  # Modbus TCP
  modbus: tcpip
  id: 1
  host: iobroker
  port: 502
  timeout: 15s
```

### Log details

```text
battery mode: modbus: exception '1' (illegal function), function '6'
```

### What type of operating system are you running?

Docker container

### Version

0.127.2

andig, 2024-06-22:
Please add modbus trace log (can be done from ui).

andig, 2024-06-24:
Ping @fritzrichter

fritzrichter, 2024-06-24:
Unterwegs, melde mich asap.

fritzrichter, 2024-06-27:
Ich kann es mir nicht erklären, aber die Tage kam keine Fehlermeldung mehr, als wäre nie was gewesen. Ich werde entsprechend das Ticket schließen in der Hoffnung dass sich das Problem langfristig in Luft aufgelöst hat :) Danke dennoch für den Support.

closed 2024-06-27 by fritzrichter

andig, 2024-06-27:
Sehr merkwürdig... danke fürs Update!