evcc-io/evcc #14514: Can't retrieve status for Peugeot car (PSA)

cbeyls, 2024-06-22:
### Describe the bug

No matter how many times I retry the procedure to generate the authentication tokens, EVCC fails to retrieve any status for the car.

### Steps to reproduce

1. Generate tokens using: `evcc token peugeot --ignore-db`
2. Fill the country code (in my case `BE`, I also tried `FR`) then open the web page.
3. Log in using e-mail and password.
4. After the web page shows a "Login successful" message, click the "OK" button under it and grab the authorization code from the web page "Location" header using Firefox developer tools.
5. EVCC output both tokens, copy them to the configuration file under the `accessToken` and `refreshToken` keys (even if the command line message says to put them as `access` and `refresh` sub-keys under a `tokens` main key).
6. Start the EVCC service, it shows an error in the web interface: "vehicle odometer: unexpected status: 404 (Not Found)".

Please note that at no point was I asked to fill in a code received by SMS.

I used to plug in a BMW car whose status could be retrieved without any problem so I'm confident that the issue is with the PSA API or the PSA integration.

### Configuration details

```yaml
chargers:
- type: template
  template: alfen
  id: 1
  host: 192.168.0.63
  port: 502
  modbus: tcpip
  name: wallbox1

vehicles:
- type: template
  template: peugeot
  title: Peugeot e-208 GT
  icon: car
  user: ***@***
  accessToken: ***
  refreshToken: ***
  capacity: 50
  name: peugeot
```

### Log details

```text
[peugeot] TRACE 2024/06/22 14:34:46 GET https://api.groupe-psa.com/connectedcar/v4/user/vehicles?client_id=1eebc2d5-5df3-459b-a624-20abfcf82530
[peugeot] TRACE 2024/06/22 14:34:47 {"_links":{"last":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles?pageToken=e15590d9a0ceccc82623bccdeb807eedAAAAAQAAAAEAAAABAAAAPAACMC0"},"self":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles?pageToken=e15590d9a0ceccc82623bccdeb807eedAAAAAQAAAAEAAAABAAAAPAACMC0"},"first":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles?pageToken=e15590d9a0ceccc82623bccdeb807eedAAAAAQAAAAEAAAABAAAAPAACMC0"}},"total":1,"totalPage":1,"currentPage":1,"_embedded":{"vehicles":[{"_links":{"collisions":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/collisions"},"alarms":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/alarms"},"callbacks":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/callbacks"},"alerts":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/alerts"},"trips":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/trips"},"self":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1"},"lastPosition":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/lastPosition"},"remotes":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/callbacks/{cbid}/remotes","templated":true},"telemetry":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/telemetry"},"user":{"href":"https://api.groupe-psa.com/connectedcar/v4/user"},"maintenance":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/maintenance"},"status":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/status"},"monitors":{"href":"https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/callbacks/{cbid}/monitors","templated":true}},"id":"108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1","vin":"[***edited out***]","motorization":"Electric","brand":"Peugeot","pictures":["https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=001","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=002","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=003","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=004","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=005","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=006","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=007","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=008","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=009","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=010","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=011","https://visuel3d-secure.peugeot.com/V3DImage.ashx?client=miseco&back=0&format=png&version=1PP2A5UZIFB0A0F4&color=0MM00NLD&trim=0PWN0RFX&OPT1=D180&OPT2=D190&OPT3=DLZ0&OPT4=DZVK&view=012"]}]}}
[peugeot] TRACE 2024/06/22 14:34:48 GET https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/status?client_id=1eebc2d5-5df3-459b-a624-20abfcf82530
[... cut ...]
[peugeot] TRACE 2024/06/22 14:35:18 GET https://api.groupe-psa.com/connectedcar/v4/user/vehicles/108e71caf0c5c678d9d0e517f9c8af52e0f0d599800b2b54c963e1414372d4815609ac5a35bcbc850525b09cb2a46f5d7713f6e6fda78bed8538edae412715ada84fe624376928e0f8f9c4e12ba08e9268b501dff44cf44942082e160895e84b1/status?client_id=1eebc2d5-5df3-459b-a624-20abfcf82530
[peugeot] TRACE 2024/06/22 14:35:19 {"uuid":"6f3f95d8-83a8-422d-bfde-34b1ff37bca4","code":40400,"message":"Not Found: We didn't find the status for this vehicle.","timestamp":"2024-06-22T12:35:19.169Z"}
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.127.2

andig, 2024-06-22:
> EVCC output both tokens, copy them to the configuration file under the accessToken and refreshToken keys (even if the command line message says to put them as access and refresh sub-keys under a tokens main key).

That's already fixed in master.

> Not Found: We didn't find the status for this vehicle

Interesting. Are you aware of any other open source software that is able to read the status? Especially, is https://github.com/TA2k/ioBroker.psa working for you?

cbeyls, 2024-06-22:
> Interesting. Are you aware of any other open source software that is able to read the status? Especially, is https://github.com/TA2k/ioBroker.psa working for you?

I didn't try yet but now that I read more about it, it looks like it's a general problem at the API or car side and has nothing to do with EVCC. Mainly because even the official app is unable to show the charging status and keeps repeating to enable connectivity and start driving a bit (which I did with no effect). My apologies.

andig, 2024-06-22:
> Mainly because even the official app is unable to show the charging status and keeps repeating to enable connectivity and start driving a bit (which I did with no effect). My apologies.

Yeah, that's a strong indicator. Closing here- feel free to reopen once the official app works and problem persists!

closed 2024-06-22 by andig