evcc-io/evcc #14515: Fahrzeugerkennung verbessern?

VolkerK62, 2024-06-22:
In dem Fall melden beide Fahrzeuge beim Connect Status B (der vom Dacia (Tiefschlaf) ist falsch, weil veraltet).
Das dann keine Erkennung möglich ist, leuchtet ein.
```
[lp-1 ] INFO 2024/06/22 18:58:53 car connected
[lp-1 ] DEBUG 2024/06/22 18:58:53 vehicle api refresh
[main ] DEBUG 2024/06/22 18:58:54 vehicle status: B (Dacia)
[main ] DEBUG 2024/06/22 18:58:59 vehicle status: B (KIAEV6)
[main ] WARN 2024/06/22 18:58:59 vehicle status: >1 matches, giving up
```
Wenn dann geladen wird, wechselt der Kia auf Status C und entspricht somit dem Status der Wallbox.
Allerdings wird das von der Erkennung wohl nicht beachtet.
```
[lp-1 ] DEBUG 2024/06/22 19:04:53 charger status: C
[lp-1 ] DEBUG 2024/06/22 19:04:53 vehicle api refresh
[main ] DEBUG 2024/06/22 19:04:53 vehicle status: B (Dacia)
[main ] DEBUG 2024/06/22 19:04:58 vehicle status: C (KIAEV6)
[main ] WARN 2024/06/22 19:04:58 vehicle status: >1 matches, giving up
```
Wäre hier eine "feinere" Differenzierung bei der Erkennung möglich?

andig, 2024-06-23:
Die Erkennung hier würde funktionieren wenn der Dacia bereits an einem Ladepunkt angeschlossen und erkannt wäre. Dann bliebe nur noch ein Fahrzeug zur Erkennung übrig.

andig, 2024-06-23:
Da die APIs relativ träge sind scheint eine feinere Auflösung des Status nicht sinnvoll.

VolkerK62, 2024-06-23:
Der Dacia könnte ja auch ganz woanders angesteckt sein.
Aber ich verstehe deinen Punkt. Man kann bei der Art der Erkennung nicht alle Möglichkeiten abfangen.
Außerdem besteht im Zweifel die Option der manuellen Auswahl.
Ich mach dann hier zu.

closed 2024-06-23 by VolkerK62