evcc-io/evcc #14517: 'checkconfig' deems configuration valid, while it's not.

rudybroersma, 2024-06-22:
### Describe the bug

I'm trying to add my heatpump as a charger to EVCC. Not much luck as of yet, seems documentation is outdated?
Anyway.. I had this:

```
  - name: heatpump
    integrateddevice: true
    icon: waterheater
    type: custom
    status:
      source: const
      value: 'C'
    enable:
      source: mqtt
      topic: homeassistant/switch/warmtepompforcedon/state
      payload: on
    maxcurrent:
      source: mqtt
      topic:
```

I then ran checkconfig:

```
$ /usr/local/src/evcc/evcc -c /home/evcc/evcc.yaml checkconfig
[main  ] INFO 2024/06/22 20:44:29 evcc 0.126.5 (4f888064)
[main  ] INFO 2024/06/22 20:44:29 using config file: /home/evcc/evcc.yaml
config valid
```

However:
```
# systemctl restart evcc
# systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/etc/systemd/system/evcc.service; enabled; preset: enabled)
     Active: active (running) since Sat 2024-06-22 20:44:34 CEST; 2s ago
   Main PID: 840550 (evcc)
      Tasks: 7 (limit: 9384)
     Memory: 30.7M
        CPU: 624ms
     CGroup: /system.slice/evcc.service
             └─840550 /usr/local/src/evcc/evcc

Jun 22 20:44:34 nuc.tozz.nl systemd[1]: Started evcc.service - evcc.
Jun 22 20:44:34 nuc.tozz.nl evcc[840550]: [main  ] INFO 2024/06/22 20:44:34 evcc 0.126.5 (4f888064)
Jun 22 20:44:34 nuc.tozz.nl evcc[840550]: [main  ] INFO 2024/06/22 20:44:34 using config file: /home/evcc/evcc.yaml
Jun 22 20:44:34 nuc.tozz.nl evcc[840550]: [main  ] INFO 2024/06/22 20:44:34 starting ui and api at :7070
Jun 22 20:44:34 nuc.tozz.nl evcc[840550]: [db    ] INFO 2024/06/22 20:44:34 using sqlite database: /home/evcc/evcc.db
Jun 22 20:44:34 nuc.tozz.nl evcc[840550]: [mqtt  ] INFO 2024/06/22 20:44:34 connecting evcc-1934116284 at tcp://192.168.178.250:1883
Jun 22 20:44:35 nuc.tozz.nl evcc[840550]: [main  ] FATAL 2024/06/22 20:44:35 cannot create charger 'heatpump': cannot create charger type 'custom': 1 error(s) decoding:
Jun 22 20:44:35 nuc.tozz.nl evcc[840550]: * '' has invalid keys: integrateddevice
Jun 22 20:44:35 nuc.tozz.nl evcc[840550]: [main  ] FATAL 2024/06/22 20:44:35 will attempt restart in: 5m0s
```

It seems I'm unable to get 'integrateddevice' working, although I seem to be following the documentation

### Steps to reproduce

Add a charger with parameter 'integrateddevice' 

### Configuration details

```yaml
- name: heatpump
    integrateddevice: true
    icon: waterheater
    type: custom
    status:
      source: const
      value: 'C'
    enable:
      source: mqtt
      topic: homeassistant/switch/warmtepompforcedon/state
      payload: on
    maxcurrent:
      source: mqtt
      topic:
```
```

### Log details

```text
Jun 22 20:44:35 nuc.tozz.nl evcc[840550]: [main  ] FATAL 2024/06/22 20:44:35 cannot create charger 'heatpump': cannot create charger type 'custom': 1 error(s) decoding:
Jun 22 20:44:35 nuc.tozz.nl evcc[840550]: * '' has invalid keys: integrateddevice
```

### What type of operating system are you running?

Linux

### Version

0.126.5

andig, 2024-06-22:
The config is invalid, checking it apparently wrong. 

> although I seem to be following the documentation

Which part?

rudybroersma, 2024-06-22:
> The config is invalid, checking it apparently wrong.
> 
> > although I seem to be following the documentation
> 
> Which part?

The documentation is a little bit unclear. I'm trying to configure a charger based on MQTT, with standbypower set and integrateddevice set to true. This seems not possible now, judging on:

https://github.com/evcc-io/evcc/issues/12496

The documentation refers to eg. standbypower with regards to Shelly sockets. I was assuming I could set the same values for a custom device as well. Looks like I can't.

What I want to achieve is to have my heatpump set up in EVCC so that when there's enough surplus (and the EV is charged) EVCC turns on my heatpump's auxillary heating to increase DHW temperature from 50 to 65 degrees C. My heatpump is a Daikin unit with a integration in my Home Assistant. I'd like to force the aux heating on using HomeAssistant

Can't figure out how to achieve this with custom MQTT chargers/meters/loadpoints/whatever

andig, 2024-06-22:
Which part of the documentation is unclear?

rudybroersma, 2024-06-22:
To me it was/is unclear that certain options such as standbypower and integrateddevice are *only* available when using certain templates. They are not available when using 'type: custom'.

For example, it seems that I cannot use "standbypower: 15" when configuring a charger using type custom with MQTT sensor readings.

andig, 2024-06-22:
You’ve mentioned docs are unclear. Which part (i.e. link to docs) is unclear?

rudybroersma, 2024-06-22:
I just answered that question 7 minutes ago?

andig, 2024-06-22:
It cannot tell you that simply because it‘s not available: https://github.com/evcc-io/evcc/issues/12496#issuecomment-1965956481