evcc-io/evcc #14519: Tariff: tax macht negativen Preis noch negativer. Richtig oder falsch?

VolkerK62, 2024-06-23:
Tax einfach positiver Betrag, immer additiv?

_Originally posted by @andig in https://github.com/evcc-io/evcc/issues/14336#issuecomment-2169188367_

@andig:  das ist noch offen, oder?

andig, 2024-06-23:
Nix Neues seit https://github.com/evcc-io/evcc/issues/14336#issuecomment-2170138450. Mir kommt das immer noch merkwürdig vor. Es bräuchte wohl mal ein Abrechnungsbeispiel aus der Realität (Tibber?).

VolkerK62, 2024-06-23:
OMG, Deutschland und die Steuern 😄  https://www.bdew.de/media/documents/3027_Anw_RC_Verfahren_im_Strom_und_Gashandel_3.Auflage.pdf

Wenn der Preis incl Gebühren am Ende immer noch negativ ist (wie realistisch ist das?) fällt keine USt an.
Wenn der Stromkunde Wiederverkäufer ist (ist ja hier im Allgemeinen nicht der Fall) wird er dadurch USt-pflichtig.

Fazit: Wenn price+charges < 0, dann kein tax.

andig, 2024-06-23:
Ich hatte sowas vermutet... Kriegen wir das noch für die anderen Länder geprüft? Sonst bräuchte es da noch mehr Differenzierung :O

VolkerK62, 2024-06-23:
>Kriegen wir das noch für die anderen Länder geprüft?

🤣  Dein Ernst?  Ich war schon froh, für Deutschland etwas gefunden zu haben.

Mein Vorschlag wäre, das obige Fazit umzusetzen und abzuwarten, ob Anmerkungen kommen.

andig, 2024-06-23:
Du bist der Chef :)

closed 2024-06-23 by andig

foto-andreas, 2024-06-25:
Ich melde hier mal Zweifel an. Ich hab das so in dem PDF-Dokument nicht gefunden. Entweder unterliegt irgendwas nicht der USt oder eben doch. Wenn etwas nicht der USt unterliegt, dann ist auch nicht der Leistungsempfänger Ust-pflichtig, es sei denn, er veräußert das wieder. Zusätzlich meine ich vorhin in dem Dokument gelesen zu haben, dass Netzentgelte nich Ust-pflichtig sind. Ich bin mir aber zu 95% sicher, dass der Verbraucher die USt darauf mit eingerechnet bekommt.

Kann es sein, dass der Text auf die Verträge zwischen Erzeuger und Lieferanten abzielt?

Wir hatten ja im letzten Jahr solche Fälle. Ich versuche mal nachzuschauen, ob das in den Abrechnungen von Tibber ersichtlich ist.

LG Andreas

foto-andreas, 2024-06-25:
[image]

[image]

foto-andreas, 2024-06-25:
[image]

foto-andreas, 2024-06-25:
In der Rechnung basiert alles auf einem errechneten Durchschnittspreis aus den Pulse-Daten. Das sieht nicht so aus, aks ob an der Stelle irgendwo die Ust-Berechnung einfließt. Die wird ja erst aus den Gesamtbetrag übrigens inkl. Netznutzung berechnet.

Meine Interpretation also: Auch bei negativem Preis wird USt eingerechnet, so wie bei anderen Gutschriften auch.

Die Preisanzeige in der App endet bei 0 in der Analyse-Ansicht.

Und bei der Buggyness der Tibber-App würde ich da eh nix drauf geben.

VolkerK62, 2024-06-25:
Ich hab soviel gesucht und gelesen, da hab ich dann wohl verschieden DInge in Bezug gebracht, die keinen Bezug haben.
Da für die USt von allem die Summe genommen wird, dürfen negative Preise nicht gesondert behandelt werden.

@andig sorry. revert?

Schön, mal so eine Tibber rechnung zu sehen.
Bei den ganzen Gebühren ist es sehr unwahrscheinlich, dass die Summe (price+charges) mal negativ ist.
Insofern ist meine Eingangsfrage eigentlich obsolet

tobe4it, 2024-06-25:
> Nix Neues seit [#14336 (comment)](https://github.com/evcc-io/evcc/issues/14336#issuecomment-2170138450). Mir kommt das immer noch merkwürdig vor. Es bräuchte wohl mal ein Abrechnungsbeispiel aus der Realität (Tibber?).

Unter developer.tibber.com kann die stundenweise Berechnung eingesehen werden:

`{
  viewer {
    homes {
      consumption(resolution: HOURLY, last: 8616) {
        nodes {
          from
          to
          cost
          unitPrice
          unitPriceVAT
          consumption
          consumptionUnit
        }
      }
    }
  }
}
`

Hier zeigt sich für den 2. Juli, bei negativen Bruttopreisen, dass Tibber keinen Unterschied macht bei der Berechnung der USt aka VAT:

`              {
                "from": "2023-07-02T14:00:00.000+02:00",
                "to": "2023-07-02T15:00:00.000+02:00",
                "cost": -3.6135460032,
                "unitPrice": -0.4259248,
                "unitPriceVAT": -0.0680048,
                "consumption": 8.484,
                "consumptionUnit": "kWh"
              }, 
`

unitPrice ist der Bruttopreis je kWh, unitPriceVAT die enthaltene USt.
[image]
[image]

Davon unbenommen führen sie auf der Rechnung die Nettopreise aller Bestandteile auf und errechnen hier die USt auf die Summe.

andig, 2024-06-25:
Also revert?!

VolkerK62, 2024-06-25:
ja.

andig, 2024-06-26:
Done.