evcc-io/evcc #14524: Fehler im Lastmanagement (Nightly)

VolkerK62, 2024-06-23:
### Describe the bug

Die angehängte Config funktioniert mit 0.127.2 wunderbar.
"overPower" wird erkannt und es wird entsprechend geregelt.

Mit 0.127.2 (f2e24171) kommt beim Start der Fehler `root circuit must not be assigned to loadpoint test`
Nimmt man den Eintrag `circuit: main` am Loadpoint weg, startet evcc.
Im Falle von "overPower" wird diese zwar angezeigt, es erfolgt allerdings keine Regelung.

### Steps to reproduce

.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

database:
  type: sqlite
  dsn: /home/wsl/.evcc/evcc.db

interval: 10s

sponsortoken: xxx

telemetry: false

log: debug
#level:
#  db: trace

#####################################################################
meters:
- name: pv
  type: template
  template: solarlog
  usage: pv
  host: 192.168.178.23
  port: 502

- name: battery
  type: custom
  power:
    source: http
    uri: http://192.168.178.21/emeter/1
    jq: .power
  soc:
    source: mqtt
    topic: Haus/Bat_SoC
    timeout: 60m
  capacity: 7.5

- name: grid_fake
  type: custom
  power:
    source: calc
    add:
    - source: mqtt
      topic: wsl/site/pvPower
      scale: -1
      timeout: 25h
    - source: mqtt
      topic: wsl/site/batteryPower
      scale: -1
      timeout: 25h
    - source: const
      value: 500
    - source: mqtt
      topic: wsl/loadpoints/2/chargePower
      timeout: 25h

##################################################################
chargers:
- name: E-Bikes
  type: template
  template: fritzdect
  ## in Garage
  uri: http://fritz.box
  user: xxx
  password: xxx
  ain: xxx
  standbypower: 10
  integrateddevice: true
  icon: bike

- name: test
  type: custom
  status:
    source: http
    uri: http://192.168.178.88/status
    jq: if .relays[].ison == true then "C" else "B" end
  enabled: 
    source: http
    uri: http://192.168.178.88/status
    jq: .relays[].ison
  enable:
    source: http
    uri: http://192.168.178.88/relay/0?turn={{if .enable}}on{{else}}off{{end}} 
  maxcurrent:
    source: script
    cmd: /bin/sh -c 'echo 1'
  features:
    - integrateddevice
    - heating

#######################################################################
vehicles:
- name: bike
  type: template #custom
  template: offline
  title: E-Bikes
  icon: bike
  capacity: 0.625
  phases: 1

#########################################################################
loadpoints:
- title: E-Bikes
  charger: E-Bikes

- title: test
  charger: test
  circuit: main

########################################################################
site:
- title: Haus
  circuit: main
  meters:
    grid: grid_fake
    pv: pv
    battery: battery

circuits:
- name: main
  title: Hauptkreis
  #maxcurrent: 10
  maxpower: 2000
  meter: grid_fake

########################################################################
tariffs:
  currency: EUR
  grid:
    type: template # fixed # custom
    template: energy-charts-api
    bzn: DE-LU
    #charges: 0.05
    tax: 0.2

  feedin:
    type: fixed
    price: 0.1095  # Preis je kWh Einspeisung
      

######################################################################

mqtt:
  broker: 192.168.178.25:1883
  topic: wsl
```

### Log details

```text
wsl@DESKTOP-7RU7VSU:~$ evcc
[main  ] INFO 2024/06/23 15:24:53 evcc 0.127.2 (f2e24171)
[main  ] INFO 2024/06/23 15:24:53 using config file: /etc/evcc.yaml
[db    ] INFO 2024/06/23 15:24:53 using sqlite database: /home/wsl/.evcc/evcc.db
[mqtt  ] INFO 2024/06/23 15:24:53 connecting evcc-2130807971 at tcp://192.168.178.25:1883
[mqtt  ] DEBUG 2024/06/23 15:24:53 tcp://192.168.178.25:1883 connected
[main  ] INFO 2024/06/23 15:24:53 listening at :7070
[circuit-main] DEBUG 2024/06/23 15:24:53 validation of max phase current disabled
[lp-1  ] DEBUG 2024/06/23 15:24:54 charge total import: 10.516kWh
[site  ] WARN 2024/06/23 15:24:54 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[main  ] FATAL 2024/06/23 15:24:54 root circuit must not be assigned to loadpoint test
[main  ] FATAL 2024/06/23 15:24:54 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

andig, 2024-06-23:
@naltatis @premultiply  Leider müssen wir das Thema LA wieder aufmachen. 0.173.3 hat ein ziemliches Chaos veranstaltet.

Aus https://github.com/evcc-io/docs/pull/567#discussion_r1644508272:

>> Alle Ladepunkte sind immer (automatisch) im Root-Circuit es sei denn sie sind explizit in einem Sub-Circuit.

> Find ich auch intuitiv aus Nutzersicht. Ich hab das im Text ergänzt.

@naltatis in der Doku steht aber:

> Hier wird der Hauptstromkreis main definiert, der eine maximale Leistung von 30kW und eine maximale Phasen-Stromstärke von 63A hat. Die Ladepunkte Garage und Carport sind diesem Stromkreis zugeordnet. 

Das widerspricht sich. Weiterhin:

@premultiply am Root Circuit dürfen aber keine WBen dran hängen. Und automatisch eingesammelt werden fehlende Boxen nicht, das ist dieses Issue. Ich bin auch gar nicht sicher, ob wir automatisch alle Boxen zuordnen sollten, oder das nicht doch dem Anwender überlassen bleiben sollte? Ich tendiere zu letzterem.

Ich denke wir müssen nochmal umbauen:
- Auch der Root Circuit kann zugewiesene Wallboxen haben
- Circuits müssen eine zugewiesene Wallbox oder einen Zähler haben

Macht das Sinn?

premultiply, 2024-06-23:
Ja, klingt für mich gut und richtig.

andig, 2024-06-23:
@VolkerK62 Nightly baut, wäre Klasse wenn du das nochmal nachtesten könntest. Vielen Dank!

closed 2024-06-23 by andig

VolkerK62, 2024-06-23:
Schon gesehen. Mach ich.

VolkerK62, 2024-06-23:
Funktioniert prima.
Nebenhinweis: `circuit: main` in der Site (aus der vorherigen Version) stört nicht und wird einfach ignoriert.