evcc-io/evcc #14526: Upgrade fails

andig, 2024-06-23:
Upgrading from older version results in:

```
raspi@raspi:~ $ sudo apt --only-upgrade install -y evcc
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Die folgenden Pakete werden aktualisiert (Upgrade):
  evcc
1 aktualisiert, 0 neu installiert, 0 zu entfernen und 255 nicht aktualisiert.
Es müssen 22,8 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 11,7 MB Plattenplatz zusätzlich benutzt.
Holen:1 https://dl.cloudsmith.io/public/evcc/stable/deb/debian bookworm/main arm64 evcc arm64 0.127.2 [22,8 MB]
Es wurden 22,8 MB in 6 s geholt (3.538 kB/s).                                                                                               
apt-listchanges: Changelogs werden gelesen...
(Lese Datenbank ... 160011 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../evcc_0.127.2_arm64.deb ...
Entpacken von evcc (0.127.2) über (0.125.0) ...
--------------------------------------------------------------------------------
ERROR: your configuration is not compatible with the new version:
[main  ] INFO 2024/06/22 17:51:02 evcc 0.127.2
[main  ] INFO 2024/06/22 17:51:02 using config file: /etc/evcc.yaml
[main  ] FATAL 2024/06/22 17:51:02 

Found systemd service database at "/var/lib/evcc/evcc.db", evcc has been invoked with database "~/.evcc/evcc.db".
Running evcc with vehicles configured in evcc.yaml may lead to expiring the yaml configuration's vehicle tokens.
This is due to the fact, that the token refresh will be saved to the local instead of the service's database.
If you have vehicles with touchy tokens like PSA or Tesla, make sure to remove vehicle configuration from the yaml file.
```

@GrimmiMeloni @nicok1 I think this is due to the config check from https://github.com/evcc-io/evcc/pull/12547 with the additional check for conflicting databases. Running evcc as service will not show this error due to `evcc.service` running with

    Environment="EVCC_DATABASE_DSN=/var/lib/evcc/evcc.db"

Imho the config check in `postremove.sh` MUST mimic the same environment:

	if ! /usr/bin/evcc checkconfig > /dev/null; then

Does that make sense? PR welcome!

GrimmiMeloni, 2024-06-24:
@andig you suggestion sounds reasonable.
I assume the fix is as simple as setting `$EVCC_DATABASE_DSN`. 
However, I'm wondering if this might conflict with a configured DB location in `evcc.yaml`.
What takes precedence (ENV or config)?

frandess, 2024-06-24:
just some more information on the case above:
- no db configured in yaml (so default)
- deleting the db in `~/.evcc` does not change the described behaviour, the output of the install process stays exactly the same (although there is no db in `~/.evcc` anymore)

andig, 2024-06-24:
@GrimmiMeloni ich glaube ENV gewinnt.

> However, I'm wondering if this might conflict with a configured DB location in evcc.yaml.

Das Problem gäbe es heute schon, unabhängig von diesem hier.

> I assume the fix is as simple as setting $EVCC_DATABASE_DSN.

Hast Du eine Ahnung, wie die Zeile dann aussehen müsste? PR welcome!

andig, 2024-08-04:
Weiss jemand wie die ENV var zusammen mit dem IF angegeben werden muss? Export fürs ganze Skript?

closed 2024-08-07 by andig