evcc-io/evcc #14531: evcc UI hängt sich auf

lapixo, 2024-06-23:
### Describe the bug

Die Oberfläche hängt sich immer wieder auf und aktualisiert die Daten nicht. 

### Steps to reproduce

Seit den letzten Versionen friert die Oberfläche immer wieder ein. Arbeitet aktuell extrem unzuverlässig. Musste die Wallbox immer wieder manuell aktivieren, um das Auto zu laden. 
Erst nach einem EVCC Neustart reagiert die Oberfläche für eine gewisse Zeit. 
Im Backend unter "Konfiguration" werden die Werte korrekt angezeigt. Kein Fehler laut Log/Trace Meldung.

[image]

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.1.111
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 40s # control cycle interval

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug

mqtt:
  broker: 192.168.1.111:1883

meters:
- name: my_grid
  type: custom
  power: # Leistung (W)
    source: mqtt
    topic: homeassistant/sensor/current_power_positiv_and_negativ/state
  energy: # Energie (kWh) - Umrechnung von Wh -> kWh
    source: mqtt
    topic: homeassistant/sensor/total_imported_energy/state
    timeout: 560m    
  currents:
    - source: mqtt
      topic: homeassistant/sensor/phase_a_current/state
      timeout: 560m      
    - source: mqtt
      topic: homeassistant/sensor/phase_b_current/state
      timeout: 560m
    - source: mqtt
      topic: homeassistant/sensor/phase_c_current/state
      timeout: 560m

- name: my_pv
  type: custom
  power:
    source: mqtt
    topic: homeassistant/sensor/pv_total_power/state 
    timeout: 560m
  energy:
    source: mqtt
    topic: homeassistant/sensor/pv_total_production/state
    timeout: 560m

- name: my_battery_1
  type: custom
  power:
    source: mqtt
    topic: homeassistant/sensor/battery_discharging_power/state
    scale: 1
    timeout: 560m
  #energy:
  #  source: mqtt
  #  topic: homeassistant/sensor/total_battery_discharge/state 
  soc:
    source: mqtt
    topic: homeassistant/sensor/battery_level_nominal/state
    timeout: 560m
  capacity: 9.6
- name: my_battery_2
  type: custom
  power:
    source: mqtt
    topic: homeassistant/sensor/battery_charging_power/state
    scale: -1
    timeout: 560m
  #energy:
  #  source: mqtt
  #  topic: homeassistant/sensor/total_battery_charge/state     
  soc:
    source: mqtt
    topic: homeassistant/sensor/battery_level_nominal/state
    timeout: 560m
  capacity: 9.6

site:
  title: Zuhause
  residualPower: 100
  meters:
    grid: my_grid
    pv:
      - my_pv
    battery:
      - my_battery_1
      - my_battery_2
```

### Log details

```text
[mqtt ] TRACE 2024/06/23 22:07:57 recv homeassistant/sensor/battery_discharging_power/state: '1063'
[mqtt ] TRACE 2024/06/23 22:07:57 recv homeassistant/sensor/phase_a_current/state: '1.6'
[mqtt ] TRACE 2024/06/23 22:07:57 recv homeassistant/sensor/phase_b_current/state: '1.6'
[mqtt ] TRACE 2024/06/23 22:07:57 recv homeassistant/sensor/phase_c_current/state: '1.6'
[mqtt ] TRACE 2024/06/23 22:08:06 recv homeassistant/sensor/current_power_positiv_and_negativ/state: '0'
[mqtt ] TRACE 2024/06/23 22:08:06 recv homeassistant/sensor/current_power_positiv_and_negativ/state: '-18'
[mqtt ] TRACE 2024/06/23 22:08:07 recv homeassistant/sensor/battery_discharging_power/state: '1142'
[mqtt ] TRACE 2024/06/23 22:08:07 recv homeassistant/sensor/phase_a_current/state: '1.5'
[mqtt ] TRACE 2024/06/23 22:08:16 recv homeassistant/sensor/current_power_positiv_and_negativ/state: '-8'
[mqtt ] TRACE 2024/06/23 22:08:17 recv homeassistant/sensor/battery_discharging_power/state: '1121'
[mqtt ] TRACE 2024/06/23 22:08:26 recv homeassistant/sensor/current_power_positiv_and_negativ/state: '-14'
[mqtt ] TRACE 2024/06/23 22:08:27 recv homeassistant/sensor/battery_discharging_power/state: '1124'
[mqtt ] TRACE 2024/06/23 22:08:27 recv homeassistant/sensor/phase_b_current/state: '1.5'
[mqtt ] TRACE 2024/06/23 22:08:27 recv homeassistant/sensor/phase_c_current/state: '1.5'
[mqtt ] TRACE 2024/06/23 22:08:36 recv homeassistant/sensor/current_power_positiv_and_negativ/state: '-4'
[mqtt ] TRACE 2024/06/23 22:08:37 recv homeassistant/sensor/battery_discharging_power/state: '1115'
[mqtt ] TRACE 2024/06/23 22:08:46 recv homeassistant/sensor/current_power_positiv_and_negativ/state: '2'
[mqtt ] TRACE 2024/06/23 22:08:47 recv homeassistant/sensor/battery_discharging_power/state: '1086'
[mqtt ] TRACE 2024/06/23 22:08:56 recv homeassistant/sensor/current_power_positiv_and_negativ/state: '0'
[mqtt ] TRACE 2024/06/23 22:08:56 recv homeassistant/sensor/current_power_positiv_and_negativ/state: '-20'
[mqtt ] TRACE 2024/06/23 22:08:57 recv homeassistant/sensor/battery_discharging_power/state: '1145'
```

### What type of operating system are you running?

Linux

### Version

0.127.2

lapixo, 2024-06-23:
Hier die Log nach einem direkten Neustart. 
[main ] INFO 2024/06/23 23:11:48 evcc 0.127.2
[main ] INFO 2024/06/23 23:11:48 using config file: /etc/evcc.yaml
[db ] INFO 2024/06/23 23:11:48 using sqlite database: /root/.evcc/evcc.db
[mqtt ] INFO 2024/06/23 23:11:48 connecting evcc-575645429 at tcp://192.168.1.111:1883
[mqtt ] DEBUG 2024/06/23 23:11:48 tcp://192.168.1.111:1883 connected
[main ] INFO 2024/06/23 23:11:48 listening at :7070
[lp-1 ] DEBUG 2024/06/23 23:11:50 charge total import: 7345.000kWh
[site ] INFO 2024/06/23 23:11:51 site config:
[site ] INFO 2024/06/23 23:11:51 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/06/23 23:11:51 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/06/23 23:11:51 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/06/23 23:11:51 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2024/06/23 23:11:51 battery 2: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2024/06/23 23:11:51 vehicles:
[site ] INFO 2024/06/23 23:11:51 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/06/23 23:11:51 loadpoint 1:
[lp-1 ] INFO 2024/06/23 23:11:51 mode: pv
[lp-1 ] INFO 2024/06/23 23:11:51 charger: power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/06/23 23:11:51 meters: charge ✓
[lp-1 ] INFO 2024/06/23 23:11:51 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2024/06/23 23:11:51 set priority soc: 100
[lp-1 ] DEBUG 2024/06/23 23:11:51 set smart cost limit: 0.2
[lp-1 ] DEBUG 2024/06/23 23:11:51 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/23 23:11:51 phase timer inactive
[lp-1 ] DEBUG 2024/06/23 23:11:51 pv timer inactive
[lp-1 ] INFO 2024/06/23 23:11:51 vehicle updated: unknown -> Tesla Y
[lp-1 ] DEBUG 2024/06/23 23:11:51 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[site ] DEBUG 2024/06/23 23:11:52 ----
[lp-1 ] DEBUG 2024/06/23 23:11:52 charge power: 0W
[lp-1 ] DEBUG 2024/06/23 23:11:52 charge currents: [0 0 0]A
[site ] DEBUG 2024/06/23 23:11:52 pv power: 0W
[site ] DEBUG 2024/06/23 23:11:52 battery 1 power: 746W
[site ] DEBUG 2024/06/23 23:11:52 battery 1 soc: 77%
[site ] DEBUG 2024/06/23 23:11:52 battery 2 power: -0W
[site ] DEBUG 2024/06/23 23:11:52 battery 2 soc: 77%
[site ] DEBUG 2024/06/23 23:11:52 battery soc: 77%
[site ] DEBUG 2024/06/23 23:11:52 battery power: 746W
[site ] DEBUG 2024/06/23 23:11:52 grid meter: 6W
[site ] DEBUG 2024/06/23 23:11:52 grid currents: [1.1 1 1.1]A

Portainer zeigt beim Container ständig "starting" an. Als ob der Dienst nicht vollständig geladen ist.

naltatis, 2024-06-23:
Wieso hast du denn bei der Werteabfrage Timeouts von `timeout: 560m` (fast 10 Stunden) stehen??

In deiner Konfiguration fehlt der Loadpoint und Charger.

lapixo, 2024-06-23:
Die Werte werden aus Homeassistant ins MQTT geschrieben. Wenn sich bestimmte Werte wie z.B. vom Akku nicht ändern (was normal ist über Tag) dann meldete EVCC früher immer direkt einen Fehler und hing sich auf.  Daher der große Timeout. 

Sry. Den Konfig Teil habe ich vergessen zu posten. 

Anbei der fehlende Teil:

chargers:
  - name: goe
    type: template
    template: go-e
    host: 192.168.10.165

loadpoints:
- title: Stellplatz
  charger: goe
  vehicle: flinmobil
  mode: pv

site:
  title: Zuhause
  meters:
    grid: my_grid
    pv:
      - my_pv
    battery:
      - my_battery_1
      - my_battery_2

lapixo, 2024-06-23:
Zur Info: Ich nutze EVCC schon seit fast einem Jahr und die Konfig mittlerweile auch über mehrere Monate. Aber seit einigen Wochen habe ich leider nur Probleme mit den o.g. Problemen, dass die EVCC UI die Werte einfach nicht mehr ausließt obwohl diese im MQTT aktuell sind wie auch im Log von EVCC zu sehen.

naltatis, 2024-06-23:
Magst du noch mal den Zustand "UI hängt sich auf" beschreiben. Bekommst du keinen neuen Werte mehr? Kannst du du andere Ansichten (Konfiguration, Ladevorgänge) weiterhin aufrufen? Friert die komplette Anwendung ein (keine weiteren Log Ausgaben)?

lapixo, 2024-06-23:
Die UI an sich lässt sich normal bedienen. Nur die Werte im Frontend werden nicht mehr aktualisiert und somit startet auch kein Ladevorgang usw. Im Backend in der experimentellen Ansicht werden die Werte meiner Ansicht nach aktualisiert.

naltatis, 2024-06-23:
Mit Backend meinst du die Konfigurationsseite in der UI, richtig?

Kannst du beim nächsten Mal, wenn du den Zustand hast, mal die Ausgabe von http://evcc.local:7070/api/state aufrufen und schauen, ob die Werte dort auch aktuell sind oder dem "eingefrorenen Stand" der UI entsprechen.

lapixo, 2024-06-23:
Ich habe zum Testen die gesamte MQTT config durch eine "direkte" Sungrow config ersetzt. Dies funktioniert einwandfrei und die UI reagiert und aktualisiert die Werte wie sie soll. 
meters:
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.1.143
    port: 502 
    timeout: 10s
  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.1.143
    port: 502 
    timeout: 10s
  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery
    modbus: tcpip
    id: 1
    host: 192.168.1.143
    port: 502 
    timeout: 10s
    
    
    Ich ändere das zurück zur MQTT config und sage gleich bescheid, was über die API ankommt.

lapixo, 2024-06-23:
"Mit Backend meinst du die Konfigurationsseite in der UI, richtig?"

Richtig.

lapixo, 2024-06-23:
Es scheint so, dass über die API auch keine neuen Werte ankommen. In der Log werden jetzt aktuell auch keine neuen Werte geholt.  Seit 5 Minuten keine Änderung im Log. 
Vor einer Stunde kamen zumindest im Log noch neue Werte an. Jetzt passiert auch hier nichts mehr. 

[main ] INFO 2024/06/23 23:47:40 evcc 0.127.2
[main ] INFO 2024/06/23 23:47:40 using config file: /etc/evcc.yaml
[db ] INFO 2024/06/23 23:47:40 using sqlite database: /root/.evcc/evcc.db
[mqtt ] INFO 2024/06/23 23:47:40 connecting evcc-634256736 at tcp://192.168.1.111:1883
[mqtt ] DEBUG 2024/06/23 23:47:40 tcp://192.168.1.111:1883 connected
[main ] INFO 2024/06/23 23:47:40 listening at :7070
[lp-1 ] DEBUG 2024/06/23 23:47:44 charge total import: 7345.000kWh
[site ] WARN 2024/06/23 23:47:44 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site ] INFO 2024/06/23 23:47:45 site config:
[site ] INFO 2024/06/23 23:47:45 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/06/23 23:47:45 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/06/23 23:47:45 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/06/23 23:47:45 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2024/06/23 23:47:45 battery 2: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2024/06/23 23:47:45 vehicles:
[site ] INFO 2024/06/23 23:47:45 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/06/23 23:47:45 loadpoint 1:
[lp-1 ] INFO 2024/06/23 23:47:45 mode: pv
[lp-1 ] INFO 2024/06/23 23:47:45 charger: power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/06/23 23:47:45 meters: charge ✓
[lp-1 ] INFO 2024/06/23 23:47:45 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2024/06/23 23:47:45 set priority soc: 100
[lp-1 ] DEBUG 2024/06/23 23:47:45 set smart cost limit: 0.26
[lp-1 ] DEBUG 2024/06/23 23:47:45 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/06/23 23:47:45 phase timer inactive
[lp-1 ] DEBUG 2024/06/23 23:47:45 pv timer inactive
[lp-1 ] INFO 2024/06/23 23:47:45 vehicle updated: unknown -> Tesla Y
[lp-1 ] DEBUG 2024/06/23 23:47:45 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[site ] DEBUG 2024/06/23 23:47:45 ----
[lp-1 ] DEBUG 2024/06/23 23:47:45 charge power: 0W
[lp-1 ] DEBUG 2024/06/23 23:47:45 charge currents: [0 0 0]A
[site ] DEBUG 2024/06/23 23:47:45 pv power: 0W
[site ] DEBUG 2024/06/23 23:47:45 battery 1 power: 585W
[site ] DEBUG 2024/06/23 23:47:45 battery 1 soc: 73%
[site ] DEBUG 2024/06/23 23:47:45 battery 2 power: -0W
[site ] DEBUG 2024/06/23 23:47:45 battery 2 soc: 73%
[site ] DEBUG 2024/06/23 23:47:45 battery soc: 73%
[site ] DEBUG 2024/06/23 23:47:45 battery power: 585W
[site ] DEBUG 2024/06/23 23:47:45 grid meter: -9W
[site ] DEBUG 2024/06/23 23:47:45 grid currents: [0.8 0.8 0.8]A

lapixo, 2024-06-23:
Ich habe einen Verdacht, dass EVCC ein Problem mit "veralteten" MQTT Werten hat und deshalb nicht richtig startet. Es sieht so aus, als ob EVCC auf eine Aktualisierung im MQTT "wartet" und einfach nicht weiter macht. 
 
Testweise habe ich Homeassistant neugestartet, um alle Werte im MQTT zu aktualisieren. Während des Neustarts, habe ich EVCC beobachtet und die Oberfläche aktualisierte sich. 

Eigentlich muss EVCC veraltete Werte "ignorieren", da sich die PV Leistung z.B. bei Sonnenuntergang nicht ändert und Homeassistant diese Werte nicht im MQTT aktualisiert. 

Ich habe das Problem schon länger (einige Wochen) beobachtet und hatte deshalb den großen Timeout gesetzt. Leider war das nur eine Notlosung und ich habe immer wieder EVCC und/oder Homeassistant neustarten müssen, damit ich zuverlässig laden kann.

lapixo, 2024-06-23:
Wollte noch mitteilen, dass das Problem für mich persönlich erstmal gelöst ist, da ich jetzt gesehen habe, dass EVCC einen Modbus Proxy anbietet. Das löst all meine Probleme, die ich versucht habe über MQTT zu lösen. 

Ich habe nun den Proxy aktiviert, damit mein Sungrow nicht von Anfragen durch mehrere Geräte "überlastet" wird. 
Meine EVCC Config habe ich nun komplett auf Modbus Kommunikation umgestellt. 

Homeassistant holt sich die Werte nun auch über den Proxy. Diese Funktion ist absolut genial! Danke für das tolle Feature!

Auch die Funktion "Verhindere Entladung im Schnell-Modus und bei geplantem Laden." ist genau das, was ich über Homeassistant damals manuell gelöst habe. Nun gibt es die Funktion direkt von EVCC, vorausgesetzt der Wechselrichter ist als modbus Gerät eingerichtet. Top! Danke auch dafür!

naltatis, 2024-06-24:
Schön, dass du hier schon mal eine Lösung hast. Aber das MQTT Verhalten, was du beobachtest, klingt ja erst einmal nicht so super.

@andig hast du eine Idee wieso das passiert?

VolkerK62, 2024-06-24:
Keine Ahnung, ob das etwas damit zu tun haben könnte, aber beim Server Start werden doch alle MQTT topics gelöscht.

premultiply, 2024-06-24:
Normalerweise müssen alle Topics einfach regelmäßig aktualisiert werden.
Ob ein Wert sich dabei ändert oder nicht ist unwichtig. Nur der dadurch aktualisierte Zeitstempel ist wichtig.